lvgl-micropython/lvgl_micropython #2: MPY compilation error

martyy665, 2024-02-09:
Hi,

seems that the latest version cannot be compiled:

```
MPY display_driver_framework.py
error compiling display_driver_framework.py:
Traceback (most recent call last):
  File "display_driver_framework.py", line 565, in _set_memory_location
ViperTypeError: can't do binary op between 'object' and 'int'
```
Or can I have done anything wrong?

kdschlosser, 2024-02-09:
I just pushed a commit that should fix it.

martyy665, 2024-02-09:
Unfortunately, it did not help. The same error is now on the line 597 instead of 565.

straga, 2024-02-09:
remove @micropython.viper, or probably need param_buf = ptr8(self._param_buf)

martyy665, 2024-02-09:
@straga Yes, removing the decorator helps. But, I am not sure about the right solution regarding speed and efficiency.

kdschlosser, 2024-02-09:
I just updated it again. I believe it should be fixed now.
I could have broken some other stuff tho

martyy665, 2024-02-09:
Yes, I use .py drivers too, as in the case I tried to use the built-in ones, the program crashed on type error  ( same as here: https://forum.lvgl.io/t/micropython-display-drivers-part-2/14131/44).

kdschlosser, 2024-02-09:
I need to see all of the code you are using and not just the traceback.

kdschlosser, 2024-02-09:
all of the parameters for the busses need to have their parameter names supplied if you don't supply them it is going to error.

martyy665, 2024-02-11:
The compilation has been successful—closing the issue.

closed 2024-02-11 by martyy665