lvgl-micropython/lvgl_micropython #4: No lv_utils

marksull, 2024-02-10:
1) Compiled for ESP32-S3 (`python make.py esp32 submodules mpy_cross BOARD=ESP32_GENERIC_S3`) (see PRs for manifest fix)
2) Deployed to board (MicroPython v1.22.1-dirty on 2024-02-10; Generic ESP32S3 module with ESP32S3)
3) Testing using https://github.com/bdbarnett/mpdisplay
 a) Tested with `mpdisplay_simpletest.py` and works perfectly
 b) Testing with `lv_touch_test.py` and I get an `ImportError: no module named 'lv_utils'`

I see tha `lv_touch_test.py` imports `lv_config` which imports `lv_utils`, with the former being present and the latter not.

```
>>> import lvgl
>>> import lv_config
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lv_config.py", line 20, in <module>
ImportError: no module named 'lv_utils'
```

Ideas on what could be causing this?

kdschlosser, 2024-02-10:
yeah lv_utils is gone. You can upload it to the MCU if you want. you will have to look back in my repo to download it. I changed the name of it and rewrote it for better functionality and also to be more pythonic. It is now called task_handler.TaskHandler which is frozen into the firmware.

The touch driver framework is no longer called "touch_framework" it is now called "pointer_framework". I don't think that @bdbarnett is using that framework. I know the display framework is not being used at all because it is specific to LVGL like the touch framework is.

What he is doing with that design is making it so that it can be used with other GUI frameworks so it's not LVGL specific.

kdschlosser, 2024-02-10:
You do realize that I have all kinds of drivers already packaged with this repo, it also has drivers for 2 different touch IC's. Those drivers don't get baked into the firmware, you need to upload them to the MCU as py files.

marksull, 2024-02-10:
I have been following the thread that highlights the collaboration you both have  been working on so I kind of assumed if there was a requirement in his library using this that is would have been called out. 🤷

I did see that lv_utils was explicitly included in the default builder/init manifest so was assuming there was a manifest oversight for the Esp32.  If this is not the case, then feel free to close this out.

bdbarnett, 2024-02-10:
Hey guys,

Sorry for not keeping up.  I've been focusing my attention on other aspects of MPDisplay and haven't kept up with Kevin's changes.  I just posted a fix to lv_config.py that should take care of it.  Please let me know how it works for you.

The code is https://github.com/bdbarnett/mpdisplay/blob/a91f8389e92416ba6873942446f5e7ef44223c5a/lv_config.py#L23-L29

```
try:
    import lv_utils
    if not lv_utils.event_loop.is_running():
        eventloop = lv_utils.event_loop(asynchronous=False, exception_sink=None)
except ImportError:
    import task_handler
    task_handler = task_handler.TaskHandler()
```

By the way, it shows I posted it 4 hours ago, but I just posted it a few minutes ago.  I use WSL and keep my repos in the Ubuntu VM.  I use VS Code, which runs with a Windows frontend and Linux backend.  When my computer goes to sleep, Windows time keeps ticking, but Linux time stands still.  So, if I don't catch it and force a time update with `sudo hwclock -s`, my code commits look like they happened in the past.  I didn't write the fix until after you pointed out the problem :)  (I setup a cron job to call `hwclock -s` once an hour, but if I wake my computer and commit something before the cron job runs, it's still backdated.)

marksull, 2024-02-10:
@bdbarnett, thanks for the quick fix. While it does resolve the issue, it just moves the exception along a little. It is not quite compatible with task_handler:

```
>>> from lib import lv_touch_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lib/lv_touch_test.py", line 13, in <module>
  File "lv_config.py", line 29, in <module>
  File "task_handler.py", line 37, in __init__
TypeError: 
```

@kdschlosser I took your advice to use the ft6x36 and st7796 drivers from this package. While indeed they both loaded without error, using the `drive/display/driver_use_exmple.py` I always raises an error with the first couple of lines :

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_use_exmple.py", line 7
SyntaxError: not a constant
```

I believe const() is meant to be a macro that replaced at compilation so I tried to manually precompile but saw the same problem:

```
./mpy-cross ../../../../driver/display/driver_use_exmple.py                                                                                                                                             
Traceback (most recent call last):
  File "../../../../driver/display/driver_use_exmple.py", line 7
SyntaxError: not a constant
```

Obviously I am misunderstanding some core concept here.

kdschlosser, 2024-02-10:
I know what that error is and it's not an issue to fix it

When I make changes I will usually make a bucket load of them at once and then I will test to see if it works. I haven't gotten to the point of testing yet because of the large number of changes I am making. I am making these changes to streamline things and also to lessen the code footprint. That is the reason why I removed the heap_caps module and moved it into the lcd_bus module.

There is going to be some issues and I apologize for that. Once the API settles down and I get things into place there will be less problems.

The driver use example has not been updated and it is just an example......

I just updated the example so it should work now.

marksull, 2024-02-10:
Thanks for your effort. I have been watching with repo for the past couple of weeks and recognise the amount of effort you are putting into this. It's super helpful.

With respect the the updated example, It fails with:

```
>>> import driver_use_exmple
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_use_exmple.py", line 51, in <module>
TypeError: 'sclk' argument required
```

If I change your example to use `sclk` in lieu of `sck` it works.

I looked at `lcd_bus.pyi` and notice that it was indeed using `sck` and I even if dropped this code onto the esp and manually loaded it, the example still reported the same issue.

I found that there is also a C implementation of this library `ext_mod/lcd_bus/esp32_src/spi_bus.c` so I guess this is being favoured over the py implementation. Just thought I would mention it so you know it needs to be fixed.

Thanks again.

kdschlosser, 2024-02-11:
don't get your hopes up too much on this repo being around for very long. See #1 as to why.

closed 2024-04-28 by marksull