lvgl-micropython/lvgl_micropython #5: Crash during DisplayDriver instantiation

martyy665, 2024-02-14:
Hi,

I am trying to switch from https://github.com/bdbarnett/mpdisplay - which is now incompatible with the latest lvgl_micropython - TO built-in (frozen) display_driver_framework and C-written bus drivers.

I am using the file  _./driver/display/driver_use_exmple.py_  with just minor changes (fix pinning, resolution and use included st7789 LCD driver).

During DisplayDriver instantiation in the driver:

`class ST7789(display_driver_framework.DisplayDriver)`

The following crash occurs:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40379a79  PS      : 0x00060030  A0      : 0x820c093f  A1      : 0x3fced6a0
A2      : 0x3c25bdb9  A3      : 0x3c25cd00  A4      : 0x3c20e598  A5      : 0xffffff30
A6      : 0x3c253a34  A7      : 0x3c1f9b58  A8      : 0x80379a79  A9      : 0x3fced680
A10     : 0x0000656c  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3c25bdb8
A14     : 0x3c25bdb4  A15     : 0x00000003  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00006570  LBEG    : 0x4038bd78  LEND    : 0x4038bd82  LCOUNT  : 0x00000000

Backtrace: 0x40379a76:0x3fced6a0 0x420c093c:0x3fced740 0x420c7759:0x3fced770 0x420c781d:0x3fced790 0x420c63f8:0x3fced7b0 0x420c622d:0x                                          3fced810 0x420c7759:0x3fced830 0x420c781d:0x3fced850 0x4037a7e1:0x3fced870 0x420c093c:0x3fced910 0x420c7759:0x3fced940 0x420c77c4:0x3f                                          ced960 0x420e6435:0x3fced9a0 0x420e6762:0x3fced9d0 0x420c7899:0x3fcedac0 0x4037aa7e:0x3fcedb00 0x420c093c:0x3fcedba0 0x420c7759:0x3fce                                          dbf0 0x420c776e:0x3fcedc10 0x420d3d3a:0x3fcedc30 0x420d407c:0x3fcedcc0 0x420db77c:0x3fcedd00

ELF file SHA256: cac8663cce55c127

Rebooting...
```
Framebuffers are left in the default state - set to None:
```
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_HIGH,
    power_pin=_POWER_PIN,
    power_on_state=st7789.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7789.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y
)
```

The previous version with an MPY-based bus driver was working.

kdschlosser, 2024-02-14:
I have not been able to spend a large amount of time working on this because of a back injury and I apologize for this. I have an appointment at a doctors to see if I am able to get some relief and then I will be able to put more time into getting things sorted out. The project is still very much in an alpha testing stage and things may not work properly. I am sorry for that.

if you can decode the backtrace information for me that would be really helpful.

martyy665, 2024-02-15:
Of course, I understand that this project is still under development and may not necessarily work flawlessly. There is the full backtrace:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40379a79  PS      : 0x00060030  A0      : 0x820c093f A1      : 0x3fced6a0
0x40379a79: mp_execute_bytecode at /home/schlosser_new/lvgl_micropython/lib/micropython/py/vm.c:407 (discriminator 3)

A2      : 0x3c25bdb9  A3      : 0x3c25cd00  A4      : 0x3c20e598 A5      : 0xffffff30
A6      : 0x3c253a34  A7      : 0x3c1f9b58  A8      : 0x80379a79 A9      : 0x3fced680
A10     : 0x0000656c  A11     : 0x00000001  A12     : 0x00000000 A13     : 0x3c25bdb8
A14     : 0x3c25bdb4  A15     : 0x00000003  SAR     : 0x00000010 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00006570  LBEG    : 0x4038bd78  LEND    : 0x4038bd82 LCOUNT  : 0x00000000
0x4038bd78: uart_ll_write_txfifo at /home/schlosser_new/lvgl_micropython/lib/esp-idf/components/hal/esp32s3/include/hal/uart_ll.h:263
 (inlined by) uart_hal_write_txfifo at /home/schlosser_new/lvgl_micropython/lib/esp-idf/components/hal/uart_hal_iram.c:40

0x4038bd82: uart_hal_write_txfifo at /home/schlosser_new/lvgl_micropython/lib/esp-idf/components/hal/uart_hal_iram.c:41

Backtrace: 0x40379a76:0x3fced6a0 0x420c093c:0x3fced740 0x420c7759:0x3fced770 0x420c781d:0x3fced790 0x420c63f8:0x3fced7b0 0x420c622d:0x3fced810 0x420c7759:0x3fced830 0x420c781d:0x3fced850 0x4037a7e1:0x3fced870 0x420c093c:0x3fced910 0x420c7759:0x3fced940 0x420c77c4:0x3fced960 0x420e6435:0x3fced9a0 0x420e6762:0x3fced9d0 0x420c7899:0x3fcedac0 0x4037aa7e:0x3fcedb00 0x420c093c:0x3fcedba0 0x420c7759:0x3fcedbf0 0x420c776e:0x3fcedc10 0x420d3d3a:0x3fcedc30 0x420d407c:0x3fcedcc0 0x420db77c:0x3fcedd00
0x40379a76: mp_execute_bytecode at /home/schlosser_new/lvgl_micropython/lib/micropython/py/vm.c:407 (discriminator 3)

0x420c093c: fun_bc_call at /home/schlosser_new/lvgl_micropython/lib/micropython/py/objfun.c:273

0x420c7759: mp_call_function_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:713

0x420c781d: mp_call_method_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:729

0x420c63f8: mp_obj_instance_make_new at /home/schlosser_new/lvgl_micropython/lib/micropython/py/objtype.c:349

0x420c622d: type_call at /home/schlosser_new/lvgl_micropython/lib/micropython/py/objtype.c:999

0x420c7759: mp_call_function_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:713

0x420c781d: mp_call_method_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:729

0x4037a7e1: mp_execute_bytecode at /home/schlosser_new/lvgl_micropython/lib/micropython/py/vm.c:1042

0x420c093c: fun_bc_call at /home/schlosser_new/lvgl_micropython/lib/micropython/py/objfun.c:273

0x420c7759: mp_call_function_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:713

0x420c77c4: mp_call_function_0 at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:687
 (inlined by) mp_parse_compile_execute at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:1620

0x420e6435: do_load_from_lexer at /home/schlosser_new/lvgl_micropython/lib/micropython/py/builtinimport.c:162
 (inlined by) do_load at /home/schlosser_new/lvgl_micropython/lib/micropython/py/builtinimport.c:258

0x420e6762: process_import_at_level at /home/schlosser_new/lvgl_micropython/lib/micropython/py/builtinimport.c:512
 (inlined by) mp_builtin___import___default at /home/schlosser_new/lvgl_micropython/lib/micropython/py/builtinimport.c:607

0x420c7899: mp_import_name at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:1525

0x4037aa7e: mp_execute_bytecode at /home/schlosser_new/lvgl_micropython/lib/micropython/py/vm.c:1235 (discriminator 3)

0x420c093c: fun_bc_call at /home/schlosser_new/lvgl_micropython/lib/micropython/py/objfun.c:273

0x420c7759: mp_call_function_n_kw at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:713

0x420c776e: mp_call_function_0 at /home/schlosser_new/lvgl_micropython/lib/micropython/py/runtime.c:687

0x420d3d3a: parse_compile_execute at /home/schlosser_new/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:123

0x420d407c: pyexec_friendly_repl at /home/schlosser_new/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:675

0x420db77c: mp_task at /home/schlosser_new/lvgl_micropython/lib/micropython/ports/esp32/main.c:159

ELF file SHA256: cac8663cce55c127

Rebooting... 
```

kdschlosser, 2024-02-23:
I injured my back and I have not been spending much time at the computer due to not being able to sit for more then a few minutes at a time. I am really sorry for the delay in this and I do want to get it fixed. IDK how long this healing process is going to be, I might have a herniated disc, have to get an MRI to know for sure. if it is a herniated disc then I will be looking at back surgery due to it causing me so much pain and being so debilitating. The funny thing is it's not my back that hurts, it's my leg that does. When I sit it lights up my leg something bad.

martyy665, 2024-02-25:
Certainly, I understand that you may have encountered some health challenges. I sincerely hope for a prompt resolution to any issues you're facing, if at all possible.

kdschlosser, 2024-02-25:
The meds are helping but not 100%. it makes the pain tolerable so it is not as debilitating. It is going to take a bit to get me closer to being pain free due to medication adjustments. Ultimately I am probably going to need back surgery to get this fixed.

straga, 2024-02-26:
Try check about Myofascial syndrome. May you need start do special exercise robber role. I do that around half year and that  help me.

kdschlosser, 2024-02-26:
I am pretty sure it is a disc issue in the sacrum from a fall I had 20 years ago. going to have an MRI done tomorrow of the lumbar and sacrum to see what is going on with the spine. Hopefully that will reveal what is happening. If it doesn't then that means it is a muscular problem that is pinching the nerve. I am crossing my fingers that it is the latter because that would mean no surgery would be involved. keeping my fingers crossed.

kdschlosser, 2024-03-24:
I believe this should now be fixed. If you want to check and see that would be great.

martytriska, 2024-03-30:
Unfortunately, there is no significant progress with current version. On this line of code (based on the example):
```
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_HIGH,
    power_pin=_POWER_PIN,
    power_on_state=st7789.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7789.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y
)
```
Every time, one of these two situations occurs:

1. The program hangs on an unhandled exception. Traceback is almost the same as described above.
2. The program "stuck" briefly, leading to the watchdog reset.

Result for case No. 1:

```
START
if not lv.is_initialized
DONE
databus is not None
self._disp_drv = lv.display_create
DONE
self._disp_drv.set_color_format
DONE
self._disp_drv.set_driver_data
DONE
framebuffer is None
allocating buffers
2056
DONE
checking framebuffer 2
data_bus.init
DONEfull_screen_siz
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037994f  PS      : 0x00060030  A0      : 0x820c2197  A1      : 0x3fced6a0  
A2      : 0x3c25b7cf  A3      : 0x3c25c700  A4      : 0x3c20ecfc  A5      : 0xffffff30  
A6      : 0x3c253668  A7      : 0x10101010  A8      : 0x3c253650  A9      : 0x10101048  
A10     : 0x3c224874  A11     : 0x3c25b7ce  A12     : 0x00000000  A13     : 0x3c253668  
A14     : 0x3c25b7c8  A15     : 0x00000002  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x10101048  LBEG    : 0x4038bd78  LEND    : 0x4038bd82  LCOUNT  : 0x00000000  

Backtrace: 0x4037994c:0x3fced6a0 0x420c2194:0x3fced740 0x420c8fb1:0x3fced770 0x420c9075:0x3fced790 0x420c7c50:0x3fced7b0 0x420c7a85:0x3fced810 0x420c8fb1:0x3fced830 0x420c9075:0x3fced850 0x4037a7e1:0x3fced870 0x420c2194:0x3fced910 0x420c8fb1:0x3fced940 0x420c901c:0x3fced960 0x420e7c8d:0x3fced9a0 0x420e7fba:0x3fced9d0 0x420c90f1:0x3fcedac0 0x4037aa7e:0x3fcedb00 0x420c2194:0x3fcedba0 0x420c8fb1:0x3fcedbf0 0x420c8fc6:0x3fcedc10 0x420d5592:0x3fcedc30 0x420d58d4:0x3fcedcc0 0x420dcfd4:0x3fcedd00
```
Result for case No. 2:
```
START                                                                                                      
if not lv.is_initialized                                                                                   
DONE                                                                                                       
databus is not None                                                                                        
self._disp_drv = lv.display_create                                                                         
DONE                                                                                                       
self._disp_drv.set_color_format                                                                            
DONE                                                                                                                                                                                                                   self._disp_drv.set_driver_data                                                                                                                                                                                         DONE                                                                                                                                                                                                                   
framebuffer is None                                                                                                                                                                                                    
allocating buffers                                                                                         
2056                                                                                                       
DONE                                                 
checking framebuffer 2                               
data_bus.init                                        
ESP-ROM:esp32s3-20210327                             
Build:Mar 27 2021                                    
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)                                                    
Saved PC:0x420b0ec7                                                                                        
SPIWP:0xee                                           
mode:DIO, clock div:1                                                                                      
load:0x3fce3810,len:0xf8c                            
load:0x403c9700,len:0xb3c                                                                                  
load:0x403cc700,len:0x2dd4                           
entry 0x403c989c                                     
W (29) boot.esp32s3: PRO CPU has been reset by WDT.  
W (29) boot.esp32s3: APP CPU has been reset by WDT.                                                        
MicroPython v1.22.1-dirty on 2024-03-30; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3 
```

martyy665, 2024-03-30:
Unfortunately, there is no significant progress with current version. On this line of code (based on the example):
```
display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_HIGH,
    power_pin=_POWER_PIN,
    power_on_state=st7789.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7789.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y
)
```
Every time, one of these two situations occurs:

1. The program hangs on an unhandled exception. Traceback is almost the same as described above.
2. The program "stuck" briefly, leading to the watchdog reset.

Result for case No. 1:

```
START
if not lv.is_initialized
DONE
databus is not None
self._disp_drv = lv.display_create
DONE
self._disp_drv.set_color_format
DONE
self._disp_drv.set_driver_data
DONE
framebuffer is None
allocating buffers
2056
DONE
checking framebuffer 2
data_bus.init
DONEfull_screen_siz
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037994f  PS      : 0x00060030  A0      : 0x820c2197  A1      : 0x3fced6a0  
A2      : 0x3c25b7cf  A3      : 0x3c25c700  A4      : 0x3c20ecfc  A5      : 0xffffff30  
A6      : 0x3c253668  A7      : 0x10101010  A8      : 0x3c253650  A9      : 0x10101048  
A10     : 0x3c224874  A11     : 0x3c25b7ce  A12     : 0x00000000  A13     : 0x3c253668  
A14     : 0x3c25b7c8  A15     : 0x00000002  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x10101048  LBEG    : 0x4038bd78  LEND    : 0x4038bd82  LCOUNT  : 0x00000000  

Backtrace: 0x4037994c:0x3fced6a0 0x420c2194:0x3fced740 0x420c8fb1:0x3fced770 0x420c9075:0x3fced790 0x420c7c50:0x3fced7b0 0x420c7a85:0x3fced810 0x420c8fb1:0x3fced830 0x420c9075:0x3fced850 0x4037a7e1:0x3fced870 0x420c2194:0x3fced910 0x420c8fb1:0x3fced940 0x420c901c:0x3fced960 0x420e7c8d:0x3fced9a0 0x420e7fba:0x3fced9d0 0x420c90f1:0x3fcedac0 0x4037aa7e:0x3fcedb00 0x420c2194:0x3fcedba0 0x420c8fb1:0x3fcedbf0 0x420c8fc6:0x3fcedc10 0x420d5592:0x3fcedc30 0x420d58d4:0x3fcedcc0 0x420dcfd4:0x3fcedd00
```
Result for case No. 2:
```
START                                                                                                      
if not lv.is_initialized                                                                                   
DONE                                                                                                       
databus is not None                                                                                        
self._disp_drv = lv.display_create                                                                         
DONE                                                                                                       
self._disp_drv.set_color_format                                                                            
DONE                                                                                                                                                                                                                   self._disp_drv.set_driver_data                                                                                                                                                                                         DONE                                                                                                                                                                                                                   
framebuffer is None                                                                                                                                                                                                    
allocating buffers                                                                                         
2056                                                                                                       
DONE                                                 
checking framebuffer 2                               
data_bus.init                                        
ESP-ROM:esp32s3-20210327                             
Build:Mar 27 2021                                    
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)                                                    
Saved PC:0x420b0ec7                                                                                        
SPIWP:0xee                                           
mode:DIO, clock div:1                                                                                      
load:0x3fce3810,len:0xf8c                            
load:0x403c9700,len:0xb3c                                                                                  
load:0x403cc700,len:0x2dd4                           
entry 0x403c989c                                     
W (29) boot.esp32s3: PRO CPU has been reset by WDT.  
W (29) boot.esp32s3: APP CPU has been reset by WDT.                                                        
MicroPython v1.22.1-dirty on 2024-03-30; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3 
```

kdschlosser, 2024-03-30:
what pins are you using?

kdschlosser, 2024-03-30:
I need to see all of the code you are using to determine what the actual problem is.

martyy665, 2024-03-30:
Init sequence is following:
```
from micropython import const

_WIDTH = const(240)
_HEIGHT = const(320)

import lcd_bus  
_DC_PIN = const(16)
_MOSI_PIN = const(6)
_MISO_PIN = const(7)
_SCLK_PIN = const(15)
_CS_PIN = const(17)
_FREQ = const(80000000)
_WP_PIN = const(-1)
_HD_PIN = const(-1)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    host=1,
    sclk=_SCLK_PIN,
    freq=_FREQ,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    cs=_CS_PIN,
    wp=_WP_PIN,
    hd=_HD_PIN,
    quad_spi=False,
    tx_only=False,
    cmd_bits=8,
    param_bits=8,
    dc_low_on_data=False,
    sio_mode=False,
    lsb_first=False,
    cs_high_active=False,
    spi_mode=0
)
import st7789  # NOQA

_RESET_PIN = const(42)
_POWER_PIN = None
_BACKLIGHT_PIN = const(18)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7789.STATE_HIGH,
    power_pin=_POWER_PIN,
    power_on_state=st7789.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7789.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

```

kdschlosser, 2024-03-30:
set your host to 2.

martyy665, 2024-03-30:
Thanks for the advice, but this does not work. On the ESP32S3, the SPI3 periphery does not have IOMUX_PINs defined as it supports only GPIO matrix mapping. Therefore, it is eliminated in lcd_bus/spi_bus.c and cannot be used on this platform (is this really the desired behaviour?)

```        
...
} else if (host == 2) {                                                                                                                                                                                          │ext_mod/lcd_bus/esp32_src/spi_bus.c:        #ifdef SPI2_IOMUX_PIN_NUM_MISO
            #ifdef SPI3_IOMUX_PIN_NUM_MISO                                                                                                                                                                               │ext_mod/lcd_bus/esp32_src/spi_bus.c:                (((args[ARG_tx_only].u_bool) && (miso == -1)) || ((!args[ARG_tx_only].u_bool) && (miso == SPI2_IOMUX_PIN_NUM_MISO))) &&
                sclk = SPI3_IOMUX_PIN_NUM_CLK;                                                                                                                                                                           │lib/esp-idf/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c:#define PIN_MISO SPI2_IOMUX_PIN_NUM_MISO
                mosi = SPI3_IOMUX_PIN_NUM_MOSI;                                                                                                                                                                          │lib/esp-idf/components/soc/esp32c3/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO 2
                                                                                                                                                                                                                         │lib/esp-idf/components/soc/esp32c3/spi_periph.c:        .spiq_iomux_pin = SPI2_IOMUX_PIN_NUM_MISO,
                if (!args[ARG_tx_only].u_bool) {                                                                                                                                                                         │lib/esp-idf/components/soc/esp32h2/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO 2
                    miso = SPI3_IOMUX_PIN_NUM_MISO;                                                                                                                                                                      │lib/esp-idf/components/soc/esp32h2/spi_periph.c:        .spiq_iomux_pin = SPI2_IOMUX_PIN_NUM_MISO,
                }                                                                                                                                                                                                        │lib/esp-idf/components/soc/esp32c2/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO 2
                if (cs == -1) {                                                                                                                                                                                          │lib/esp-idf/components/soc/esp32c2/spi_periph.c:        .spiq_iomux_pin = SPI2_IOMUX_PIN_NUM_MISO,
                    cs = SPI3_IOMUX_PIN_NUM_CS;                                                                                                                                                                          │lib/esp-idf/components/soc/esp32s2/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO 13
                }                                                                                                                                                                                                        │lib/esp-idf/components/soc/esp32s2/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO_OCT 37
                if (args[ARG_quad_spi].u_bool) {                                                                                                                                                                         │lib/esp-idf/components/soc/esp32s2/include/soc/spi_pins.h:#define FSPI_IOMUX_PIN_NUM_MISO SPI2_IOMUX_PIN_NUM_MISO
                    if (wp == -1) {                                                                                                                                                                                      │lib/esp-idf/components/soc/esp32/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO HSPI_IOMUX_PIN_NUM_MISO
                        wp = SPI3_IOMUX_PIN_NUM_WP;                                                                                                                                                                      │lib/esp-idf/components/soc/esp32s3/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO 13
                    }                                                                                                                                                                                                    │lib/esp-idf/components/soc/esp32s3/include/soc/spi_pins.h:#define SPI2_IOMUX_PIN_NUM_MISO_OCT 37
                    if (hd == -1) {                                                                                                                                                                                      │lib/esp-idf/components/soc/esp32s3/spi_periph.c:        .spiq_iomux_pin = SPI2_IOMUX_PIN_NUM_MISO,
[... cut ...]   │lib/esp-idf/components/driver/test/include/test/test_common_spi.h:#define PIN_NUM_MISO            SPI2_IOMUX_PIN_NUM_MISO
                mp_raise_msg(&mp_type_ValueError, MP_ERROR_TEXT("host 2 is not available for this board"));                                                                                                              │lib/esp-idf/components/driver/test/include/test/test_common_spi.h:#define MASTER_IOMUX_PIN_MISO   SPI2_IOMUX_PIN_NUM_MISO
            #endif
...
```

[image]

martyy665, 2024-04-02:
What is the real reason SPI3 (host=2) cannot be used? Routing signals via the GPIO matrix on ESP32-S3 should not impact maximal bus speed which is still 80 MHz (on some older ESPs, it was limited to 40 MHz with a bit of latency added, but it's not this case).

Moreover, has anybody SPI-driven LCD working smoothly with in-compiled bus drivers using the SPI2 (host=1)?

kdschlosser, 2024-04-02:
Try compiling it again. I have removed a large chunk of code from the driver. see if that fixes your issue.

And to my understanding is that SPI0 and SPI1 are typically used for the SPI flash memory and the SPI RAM which is the reason why I recommended using SPI2

This is directly from the datasheet for the ESP32-S3 and the reason why I suggested the host get set to 2

> ESP32-S3 has the following SPI interfaces:
> • SPI0 used by ESP32-S3’s GDMA controller and cache to access in-package or off-package flash/PSRAM
> • SPI1 used by the CPU to access in-package or off-package flash/PSRAM
> • SPI2 is a general purpose SPI controller with access to a DMA channel allocated by the GDMA controller
> • SPI3 is a general purpose SPI controller with access to a DMA channel allocated by the GDMA controller

https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf#subsubsection.3.5.2

kdschlosser, 2024-04-02:
also if you look here it says that SPI0 and SPI1 are locked to specific pins where as SPI2 and SPI3 can be used on any GPIO pin.

https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf#page.53

martyy665, 2024-04-02:
Fine, I try to recompile it - the source seems to be simplified very massively ;).

kdschlosser, 2024-05-09:
This problem should be solved so I am closing the issue.

closed 2024-05-09 by kdschlosser