lvgl-micropython/lvgl_micropython #8: lvgl trying to allocate 1 gigabyte of memory

mirkokral, 2024-03-01:
Hello, i get this error:
```
MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 50, in <module>
  File "display_driver_framework.py", line 178, in __init__
MemoryError: memory allocation failed, allocating 1070255496 bytes
```
When running this code:
```python
import sys
sys.path.append('.')
import lcd_bus
import st7789

import machine

display_bus = lcd_bus.SPIBus(
    dc=11,
    host=1,
    sclk=40,
    freq=80000000,
    mosi=41,
    miso=38,
    cs=12,
    quad_spi=False,
    tx_only=False,
    cmd_bits=8,
    param_bits=8,
    dc_low_on_data=False,
    sio_mode=False,
    lsb_first=False,
    cs_high_active=False,
    spi_mode=0
)

perion = machine.Pin(10, machine.Pin.OUT)
perion.value(1)

if 0:
    # with DMA, the repaints seem to be too slow? To be investigated
    # we seem to be fine performance-wise without DMA with 320x240 anyway
    import rp2_dma
    rp2_dma=rp2_dma.DMA(0)
else: rp2_dma=None

import lvgl as lv
lv.init()

lcd=st7789.ST7789(
    data_bus=display_bus,
    display_width=240,
    display_height=320,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    backlight_pin=42,
    backlight_on_state=st7789.STATE_HIGH,
    offset_x=0,
    offset_y=0
)
display.init()
scr=lv.obj()
btn=lv.button(scr)
lbl=lv.label(btn)
lbl.set_text("Press me!")
btn.center()
btn.add_event(lambda event: print('Button clicked!'),lv.EVENT.CLICKED,None)
lv.screen_load(scr)
```

mirkokral, 2024-03-01:
For convenience:
This is the exact line at the file at the end of the traceback: 
```python
178 |            self._disp_drv = lv.display_create(display_width, display_height)
```

kdschlosser, 2024-03-24:
This should now be fixed, can you please test and let me know.

acowls, 2024-04-10:
@mirkokral - wondering if you can help as I am trying to build lvgl on a RP2020 PICO W board. Is that your target board ? - I ask since you had a reference to rp2_dma in the code snippet. 

I get this error:
```
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
NotImplementedError: LCD SPI but is not available for this MCU
```

code :
```
import sys
sys.path.append('.')
import lcd_bus

import machine

display_bus = lcd_bus.SPIBus(
    dc=11,
    host=1,
    sclk=40,
    freq=80000000,
    mosi=41,
    miso=38,
    cs=12,
    quad_spi=False,
    tx_only=False,
    cmd_bits=8,
    param_bits=8,
    dc_low_on_data=False,
    sio_mode=False,
    lsb_first=False,
    cs_high_active=False,
    spi_mode=0
)
```

lvgl_micropython build from master on 2024-04-09 
```
MicroPython v1.22.1-dirty on 2024-04-09; Raspberry Pi Pico W with RP2040
```

kdschlosser, 2024-04-11:
I have not hammered out getting it to compile for the PI boards yet.

kdschlosser, 2024-04-11:
OK those problems are fixed m8

kdschlosser, 2024-05-09:
This problem should be solved so I am closing the issue.

closed 2024-05-09 by kdschlosser