lvgl-micropython/lvgl_micropython #10: 'SPI_Bus' object has no attribute 'MAXIMUM_BUFFER_SIZE'

cheops, 2024-03-08:
when defining the buffers in python code and giving them as argument to the descriptor gives an error

```
import lcd_bus
import st7789
from micropython import const

_MOSI = const(6)
_MISO = const(8)
_SCLK = const(7)
_CS = const(5)
_DC = const(4)
_RESET = const(48)
_FREQ = const(25_000_000)

_WIDTH = const(296)
_HEIGHT = const(240)

gc.collect()
buf1 = bytearray(_WIDTH*_HEIGHT*2)
COLOR_FORMAT_RGB565 = const(18)

display=st7789.ST7789(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET,
    frame_buffer1=buf1,
    frame_buffer2=None,
    color_byte_order=st7789.BYTE_ORDER_BGR,
    color_space=COLOR_FORMAT_RGB565
)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)

```
gives the following error
```
  File "display_driver_framework.py", line 242, in __init__
AttributeError: 'SPI_Bus' object has no attribute 'MAXIMUM_BUFFER_SIZE'
```

kdschlosser, 2024-03-24:
This should be fixed, can you please test and let me know.

cheops, 2024-03-24:
tested and no more error!

closed 2024-03-24 by cheops

kdschlosser, 2024-03-24:
If you want to see some speed. this is what you need to do.

```python

import lcd_bus
from micropython import const

_MOSI = const(6)
_MISO = const(8)
_SCLK = const(7)
_CS = const(5)
_DC = const(4)
_RESET = const(48)
_FREQ = const(25_000_000)

_WIDTH = const(296)
_HEIGHT = const(240)

import lcd_bus

bus = lcd_bus.SPIBus(
    dc=_DC,
    host=1,
    mosi=_MOSI,
    miso=_MISO,
    sclk=_SCLK,
    cs=_CS,
    freq=_FREQ
)

buf_size = int(_WIDTH *_HEIGHT * 2 / 10)
buf1 = bus.allocate_framebuffer(buf_size, lcd_bus.MEMORY_INTERNAL |  lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(buf_size, lcd_bus.MEMORY_INTERNAL |  lcd_bus.MEMORY_DMA)

import st7789
import lvgl as lv

display=st7789.ST7789(
    data_bus=bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

That will more then double the speed of the display redrawing.