lvgl-micropython/lvgl_micropython #12: set_buffers has been changed to set_draw_buffers

yusuhua, 2024-03-12:
https://github.com/kdschlosser/lvgl_micropython/blob/main/driver/frozen/display/display_driver_framework.py
There are two set_buffers here, lvgl has been changed to set_draw_buffers.

See: https://github.com/lvgl/lvgl/blob/cb800cb793aa9062ca980cbabc61dc59306912f9/src/display/lv_display.h

yusuhua, 2024-03-12:
I saw similar issues discussed:
https://github.com/kdschlosser/lvgl_micropython/pull/9

kdschlosser, 2024-03-12:
I have updated the LVGL submodule so this problem should be gone.

kdschlosser, 2024-03-12:
`set_draw_buffers` is only used for setting lvgl buffers not raw buffers. we are dealing with raw buffers so the current code is correct. I forgot to update the lvgl submodule.

kdschlosser, 2024-03-13:
OK good.