lvgl-micropython/lvgl_micropython #14: please open the discussion section on the repo

fabse-hack, 2024-03-15:
Hello,

first: NICE WORK!!!!!!!!!!! THANK YOUUUU!!

second: i'm still looking all few days to you work, and looking what you are doing. I'm beginner and not understand everything in the complexy of lvgl + micropython.
Question: **can you open the discussion part of the repo here? ** i think that would be help. And maybe in the future we need some youtube tutorials / examples / projects :)

Greetings

fabse-hack, 2024-03-15:
ah, i have an idea: Is possibe to make a cbuild with the github actions? ... normally can use a trigger, if a new micropython version is published, then it will burn the lvgl for it?

kdschlosser, 2024-03-15:
OK it's opened

kdschlosser, 2024-03-15:
I can set up github actions but the problem is the CI is slow and I would not be able to compile for all of the MCU's that are supported in the allotted time. 

Also triggering with new releases of MicroPython is not a good idea because the API is always changing in MicroPython. Dependencies are always changing as well. I know they are working on getting the ESP32 port to work with ESPIDF 5.2+.. So I would also need to update the submodule dependencies and also change some of the build code in order for it to work. So it is best to make a manual change. 

You can choose whatever MicroPython version you want by using the following in your shell
```shell
cd lib/micropython
git checkout {release tag name}
git pull
```

where `"{release tag name}"` is the literal name that is set for a release. This is typically the version prefixed with a "v"

kdschlosser, 2024-03-15:
I could trigger a build for all of the MCU's that would use my computer that is here with me. It would be able to run the compilations on all 175 of the MCU's in ~60 minutes. I am not sure that I would be able to store that many artifacts though. That would be close to a 1/2 a gig worth of data. about 2.5mb per MCU

fabse-hack, 2024-03-15:
hmm yes, and i'm not understand 100% the submodules and the mpy_cross. This are not taken into this builds (because there are needed individually?) .... 

and a tip: the micropython discord server have an #display channel too, i always ask matty there of the status of lvgl :)

kdschlosser, 2024-03-15:
You have to read the readme file. It tell you hot to build

It is easier to build than using MicroPython's build system. The one that is included with this repo takes care of almost everything for you. You have to install things like gcc and build-essentials but after that it's a single command you type in and you are off and running. It takes care of everything else. It will even adjust the partition size if you are compiling for ESP32. It will make the partition the firmware gets put onto as small as possible to fit the firmware. 

Best bet is to go over that readme file. It will explain everything. If you are not sure on something ask away.

kdschlosser, 2024-03-24:
This is finished

closed 2024-03-24 by kdschlosser