lvgl-micropython/lvgl_micropython #16: How to drive the touch screen whose control chips are ili9341 and xpt2046?

kk8900, 2024-03-17:
How to drive the touch screen whose control chips are ili9341 and xpt2046?
I can't import the corresponding library normally with the following code:
```
from espidf import VSPI_HOST
from ili9XXX import ili9341
from xpt2046 import xpt2046
```

kdschlosser, 2024-03-17:
I have not yet written a driver for the xpt2046 touch.

kdschlosser, 2024-03-17:
I just added the driver for the xpt2046.

When using this driver if it shares the same SPI as the display then you must set all the pin numbers the same as the bus driver and you MUST also set the host to be the same number as well. what the host number is going to be will depend on the MCU you are using.

For the ESP32 I suggest using host 2. the host numbers that are lower than that can be used for RAM and for the flash. The one thing that is going to be different between the 2 is going to be the CS pin. All pins MUST be connected in order for the touch driver to work.. mios, mosi, clk and cs

kdschlosser, 2024-03-24:
I did add the driver for the touch display. can you please test and let me know

kk8900, 2024-03-25:
> I did add the driver for the touch display. can you please test and let me know

Sorry, my firmware was compiled successfully, but I haven't had time to test it yet. I will test and reply you as soon as possible.

kdschlosser, 2024-03-25:
no worries take your time. Just didn't want to forget this is an open issue.

kk8900, 2024-03-27:
Maybe I'm not familiar with micropython, and my test failed. The following is our test code:
`import lcd_bus
import ili9341
import xpt2046
import lvgl as lv

bus = lcd_bus.SPIBus(
    dc=13,
    sclk=18,
    mosi=17,
    miso=16,
    cs=14,
    host=1,
    freq=5000000,
)

display = ili9341.ILI9341(
    data_bus=bus,
    display_width=240,
    display_height=320,
    reset_pin=15,
    #power_pin=-1,
    backlight_pin=4,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(1)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.init()
display.set_backlight(100)

`
Prompt the following error, how to solve it?
`Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
  File "display_driver_framework.py", line 178, in __init__
MemoryError: memory allocation failed, allocating 1107657908 bytes`

kdschlosser, 2024-03-27:
Lets see if allocating the frame buffers externally allow it to run properly.

```python
# it is best to stagger the imports like I have to minimize memory use before 
# you allocate the frame buffer. Nor things loaded = more memory used and you 
# may not have enough to allocate the frame buffers. I am using 2 frame buffers
# because this is going to give you the best performance. That is if you are 
# using an ESP2 I didn't check. If you are not using an ESP32 then you can
# change buf2 to None

import lcd_bus

bus = lcd_bus.SPIBus(
    dc=13,
    sclk=18,
    mosi=17,
    miso=16,
    cs=14,
    host=1,
    freq=80000000,
)

buf1 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341  # NOQA
import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=bus,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    display_width=240,
    display_height=320,
    reset_pin=15,
    power_pin=None,
    backlight_pin=4,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(1)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

import xpt2046  # NOQA

touch = xpt2046.XPT2046(
    miso=16,
    mosi=17,
    clk=18,
    cs=0,  # THIS NEEDS TO BE CHANGED
    # if the same bus is used for the display you MUST set the same 
    # host as what was used for the display
    host=1,
    freq=5000000,
    touch_cal=None
)
```

kk8900, 2024-03-28:
> Lets see if allocating the frame buffers externally allow it to run properly.
>
> ```python
> # it is best to stagger the imports like I have to minimize memory use before
> # you allocate the frame [...]

When I run the above code, ESP32S3 restarts and the screen is not lit.
Could it be the problem of ili9341.py?

ancebfer, 2024-04-05:
I'm also trying to drive a ili9341 (without touch screen) with an ESP32S3 without success.

As I'm using a ESP32-S3-WROOM-2-N32R8V modules, I have modified "ESP32_GENERIC_S3/sdkconfig.board" file:

```
CONFIG_ESPTOOLPY_OCT_FLASH=y
CONFIG_ESPTOOLPY_FLASHMODE_OPI=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y

CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE_32MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-32MiB.csv"
```

I have generated the image with:

`python3 make.py esp32 mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=ili9341 LV_CFLAGS="-DLV_COLOR_DEPTH=16"`

And I'm using this example based on code taken from other responses:

```
import lcd_bus

bus = lcd_bus.SPIBus(
    dc=7,
    sclk=5,
    mosi=6,
    miso=4,
    cs=16,
    host=1,
    freq=40000000
)

buf1 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341
import lvgl as lv

display = ili9341.ILI9341(
    data_bus=bus,
    display_width=240,
    display_height=320,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=15,
    power_pin=None,
    backlight_pin=None,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(1)
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.init()
display.set_backlight(100)

btn = lv.button(lv.screen_active())
btn.set_size(100, 100)
lbl = lv.label(btn)
lbl.set_text("Press me, I'm a button")
btn.center()

label2 = lv.label(lv.screen_active())
label2.set_long_mode(lv.label.LONG.SCROLL_CIRCULAR) # Circular scroll
label2.set_width(150)
label2.set_text("It is a circularly scrolling text. ")
label2.align(lv.ALIGN.CENTER, 0, 40)

import time

while True:
    time.sleep_ms(1)
    lv.tick_inc(1)
    lv.task_handler()
```

The example runs without errors but nothing is displayed in the LCD.

There are no hardware related issues because example:

https://github.com/rdagger/micropython-ili9341/blob/master/demo_bouncing_boxes.py

Runs right adapting LCD pinout:

```
        spi = SPI(1, baudrate=40000000, sck=Pin(5), mosi=Pin(6))
        display = Display(spi, dc=Pin(7), cs=Pin(16), rst=Pin(15), width=240, height=320, rotation=270)
```

Any idea about how to make it work?

kk8900, 2024-04-26:
> Lets see if allocating the frame buffers externally allow it to run properly.
>
> ```python
> # it is best to stagger the imports like I have to minimize memory use before
> # you allocate the frame [...]
Hello, I tried to drive ST7789 with the following code, but it failed. Can you help me see what the problem is?
‘’‘
import lcd_bus

bus = lcd_bus.SPIBus(
    dc=6,
    sclk=3,
    mosi=5,
    miso=-1,
    cs=4,
    host=1,
    freq=80000000,
)

buf1 = bus.allocate_framebuffer(int(204 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(int(204 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7789  # NOQA
import lvgl as lv  # NOQA

display = st7789.ST7789(
    data_bus=bus,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    display_width=204,
    display_height=320,
    reset_pin=None ,
    power_pin=None,
    backlight_pin=2,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(1)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)
’‘’
The error message is:
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
  File "display_driver_framework.py", line 240, in __init__
ValueError: 259(spi_bus_initialize)

ste7anste7an, 2024-05-05:
I use an ili9341 SPI display with xpt2046 touch controller, connected to the same SPI (with seperate CS, of course). I have an ESP32-WROVER with 4MByte of PSRAM.

I compiled my firmware based on the latest repo using this command:

```
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046 LV_CFLAGS="-DLV_COLOR_DEPTH=16"
```

Then I used a program similar to what is suggested in this issue, with my specific PINs (no power nor backlight Pin). I commented the touch for the time being and rfeduced the SPI clock to 10MHz (also with 80Mhz I get the same error).

```
import lcd_bus
_="""disp = ili9341(miso=12, mosi=13, clk=14, cs=15, dc=23, rst=25,
               backlight=-1,power=-1,
               width=320, height=240, rot=LANDSCAPE)
# use same SPI as display, init touch
touch = xpt2046(spihost=esp.HSPI_HOST,cs=26,transpose=False,
                cal_x0=3865, cal_y0=329, cal_x1=399, cal_y1=3870)

"""
bus = lcd_bus.SPIBus(
    dc=23,
    sclk=14,
    mosi=13,
    miso=12,
    cs=15,
    host=1,
    freq=10000000,
)

buf1 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341
import lvgl as lv

display = ili9341.ILI9341(
    data_bus=bus,
    display_width=240,
    display_height=320,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    reset_pin=25,
    power_pin=None,
    backlight_pin=None,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

#display.set_power(1)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
display.set_backlight(100)

import xpt2046  # NOQA

_="""
touch = xpt2046.XPT2046(
    miso=12,
    mosi=13,
    clk=14,
    cs=26,  # THIS NEEDS TO BE CHANGED
    # if the same bus is used for the display you MUST set the same 
    # host as what was used for the display
    host=2,
    freq=5000000,
    touch_cal=None
)
"""

```

This results in this error:
```
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
  File "display_driver_framework.py", line 240, in __init__
ValueError: 258(esp_lcd_new_panel_io_spi)
```

I can manually configure this SPI bus:
```
>>> from machine import SPI,Pin
hspi = SPI(1, 10000000, sck=Pin(14), mosi=Pin(13), miso=Pin(12))

>>> hspi
SPI(id=1, baudrate=10000000, polarity=0, phase=0, bits=8, firstbit=0, sck=14, mosi=13, miso=12
)
```

Any idea what is going wrong here?

closed 2024-05-14 by kk8900