lvgl-micropython/lvgl_micropython #19: About display driver, pure python or hybrid.

sam0910, 2024-03-23:
Greetings!

Always thanks to @kdschlosser for maintaining this great project.
you are the savior of micropython bindings.

I found a LVGL blog about [Display driver, pure python or hybrid](https://blog.lvgl.io/2019-08-05/micropython-pure-display-driver#pure-micropython-driver)
[lvgl/lv_micropython_biding](https://github.com/lvgl/lv_binding_micropython/blob/dc5bb8fcf183c532c1bfa789e0a4bd364a68c3e3/driver/esp32/ili9XXX.py) is still using C callback for ili9341, ili9488, ili9488g, gc9a01, st7789 for the performance.

I know you are busy on many other great things and also this repo is on going,
Could I ask, is there a plan to use C callback or viper emitter?

Great thanks again.

kdschlosser, 2024-03-23:
The reason why with the current official binding there is a "pure python" or "hybrid" is because of how the official binding deals with flushing the frame buffer and the copying of the frame buffer data. In my version of the binding I have eliminated the need to copy anything. The only thing that can take a bit of time to do is swapping the 2 bytes for each pixel when using RGB565. This is not something that has to be done for all displays and it is an option that can be turned on and off when constructing the display driver. 

I am guessing you are using the ESP32. and when working with the ESP32 the official binding exposes parts of the ESP32 SDK so it can be accessed from MicroPython. There are 2 components when working with a display, you have the chip that controls the display itself and you have the connection method to that chip. That connection method is what we call a "bus" which is short for data bus. The way the MCU connects to the display is going to be a standardized for of communication it will be SPI or i8080 or an RGB connection. These are the methods that the ESP32 is able to use. There are others but they are not supported by the ESP32.

Because the bus is standardized that means it is always go to work the same no matter what brand of display you use. The chip that controls the display is going to be different depending on what display you are using. In the official binding the bus and the code for the control chip are all put together. There is a problem with doing that. A single chip can support more than one connection type. it can support SPI , I8080 and RGB and depending on how it is connected to the ESP32 any of those can be used to communicate with the display.

What I did was I separated the bus from the display chip code. They are 2 separate things. The code for the bus is written in C code where as the code for the display IC is written in Python. The bus is what does the sending of the data and it need to be in C code for performance purposes. I came up with an API for the different busses that allow any display IC driver to work with any bus driver. This is because the function names to do things are the same and the data that gets passed to those functions is also the same. The display IC driver shouldn't need to have any knowledge and it shouldn't even care what bus is being used. That is how I wrote the code.

There is no reason to use the viper code emitter because anything that is written that needs to have speed is written in C code. The only thing the display IC driver does is pass the initial startup commands to the display which is a one shot and done thing so it only happens on startup. The other thing the display driver does is it acts as a go between when data is being passed from LVGL to the display. There is nothing that needs to be done during that process other than sending a memory address to the display before sending the frame buffer data to the display. The code in the display IC driver which is written in python simply passes the commands to the bus driver which is written in C. so once again the speed is at C code speed so no need for using the viper code emitter because it's not going to increase performance at all. It will end up using more memory.

This is the code that is used to handle passing the data from LVGL to the display. I will mark what is happening in the code for you to understand a little better.

```python
    def _flush_cb(self, _, area, color_p):
        # this is a simple math and is not CPU intensive
        # some displays have bezel around them and that bezel might 
        # overlap a little bit of the display. This allows us to shift the data
        # so it is not under the bezel.
        x1 = area.x1 + self._offset_x
        x2 = area.x2 + self._offset_x

        y1 = area.y1 + self._offset_y
        y2 = area.y2 + self._offset_y

        # This is how the original binding was set up and I have not 
        # changed this portion. It will be something that I will change
        # in the future because it is a step that really doesn't need to take place
        # The sole purpose of this is to convert the buffer data into a memory view object
        # so we can pass it to the bus driver to be transmitted
        size = (
            (x2 - x1 + 1) *
            (y2 - y1 + 1) *
            lv.color_format_get_size(self._color_space)
        )
       
       # I separated sending any commands to the display when flushing the frame buffer because 
       # I didn't want to have to repeat code for flushing the frame buffer.  Some displays 
       # you don't have to send any commands to when flushing the frame buffer and that is able to be
       # done without having to rewrite the code in this function. 
       cmd = self._set_memory_location(x1, y1, x2, y2)

        # we have to use the __dereference__ method because this method is
        # what converts from the C_Array object the binding passes into a
        # memoryview object that can be passed to the bus drivers
        data_view = color_p.__dereference__(size)

        # This is the call where we pass the data to the display. The actual code that does this is written in C code.
        self._data_bus.tx_color(cmd, data_view, x1, y1, x2, y2)
```

if this function is put into the viper code emitter the only thing you are going to make faster is the math that is done. There really isn't all that much math being done int he flush function. There is always a cost when [... cut ...] y will need to be updated. Since there is no knowledge of LVGL in the binding I wrote when sending data to the display I will never have to alter the code at all when LVGL gets updated. What that means is when I want to update to a new LVGL version all I have to do is pull the new code in from the LVGL repository and maybe make a few small changes in the display drivers written in Python if there is an API naming change. I don't have to worry about large modifications in order to get it to work.

sam0910, 2024-03-24:
Thank you for the detail reply, 
That is indeed a superior approach, I am amazed that you've done.

Great thanks indeed!

closed 2024-03-24 by sam0910