lvgl-micropython/lvgl_micropython #20: RGBDisplay frame buffer allocation failure on ESP32-S3

bland328, 2024-03-24:
I have a [Sunton ESP32-8048S043 board](https://macsbug.files.wordpress.com/2022/10/4280s043_layout-2.pdf) with ESP32-S3 (16MB QIO flash / 8MB OPI SPIRAM), 800×480 parallel IPS display, and GT911 capacitive touch digitizer.

`lvgl_micropython` is building, installing and fundamentally working, but I'm running into a problem allocating the `RGBDisplay` frame buffer.

Here's how I'm building the firmware under macOS Sonoma:
`python3 make.py esp32 clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=rgb_display INDEV=GT911`

Here's my `main.py`:
```py
import micropython
micropython.mem_info()

import lcd_bus
display_bus = lcd_bus.RGBBus(
    hsync = 39,
    vsync = 41,
    de = 40,
    disp = -1,
    pclk = (12 * 1000 * 1000),
    data0 = 8, data1 = 3, data2 = 46, data3 = 9, data4 = 1,  # B
    data5 = 5, data6 = 6, data7 = 7, data8 = 15, data9 = 16, data10 = 4,  # G
    data11 = 45, data12 = 48, data13 = 47, data14 = 21, data15 = 14,  # R
    hsync_front_porch = 8,
    hsync_back_porch = 8,
    hsync_pulse_width = 4,
    hsync_idle_low = True,
    vsync_front_porch = 8,
    vsync_back_porch = 8,
    vsync_pulse_width = 4,
    vsync_idle_low = True,
    de_idle_high = False,
    pclk_idle_high = False,
    pclk_active_low = True,
    disp_active_low = False,
)

import rgb_display
import lvgl as lv
display = rgb_display.RGBDisplay(
    data_bus = display_bus,
    display_width = 800,
    display_height = 480,
    color_byte_order = rgb_display.BYTE_ORDER_BGR,
    color_space = lv.COLOR_FORMAT.RGB565,
)
```

And here's the resulting output:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0xf8c
load:0x403c9700,len:0xb3c
load:0x403cc700,len:0x2dd4
entry 0x403c989c
stack: 656 out of 15360
GC: total: 64000, used: 3120, free: 60880, max new split: 8257536
 No. of 1-blocks: 20, 2-blocks: 6, max blk sz: 30, max free sz: 3704
E (5974) lcd_panel.rgb: lcd_rgb_panel_alloc_frame_buffers(156): no mem for frame buffer
E (5974) lcd_panel.rgb: esp_lcd_new_rgb_panel(285): alloc frame buffers failed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420bb3ac  PS      : 0x00060030  A0      : 0x820bbf8e  A1      : 0x3fced740
A2      : 0x3fcac9a0  A3      : 0x3c1f7a5c  A4      : 0x3c1f7e3c  A5      : 0x00001756
A6      : 0x3c1f7a5c  A7      : 0x3c1f7f74  A8      : 0x00000000  A9      : 0x7fffffff
A10     : 0x0000005a  A11     : 0x3fcedefc  A12     : 0x3fced720  A13     : 0x0000000c
A14     : 0x3c1f7a5c  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffa

Backtrace: 0x420bb3a9:0x3fced740 0x420bbf8b:0x3fced760 0x420034bd:0x3fced7f0 0x42001cc1:0x3fced850 0x4200164b:0x3fced8a0 0x420c295d:0x3fced8f0 0x420c9719:0x3fced920 0x420c97dd:0x3fced940 0x4037a7e1:0x3fced960 0x420c28fc:0x3fceda00 0x420c9719:0x3fceda30 0x420c97dd:0x3fceda50 0x420c83b8:0x3fceda70 0x420c81ed:0x3fcedad0 0x420c9719:0x3fcedaf0 0x420c97dd:0x3fcedb10 0x4037a7e1:0x3fcedb30 0x420c28fc:0x3fcedbd0 0x420c9719:0x3fcedc00 0x420c972e:0x3fcedc20 0x420d5cfa:0x3fcedc40 0x420d609d:0x3fcedcd0 0x420dd713:0x3fcedd00
```

Any thoughts on what I'm getting wrong, or what I might try next?

kdschlosser, 2024-03-25:
I need to make a change to the code to handle that issue. This is due to how the RGB driver works.

In the meantime here is what you can do.

```python
import micropython
micropython.mem_info()

import lcd_bus
display_bus = lcd_bus.RGBBus(
    hsync = 39,
    vsync = 41,
    de = 40,
    disp = -1,
    pclk = (12 * 1000 * 1000),
    data0 = 8, data1 = 3, data2 = 46, data3 = 9, data4 = 1,  # B
    data5 = 5, data6 = 6, data7 = 7, data8 = 15, data9 = 16, data10 = 4,  # G
    data11 = 45, data12 = 48, data13 = 47, data14 = 21, data15 = 14,  # R
    hsync_front_porch = 8,
    hsync_back_porch = 8,
    hsync_pulse_width = 4,
    hsync_idle_low = True,
    vsync_front_porch = 8,
    vsync_back_porch = 8,
    vsync_pulse_width = 4,
    vsync_idle_low = True,
    de_idle_high = False,
    pclk_idle_high = False,
    pclk_active_low = True,
    disp_active_low = False,
)

buf1 = display_bus.allocate_framebuffer(800 * 480 * 2, lcd_bus.MEMORY_SPIRAM)
buf2 = display_bus.allocate_framebuffer(800 * 480 * 2, lcd_bus.MEMORY_SPIRAM)

import rgb_display
import lvgl as lv
display = rgb_display.RGBDisplay(
    data_bus = display_bus,
    display_width = 800,
    display_height = 480,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    color_byte_order = rgb_display.BYTE_ORDER_BGR,
    color_space = lv.COLOR_FORMAT.RGB565,
)

```

kdschlosser, 2024-03-25:
I updated the code so clone it again and give it a try the problem should be fixed without you needing to manually create the buffers like in my last post.

bland328, 2024-03-25:
Thanks very much for that, @kdschlosser!

Sadly, either way I do it (manually or the newly-updated way), I get this failure immediately after the `rgb_display.RGBDisplay()` call:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420bbe78  PS      : 0x00060030  A0      : 0x8200356c  A1      : 0x3fced760
A2      : 0x3c251b38  A3      : 0x3fcac9a0  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x60009050  A7      : 0x00000000  A8      : 0x820bbe4d  A9      : 0x3c1f5540
A10     : 0x00b71b00  A11     : 0x0000001f  A12     : 0x00000000  A13     : 0x00000000
A14     : 0xffffffc0  A15     : 0x3c3d7b0c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420bbe75:0x3fced760 0x42003569:0x3fced7f0 0x42001ccd:0x3fced850 0x42001657:0x3fced8a0 0x420c2a09:0x3fced8f0 0x420c97c5:0x3fced920 0x420c9889:0x3fced940 0x4037a7e1:0x3fced960 0x420c29a8:0x3fceda00 0x420c97c5:0x3fceda30 0x420c9889:0x3fceda50 0x420c8464:0x3fceda70 0x420c8299:0x3fcedad0 0x420c97c5:0x3fcedaf0 0x420c9889:0x3fcedb10 0x4037a7e1:0x3fcedb30 0x420c29a8:0x3fcedbd0 0x420c97c5:0x3fcedc00 0x420c97da:0x3fcedc20 0x420d5da6:0x3fcedc40 0x420d6149:0x3fcedcd0 0x420dd7bf:0x3fcedd00
```

kdschlosser, 2024-03-25:
Lets decode the backtrace.

```
~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32-elf-addr2line -e /lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.elf 0x420bbe75:0x3fced760 0x42003569:0x3fced7f0 0x42001ccd:0x3fced850 0x42001657:0x3fced8a0 0x420c2a09:0x3fced8f0 0x420c97c5:0x3fced920 0x420c9889:0x3fced940 0x4037a7e1:0x3fced960 0x420c29a8:0x3fceda00 0x420c97c5:0x3fceda30 0x420c9889:0x3fceda50 0x420c8464:0x3fceda70 0x420c8299:0x3fcedad0 0x420c97c5:0x3fcedaf0 0x420c9889:0x3fcedb10 0x4037a7e1:0x3fcedb30 0x420c29a8:0x3fcedbd0 0x420c97c5:0x3fcedc00 0x420c97da:0x3fcedc20 0x420d5da6:0x3fcedc40 0x420d6149:0x3fcedcd0 0x420dd7bf:0x3fcedd00
```

You may have to adjust the path to xtensa-esp32-elf-addr2line. I am not sure where it gets put on MACOS

kdschlosser, 2024-03-25:
download the attached file and unzip it. The upload it to your MCU. When the driver framework loads it will use this one ionstead of the one that is baked into the firmware automatically.

[display_driver_framework.zip](https://github.com/kdschlosser/lvgl_micropython/files/14740005/display_driver_framework.zip)

I added a bunch of print statements to see where it is tanking. The backtrace stuff isn't going to give any insight into location for the python code so this will help locate the function call that is tanking it.

bland328, 2024-03-25:
Here's the output from your alternate `display_driver_framework.zip`:

```
stack: 656 out of 15360
GC: total: 64000, used: 3472, free: 60528, max new split: 8257536
 No. of 1-blocks: 24, 2-blocks: 12, max blk sz: 30, max free sz: 3666
START
if not lv.is_initialized
DONE
databus is not None
self._disp_drv = lv.display_create
DONE
self._disp_drv.set_color_format
DONE
self._disp_drv.set_driver_data
DONE
framebuffer is None
allocating buffers
2056
1032
DONE
checking framebuffer 2
data_bus.init
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420bbe78  PS      : 0x00060030  A0      : 0x8200356c  A1      : 0x3fced760
A2      : 0x3c251b38  A3      : 0x3fcac9a0  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x60009050  A7      : 0x00000000  A8      : 0x820bbe4d  A9      : 0x3c1f5540
A10     : 0x00b71b00  A11     : 0x0000001f  A12     : 0x00000000  A13     : 0x00000000
A14     : 0xffffffc0  A15     : 0x3c3d7b0c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x420bbe75:0x3fced760 0x42003569:0x3fced7f0 0x42001ccd:0x3fced850 0x42001657:0x3fced8a0 0x420c2a09:0x3fced8f0 0x420c97c5:0x3fced920 0x420c9889:0x3fced940 0x4037a7e1:0x3fced960 0x420c29a8:0x3fceda00 0x420c97c5:0x3fceda30 0x420c9889:0x3fceda50 0x420c8464:0x3fceda70 0x420c8299:0x3fcedad0 0x420c97c5:0x3fcedaf0 0x420c9889:0x3fcedb10 0x4037a7e1:0x3fcedb30 0x420c29a8:0x3fcedbd0 0x420c97c5:0x3fcedc00 0x420c97da:0x3fcedc20 0x420d5da6:0x3fcedc40 0x420d6149:0x3fcedcd0 0x420dd7bf:0x3fcedd00
```

bland328, 2024-03-25:
I just realized that this line from my `main.py`:

`pclk = (12 * 1000 * 1000),`

was written under the assumption that `pclk` is a frequency, but now I'm realizing it should likely be the number of the GPIO pin which runs to the `DCLK` line of the display.  Does this sound right?

kdschlosser, 2024-03-25:
Yup I just saw that too.

kdschlosser, 2024-03-25:
I mean JUST now I saw that... LOL..

kdschlosser, 2024-03-25:
pclk you set to 42
and freq you set to 12000000

kdschlosser, 2024-03-25:
Everything else looks to be correct.

kdschlosser, 2024-03-25:
You might have to dink about with the freq setting a bit if your display image is not centered.. You might be able to push it up to a higher clock frequency than 12mhz. you have to test it to find out.

kdschlosser, 2024-03-25:
It will work like it should once you make those changes.

bland328, 2024-03-25:
The crash is gone, and the output with your alternate `display_driver_framework.py` still installed is this:

```
START
if not lv.is_initialized
DONE
databus is not None
self._disp_drv = lv.display_create
DONE
self._disp_drv.set_color_format
DONE
self._disp_drv.set_driver_data
DONE
framebuffer is None
allocating buffers
2056
1032
DONE
checking framebuffer 2
data_bus.init
DONE
self._disp_drv.set_flush_cb
DONE
self._disp_drv.set_buffers
DONE
data_bus.register_callback
INIT DONE
```

I added `display.init()` to the end of `main.py`, hoping I'd at least see the backlight come on, but no luck so far.

It's very late for me here, so I'll tackle this again after I've slept.  Thanks very much for your help to this point!

kdschlosser, 2024-03-25:
display = rgb_display.RGBDisplay(
    data_bus = display_bus,
    display_width = 800,
    display_height = 480,
    color_byte_order = rgb_display.BYTE_ORDER_BGR,
    color_space = lv.COLOR_FORMAT.RGB565,
)

bland328, 2024-03-25:
Thanks so much, @kdschlosser, both for your time creating `lvgl_micropython` and for your generous assistance.

At this point, the backlight is on and I have a white screen, but nothing renders.

Would you guess I'm up against a problem in my definition of either `lcd_bus.RGBBus` or `rgb_display.RGBDisplay`?

Or is it more likely I'm mishandling `lvgl` itself somehow?

kdschlosser, 2024-03-25:
By looking at the specs for the display I am going to say this is what you need to be running..

```
import time, micropython
micropython.mem_info()

import lcd_bus
display_bus = lcd_bus.RGBBus(
    hsync = 39,
    vsync = 41,
    de = 40,
    disp = -1,
    pclk = 42,  # to TFT DCLK
    data0  = 8,  data1  = 3,  data2  = 46, data3  = 9,  data4  = 1,  # B
    data5  = 5,  data6  = 6,  data7  = 7,  data8  = 15, data9  = 16, data10 = 4,  # G
    data11 = 45, data12 = 48, data13 = 47, data14 = 21, data15 = 14,  # R
    freq = 12000000,
    hsync_front_porch = 8,
    hsync_back_porch = 8,
    hsync_pulse_width = 4,
    hsync_idle_low = True,
    vsync_front_porch = 8,
    vsync_back_porch = 8,
    vsync_pulse_width = 4,
    vsync_idle_low = True,
    de_idle_high = False,
    pclk_idle_high = False,
    pclk_active_low = True,
    disp_active_low = False,
)

import rgb_display
import lvgl as lv
display = rgb_display.RGBDisplay(
    data_bus = display_bus,
    display_width = 800,
    display_height = 480,
    backlight_pin = 2,
    color_space = lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()
display.set_backlight(100)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(200, 50)
slider.center()

while True:
    time.sleep_ms(1)
    lv.tick_inc(1)
    lv.task_handler()
```

kdschlosser, 2024-03-25:
I just updated the code to deal with using rgb565_byte_swap in a different manner. It will increase the performance by about 30%.

bland328, 2024-03-25:
I'm now running precisely the code you provided, but I'm still getting just a blank white display.

I'll do more digging for display specs.

FYI, a couple sources assert that [this is the screen being used](https://www.waveshare.com/4.3inch-Capacitive-Touch-LCD.htm), and at first glance it certainly appears consistent with what's mounted on the board.

kdschlosser, 2024-03-25:
turn the freq down to 14mhz or even 13mhz and that will fix the slider moving.
I ran into this very same problem with the waveshare display I have.

kdschlosser, 2024-03-25:
I believe the flickering will go away once you get the frequency dialed in

bland328, 2024-03-25:
> I believe the flickering will go away once you get the frequency dialed in

And so it did!  13mhz appears solid.  Very exciting, thank you. 

Your work on this is well-conceived, nicely executed, and _incredibly_ appreciated--is there a link I can use to buy you a coffee, beer or microcontroller? :)

kdschlosser, 2024-03-25:
Quit drinking over a decade ago so beer is not something I would use.
Coffee is not something I buy, I will make it at home and I only drink 1/2 a cup in the morning. So that's probably not ideal either.

An MCU would be nice. can never have too many of those.

https://www.paypal.com/ncp/payment/4NZXTMZD25PMJ

kdschlosser, 2024-03-25:
yup that's a fat finger. lemme fix it.

kdschlosser, 2024-03-25:
all fixed

kdschlosser, 2024-03-25:
I am super happy we got you up and running. I knew the RGB bus driver worked but you having issues with it exposed the memory issue which is now fixed and it also exposed a better way of handling the RGB565 byte swap that should be a HUGE improvement in performance.

I still have to tell you how to get the gt911 driver working...

you need to change the `INDEV=GT911` in your build command to `INDEV=gt911` I know it wouldn't matter on windows because file names are not case sensitive and I cannot remember if I added code to make the filename all lower case or not. so just to make sure I would make it all lowercase.

Now I added a special way of dealing with I2C in MicroPython. it actually makes it easier to use and it will deal with using I2C from more than one thread. The ESP32 supports threading and if you have any attached I2C devices you are able to collect or send data to those devices in a thread.

so this is how the gt911 driver works

```python
import i2c
import gt911

i2c_bus = i2c.I2CBus(scl=10, sda=11, freq=4000000, host=None, use_pullups=False, use_locks=False)

touch_driver = gt911.GT911(i2c_bus=i2c_bus, reset_pin=None, interrupt_pin=None, touch_cal=None)
```

You can change the host if you like and you can specify to use built in pullups if you like. The `use_locks` you set to `True` if you are going to use threading. 

The `reset_pin` you can use. The `interrupt_pin` doesn't get used as an interrupt, it is required in order to perform a hardware reset of the touch driver IC.

`touch_cal` is something I have to finish up making. I wrote into the firmware the ability to calibrate the touch display. On the ESP32 the calibration data gets saved to NVRAM. This is nice because it would give you the ability to calibrate the touch at runtime and not have to hard code the calibration data. It allows for real time calibrations to take place for the touch screen while the MCU is running and you will not need to perform a reboot in order for the settings to take effect.

bland328, 2024-03-25:
Works fine now, thanks!  And I've sent a gift your way 🎁

Two quick notes:

1. In the `To flash firmware:` instructions that appear after building, the `esptool.py,` trailing comma fat-fingering appears again.

2. Also, FWIW, the suggested `erase_flash` command specifies a speed of `460800`, whereas the `write_flash` command specifies a speed of `921600`.  Perhaps that's correct in many cases, but for me, I'm only able to successfully write flash if I decrease the speed to `460800`.

bland328, 2024-03-25:
> I still have to tell you how to get the gt911 driver working...

Wow...that's really above and beyond. Thanks so very much--I'll give it a try in the next 30 mins or so!

bland328, 2024-03-25:
After eliminating one error by dropping one of the zeros in `i2c.I2CBus(...freq=4000000,...`, I then get this error (line 57 being your unaltered `touch_driver =...` line):

```
Traceback (most recent call last):
  File "main.py", line 57, in <module>
  File "gt911.py", line 91, in __init__
  File "pointer_framework.py", line 58, in __init__
AttributeError: 'lv_indev_t' object has no attribute 'init'
```

I'm digging into it, but wanted to bounce it off you in case you know what's up.

kdschlosser, 2024-03-25:
I just updated the code, give it s shot now.

kdschlosser, 2024-03-25:
I am not sure if the gt911 uses a 4mhz frequency or a 1mhz frequency. You might have to adjust that if there is an issue like it is not able to read from the device.

bland328, 2024-03-25:
I found in [this datasheet](https://www.crystalfontz.com/controllers/GOODIX/GT911ProgrammingGuide/478/) some GT911 details suggesting the I2C interface transmission rate should be between 200Kbps and 400Kbps:

> GT911 interfaces with the host via 6 pins: VDD, GND, SCL, SDA, INT and RESET. The INT pin of the host can be rising/falling-edge Triggered. In addition, when INT is in input state, the host should set INT to be floating, with no internal pull-up or pull-down; the host controls the RESET pin of the GT911 by outputting high or low level. To ensure reliable reset, it is recommended that RESET pin outputs low for longer than 100μs.
>
> _GT911 communicates with the host via standard I2C interface, with a maximum transmission rate of 400K bps.
When the host communicates at rates exceeding 200K bps, it is required to pay special attention to the resistance of the external pull-up resistor of I2C interface to ensure the edges of SCL and SDA are steep enough._ GT911 invariably serves as slave device in communication and its I2C device address consists of 7 device address bits and 1 Read/Write control bit. The high 7 bits are device address while bit 0 is Read/Write control bit. GT911 supports two slave device addresses which are shown below...

Is this applicable to the `i2c.I2CBus(...freq=...)` parameter?  Or do I misunderstand that parameter?

I've tried `200000`, `300000` and `400000` as values for the `freq` param, but in every case I end up with this error:

```
Traceback (most recent call last):
  File "main.py", line 59, in <module>
  File "gt911.py", line 106, in __init__
  File "i2c.py", line 146, in write
  File "i2c.py", line 88, in writeto
OSError: [Errno 19] ENODEV
```

kdschlosser, 2024-03-25:
It says it right here...

> GT911 communicates with the host via standard I2C interface, with a maximum transmission rate of 400K bps.

so 400000 is what you need to set the freq to it would appear.

kdschlosser, 2024-03-25:
after the I2C bus is set up but before you set up the driver add this line of code.

```python
print(i2c_bus.scan())
```

That will print out all of the device ID's that are attached to the bus.

paste trhat for me so I can check and make sure I am using the right device id's

bland328, 2024-03-25:
I've been chasing this for hours, but no luck.

We're agreed that `freq` should be `400000`.  Plus, after examining both the [board layout](https://macsbug.wordpress.com/2022/11/29/esp32-8048s043) and [other projects](https://github.com/HASwitchPlate/openHASP/blob/6cf4262f2356eaa686abe045ab01a5467a6963b1/user_setups/esp32s3/sunton-esp32-s3-tft.ini#L123) compatible with this board,  I'm 99.9% confident `scl` should be `20` and `sda` should be `19`.  Also, it looks like `reset_pin` should be `38`, and `interrupt_pin` should be `None` _(because the interrupt trace appears to be interrupted by an open pad, for reasons that are a mystery to me)_.

However, this code...

```py
i2c_bus = i2c.I2CBus(scl=20, sda=19, freq=400000, host=None, use_pullups=False, use_locks=False)
print(i2c_bus.scan())
```

never returns from `i2c_bus.scan()`.

FWIW, I've also experimented with `machine.I2C()` and `machine.SoftI2C()`, and get unexpected results in every case--that is, either `scan()` never returns, or returns an empty list.

If I didn't know better, I'd say the board has a hardware issue.  But I do know better, because the original demo firmware that shipped on the board works fine.

So, I'm utterly stuck at the moment, sadly.

kdschlosser, 2024-03-26:
set the frequency to 1mhz and see what happens

kdschlosser, 2024-03-26:
what are you using for a host? If it's not set, set it to 1.

kdschlosser, 2024-03-26:
What is the model number of the display you have? Does it end in a "C". If it doesn't then we are using the wrong touch driver and that could be the problem... There are 2 different touch IC's that are used with this display. The "C" means Capacitive which is the GT911 and without the "C" is the resistive touch and that is the XPT2046 IC. completely different driver.

kdschlosser, 2024-03-26:
It actually appears like they may have used 3 different drivers. the other one is the FT6x36 IC

bland328, 2024-03-26:
> What is the model number of the display you have? Does it end in a "C"

I certainly ordered the capacitive version, and though the board isn't literally _marked_ "-C", running to the TFT there is a flexible flat cable with a visible GT911 chip mounted on it, as seen in [this marked-up photo](https://macsbug.files.wordpress.com/2022/10/4280s043_layout-2.pdf).

bland328, 2024-03-26:
> what are you using for a host? If it's not set, set it to 1.

I was originally using None for the host value, but did switch to 1 along the way, after some research suggested that was appropriate.

bland328, 2024-03-26:
> set the frequency to 1mhz and see what happens

I made a `boot.py` file containing only this:

```py
import i2c
i2c_bus = i2c.I2CBus(scl=20, sda=19, freq=1000000, host=1, use_pullups=False, use_locks=False)
print(i2c_bus.scan())
```

It hangs on the scan for about 1m50s, then prints an empty list.  FWIW, using pullup resistors makes no difference.

kdschlosser, 2024-03-26:
I will mess about with the I2C driver on my end and see if I am able to get the gt911 driver to run. I have a display here that has that IC.

kdschlosser, 2024-03-26:
Ya know what it could be...

I think I know what it is. There is a bug in MicroPython with I2S and I wonder if there is something with that bug and this problem...

hmmm..... I can't remember if 1.19 had support for the S3 MCU. I think it did. I know it used espidf 4.something.. The I2S issue came about with idf 5.0.x support being added to micropython.

I will compile a binary file of 1.19 for ya just to test and see if I2C is working. If it is and you get a list of device ids when performing a scan then we know the problem is upstream and it needs to be fixed... If we get the same result then your board is going to be no good.

bland328, 2024-03-26:
> I think I know what it is. There is a bug in MicroPython...

That's very interesting!  This certainly does feel like "that sort of a problem," because again, the touch screen _does_ work with the demo firmware.

Thanks very offering to provide that build!

> on a happier note, I added generating a stub file for lvgl...

Oh! I love that. I do use VSCode with Pylance, so I will certainly make good use of the stub file.

bland328, 2024-03-26:
> I think I know what it is. There is a bug in MicroPython...

Just now, I installed Arduino IDE and configured [this simple I2C scanning sketch](https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide) to use GPIO 20 for `scl` and GPIO 19 for `sda`.

When I run it, I get `I2C device found at address 0x5D`, which is the expected address for the GT911 device.

So, yes, certainly appears to be a MicroPython bug.  I'm hoping with enough digging and experimenting I might be able to find some sort of workaround to get me by for now.

kdschlosser, 2024-03-26:
I am going to open an issue in MicroPython's repo about this.

kdschlosser, 2024-03-26:
https://github.com/micropython/micropython/issues/11834

the issue has been open for a while now and nothing has been done about it. 

I don't get any more attention than anyone else. You can chime in and post that you are also having the same issue as well. Technically speaking while you are having the issue I am also having the same issue as well. I tested it on one of my MCU's last night. Exact same thing is happening as you.

The more people that report a problem the more attention there will be to get it fixed.

kdschlosser, 2024-03-26:
The last time I reported a bug it took nearly a year to get it solved. I might have to come up with a way to fix this problem myself. waiting a year to get it fixed is not acceptable.

bland328, 2024-03-26:
> The last time I reported a bug it took nearly a year to get it solved. I might have to come up with a way to fix this problem myself. waiting a year to get it fixed is not acceptable.

Seems bizarre to me that something as fundamental as I2C could be broken on such a popular MCU without getting more attention from the MicroPython community.

If you do end up tackling this, it's much appreciated, and I'm happy to run any tests that might help along the way.

kdschlosser, 2024-03-26:
I found the problem...

Whoever coded the I2C implementation in MicroPython is making general assumptions..

```
i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = self->sda,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,   <------- HERE
        .scl_io_num = self->scl,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,   <------ AND HERE
        .master.clk_speed = freq,
    };
```

the internal pull-up values added to the 4.7K ohm resistors that are in place on the MCU are more than likely causing the problem. 

The higher the frequency the lower the value for the resistance. The ideal window is 1K ohm to 5K ohm so chances are what is happening is the addition of the pullup being added in parallel to the resistors onboard are moving the resistance value out of where it should be.

bland328, 2024-03-26:
I haven't yet looked at that function in context, but are you saying that the pull-up resistors are being used regardless of whether the caller wants them?

kdschlosser, 2024-03-26:
Go ahead and edit the source file in MicroPython for the I2C and change those 2 pull-up settings to a zero and compile it and give it a try again.

It may or may not be what is causing the problem but it is good to eliminate it as a possible cause because the board you are using already has pull-up resistors in place so we don't need to have the internal one enabled anyhow.

It will eliminate it as a possible cause if it doesn't solve the problem.

kdschlosser, 2024-03-26:
The IC for the ESP32 in MicroPython needs a complete rewrite. It is not wrong in a small way, it is really wrong. When they upgraded from using ESP-IDF 4.4 to 5.0 they never checked the I2C code across all of the MCU's the way they have it will work for an ESP32 but not for an ESP32-S3. They didn't do what was needed when they did that upgrade.

I can submit a PR but it will never get merged. I have done this before. It is a complete waste of time for me to write the code to submit to them.

I will write a new one and have the build system replace the files. It is going to be easier that way. I pointed out all of the problems that exist with the current I2C to the developers of MicroPython. It is up to them to fix it and I know they won't. I am already getting flack about the pull-up resistors being enabled and being told that it's OK for them to be enabled even tho they will alter the resistance if the user has external ones. I have seen this kind of response with a lot of bug reports that have been opened for MicroPython. and most bugs never get fixed because of it.  It's the "It's working fine when I use it" kind of a response. No investigation done to see if there might be an issue.

kdschlosser, 2024-03-26:
So give me some time to write a new I2C driver. Depending on how I feel I might be able to get it done tonight.

bland328, 2024-03-27:
> The IC for the ESP32 in MicroPython needs a complete rewrite.

Well, that's not good, but at least it explains why neither the change you suggested nor a bunch of other experiments I just ran managed to fix anything.  Nor even _change_ anything.  I even resorted to inserting a null pointer reference just to confirm that code path was even being called. 😅

>  It is up to them to fix it and I know they won't.

That's a shame--MicroPython seems like a great project in so many ways.

> I am already getting flack about the pull-up resistors being enabled...

Ugh. If it were fine for pull-ups to always be active., then there wouldn't a hardware option to disable them.

kdschlosser, 2024-03-27:
OK I am done. I just committed and pushed the code changes.

I have not tested it to even see if it will compile. Fingers crossed....

I wrote it so it is fully compatible with the original I2C driver so hopefully this will do the trick.

bland328, 2024-03-27:
Wow--an impressive amount of work for one evening!

It doesn't compile, owing mostly to undeclared `I2C_TRANS_BUF_MINIMUM_SIZE` and `self_in`.

Still, I look forward to testing it soon :)

kdschlosser, 2024-03-27:
I was able to finagled it so I didn't have to do a complete rewrite of it. It took me a while to read the MicroPython code to figure out what what in the world they were doing. I cannot believe the sheer amount of mechanics involved in sending a simple I2C message. It is mind blowing actually.

There is something I did want to look at which would make dealing with the gt911 driver a little easier. The gt911 IC uses 16 bit memory addresses and I wanted to take a look and see if MicroPython encodes the addresses LSB first. If it does then I can simplify the gt911 driver a pretty decent amount by using the write to mem and read from mem

OK cool both of those are super easy to fix.

bland328, 2024-03-27:
Oh, and there's also an undeclared `i2c_num` in `i2c_write()`.

[110 more replies]

closed 2024-05-09 by kdschlosser