lvgl-micropython/lvgl_micropython #21: Build failure with STM32, 'lib/micropython/lib/stm32lib' did not match any file(s) known to git

OhNarts, 2024-03-25:
I'm on Fedora 38 and I'm trying to build for the NUCLEO_F767ZI board. On a fresh clone of the repository I ran the command

`python3 make.py stm32 submodules clean mpy_cross BOARD=NUCLEO_F767ZI`

and I get the error

`error: pathspec 'lib/micropython/lib/stm32lib' did not match any file(s) known to git`

Here's the full output log
```
git submodule update --init -- lib/micropython
Submodule path 'lib/micropython': checked out '9b8c64c9cee8203be167e6bffc74e186ae2fc958'

cd lib/micropython && git checkout v1.22.2
Previous HEAD position was 9b8c64c9c all: Bump version to 1.22.1.
HEAD is now at 8cd15829e all: Bump version to 1.22.2.

git submodule update --init -- lib/lvgl
Submodule path 'lib/lvgl': checked out '2722e390e4671ff43d3dd0f65609dc432d99dfc3'

cd lib/lvgl && git checkout master
Your branch is up to date with 'origin/master'.
Previous HEAD position was 2722e390e chore(benchmark): add warning for probably low memory (#5797)
Switched to branch 'master'

git submodule update --init -- lib/pycparser
git submodule update --init -- lib/micropython/lib/stm32lib
error: pathspec 'lib/micropython/lib/stm32lib' did not match any file(s) known to git
```
Any guidance on how to remedy this issue would be much appreciated

kdschlosser, 2024-03-25:
You are going to be the guinea pig with using an STM32 MCU. I don't have one of these MCU's to test with and I have not had any exposure to them at all. It is going to take some work to get everything working properly.

I wanted to let you know that ahead of time to see if you are interested in putting in that effort to get it running.

Currently the only types of displays that will be supported are I8080 and SPI displays. The driver used for the i8080 is a bit-bang driver so the performance is not going to be as fast as it could be. If you have an RGB display there is currently no driver available. I would love to add STM32 specific drivers for MIPI, I8080, RGB and SPI and if you know how to do that in C code would you be willing to lend a hand?

I am going to run some tests to see what is gong on with the STM32 compilation and I will get that portion fixed. Give me a few hours to do that.

OhNarts, 2024-03-25:
Not sure how much help I would be in that department, I'm fairly new to this repository.

kdschlosser, 2024-05-04:
I am setting up a CI to test the build systems. It is still a WIP but it will help nail down problems like this one. I am sorry for the delay in getting this fixed for you. I just went through spine surgery 6 weeks ago and the amount of time I am able to spend in front of the PC has been limited.