lvgl-micropython/lvgl_micropython #28: building unix fails with unknown SDL class members

TecDroiD, 2024-04-12:
Hi.. next thing... ;) I try to build for unix just to get a little more used to lvgl without having my display permanently on.. 

so my call is
` python make unix VARIANT=standard`

and the result is
```
...
/home/tecdroid/git/lvgl_micropython/ext_mod/lcd_bus/sdl_bus/sdl_bus.c: In function ‘event_filter_cb’:
/home/tecdroid/git/lvgl_micropython/ext_mod/lcd_bus/sdl_bus/sdl_bus.c:580:93: error: ‘SDL_TouchFingerEvent’ has no member named ‘which’
580 |                 mp_obj_dict_store(dict, MP_OBJ_NEW_QSTR("id"), mp_obj_new_int(event->tfinger.which));
|                                                                                             ^
/home/tecdroid/git/lvgl_micropython/ext_mod/lcd_bus/sdl_bus/sdl_bus.c:598:103: error: ‘SDL_ControllerTouchpadEvent’ has no member named ‘fingerId’; did you mean ‘finger’?
598 |                 mp_obj_dict_store(dict, MP_OBJ_NEW_QSTR("finger_id"), mp_obj_new_int(event->ctouchpad.fingerId));
|                                                                                                       ^~~~~~~~
|                                                                                                       finger
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [../../py/mkrules.mk:90: build-standard/lcd_bus/sdl_bus/sdl_bus.o] Fehler 1
make: *** Es wird auf noch nicht beendete Prozesse gewartet....
make: Verzeichnis „/home/tecdroid/git/lvgl_micropython/lib/micropython/ports/unix“ wird verlassen
``` 

at least, according to my SDL_event.h, this is correct. there's no which in SDL_TouchFingerEvent (but in SDL_ControllerTouchpadEvent) and no fingerId in SDL_ControllerTouchpadEvent (but, fun fact, in SDL_TouchFingerEvent)

any idea?

kdschlosser, 2024-04-12:
The unix port is still under development. I am wanting to keep the API the same for the display drivers and the indev drivers across all ports. This will allow a user to have code like the following

```python

import sys
import lcd_bus

if sys.platform in ('linux', 'darwin'):
    bus = lcd_bus.SDLBus(...)
    
    import sdl_display

    display = sdl_display.SDLDisplay(bus, ...)
    
    import sdl_mouse

    indev = sdl_mouse.SDLMouse(...)

else:
    # code here for a display that is attached to an MCU
    ...
```

doing that will allow a user to run the same exact code on the unix port or an MCU. This makes it a whole lot easier and faster when developing. the same file that runs in MicroPython on the desktop is used on the MCU.

kdschlosser, 2024-04-13:
I wanted to let you know that I got it to compile properly for the unix build. I have not updated the code here on the repo yet. I want to test it out to see if it works.

kdschlosser, 2024-04-15:
making progress. I got it to display properly. I have to test the mouse still. Gonna do that tonight. It should be working in the next few days

TecDroiD, 2024-04-15:
well.. thank you for the great job you're doing here :)

kdschlosser, 2024-04-17:
I am working on getting the the mouse driver working. I didn't want to use polling in the main loop of the program as that loop could take a while to complete. I was going to use a separate thread to poll the events and schedule a callback using the micropython scheduler but apparently if `time.sleep_ms` is used that stops the scheduler as well. I don't know why it would. It should be written so a scheduled task is able to break in during that time.

kdschlosser, 2024-04-18:
OK it should be working. The README.md file has all of the information needed to compile it. I have provided the build command, example code and also the requirements. There is a long list of requirements due to compiling a new version of SDL. So make sure you get those installed prior to compiling

TecDroiD, 2024-04-18:
i have libsdl 2.26.5+dfsg-1 installed which is standard for current debian. 
currently, it looks for an import named SDL.h in sdl_bus.h as well as in sdl_bus.c. Since it seems that both are only compiled under same conditions (MP_PORT_UNIX), I removed the entry in sdl_bus.c. 
Sadly, in debian, SDL2 lies under /usr/include/SDL2 so I would either have to change the path in sdl_bus.h (what I did) or in the includes path (which i should have done)
Now it compiled and works fine.

kdschlosser, 2024-04-18:
Ubuntu is the most popular Unix OS in use. This is what I used for the purposes of deciding how to handle SDL2. SDL2 2.0.20 is the version of SDL2 that is available to the latest release of Ubuntu. I also want to use the most recent version of SDL2 because of bug fixes and what have you. In order to do that SDL2 MUST be compiled. the most recent release of SDL2 is  2.30.2.

If you are having issues I recommend uninstalling SDL2 from your OS and letting the build system do what it needs to do to compile. For some damned reason I am not able to override the include path that is added to the compiler for installed packages. That one ALWAYS seems to be the priority without any way to override that behavior that isn't OS dependent. It's a silly design if you ask me.

I have to test and see if the static linking I am doing with the binding is what is actually being used. I believe it is, however it may not be and I have also changed the search order for the so in the executable so the first folder searched is the parent folder where the executable is located. The compiled SDL2 shared library is moved into the parent folder where the executable is. So you should still be able to install SDL2 into the OS using the OS package manager once the binding has been compiled.

If you know of a way to force the priority of the include paths that is not OS dependent I am all ears. If you want to spend the time to figure that out then have it at. I didn't want to spend too much time dealing with it as I wanted top get it to compile and run properly and figured if someone had an issue with it they would let me know how to go about fixing it.

kdschlosser, 2024-04-18:
The other animal we run into with SDL2 is the include path. This is not the same for all OS's. The headers are assumed to be under the folder of SDL2 but this is not always the case. it can be under SDL or some other folder depending on the OS used. If I compile SDL2 I am able to mitigate that problem because I know exactly where the include folder is going to be and it's not OS dependent.

kdschlosser, 2024-04-20:
I believe I have fixed the include path issue with the build system.. If you want to give it a try you are more then welcome to. I have not tested it yet but the fix was kind of silly actually.

if an include is wrapped in brackets GCC will search the system includes only, it will not use the include paths passed to the compiler using the `-I` switch. If the include is wrapped in double quotes GCC searches the directory of the current file then it searches the directories give using the `-I` in the same order in which they are passed and then lastly it will search the system paths.

The includes for SDL I had with brackets. so I have changed them to double quotes so it should now work correctly even if you have SDL already installed into the system.

TecDroiD, 2024-04-21:
Well, at least, it builds now :)

kdschlosser, 2024-04-21:
so it compiles but doesn't run?

I still haven't tested it yet. I will later on today.

TecDroiD, 2024-04-21:
don't panic. it builds and runs :)

kdschlosser, 2024-04-21:
I believe it is linked statically. I was messing about with it so I would have to go and look to see if I did a static linking. I was messing about with the dynamic and ran into the include issue and that messed about with the static and was trying to get the thing to link correctly and get the includes to work properly. so when it did I left it alone. I don't remember off the top of my head what it is doing at this point in time. I have to go and check.

kdschlosser, 2024-04-21:
This is from the ld man pages.

> Add archive file archive to the list of files to link. This option may be used any number of times. ld will search its path-list for occurrences of libarchive.a for every archive specified. On systems which support shared libraries, ld may also search for libraries with extensions other than .a. Specifically, on ELF and SunOS systems, ld will search a directory for a library with an extension of .so before searching for one with an extension of .a. By convention, a .so extension indicates a shared library. The linker will search an archive only once, at the location where it is specified on the command line. If the archive defines a symbol which was undefined in some object which appeared before the archive on the command line, the linker will include the appropriate file(s) from the archive. However, an undefined symbol in an object appearing later on the command line will not cause the linker to search the archive again. See the -( option for a way to force the linker to search archives multiple times. You may list the same archive multiple times on the command line. This type of archive searching is standard for Unix linkers. However, if you are using ld on AIX, note that it is different from the behaviour of the AIX linker. 

It sounds like as a default it is going to statically link and if it is unable to find the library's archive file then it will dynamically link.

I am going to mess about with it and see if I am able to attach the library name directly to the end of the GCC command.

kdschlosser, 2024-04-21:
OK so it is now linking statically. I pushed the changed to the repo so that should be good to go now.

samjt, 2024-04-28:
I’m trying to build on Mac, I see homebrew has SDL2 2.30.2, but I guess the include path is not what you’re expecting with make.py as I also get header missing errors. I’m not sure where to set that path though.

I also see the submodules command throws if the build platform isn’t Linux, but I’m not sure what other dependencies are required that aren’t buildable?

kdschlosser, 2024-04-28:
I really need to set up a CI so that way I will now if it will compile on macOS or not. I am going to mess about with doing that this evening. I am only able to test on linux and I will like it to be able to work across all platforms for any of the MCU's. It is going to take a bit to get to that point though.

samjt, 2024-04-29:
I’m happy to help testing builds on macOS, but I’m not sure where you need help?

kdschlosser, 2024-05-09:
OK CI integration is done. I updated the readme to list the dependencies and how to build for macOS and also how to build on macOS.

I am going to close this issue.

closed 2024-05-09 by kdschlosser