lvgl-micropython/lvgl_micropython #29: i can't use I80bus

sasakihin, 2024-04-15:
i got a [WT32SC01 plus ](https://doc.riot-os.org/group__boards__esp32s3__wt32__sc01__plus.html#esp32s3_wt32_sc01_plus_board_configuration)
also referred to russhughes/wt32sc01py: WT32SC01 Plus MicroPython Display Driver (github.com), and corrected the code.

`from micropython import const
import lcd_bus
import st7796
import lvgl as lv

_WIDTH = const(480)
_HEIGHT = const(320)

_DC_PIN = const(0)
_WR_PIN = const(47)
_DATA0_PIN = const(9)
_DATA1_PIN = const(46)
_DATA2_PIN = const(3)
_DATA3_PIN = const(8)
_DATA4_PIN = const(18)
_DATA5_PIN = const(17)
_DATA6_PIN = const(16)
_DATA7_PIN = const(15)
_CS_PIN = const(6)

_RESET_PIN = const(4)
_POWER_PIN = const(45)
_BACKLIGHT_PIN = const(45)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

display_bus = lcd_bus.I80Bus(
    dc=_DC_PIN,
    wr=_WR_PIN,
    data0=_DATA0_PIN,
    data1=_DATA1_PIN,
    data2=_DATA2_PIN,
    data3=_DATA3_PIN,
    data4=_DATA4_PIN,
    data5=_DATA5_PIN,
    data6=_DATA6_PIN,
    data7=_DATA7_PIN,
    )

lv.init()
print('lv.init()')

display = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7796.STATE_HIGH,
    power_pin=None,
    power_on_state=st7796.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7796.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True)

#display.init()
display.set_backlight(100)
display.init()

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)`

There are no errors, but nothing shows up on the screen.
How can i fixit?

kdschlosser, 2024-04-15:
```
from micropython import const
import lcd_bus
import st7796
import lvgl as lv

_WIDTH = const(480)
_HEIGHT = const(320)

_DC_PIN = const(0)
_WR_PIN = const(47)
_DATA0_PIN = const(9)
_DATA1_PIN = const(46)
_DATA2_PIN = const(3)
_DATA3_PIN = const(8)
_DATA4_PIN = const(18)
_DATA5_PIN = const(17)
_DATA6_PIN = const(16)
_DATA7_PIN = const(15)
_CS_PIN = const(6)

_RESET_PIN = const(4)
_POWER_PIN = const(45)
_BACKLIGHT_PIN = const(45)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

display_bus = lcd_bus.I80Bus(
dc=_DC_PIN,
wr=_WR_PIN,
data0=_DATA0_PIN,
data1=_DATA1_PIN,
data2=_DATA2_PIN,
data3=_DATA3_PIN,
data4=_DATA4_PIN,
data5=_DATA5_PIN,
data6=_DATA6_PIN,
data7=_DATA7_PIN,
)

lv.init()
print('lv.init()')

display = st7796.ST7796(
data_bus=display_bus,
display_width=_WIDTH,
display_height=_HEIGHT,
# we are going to let the driver handle the allocation of the frame buffers
frame_buffer1=None,
frame_buffer2=None,
reset_pin=_RESET_PIN,
reset_state=st7796.STATE_HIGH,
power_pin=None,
power_on_state=st7796.STATE_HIGH,
backlight_pin=_BACKLIGHT_PIN,
backlight_on_state=st7796.STATE_HIGH,
offset_x=_OFFSET_X,
offset_y=_OFFSET_Y,
color_space=lv.COLOR_FORMAT.RGB565,
rgb565_byte_swap=True)

#display.init()
display.set_backlight(100)
display.init()

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)

import task_handler

th = task_handler.TaskHandler()

```

sasakihin, 2024-04-15:
> ```
> from micropython import const
> import lcd_bus
> import st7796
> import lvgl as lv
>
> _WIDTH = const(480)
> _HEIGHT = const(320)
>
> _DC_PIN = const(0)
> _WR_PIN = const(47)
> _DATA0_PIN = const(9)
> _DATA1_PIN = [...]

nothing happen....it's still black(no back light)...

kdschlosser, 2024-04-15:
give this a try

```python
from micropython import const
import lcd_bus
import st7796
import lvgl as lv

_WIDTH = const(480)
_HEIGHT = const(320)

_DC_PIN = const(0)
_WR_PIN = const(47)
_DATA0_PIN = const(9)
_DATA1_PIN = const(46)
_DATA2_PIN = const(3)
_DATA3_PIN = const(8)
_DATA4_PIN = const(18)
_DATA5_PIN = const(17)
_DATA6_PIN = const(16)
_DATA7_PIN = const(15)

_RESET_PIN = const(4)
_POWER_PIN = None
_BACKLIGHT_PIN = const(45)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

display_bus = lcd_bus.I80Bus(
    dc=_DC_PIN,
    wr=_WR_PIN,
    data0=_DATA0_PIN,
    data1=_DATA1_PIN,
    data2=_DATA2_PIN,
    data3=_DATA3_PIN,
    data4=_DATA4_PIN,
    data5=_DATA5_PIN,
    data6=_DATA6_PIN,
    data7=_DATA7_PIN,
    freq=40000000,
    dc_idle_high=False,
    dc_cmd_high=False,
    dc_data_high=True,
    dc_dummy_high=False,
    pclk_idle_low=False
)

lv.init()
print('lv.init()')

display = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7796.STATE_HIGH,
    power_pin=None,
    power_on_state=st7796.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7796.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

#display.init()
display.set_backlight(100)
display.init()

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)

import task_handler

th = task_handler.TaskHandler()
```

sasakihin, 2024-04-15:
> give this a try
>
> ```python
> from micropython import const
> import lcd_bus
> import st7796
> import lvgl as lv
>
> _WIDTH = const(480)
> _HEIGHT = const(320)
>
> _DC_PIN = const(0)
> _WR_PIN = const(47)
> _DATA0_PIN = [...]

it's also black....

I successfully lit up the screen with the following code, but the backlight PWM doesn't seem to be working. Even with a value of 1, it's fully bright.

```

from lcd_bus import I80Bus
import st7796
import i2c
from machine import  Pin, freq
import ft6x36
import lvgl as lv

reset=Pin(4, Pin.OUT, value=1)# WHY???

display_bus = I80Bus(
    dc=0,
    wr=47,
    cs=6,
    data0=9,
    data1=46,
    data2=3,
    data3=8,
    data4=18,
    data5=17,
    data6=16,
    data7=15,
)

display = st7796.ST7796(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    backlight_pin=45,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

display.init()
display.set_backlight(1) #0 is close backlight 

i2c_bus = i2c.I2CBus(scl=5,sda=6,freq=100000,use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

scr = lv.obj()
scr.set_style_bg_color(lv.color_hex(0x00FF00), 0)
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)

import task_handler

th = task_handler.TaskHandler()
```

kdschlosser, 2024-04-16:
to be able to alter the backlight brightness you have to set the `backlight_on_state` parameter to `STATE_PWM`

```python

from micropython import const
import lcd_bus
import st7796
import lvgl as lv

_WIDTH = const(480)
_HEIGHT = const(320)

_DC_PIN = const(0)
_WR_PIN = const(47)
_DATA0_PIN = const(9)
_DATA1_PIN = const(46)
_DATA2_PIN = const(3)
_DATA3_PIN = const(8)
_DATA4_PIN = const(18)
_DATA5_PIN = const(17)
_DATA6_PIN = const(16)
_DATA7_PIN = const(15)

_RESET_PIN = const(4)
_POWER_PIN = None
_BACKLIGHT_PIN = const(45)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

display_bus = lcd_bus.I80Bus(
    dc=_DC_PIN,
    wr=_WR_PIN,
    data0=_DATA0_PIN,
    data1=_DATA1_PIN,
    data2=_DATA2_PIN,
    data3=_DATA3_PIN,
    data4=_DATA4_PIN,
    data5=_DATA5_PIN,
    data6=_DATA6_PIN,
    data7=_DATA7_PIN,
    freq=40000000,
    dc_idle_high=False,
    dc_cmd_high=False,
    dc_data_high=True,
    dc_dummy_high=False,
    pclk_idle_low=False
)

lv.init()
print('lv.init()')

display = st7796.ST7796(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=st7796.STATE_HIGH,
    power_pin=None,
    power_on_state=st7796.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7796.STATE_PWM,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

#display.init()

import time

for i in range(0, 101):
    display.set_backlight(i)
    time.sleep_ms(3)

display.init()

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)

import task_handler

th = task_handler.TaskHandler()
```

sasakihin, 2024-04-17:
> ```python
> STATE_PWM
> ```

``display = st7796.ST7796(
    data_bus=display_bus,
    display_width=320,
    display_height=480,
    frame_buffer1=None,
    frame_buffer2=None,
    #reset_pin=_RESET_PIN, 
    #reset_state=st7796.STATE_HIGH,
    power_pin=None,
    power_on_state=st7796.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=st7796.STATE_PWM,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

some error :
MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 64, in <module>
  File "display_driver_framework.py", line 564, in set_backlight
TypeError: can't convert float to int

kdschlosser, 2024-04-17:
OK that I can fix no worries.

marksull, 2024-05-05:
Quick question. I have the same board/setup and managed to get it running with this thread above. The only issue is any colour that contains RED, is wrong. For example, if I set the colour to RED I get a yellow.

Looking at the specs for the board it definitely says it is RGB565.

Looking at your  ST7796 driver, I see it explicitly calls out in the  method doc that you need to have LV_COLOR_DEPTH=16 for RGB565, so when I am building from your library I am using `python make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC_S3 DISPLAY=st7796 INDEV=ft6x36 LV_CFLAGS="-DLV_COLOR_DEPTH=16"` but the problem persists. Any thoughts on where I could be going wrong?

kdschlosser, 2024-05-05:
You need to pass `rgb565_byte_swap=True` to the display driver constructor...

This is your build command. Setting LV_COLOR_DEPTH doesn't do anything anymore.
```
python make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC_S3 DISPLAY=st7796 INDEV=ft6x36
```

You're code should look something like this...

```python

import lcd_bus

data_bus = lcd_bus.I80Bus(
    data0=1,
    data0=2,
    ...
)

import st7796

display = st7796 .ST7796(
    data_bus,
    width=480,
    height=320,
    ....
    rgb565_byte_swap=True,    < you need to add this.
```

kdschlosser, 2024-05-05:
the LV_COLOR_DEPTH macro no longer sets the color depth this is done using the the COLOR_FORMAT values at runtime. The reason why this change was made was to support alpha channels in the rendering aspects. The color depth is simply the number of bytes that get written for each pixel and it should not change how the rendering is done. That was how LVGL v8 worked. So it was changed and the change now allow the color depth to be set at runtime instead of compile time. It allows for you to be able to change the display you are using without having to compile and flash the firmware over again.

marksull, 2024-05-05:
Yes, definitely have both the `color_space` and the `rgb565_byte_swap` as per the above examples. I did try flipping the `rgb565_byte_swap` but it just made things worse.

kdschlosser, 2024-05-05:
ok so for your test code keep it simple and only set the background color of the screen.

```python

import time

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
lv.refr_now(lv.display_get_default())

```

Use that for the test code and tell me what you get for colors. There is going to be a 5 second pause between each color change. A total of 3 colors will be shown.

In the bus drover do you have `reverse_color_bits`  or `swap_color_bytes` set to `True`???

marksull, 2024-05-05:
I replaced `scrn = lv.screen_active(lv.display_get_default())` with `scrn = lv.screen_active()` because of the error `TypeError: function takes 0 positional arguments but 1 were given` - not sure this is relevant so included it.

The colours presented in order were; yellow, pink(ish), very light blue.

The bus driver is not initialised with either of those arguments. But I added them for a quick test and the only noticeable change was that the order of the colours reversed.

kdschlosser, 2024-05-05:
what display are you using? is it a board that has the ESP32 already built onto it?

marksull, 2024-05-05:
WT32-SC01Plus - yes, already built in.: https://www.antratek.com/media/wysiwyg/pdf/WT32-SC01-Plus-V1.3-EN.pdf

I am thinking I may have built the image wrong which may be the issue. The example you gave me above includes `lv.display_get_default()` which does not work on my MicroPython. This is the same issue I was having with my SqaureLine project export which makes me think there is another issue. What is confusing is that the disp_get_default method is definitely there and the display driver and touch interface work perfectly (other than the colour) when I create my own lv objects. So, not sure what is going on. Probably a me problem. I might try rebuilding with an earlier version of lvgl and see if that helps.

```
MicroPython >>> lv.disp_get_default()
AttributeError: 'module' object has no attribute 'disp_get_default'

MicroPython >>> "display_get_default" in dir(lv)
True

MicroPython >>> lv.version_info()
'dev'

MicroPython >>> lv.version_major()
9

MicroPython >>> lv.version_minor()
0

MicroPython >>> lv.version_patch()
1
```

kdschlosser, 2024-05-05:
display_get_default NOT disp_get_default. Those are 2 completely different API's This binding uses the master branch of LVGL which is 9.1.1. The disp_get_default is from LVGL 8.x

I am also not sure why LVGL  9.0.1 is showing up because that version doesn't exist.

paste your python code here for the entire setup. setting up the bus and also setting up the display.

marksull, 2024-05-05:
Arhhh, yes, didn't notice the difference between LVGL 9.x and 8.x. I am now running 9.1.1. My init is:

```
import i2c
import lvgl as lv
import task_handler

from lcd_bus import I80Bus
from machine import Pin

import ft6x36
import st7796

reset = Pin(4, Pin.OUT, value=1)

display_bus = I80Bus(
        dc=0,
        wr=47,
        cs=6,
        data0=9,
        data1=46,
        data2=3,
        data3=8,
        data4=18,
        data5=17,
        data6=16,
        data7=15,
)

display = st7796.ST7796(
        data_bus=display_bus,
        display_width=320,
        display_height=480,
        backlight_pin=45,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True,
)

display.init()
display.set_backlight(1)

i2c_bus = i2c.I2CBus(scl=5, sda=6, freq=100000, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)
parent = lv.obj()
th = task_handler.TaskHandler()
```

kdschlosser, 2024-05-05:
try running this code exactly how you so it. Don't change anything with it.

```python

import lcd_bus
display_bus = lcd_bus.I80Bus(
        dc=0,
        wr=47,
        cs=6,
        data0=9,
        data1=46,
        data2=3,
        data3=8,
        data4=18,
        data5=17,
        data6=16,
        data7=15,
)

fb1 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796

from machine import Pin
reset = Pin(4, Pin.OUT, value=1)

display = st7796.ST7796(
        data_bus=display_bus,
        frame_buffer1=fb1,
        frame_buffer2=fb2,
        display_width=320,
        display_height=480,
        backlight_pin=45,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True,
)

display.init()
display.set_backlight(1)

import i2c
import lvgl as lv
import task_handler

import ft6x36
import time

i2c_bus = i2c.I2CBus(scl=5, sda=6, freq=100000, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn = lv.screen_active(lv.display_get_default())
scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
lv.refr_now(lv.display_get_default())

th = task_handler.TaskHandler()
```

marksull, 2024-05-05:
1) Had to move `import lvgl as lv` to the top because `lv` was used in `color_space=lv.COLOR_FORMAT.RGB565`
2) Had to change `lv.screen_active(lv.display_get_default())` to `lv.screen_active()`as `lv.screen_active` takes no arguments

The results are the same; yellow, pink(ish), very light blue.

kdschlosser, 2024-05-05:
```python
import lcd_bus
display_bus = lcd_bus.I80Bus(
        dc=0,
        wr=47,
        freq=20000000,
        data0=9,
        data1=46,
        data2=3,
        data3=8,
        data4=18,
        data5=17,
        data6=16,
        data7=15
)

fb1 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796

from machine import Pin
reset = Pin(4, Pin.OUT, value=1)

display = st7796.ST7796(
        data_bus=display_bus,
        frame_buffer1=fb1,
        frame_buffer2=fb2,
        display_width=320,
        display_height=480,
        backlight_pin=45,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True,
)

display.init()
display.set_backlight(1)

import i2c
import lvgl as lv
import task_handler

import ft6x36
import time

i2c_bus = i2c.I2CBus(scl=5, sda=6, freq=100000, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
lv.refr_now(lv.display_get_default())

th = task_handler.TaskHandler()

```

marksull, 2024-05-05:
Thanks for your help, the same colours; yellow, pink(ish), very light blue. You know what, I have enough colour to do what I need so I will just keep on with my project. No need to chase this for me.

kdschlosser, 2024-05-06:
I am curious to know if it is a problem with the code or if you have a hardware issue. I am thinking that you might have a hardware issue. I don't have one of those displays to test with, but I am sure that someone will come along that does have one at some point..

marksull, 2024-05-06:
I have a second unboxed one (I always buy in pairs for potential hardware issues or I screw up ;-) ) so let me test it and I will confirm

marksull, 2024-05-06:
Same colours as mentioned above on second unit.

sasakihin, 2024-05-06:
some body call me? ^_^

MicroPython v1.22.1-dirty on 2024-04-14

`
import lcd_bus
import i2c
import lvgl as lv
import task_handler
display_bus = lcd_bus.I80Bus(
        dc=0,
        wr=47,
        freq=20000000,
        data0=9,
        data1=46,
        data2=3,
        data3=8,
        data4=18,
        data5=17,
        data6=16,
        data7=15
)

fb1 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796

from machine import Pin
reset = Pin(4, Pin.OUT, value=1)

display = st7796.ST7796(
        data_bus=display_bus,
        frame_buffer1=fb1,
        frame_buffer2=fb2,
        display_width=320,
        display_height=480,
        backlight_pin=45,
        color_space=lv.COLOR_FORMAT.RGB565,
        rgb565_byte_swap=True,
)

display.init()
display.set_backlight(1)

import i2c
import lvgl as lv
import task_handler

import ft6x36
import time

i2c_bus = i2c.I2CBus(scl=5, sda=6, freq=100000, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
label = lv.label(scrn)
label.set_text('0xFF0000')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
label.set_text('0x00FF00')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
label.set_text('0x0000FF')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

mask=0xFFFFFF

scrn = lv.screen_active()
blue = mask ^ 0xFF0000
scrn.set_style_bg_color(lv.color_hex(blue), 0)
label.set_text(f"blue: {hex(blue)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

green = mask ^ 0x00FF00
scrn.set_style_bg_color(lv.color_hex(green), 0)
label.set_text(f"green: {hex(green)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

red = mask ^ 0x0000FF
scrn.set_style_bg_color(lv.color_hex(red), 0)
label.set_text(f"red: {hex(red)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

th = task_handler.TaskHandler()
`

https://github.com/kdschlosser/lvgl_micropython/assets/41147444/4b384877-b778-45e4-be87-918cf8d1fc83

kdschlosser, 2024-05-06:
so all of the bits are flipped it seems???

wanna know what would cause that???

kdschlosser, 2024-05-06:
```python
import lcd_bus
import i2c
import lvgl as lv
import task_handler
display_bus = lcd_bus.I80Bus(
dc=0,
wr=47,
freq=20000000,
data0=9,
data1=46,
data2=3,
data3=8,
data4=18,
data5=17,
data6=16,
data7=15
)

fb1 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(int(480 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796

from machine import Pin
reset = Pin(4, Pin.OUT, value=1)

display = st7796.ST7796(
data_bus=display_bus,
frame_buffer1=fb1,
frame_buffer2=fb2,
display_width=320,
display_height=480,
backlight_pin=45,
color_space=lv.COLOR_FORMAT.RGB565,
rgb565_byte_swap=True,
)

display.init()
display.invert_colors(True)
display.set_backlight(1)

import i2c
import lvgl as lv
import task_handler

import ft6x36
import time

i2c_bus = i2c.I2CBus(scl=5, sda=6, freq=100000, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
label = lv.label(scrn)
label.set_text('0xFF0000')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
label.set_text('0x00FF00')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
label.set_text('0x0000FF')
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

mask=0xFFFFFF

scrn = lv.screen_active()
blue = mask ^ 0xFF0000
scrn.set_style_bg_color(lv.color_hex(blue), 0)
label.set_text(f"blue: {hex(blue)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

green = mask ^ 0x00FF00
scrn.set_style_bg_color(lv.color_hex(green), 0)
label.set_text(f"green: {hex(green)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

red = mask ^ 0x0000FF
scrn.set_style_bg_color(lv.color_hex(red), 0)
label.set_text(f"red: {hex(red)}")
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

th = task_handler.TaskHandler()
```

sasakihin, 2024-05-07:
display.invert_colors(True)
    AttributeError: unreadable attribute

kdschlosser, 2024-05-07:
anything more than that? what is the full traceback?

kdschlosser, 2024-05-07:
DOAH!!!

You need to use it like a property. I missed it being made into a property...
`display.invert_colors = True`

sasakihin, 2024-05-07:
`display.invert_colors = True`

it works!

[image]
[image]
[image]

kdschlosser, 2024-05-07:
not quite.  the RGB is backwards. if you have the color swap set in the bus unset it. 

your RGB is coming out as BGR. If you don't have that set in the bus then you need to change the `color_byte_order` in the display to `st7796.BYTE_ORDER_BGR`

kdschlosser, 2024-05-09:
This is now solved so I am closing the issue.

closed 2024-05-09 by kdschlosser