lvgl-micropython/lvgl_micropython #42: Examples and background of lvgl_micropython

ste7anste7an, 2024-05-09:
This is where I have been waiting for for a long time! A nice micropython based lvgl implementation that can be compiled for the newest MicroPython version for all kind of different platforms. I have displays based on ESP32, ESP32-SPIRAM, ESP32S3 and ESP32-C3. Only for the normal ESP32 I am able to use a 1.20 version of LVGL Micropython. 
Using this repo, I am able to compile firmware for all of my platforms. Great!

Unfortunately, mainly becuase of the lack of understanding, I do not get an example running on any of my platforms I have (based on ili9341, ST7789 and gc9a01 screen drivers and  xpt2046 and CST820 touchpad drivers).

So my questions would be: is it possible to give some background information, e.g. an outline of how the lvgl as a native module is implemented, how does it interact with the esp_lcd library, how does the DMA and bufffering works (none, single or double).  Some of this I already read in remarks and issues posted here and on the original LVGL guthub repo, but I can not make a coherent picture out of it.

What also would be helpfull, would be some basic examples for common combinations of busses (SPI, i80),, display drivers (e.g. ili9341 or ST7789) and touch controllers (xpt2048) with some hints of how to debug common problems such as running in SPI-related issues. (see e.g.  [issue #16](https://github.com/kdschlosser/lvgl_micropython/issues/16) )

I would be more than happy, to contribute to a section where we collect examples for different platforms once I get things running. In that way, the effort of @kdschlosser could be more focussed on common problems people have, rather than haphazzerdly helping people out.

kdschlosser, 2024-05-09:
We are having the same thought.. I just set this up yesterday...

https://github.com/kdschlosser/lvgl_micropython/discussions/categories/board-specific

This is where people can post their setup code for specific boards (display + MCU combinations) 

If you looks at the one that I made yesterday in the first post is the code used to get the board up and running. It will tell you exactly how to get things going.

I have only touched base with the documentation mainly because I am still working on getting everything fully running and the API has been jumping around somewhat. The API is getting more stable and there is a lot less changing going on with it. It sucks to document something and then have to change the documentation 10 times in as many days because of the API moving about.

I have given an example in the README.md file in the root of the repository. It goes into some detail on how to build and command line arguments and there is an example at the bottom.

I have not updated the README in a bit and I have added some additional command line parameters specifically for the ESP32. I have also provided the ability to package the display driver and the touch driver into the firmware that gets flashed.

For the ESP32 series of MCU's when you compile the application partition will automatically get sized properly, so no having to go into those csv files to dink about with the numbers to get things to fit. I set it up to do it for you and it does it keeping the partition size as close to the firmware size as possible, no wasted space at all.

I have also taken into consideration the ESP32-S3-N32R8. This MCU is not officially supported by micropython. I have added some command line arguments that allow you to set the flash size to 32mb and to set to flash to octal. when this is done the config options are added to alter the mode in which the flash is controlled with. The correct partition file for that flash size gets used and modified.

I am trying to keep everything as simple to use as possible. There is a single entry point for building for all ports. All of the display drivers work exactly the same across all MCU's. I have everything set up so you can develop on a desktop and test the code running micropython with LVGL on a desktop and then transfer the same script that was running on the desktop directly to the MCU and it will run with you having to make zero changes to your script (other than hardware specific stuff like accessing pins). I am still working on getting to to compile on Windows, macOS and Unix are both good to go. I have to merge the branch I have been messing with to the master for that feature to become official. I plan on doing that in the next day or 2.

kdschlosser, 2024-05-09:
The API I have written makes everything work exactly the same across all platforms. I put in a lot of work hammering that out. I didn't want users to have to make wild code changes in order to move a script from say an ESP32 over to an STM32. The only thing that would need to change are the pins. STM32 uses strings for the pins where as ESP32 uses integers.

That's all that would need to be done. This can even be coded into the script so it will know what MCU it is running on and you can have it set the pins accordingly.

kdschlosser, 2024-05-09:
I have not fully tested the SPI yet. I have to dig out my display and give it a go to see if it works. I know the RGB and the I80 work for the ESP32. and I was told that the SPI works for the STM32 (don't know if any changes were made at all to get it working)

The only MCU's that support DMA are the ESP32 line of SOC's

ste7anste7an, 2024-05-09:
Thank you for the comment and initiative to have a section on board specific configs. That will be of great help. 
I have here an RGB display laying around somewhere. I will try to get that working. In the mean time I am happy to do some tests with SPI (see issue #16) if you have ideas what I can try. I would like to get this firmware working for a basic ESP with and without PSRAM and using a normal ili9341 SPI screen with SPI touch driver XPT2046.

kdschlosser, 2024-05-09:
It should be as easy as using this command to compile...

```
python3 make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC DISPLAY=ili9341 INDEV=xpt2046
```

That should compile the firmware for ya. I would need for information about the display in order to put together the python code for ya. resolution, color depth, pin connections.. that kind of stuff.

kdschlosser, 2024-05-09:
I am going to be pushing a rather large commit. It is for adding github actions to run the builds to make sure they compile properly. I don't have a way to test the individual MCU's but at least compiling will be tested.

I reverted some recent changes I made to the SPI driver, hopefully that will fix the error you were getting.

ste7anste7an, 2024-05-09:
> It should be as easy as using this command to compile...
>
> ```
> python3 make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC DISPLAY=ili9341 INDEV=xpt2046
> ```
>
> That should compile the firmware [...]

I used exact the same line for building my formware for a unit with SPI RAM, so I used an extra parameter BOARD_VARIANT=SPIRAM 
My display is a 320x240 TFT with ILI9341 controller connected to SPI with pins:
miso=12, mosi=13, clk=14, cs=15, dc=23, rst=25
It has a hardwired backlight and power.
It uses an XPT2046 touch sensor conneced to the same SPI except for the CS=26

I already tried the example here in the issues, but regardless of which SPI controller I choose, I get SPI errors.

kdschlosser, 2024-05-09:
I am going to merge the commit once the tests pass. That will hopefully fix the error you are getting.

ste7anste7an, 2024-05-10:
I just pulled a fresh  clone of the repo and build using:
```
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046
```
It compiles fine and I flashed my board and then I use this code:
```
import lcd_bus

bus = lcd_bus.SPIBus(
    dc=23,
    sclk=14,
    mosi=13,
    miso=12,
    cs=15,
    host=2,
    freq=12000000,
)

import gc
gc.collect()
buf1 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
buf2 = bus.allocate_framebuffer(int(240 * 320 * 2 / 10), lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import ili9341
import lvgl as lv

display = ili9341.ILI9341(
    data_bus=bus,
    frame_buffer1=buf1,
    frame_buffer2=buf2,
    display_width=240,
    display_height=320,
    reset_pin=25,
    power_pin=None,
    backlight_pin=None,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

#display.set_power(1)
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
```

With host=2 I get this error:
```
Setting the bus host to 1 will give better performance
Traceback (most recent call last):
  File "<stdin>", line 41, in <module>
  File "display_driver_framework.py", line 240, in __init__
ValueError: 258(spi_bus_initialize)
```

Setting host=1 results in this error:
```
Traceback (most recent call last):
  File "<stdin>", line 41, in <module>
  File "display_driver_framework.py", line 240, in __init__
ValueError: 258(esp_lcd_new_panel_io_spi)
```

kdschlosser, 2024-05-10:
do me a favor and clone the repo again and give it a test. I added some output that should shed some light as to what is going on.

kdschlosser, 2024-05-10:
ok so scratch that last comment. I am pretty sure I found the issue. clone the repo and give it another go.

ste7anste7an, 2024-05-10:
Thanks. I have no display at hand, but I do have an esp32 to try. When I run my demo program I still get:
Traceback (most recent call last):
  File "<stdin>", line 30, in <module>
  File "display_driver_framework.py", line 240, in __init__
ValueError: 258(spi_bus_initialize)

I will try later with an LCD attached.

ste7anste7an, 2024-05-10:
Unfortunately, with the LCD display connected the outcome is exactly the same. With the same error: ValueError: 258(spi_bus_initialize)

If you could enable some debugging, then we could find where things are not going wright. I am more than happy to try that.

kdschlosser, 2024-05-10:
That error is "Invalid parameter" and I am trying to figure out what parameter is invalid. They all have the correct types being passed which is what is baffling.

It does a test for the type and also if something is `NULL`.. I have to go and look at the IDF code to see exactly where it is kicking out the invalid parameter error.

kdschlosser, 2024-05-10:
what are you using for an ESP? are you using the ESP32? or the ESP32-S3?

kdschlosser, 2024-05-10:
I just pushed a change that will hopefully fix the problem. It is rather stupid what was done and it is masking what is happening under the hood. 

I hope it will fix the problem.

ste7anste7an, 2024-05-10:
I use an ESP32-WROVER (a normal ESP32 with 4MBYte flash and 4Mbyte SPIRAM).

I will give it a try with the fix you suggest.

kdschlosser, 2024-05-10:
You should step up into the ESP32-S3. The reason why I say this is if you get the version that has 8mb of SPIRAM the ram is going to be 2 times faster than the ESP32. That makes quite a bit of difference when dealing with displays. This is because of the buffer sizes being as large as they are. You also get a hell of a lot more SRAM that is DMA'able. The SPIRAM on the S3 is also DMA'able where it is not on the ESP32. You are really limited in the frame buffer size if using the ESP32, something like a 480 x 320 x 16bit display is the maximum you can have with using double buffering. There is not enough DMA memory available to do much more than that.

ste7anste7an, 2024-05-10:
I am afraid I have not much choice, as we want to have bluetooth classic support for the hardware (not in micrpopython though), that is the main reason for using the normal ESP. This is a self-developed board for extending Lego robots.  The LVGL 1.19 micropython version from the lvgl github runs fine and fast enough on this msall 320x240 display. We want to move to 1.22 micropython, and that is the reason I thought using your nice universal project. 

Unfortunaletly, the last fix you made still results in the same error.

ste7anste7an, 2024-05-10:
I get a compilation warning/error:
```
lvgl_micropython/lib/esp-idf/components/esp_lcd/include/esp_lcd_panel_io.h:156:141: note: expected 'struct esp_lcd_panel_io_t **' but argument is of type 'esp_lcd_panel_io_handle_t' {aka 'struct esp_lcd_panel_io_t *'}
156 | esp_err_t esp_lcd_new_panel_io_spi(esp_lcd_spi_bus_handle_t bus, const esp_lcd_panel_io_spi_config_t *io_config, esp_lcd_panel_io_handle_t *ret_io);
```
Should there be an & before the last argument, in order to have it return the esp_lcd_panel_io_handle?

kdschlosser, 2024-05-10:
I believe I know what I need to do to get it to work properly...

kdschlosser, 2024-05-10:
I am going to fix the & thing. and I want you to use this for the SPIBus code

```
bus = lcd_bus.SPIBus(
    dc=23,
    sclk=14,
    mosi=13,
    miso=12,
    cs=15,
    host=-1,
    freq=80000000,
)
```

give me 3 minutes to correct the & thing

kdschlosser, 2024-05-11:
OK so the last change I made should fix i, hopefully. Instead of trying to store the host as a void pointer I am storing it as an int and then casting it to the void pointer when the function to set the bus up gets called. Little bit different plan of attach but I think it should do the trick. I am hoping it does. I will know once the CI is finished running

ste7anste7an, 2024-05-11:
The host=-1 does not work either. I tried host=1,2 with your latest fix. Still get ValueError: 258(esp_lcd_new_panel_io_spi) or ValueError: 259(spi_bus_initialize) depending on which spi host I choose. 

Does the firmware runs on any of your SPI lcd boards? Or just on a bare ESP32, to try whether the spi bus can be initialized?

kdschlosser, 2024-05-11:
I will flash an MCU later on today to see what is gong on. I have an ESP32 and also an ESP32-S3 and I will try it on both.

demitrix, 2024-05-13:
I was able to flash a recent clone and initialize everything with no errors. However all I get is a blank white screen.

ste7anste7an, 2024-05-13:
@demitrix , what kind of setup do you have? Esp32 type and which bus controller and display?

demitrix, 2024-05-13:
@ste7anste7an Generic ESP32 dev kit, Generic Red ili9341 with xpt2046, via SPI. Reflashed this morning and everything loads as it should but screen remains blank. xpt2046 is still in dev but is broken as well, get a 'bus already initialized' error if trying to use the same SPI bus.

kdschlosser, 2024-05-13:
let me see the exact error you are getting.

demitrix, 2024-05-15:
The error only relates to xpt2046, which is lower priority than a working SPI driver for the ili9341. I reflashed today in hopes of getting something displayed. Still stuck on a blank screen, and no solid examples of working code. Seems like everyone is doing everything differently lv.tick() vs taskhandler, lv.obj(scr) vs get_active_screen().

ste7anste7an, 2024-05-15:
@demitrix, you seem to get more working than I do. I keep getting SPI errors when I want to initiate the display. Could you please show your code you are using. With that I might get passed the SPI error.

demitrix, 2024-05-15:
`from micropython import const

_WIDTH = const(320)
_HEIGHT = const(240)

import lcd_bus  # NOQA
_DC_PIN = const(2)
_MOSI_PIN = const(23)
_MISO_PIN = const(19)
_SCLK_PIN = const(18)
_CS_PIN = const(15)
_FREQ = const(40000000)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    host=2,
    sclk=_SCLK_PIN,
    freq=_FREQ,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    cs=_CS_PIN,
)

import ili9341  # NOQA

_RESET_PIN = const(4)
_BACKLIGHT_PIN = const(26)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=ili9341.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=ili9341.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()
display.set_backlight(100)

import time

print('red')
scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

print('green')
scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

print('blue')
scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
lv.refr_now(lv.display_get_default())

time_passed = 1000000

while True:
    start_time = time.time_ns()
    time.sleep_ms(1000000)
    lv.tick_inc(int(time_passed // 1000000))
    time_passed -= int(time_passed // 1000000) * 1000000
    lv.task_handler()
    end_time = time.time_ns()
    time_passed += time.ticks_diff(end_time, start_time)`

I've also tried SPI host -1,0,1,2. -1 and 0 generate "ValueError: Unable to determine SPI host, please supply a host number". Also tried with and without external frame buffers.

ste7anste7an, 2024-05-15:
I run it on a regular ESP32 with ILI9341 SPI display. When usin host=1 it works better than before.  I get a grayish background with a regular pattern when running @demitrix 's code (with adapted pin nubering for my hardware). and no SPI error. When I issue a 
 ```
 lv.refr_now(lv.display_get_default())
 ```
 The background gets completely white without a pattern.

ste7anste7an, 2024-05-15:
With a ESP32-WROVER with 4 MByte of PSRAM, I get 
host=2: `ValueError: 258(spi_bus_initialize)`
host=1: `ValueError: 258(esp_lcd_new_panel_io_spi)`
host=0: `ValueError: 259(spi_bus_initialize)`
host=-1:` ValueError: 258(esp_lcd_new_panel_io_spi)`

kdschlosser, 2024-05-16:
Try it again. I made some more changes that will hopefully correct it.

demitrix, 2024-05-16:
No change on my ESP32-WROOM.

ste7anste7an, 2024-05-17:
I removed the lcgl_micropython directory and cloned the latest version. If I try to compile my SPIRAM version, I encounter this error:
```
/home/stefan/projects/esp32/lvgl_micropython/lib/lvgl/src/core/../core/../misc/../font/../misc/cache/../../osal/lv_freertos.h:31:10: fatal error: FreeRTOS.h: No such file or directory
31 | #include "FreeRTOS.h"
|          ^~~~~~~~~~~~
compilation terminated.
```

This is not present when using the commit 820cff5018412a902a0d910a3259444be2018eb7. 

Can I just do a git pull to get to the newest version, or whould I remove the directory and clone the complete repo?

demitrix, 2024-05-18:
Same result on a ES32-S3 N16R8. Blank white screen no errors.

kdschlosser, 2024-05-18:
@ste7anste7an 
 
That compile error should be fixed now.

ste7anste7an, 2024-05-19:
@kdschlosser Yes, it compiles just fine right now. But I remain getting the SPI related errors, regardless of which host interface number I choose on my ESP32-WROVER  with PSRAM.

demitrix, 2024-05-20:
> @kdschlosser Yes, it compiles just fine right now. But I remain getting the SPI related errors, regardless of which host interface number I choose on my ESP32-WROVER  with PSRAM.
> 

What errors are you getting? I've compiled and reflashed and ran multiple variations of my example above and get new errors unless using a SPI bus that's already in use. This is on an ESP32-WROOM and an ESP32S3-N16R8.

ste7anste7an, 2024-05-20:
> With a ESP32-WROVER with 4 MByte of PSRAM, I get host=2: `ValueError: 258(spi_bus_initialize)` host=1: `ValueError: 258(esp_lcd_new_panel_io_spi)` host=0: `ValueError: 259(spi_bus_initialize)` host=-1:` ValueError: 258(esp_lcd_new_panel_io_spi)`
> 
> Both my displays work with the older lv_micropython 1.19 default firmware.

@demitrix  The very same errors as above with the latest version of this repo.

demitrix, 2024-05-24:
Found a way to get a new error. Using my S3 I get no SPI errors but when using the actual FSPI pins I get 
`
W (29) boot.esp32s3: PRO CPU has been reset by WDT.
W (29) boot.esp32s3: APP CPU has been reset by WDT.
`
When display_driver_framework.py tries to initialize the display from calling 
`
display = ili9341.ILI9341(...)
`

kdschlosser, 2024-05-25:
The hosts are numbered like this...

```c
typedef enum {
//SPI1 can be used as GPSPI only on ESP32
    SPI1_HOST=0,    ///< SPI1
    SPI2_HOST=1,    ///< SPI2
#if SOC_SPI_PERIPH_NUM > 2
    SPI3_HOST=2,    ///< SPI3
#endif
    SPI_HOST_MAX,   ///< invalid host value
} spi_host_device_t;
```

SPI1_HOST = 0
SPI2_HOST = 1
SPI3_HOST = 2

and then we have these aliases.

```c
#ifdef CONFIG_IDF_TARGET_ESP32
#define SPI_HOST    SPI1_HOST
#define HSPI_HOST   SPI2_HOST
#define VSPI_HOST   SPI3_HOST
#elif CONFIG_IDF_TARGET_ESP32S2
// SPI_HOST (SPI1_HOST) is not supported by the SPI Master and SPI Slave driver on ESP32-S2 and later
#define SPI_HOST    SPI1_HOST
#define FSPI_HOST   SPI2_HOST
#define HSPI_HOST   SPI3_HOST
#endif
```

for the ESP32

SPI_HOST = 0
HSPI_HOST = 1
VSPI_HOST = 2

and for the ESP32-S2
SPI_HOST = 0
FSPI_HOST = 1
HSPI_HOST = 2

now on the ESP32 the naming makes sense.

HSPI is Hardware SPI
VSPI is Virtual SPI (software based)

On all ESP32's host 0 is not usable at all because this is reserved for flash and RAM.

The ESP32, ESP32-S2 and ESP32-S3 are the only MCU's that have an SPI2_HOST. all others do not. SPI3 which is numerically a 2 runs through the GPIO matrix. this is going to be slower because it is software based. they are not hardware pins even tho the ESP-IDF has specific pin definitions set for SPI3. It doesn't make any sense to me. I completely forgot about that when I wrote the code and I am setting a flag that says IOMUX pins are being used even when SPI3 is supplied as the host and the pins align with the ones that are defines in the ESP-IDF for SPI3.

I am gong to remove the code that sets the IOMUX flag as I believe that this is something that might get handled internally in the IDF so me doing it is pointless and me doing it incorrectly is probably what is causing the issues.

I am going to remove that code now and push the commit. I will let you know when I am done.

kdschlosser, 2024-05-25:
OK I pushed the commit. I did spot another issue and I corrected that as well. Give it a shot and see if it works.

kdschlosser, 2024-05-25:
There is something that is getting pissed off about what I am doing and IDK what it is. rather frustrating actually.

I am going to strip more out of the driver to see if we can narrow down the cause.

demitrix, 2024-05-26:
I'm not really proficient in ESP dev, but I threw some print statements in the esp32 spi_bus.c and its dying at this point
`ret = esp_lcd_new_panel_io_spi(self->bus_handle, &self->panel_io_config, &self->panel_io_handle.panel_io);`

kdschlosser, 2024-05-27:
did you pull the latest code and test it using that?

demitrix, 2024-05-27:
I pulled yesterday, I just compiled the non spiram s3 version to see what happens. I'll do a fresh pull now.

demitrix, 2024-05-28:
This may be a micropython issue https://github.com/micropython/micropython/issues/10937

kdschlosser, 2024-05-28:
@demitrix 
Change your code to this.

```Python
from micropython import const

_WIDTH = const(320)
_HEIGHT = const(240)

import lcd_bus  # NOQA
_DC_PIN = const(2)
_MOSI_PIN = const(23)
_MISO_PIN = const(19)
_SCLK_PIN = const(18)
_CS_PIN = const(15)
_FREQ = const(40000000)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    host=2,
    sclk=_SCLK_PIN,
    freq=_FREQ,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    cs=_CS_PIN,
)

import ili9341  # NOQA

_RESET_PIN = const(4)
_BACKLIGHT_PIN = const(26)
_OFFSET_X = const(0)
_OFFSET_Y = const(0)

import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=_RESET_PIN,
    reset_state=ili9341.STATE_HIGH,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=ili9341.STATE_HIGH,
    offset_x=_OFFSET_X,
    offset_y=_OFFSET_Y,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.init()
display.set_backlight(100)

import time

print('red')
scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0xFF0000), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

print('green')
scrn.set_style_bg_color(lv.color_hex(0x00FF00), 0)
lv.refr_now(lv.display_get_default())
time.sleep_ms(5000)

print('blue')
scrn.set_style_bg_color(lv.color_hex(0x0000FF), 0)
lv.refr_now(lv.display_get_default())

import task_handler

th = task_handler.TaskHandler()
```

kdschlosser, 2024-05-28:
There is an issue in MicroPython with using sleep and it not releasing the FreeRTOS task to allow the WDT to reset itself. That causes the WDT to think the system has hung

demitrix, 2024-05-28:
Above code results in the same CPU resets since we aren't even making it past 
`display = ili9341.ILI9341(...)`

demitrix, 2024-05-28:
Switched out to 5.1.4 and got this 
```
make[1]: warning: -j14 forced in submake: resetting jobserver mode.
make[3]: /bin/sh: Argument list too long
make[3]: *** [esp-idf/main_esp32s3/CMakeFiles/BUILD_FROZEN_CONTENT.dir/build.make:636: genhdr/qstr.i.last] Error 127
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:4877: esp-idf/main_esp32s3/CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
Compiler supported targets: xtensa-esp32s3-elf

make[1]: *** [Makefile:136: all] Error 2
```

kdschlosser, 2024-05-28:
did you forget to add the mpy_cross argument?

kdschlosser, 2024-05-28:
and how did you go about switching it out? It's not a simple as running git against the submodule to checkout a different version.

demitrix, 2024-05-28:
Updated the relevant python scripts make.py and esp32.py, switched submodule version and updated its submodules.

kdschlosser, 2024-05-28:
what board are you using?

demitrix, 2024-05-28:
Exact board here
https://www.amazon.com/dp/B0CKXJLP4B?ref=ppx_yo2ov_dt_b_product_details&th=1

kdschlosser, 2024-05-28:
you made a custom board file and config file for it?

[26 more replies]

closed 2024-06-19 by kdschlosser