lvgl-micropython/lvgl_micropython #44: Touch control fails after rotation

sasakihin, 2024-05-14:
yesterday i build for WT32-SC01 Plus 
use discussions [WT32-SC01 Plus](https://github.com/kdschlosser/lvgl_micropython/discussions/40) code to test
```
Traceback (most recent call last):
  File "<stdin>", line 98, in <module>
  File "pointer_framework.py", line 36, in calibrate
  File "touch_cal_data.py", line 98, in reset
  File "touch_cal_data.py", line 78, in top
OverflowError: overflow converting long int to machine word
```
then i test ratation and touch
```
# WT32-SC01-Plus

import lcd_bus

from micropython import const

# display settings
_WIDTH = const(320)
_HEIGHT = const(480)
_BL = const(45)
_RST = const(4)
_DC = const(0)
_WR = const(47)
_FREQ = const(20000000)
_DATA0 = const(9)
_DATA1 = const(46)
_DATA2 = const(3)
_DATA3 = const(8)
_DATA4 = const(18)
_DATA5 = const(17)
_DATA6 = const(16)
_DATA7 = const(15)
_BUFFER_SIZE = const(30720)

_SCL = const(5)
_SDA = const(6)
_TP_FREQ = const(100000)

display_bus = lcd_bus.I80Bus(
    dc=_DC,
    wr=_WR,
    freq=_FREQ,
    data0=_DATA0,
    data1=_DATA1,
    data2=_DATA2,
    data3=_DATA3,
    data4=_DATA4,
    data5=_DATA5,
    data6=_DATA6,
    data7=_DATA7
)

# you can optionally allocate the frame buffers like seen below. Or you can
# let the display driver make them the choice is yours. The display driver 
# is going to make decisions based in the display bus being used as well as 
# the display size to figure out what the best buffer size needs to be. It 
# will use double buffering when possible.

fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

import st7796  # NOQA
import lvgl as lv  # NOQA

lv.init()

display = st7796.ST7796(
    data_bus=display_bus,
    frame_buffer1=fb1,
    frame_buffer2=fb2,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    backlight_pin=_BL,
    reset_pin=_RST,
    reset_state=st7796.STATE_LOW,
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=st7796.BYTE_ORDER_BGR,
    rgb565_byte_swap=True,
)

# send the initialization commands
display.init()

# rotate the display to get landscape.
display.set_rotation(lv.DISPLAY_ROTATION._90)

# Turn on the display backlight.
display.set_backlight(100)

display.invert_colors = True

import i2c  # NOQA
import task_handler  # NOQA
import ft6x36  # NOQA
import time  # NOQA

# the indev drivers use a special form of i2c that must be used.
i2c_bus = i2c.I2CBus(scl=_SCL, sda=_SDA, freq=_TP_FREQ, use_locks=False)
indev = ft6x36.FT6x36(i2c_bus)

th = task_handler.TaskHandler()

def before_task_handler_cb(evt, user_data):
    # place code here that you want to run before the task handler gets called.
    # returning nothing or True will run the task handler, returning False
    # will skip running the task handler
    return True  

def after_task_handler_cb(evt, user_data):
    # place code here that you want to run after the task handler gets called.
    pass
    

#th.add_event_cb(task_handler.TASK_HANDLER_STARTED, None)
#th.add_event_cb(task_handler.TASK_HANDLER_FINISHED, None)

# The task handler above handles collecting the touch input and flushing 
# the frame buffer. If you need to have looping code to check pin states or 
# to perform other tasks you can either use a timer that is built into LVGL, 
# or the task handler callbacks above or you can create your own loop 
# after the task handler has started. 
# 
# Use of the task handler above is optional and you can create your own 
# loop that handles incrementing the tick and calling the task handler. 
# The choice is yours.  

# replace the code below with your own code. 
scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(300, 50)

slider.center()

label = lv.label(scrn)
label.set_text('HELLO WORLD!')
label.align(lv.ALIGN.CENTER, 0, -50)
```
The touch control still doesn’t rotate, and the colors are wrong.

https://github.com/kdschlosser/lvgl_micropython/assets/41147444/97c2df24-6be4-4eab-8e20-c314b45e74bb

kdschlosser, 2024-05-15:
yes I know the rotation is off and I am going to be working on that shortly. I am am not sure why it is not working. maybe the event isn't registering correctly... I am not 100% sure. I will test it this evening to see what is going on. I submitted an issue in MicroPython about stuffing a signed long into a signed int and a missing cast that should be there I am not sure when it will get fixed.

kdschlosser, 2024-05-15:
https://github.com/micropython/micropython/issues/14488

kdschlosser, 2024-05-15:
the rotation portion I am messing about with now to see why it's not working.

kdschlosser, 2024-05-16:
I changed the rotation see if it works now.

sasakihin, 2024-05-23:
> I changed the rotation see if it works now.

i build *.bin today ,it worse regardless of whether it rotates or not,touch no response

kdschlosser, 2024-05-23:
OK don't worry about the touch for the time being. I am still trying to get the bus drivers working properly. There are changes that have been made to the RGB driver and also the SPI driver that I need to know if they are functioning properly now.

kdschlosser, 2024-05-23:
I will get the touch situated once I know the bus drivers are working. Cannot really mess with the touch if there is nothing being displayed right?

marksull, 2024-06-09:
@sasakihin did you happen to test this recently with all the changes made over the last month? Does the touch layer now match the display layer because I was seeing the same issue? I might get it a build and test today...

kdschlosser, 2024-06-09:
The rotation has been tested and it is functioning properly.

https://github.com/kdschlosser/lvgl_micropython/issues/47#issuecomment-2152675574

marksull, 2024-06-09:
I am seeing the touch layer is out by 180 degrees when the using the landscape modes of `display.set_rotation(lv.DISPLAY_ROTATION._90)` and `display.set_rotation(lv.DISPLAY_ROTATION._270)`. @sasakihin do you see the same on the WT32-SC01 Plus?

This is with the latest build, with MP 1.23.0 with LV 9.0.1.

kdschlosser, 2024-06-09:
and how are you using LV 9.0.1?? the build system handles using the latest master branch code. If you are initializing the submodules you shouldn't be. If you are using an old version of LVGL that could be the cause of the rotation being off by 180°

kdschlosser, 2024-06-09:
I also want to make sure you are not doing something like changing the width and height at all when you do the rotation. Those need to remain at the native resolution for rotation 0. if those get changed for some reason it could cause an issue as well. so make sure you are setting the correct width and height when constructing the display driver and there is no reason to ever change those values. It gets done internally as needed.

kdschlosser, 2024-06-09:
Now that I am thinking about it I have to check and see what direction the display display drivers are doing the rotation compared to LVGL. If LVGL rotates to the right but the display driver is rotating to the left this could cause the issue you are having. 

Not sure I have too much of a choice with that though. Might need to have LVGL updated so it can be instructed which way the rotation should occur.

marksull, 2024-06-09:
> and how are you using LV 9.0.1?? the build system handles using the latest master branch code. If you are initializing the submodules you shouldn't be. If you are using an old version of LVGL that could be the cause of the rotation being off by 180°

Bad (or good) habit. Always lock my versions. Good call. I will update and test again. Sorry about that.

kdschlosser, 2024-06-09:
I am running against the master branch because there are issues that have come up that need fixing in LVGL in order for this binding to work properly. If I could lock it to a version where there were no issues I would. haven't gotten to that point yet.

kdschlosser, 2024-06-09:
I am all for locking it to a known good version, that is the best way to go. Just haven't been able to get there yet.

kdschlosser, 2024-06-09:
this is the code in the master branch of LVGL that hands the remapping of the touch coordinates. for some crazy reason this looks wrong to me and I have not been able to put my finger on why.

```c
if(disp->rotation == LV_DISPLAY_ROTATION_180 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    data->point.x = disp->hor_res - data->point.x - 1;
    data->point.y = disp->ver_res - data->point.y - 1;    // this is not supposed to be done for 270 rotation

if(disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    int32_t tmp = data->point.y;
    data->point.y = data->point.x;
    data->point.x = disp->ver_res - tmp - 1;     // gets flipped back
}
```

kdschlosser, 2024-06-09:
besides the fact that it is doing extra work for no really good reason. Maybe that is what my brain is not liking.

kdschlosser, 2024-06-09:
this code would solve that issue.

```c
if (disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_180) {
    data->point.y = disp->ver_res - data->point.y - 1;
}

if (disp->rotation == LV_DISPLAY_ROTATION_180 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    data->point.x = disp->hor_res - data->point.x - 1;
}

if (disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    int32_t tmp = data->point.y;
    data->point.y = data->point.x;
    data->point.x = tmp;
}
```

marksull, 2024-06-09:
Confirmed with latest LVGL 9.1.0 touch layer is still inverted.

kdschlosser, 2024-06-09:
When you rotate the display 90° does it rotate the display to the left or the right?

marksull, 2024-06-09:
Right (clockwise)

kdschlosser, 2024-06-09:
I updated the display drivers. It should correct the problem (hopefully)

marksull, 2024-06-09:
Recompiled, and reran my test code, but now get this:
```
>>> import version
LVGL Version: 9.1.0
>>> import rotation_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rotation_test.py", line 52, in <module>
  File "st7796.py", line 34, in <module>
  File "st7796.py", line 58, in ST7796
AttributeError: 'module' object has no attribute 'MADCTL_MX'
```

kdschlosser, 2024-06-09:
give it a try again. I just updated it.

marksull, 2024-06-09:
Similar, but not the same:

kdschlosser, 2024-06-10:
doah!!! I am in idiot. I forgot those parameters are const and cannot be used between modules. I have to duplicate them. Not a big deal. just gotta copy and paste them.

kdschlosser, 2024-06-10:
ok it now should be fixed

marksull, 2024-06-10:
This has had an unintended side-effect. The screen is now back-to-front. So the words are now displayed backwards (as if I am looking from the wrong side of the screen, if that makes sense). Just to make sure this wasn't an issue on the previous code and I just didn't notice it, I went back to commit a2ec3a1596618b9b5f6eab368109297d58e83ebb (just before the changes) and rebuilt an confirm that the words are displayed in the correct direction

kdschlosser, 2024-06-19:
want to give this another go and see if it works now?

marksull, 2024-06-19:
The text direction is still displayed backwards when rotated 90 degrees, although, the original problem which was the touch layer wasn't rotated is now fixed!!

[image]

marksull, 2024-06-22:
Here is a summary of Touch/Display layers for the ST7796 compared to the latest commit and the commit immediately before all the layer changes a couple weeks back when you first started looking at this issue. I didn't realise that the everything is displayed backwards on all rotations (because I was only testing on 90 to see if the problem was fixed) but when I tested all rotations it became evident.

[image]

kdschlosser, 2024-06-22:
shoot me a link to the commit where the text direction was correct.

marksull, 2024-06-22:
It is linked above a couple of comments back, commit a2ec3a1

marksull, 2024-07-25:
@kdschlosser this is the issue that was discussing the backward text before you summarised it in https://github.com/kdschlosser/lvgl_micropython/issues/65. It was for the ST7796.

kdschlosser, 2024-12-03:
I believe these issues have been sorted out and I am closing this issue. If the issue still exists lemme know and I will open the issue back up.

closed 2024-12-03 by kdschlosser