lvgl-micropython/lvgl_micropython #46: Build errors

AndreoBotelho, 2024-05-29:
While building for STM32 I got the errors:
```
Aborting
fatal: Unable to checkout '928df866e4d287ebc3c60726151513ebee609128' in submodule path '../../lib/stm32lib'
make: *** [../../py/mkrules.mk:254: submodules] Error 1
make: *** Waiting for unfinished jobs....
```
fixed with manual checkout and then

```
build/lv_mpy.c:33:10: fatal error: /home/andre/stm32/new/lvgl_micropython/build/lvgl_header.h: No such file or directory
33 | #include "/home/andre/stm32/new/lvgl_micropython/build/lvgl_header.h"
|          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

fixed copying the folder from old repo and then

```
/usr/bin/ld: build/py/qstr.o: warning: relocation against `mp_qstr_frozen_const_pool' in read-only section `.text.qstr_init'
/usr/bin/ld: build/py/qstr.o: in function `qstr_init':
qstr.c:(.text.qstr_init+0x7): undefined reference to `mp_qstr_frozen_const_pool'
```

mpy_cross is not building, I dont know how to fix.

kdschlosser, 2024-05-29:
are you adding mpy_cross to the build command?

`python3 make.py stm32 clean submodules mpy_cross BOARD=.... DISPLAY=.... INDEV=....`

SPI touch interfaces are not going to work because of how the SPI driver has been written in MicroPython. I am dealing with that now in the esp32 port which has caused me to write a new SPI driver and I am going to be writing a new SDCard driver as well. It has to do with MicroPython's SPI needing to initialize the SPI Bus. If it doesn't do that and something else does it will spit back an error.

AndreoBotelho, 2024-05-29:
> are you adding mpy_cross to the build command?
>
> `python3 make.py stm32 clean submodules mpy_cross BOARD=.... DISPLAY=.... INDEV=....`
>
> SPI touch interfaces are not going to work because of how the [...]

this error is during mpy-cross compilation

```
/usr/bin/ld: build/py/qstr.o: warning: relocation against `mp_qstr_frozen_const_pool' in read-only section `.text.qstr_init'
/usr/bin/ld: build/py/qstr.o: in function `qstr_init':
qstr.c:(.text.qstr_init+0x7): undefined reference to `mp_qstr_frozen_const_pool'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[1]: *** [../py/mkrules.mk:233: build/mpy-cross] Error 1
make: *** [../../py/mkrules.mk:179: ../../mpy-cross/build/mpy-cross] Error 2
```

AndreoBotelho, 2024-05-29:
the problem was previous compilation, make.py doesnt clean the mpy-cross build so I had to clean with make and compile again to fix the linking step. The build worked fine after fix these problems.

kdschlosser, 2024-05-29:
Ahhh OK. so I need to also call clean on the myp_cross if clean and mpy_cross are both in the command. I gotcha. I didn't even think about cleaning mpy_cross at all. 

I will fix that now.

kdschlosser, 2024-05-29:
OK that should now be fixed. so when `clean` and `mpy_cross` are provided as build arguments it will also clean and rebuild mpy_cross. If only `clean` is given then only the build will be cleaned and rebuilt not mpy-cross

closed 2024-06-19 by kdschlosser