lvgl-micropython/lvgl_micropython #47: Adapt XPT2046 driver for custom SPI

demitrix, 2024-05-30:
Due to work done under https://github.com/kdschlosser/lvgl_micropython/issues/42 XPT2046 driver needs to be adapted to use new custom SPI implementation. Creating this issue as a place to track that progress.

kdschlosser, 2024-05-30:
OK cool. I am working on it now. I am updating the lcd_bus driver so instead of passing pin numbers to it you are going to pass an spi.Bus instance to it. I am also going to do the same thing for any touch drivers that use SPI. I have to figure out a way to stall the main loop until a response is gotten form the SPI. I am thinking a lock object will do the trick but the issue there is that ends up being ESP specific python code. I have to figure out what MCU's support the threading module.

demitrix, 2024-05-31:
Quite a bit of low level protocol is over my head, but I have noticed that CircuitPython has a try_lock() method of their SPI class. Might be able to lend you some insight.

kdschlosser, 2024-05-31:
I have pushed some code changes is you want to give it a try. I am sure there is bound to be errors and I will fix those as we come across them. It is compiling as per the CI.

demitrix, 2024-05-31:
Firmware builds. I dug in a bit, but I'm missing how to instantiate the spi bus object to pass to lcd_bus and xpt2046. It's not this
```
spi_bus = spi.Bus(
    host=1,
    sclk=_SCLK_PIN,    
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    spi_bus = spi_bus,
    freq=_FREQ,
    cs=_CS_PIN,
)
```

kdschlosser, 2024-05-31:
OoOO you have backtrace data. That's good. Lets decode that and see what it says.

```
~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-addr2line -fe lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.elf 0x420adc45:0x3fced3f0 0x420c9b4d:0x3fced4a0 0x420d0909:0x3fced4d0 0x420d09cd:0x3fced4f0 0x4037b0a9:0x3fced510 0x420c9aec:0x3fced5b0 0x420d0909:0x3fced5e0 0x420d09cd:0x3fced600 0x4037b0a9:0x3fced620 0x420c9aec:0x3fced6c0 0x420d0909:0x3fced6f0 0x420d09cd:0x3fced710 0x4037b0a9:0x3fced730 0x420c9aec:0x3fced7d0 0x420d0909:0x3fced800 0x420fc6ab:0x3fced820 0x420fc6fd:0x3fced860 0x420d0909:0x3fced880 0x420270a1:0x3fced8a0 0x42051a6b:0x3fced8e0 0x42051aca:0x3fced910 0x42051a81:0x3fced960 0x420839f1:0x3fced990 0x420835e1:0x3fced9d0 0x42006bdd:0x3fceda10 0x42027f2d:0x3fceda30 0x42006d13:0x3fceda70 0x420d0909:0x3fcedab0 0x420d09cd:0x3fcedad0 0x4037b0a9:0x3fcedaf0 0x420c9aec:0x3fcedb90 0x420d0909:0x3fcedbc0 0x420fc6ab:0x3fcedbe0 0x420fc6fd:0x3fcedc20 0x420d0909:0x3fcedc40 0x420d098d:0x3fcedc60 0x420d1530:0x3fcedc90 0x420d16a5:0x3fcedd10 0x420e4c04:0x3fcedd30 0x421044ed:0x3fcedd50 0x420dd143:0x3fcedd70 0x420e4924:0x3fceddb0
```

copy and paste the above line to the shell prompt. This needs to be done from the root directory of the repo. Then paste the output to me and lets see what it says.

demitrix, 2024-05-31:
```
esp32_hw_spi_dev_comm
??:?
fun_builtin_var_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objfun.c:114
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:729
mp_execute_bytecode
/home/demitrix/lvgl_micropython/lib/micropython/py/vm.c:1042
fun_bc_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objfun.c:273
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:729
mp_execute_bytecode
/home/demitrix/lvgl_micropython/lib/micropython/py/vm.c:1042
fun_bc_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objfun.c:273
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:729
mp_execute_bytecode
/home/demitrix/lvgl_micropython/lib/micropython/py/vm.c:1042
fun_bc_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objfun.c:273
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_self_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/objboundmeth.c:70
bound_meth_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objboundmeth.c:83
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
lv_indev_t_read_cb_callback
lv_mp.c:?
indev_read_core
lv_indev.c:?
lv_indev_read
??:?
lv_indev_read_timer_cb
??:?
lv_timer_exec
lv_timer.c:?
lv_timer_handler
??:?
lv_task_handler
lv_mp.c:?
mp_lv_task_handler
lv_mp.c:?
lv_fun_builtin_var_call
lv_mp.c:?
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:729
mp_execute_bytecode
/home/demitrix/lvgl_micropython/lib/micropython/py/vm.c:1042
fun_bc_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objfun.c:273
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_method_self_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/objboundmeth.c:70
bound_meth_call
/home/demitrix/lvgl_micropython/lib/micropython/py/objboundmeth.c:83
mp_call_function_n_kw
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:713
mp_call_function_1
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime.c:691
mp_call_function_1_protected
/home/demitrix/lvgl_micropython/lib/micropython/py/runtime_utils.c:33
mp_sched_run_pending
/home/demitrix/lvgl_micropython/lib/micropython/py/scheduler.c:111
mp_hal_stdin_rx_chr
/home/demitrix/lvgl_micropython/lib/micropython/ports/esp32/mphalport.c:124
readline
/home/demitrix/lvgl_micropython/lib/micropython/shared/readline/readline.c:560
pyexec_friendly_repl
/home/demitrix/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:610
mp_task
/home/demitrix/lvgl_micropython/lib/micropython/ports/esp32/main.c:159
```

kdschlosser, 2024-06-01:
OK so the error is happening in the esp32_hw_spi_dev_comm function.

kdschlosser, 2024-06-01:
hey, I made some changes to how things work. I am attempting to work around the issues with the SPI. I believe this should work.

Here is some example code. Things need to be done in this exact order so don't move anything around at all. You need to set the CS and the interrupt pins for the touch IC, if applicable. 

``` python
from micropython import const  # NOQA
import machine # NOQA

_WIDTH = const(320)
_HEIGHT = const(240)

_TP_FREQ = const(1_000_000)
_TP_CS = const(-1)  # CHANGE THIS TO THE CS PIN FOR THE TOUCH PANEL
_TP_INT = const(-1)  # CHANGE THIS TO THE INTERRUPT PIN FOR THE TOUCH PANEL

_SPI_HOST = const(1)
_DC_PIN = const(7)
_MOSI_PIN = const(11)
_MISO_PIN = const(13)
_SCLK_PIN = const(12)
_CS_PIN = const(10)
_RESET_PIN = const(6)
_BACKLIGHT_PIN = const(5)
_FREQ = const(40_000_000)

import lcd_bus  # NOQA

spi_bus = machine.SPI(
    _SPI_HOST,
    _TP_FREQ,
    miso=_MISO_PIN,
    mosi=_MOSI_PIN,
    sck=_SCLK_PIN,
)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    host=_SPI_HOST,
    sclk=_SCLK_PIN,
    freq=_FREQ,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    cs=_CS_PIN,
)

import ili9341  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET_PIN,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=ili9341.STATE_HIGH,
)

display.init()
display.set_backlight(100)

import xpt2046  # NOQA

indev = xpt2046.XPT2046(spi_bus, _TP_CS, _TP_INT)

import lvgl as lv  # NOQA

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(_WIDTH - 50, 50)
slider.center()

import task_handler  # NOQA

th = task_handler.TaskHandler()
```

demitrix, 2024-06-01:
Everything runs, but I can't tell if touch is doing anything, There is no console output when touching the screen. I assume at the least I should see some SPI debug output for devices locking and releasing the bus.

kdschlosser, 2024-06-01:
Please copy and paste the code you are running into a post here.

kdschlosser, 2024-06-01:
and are you getting anything on the display? can you see the slider being displayed when running the code?

demitrix, 2024-06-02:
Yeah slider loads on the screen. I'm using the code you posted above, only I swapped height and width and added rotation.

kdschlosser, 2024-06-02:
don't add any rotation at all. 

is the height and width correct if you don't change the rotation

kdschlosser, 2024-06-02:
the slider might not be working because of a glitch in the rotation end of things. I just want to know at this point is the touch is working or not. I will deal with the rotation later on.

kdschlosser, 2024-06-02:
I have no doubt the hardware works fine. I am working through bugs in the binding. Things are handled differently in this binding then using something like the Arduino IDE. That's because there is no DMA memory use and the espiodf port for the arduino IDE is an antique so the esp_lcd component is not being used. A lot of the code in the tft_espi library is not going to offer the best performance wise.

demitrix, 2024-06-03:
[image]

This is the result with no changes to code.

[image]

Flipping width and height, no rotation applied.

kdschlosser, 2024-06-03:
ok so if you ONLY swap the width and height what happens?

kdschlosser, 2024-06-03:
OK then that is what the native width and height are with the rotation set to 0 degrees. That is what your code should be always when you start the display.

Now.. There is still no touch inputs taking place. Are you using the interrupt pin?

demitrix, 2024-06-03:
Yes, I tried with interrupt defined and set to -1. I haven't looked to see if theres any logging around the interrupt function, so I'm not sure its being caught.

kdschlosser, 2024-06-03:
OK we are going to mess around with the driver some.

run this code to test if we get anything on the touch. The display will be blank but you should still be able to get touch input

```python
"""XPT2046 Touch module."""
from time import sleep
from micropython import const

_WIDTH = const(240)
_HEIGHT = const(320)

_TP_FREQ = const(1_000_000)
_TP_CS = const(-1)  # CHANGE THIS TO THE CS PIN FOR THE TOUCH PANEL
_TP_INT = const(-1)  # CHANGE THIS TO THE INTERRUPT PIN FOR THE TOUCH PANEL

_SPI_HOST = const(1)
_MOSI_PIN = const(11)
_MISO_PIN = const(13)
_SCLK_PIN = const(12)

class Touch(object):
    """Serial interface for XPT2046 Touch Screen Controller."""

    # Command constants from ILI9341 datasheet
    GET_X = const(0xD0)  # X position
    GET_Y = const(0x90)  # Y position
    GET_Z1 = const(0xB0)  # Z1 position
    GET_Z2 = const(0xC0)  # Z2 position
    GET_TEMP0 = const(0x80)  # Temperature 0
    GET_TEMP1 = const(0xF0)  # Temperature 1
    GET_BATTERY = const(0xA0)  # Battery monitor
    GET_AUX = const(0xE0)  # Auxiliary input to ADC

    def __init__(self, spi, cs, int_pin=None, int_handler=None,
                 width=240, height=320,
                 x_min=100, x_max=1962, y_min=100, y_max=1900):
        """Initialize touch screen controller.

        Args:
            spi (Class Spi):  SPI interface for OLED
            cs (Class Pin):  Chip select pin
            int_pin (Class Pin):  Touch controller interrupt pin
            int_handler (function): Handler for screen interrupt
            width (int): Width of LCD screen
            height (int): Height of LCD screen
            x_min (int): Minimum x coordinate
            x_max (int): Maximum x coordinate
            y_min (int): Minimum Y coordinate
            y_max (int): Maximum Y coordinate
        """
        self.spi = spi
        self.cs = cs
        self.cs.init(self.cs.OUT, value=1)
        self.rx_buf = bytearray(3)  # Receive buffer
        self.tx_buf = bytearray(3)  # Transmit buffer
        self.width = width
        self.height = height
        # Set calibration
        self.x_min = x_min
        self.x_max = x_max
        self.y_min = y_min
        self.y_max = y_max
        self.x_multiplier = width / (x_max - x_min)
        self.x_add = x_min * -self.x_multiplier
        self.y_multiplier = height / (y_max - y_min)
        self.y_add = y_min * -self.y_multiplier

        if int_pin is not None:
            self.int_pin = int_pin
            self.int_pin.init(int_pin.IN)
            self.int_handler = int_handler
            self.int_locked = False
            int_pin.irq(trigger=int_pin.IRQ_FALLING | int_pin.IRQ_RISING,
                        handler=self.int_press)

    def get_touch(self):
        """Take multiple samples to get accurate touch reading."""
        timeout = 2  # set timeout to 2 seconds
        confidence = 5
        buff = [[0, 0] for x in range(confidence)]
        buf_length = confidence  # Require a confidence of 5 good samples
        buffptr = 0  # Track current buffer position
        nsamples = 0  # Count samples
        while timeout > 0:
            if nsamples == buf_length:
                meanx = sum([c[0] for c in buff]) // buf_length
                meany = sum([c[1] for c in buff]) // buf_length
                dev = sum([(c[0] - meanx)**2 +
                          (c[1] - meany)**2 for c in buff]) / buf_length
                if dev <= 50:  # Deviation should be under margin of 50
                    return self.normalize(meanx, meany)
            # get a new value
            sample = self.raw_touch()  # get a touch
            if sample is None:
                nsamples = 0    # Invalidate buff
            else:
                buff[buffptr] = sample  # put in buff
                buffptr = (buffptr + 1) % buf_length  # Incr, until rollover
                nsamples = min(nsamples + 1, buf_length)  # Incr. until max

            sleep(.05)
            timeout -= .05
        return None

    def int_press(self, pin):
        """Send X,Y values to passed interrupt handler."""
        if not pin.value() and not self.int_locked:
            self.int_locked = True  # Lock Interrupt
            buff = self.raw_touch()

            if buff is not None:
                x, y = self.normalize(*buff)
                self.int_handler(x, y)
            sleep(.1)  # Debounce falling edge
        elif pin.value() and self.int_locked:
            sleep(.1)  # Debounce rising edge
            self.int_locked = False  # Unlock interrupt

    def normalize(self, x, y):
        """Normalize mean X,Y values to match LCD screen."""
        x = int(self.x_multiplier * x + self.x_add)
        y = int(self.y_multiplier * y + self.y_add)
        return x, y

    def raw_touch(self):
        """Read raw X,Y touch values.

        Returns:
            tuple(int, int): X, Y
        """
        x = self.send_command(self.GET_X)
        y = self.send_command(self.GET_Y)
        if self.x_min <= x <= self.x_max and self.y_min <= y <= self.y_max:
            return (x, y)
        else:
            return None

    def send_command(self, command):
        """Write command to XT2046 (MicroPython).

        Args:
            command (byte): XT2046 command code.
        Returns:
            int: 12 bit response
        """
        self.tx_buf[0] = command
        self.cs(0)
        self.spi.write_readinto(self.tx_buf, self.rx_buf)
        self.cs(1)

        return (self.rx_buf[1] << 4) | (self.rx_buf[2] >> 4)

import machine

if _TP_INT != -1:
    i_pin = machine.Pin(_TP_INT, machine.Pin.IN)
else:
    i_pin = None

if _TP_CS != -1:
    cs_pin = machine.Pin(_TP_CS, machine.Pin.OUT)
    cs_pin.value(1)
else:
    cs_pin = None
    
    
def callback(x, y):
    print(x, y)
    

spi_bus = machine.SPI(_SPI_HOST, _TP_FREQ, miso=_MISO_PIN, mosi=_MOSI_PIN, sck=_SCLK_PIN)
tp = Touch(spi_bus, cs_pin, i_pin, callback, width=_WIDTH, height=_HEIGHT)
```

kdschlosser, 2024-06-03:
does your touch screen have a reset pin?

demitrix, 2024-06-03:
No, its just the generic red ili9341/xpt2046 combo
http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807

kdschlosser, 2024-06-03:
ok I updates the repo again. clone it and recompile and then run it again using the code below and see if it works 

```python
from micropython import const  # NOQA
import machine # NOQA

_WIDTH = const(240) 
_HEIGHT = const(320)

_TP_FREQ = const(1_000_000)
_TP_CS = const(-1)  # CHANGE THIS TO THE CS PIN FOR THE TOUCH PANEL

_SPI_HOST = const(1)
_DC_PIN = const(7)
_MOSI_PIN = const(11)
_MISO_PIN = const(13)
_SCLK_PIN = const(12)
_CS_PIN = const(10)
_RESET_PIN = const(6)
_BACKLIGHT_PIN = const(5)
_FREQ = const(40_000_000)

import lcd_bus  # NOQA

spi_bus = machine.SPI(
    _SPI_HOST,
    _TP_FREQ,
    miso=_MISO_PIN,
    mosi=_MOSI_PIN,
    sck=_SCLK_PIN,
    cs=_TP_CS
)

display_bus = lcd_bus.SPIBus(
    dc=_DC_PIN,
    host=_SPI_HOST,
    sclk=_SCLK_PIN,
    freq=_FREQ,
    mosi=_MOSI_PIN,
    miso=_MISO_PIN,
    cs=_CS_PIN,
)

import ili9341  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_RESET_PIN,
    backlight_pin=_BACKLIGHT_PIN,
    backlight_on_state=ili9341.STATE_HIGH,
)

display.init()
display.set_backlight(100)

import xpt2046  # NOQA

indev = xpt2046.XPT2046(spi_bus)

import lvgl as lv  # NOQA

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(_WIDTH - 50, 50)
slider.center()

import task_handler  # NOQA

th = task_handler.TaskHandler()
```

kdschlosser, 2024-06-03:
I changed the code to use polling instead of interrupt based. The interrupt adds a layer of complexity that is hard to work around. So lets give this a try and see if it works.

demitrix, 2024-06-03:
Small typo on line 252 of micropy_updates/esp32/machine_hw_spi.c, you have : at the end. 

No-go on that. The xpt device is locking the SPI and not releasing it.
`D (3148) spi_master: device0 locked the bus`
Is the last console line, nothing gets drawn on the screen and console is unresponsive.

kdschlosser, 2024-06-04:
ahh ok I forgot to free the bus.

kdschlosser, 2024-06-04:
ok I just pushed a commit that should fix those issues.

demitrix, 2024-06-04:
Same issues here. I changed out for ticks.us() and set time to the equivalent 5000. Everything runs, I added logging to get_coords, and it is returning coords. Can't interact with the slider though, not sure if its a calibration issue, screen update issue, or something else. Might try to replcae the slider with a button and see if even a simple click registers.

kdschlosser, 2024-06-04:
So you are seeing the driver respond to touches now? Correct?

If yes that is a step in the right direction and the problem isn't going to be specific to that driver anymore. I should be able to test locally now and see what is going on and if the problem is happening with other drivers.

You aren't rotating the screen are you? I know that the rotation is not working properly at the moment and that does need fixing I am trying to just get the touch working and then I will mess with why it's breaking when rotating.

demitrix, 2024-06-04:
Yeah I had to throw in a print line in get_coords to catch it, but its sending back values and not Nones so its working. No rotation set, wasn't really expecting touch interactivity to work anyhow, theres calibration and other other issues at play. I need to put together a trial to make sure the screen can still update while touch is working.

kdschlosser, 2024-06-04:
I am curious to know what is being recorded as the touch coordinates. Do you have an example of that?

demitrix, 2024-06-04:
```
print(self.PRESSED, x, y)
return self.PRESSED, x, y
```
Just added the print statement to see under the hood. Using my displays "native" orientation
[image]
| Position  | Approximate X | Approximate Y |
| ------------- | ------------- | ------------- |
| TL  | 20 | 295 |
| TR  | 210 | 295 |
| BR  | 210  | 40 |
| BL  | 20 | 40 |

kdschlosser, 2024-06-04:
It's not supposed to be returning values if you are not touching the display. So that's an issue I need to figure out why it is doing it.

It has to be something with setting up the touch panel. I have to dig through the data sheet for the touch IC to see how to turn off interrupt mode or how to check to see if there is touch that is actually taking place.

demitrix, 2024-06-04:
It's not sending anything without touches. It seems to be working just fine.

kdschlosser, 2024-06-05:
I did update the driver so now it will check to see if there is a Z axis. If there is no Z axis above a certain threshold then it will not register as a touch input.

As far as a string of coordinates being generated that is fine to have happen. What happens is LVGL loops checking the touch panel and updating the display on each loop. It doesn't redraw the display on each loop it will only read the input and change values for widgets etc... I can tell it not to do this but I want to see how it performs the way that it is now. If the touch input is kind of crappy then I will tell it to exit the loop.

ste7anste7an, 2024-06-05:
I tried to build the latest clean repo this morning using this command:
```
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046
```
and I get this error:

```
MPY pointer_framework.py
error compiling pointer_framework.py:
Traceback (most recent call last):
  File "pointer_framework.py", line 12, in _remap
ViperTypeError: binary op  not implemented
```

Any suggestions what could be wrong?

ikkesia, 2024-06-05:
Error whit last commit and the usual "slider" test code
```
Traceback (most recent call last):
  File "<stdin>", line 66, in <module>
  File "xpt2046.py", line 4, in <module>
  File "pointer_framework.py", line 4, in <module>
ImportError: no module named 'lcd_utils'
```

demitrix, 2024-06-05:
So the new xpt2046 module doesn't work for me at all. I added a line to print coords got nothing, so I added a line to print the z value before comparing it to the threshold. That ran once and then everything hangs. I rolled xpt2046.py back to https://github.com/kdschlosser/lvgl_micropython/blob/e10678cb7feb975712e3813a18af0f368253f9a4/api_drivers/common_api_drivers/indev/xpt2046.py and ran the demo and now the slider is working.

ikkesia, 2024-06-05:
@demitrix didn't you have a run time error on 'lcd utils'?

demitrix, 2024-06-05:
>  @demitrix didn't you have a run time error on 'lcd utils'?

I did not. Make sure you have a clean pull, or delete and reclone the repo.

kdschlosser, 2024-06-05:
OK I rolled the driver back to that PR and fixed the  ticks_ns problem.

Let me know if the slider now works properly.

kdschlosser, 2024-06-05:
If the slider is working then try adding the rotate to it. You have to do this after the touch driver has been created. I am trying to set the rotation up so it is something that can be done real time. so if someone wanted to add a mems sensor to detect orientation they could and it would rotate the screen based on the device position.

demitrix, 2024-06-05:
```
display.init()
display.set_backlight(100)

import xpt2046  # NOQA

indev = xpt2046.XPT2046(spi_bus)

import lvgl as lv  # NOQA

display.set_rotation(lv.DISPLAY_ROTATION._90)

scrn = lv.screen_active()
scrn.set_style_bg_color(lv.color_hex(0x000000), 0)

slider = lv.slider(scrn)
slider.set_size(_WIDTH - 50, 50)
slider.center()

import task_handler  # NOQA

th = task_handler.TaskHandler()
```
Touch is working but is not rotated. I can move the slider by sliding where it was before rotation. Also I was incorrect, the previous version of xpt2046 is reporting phantom touches. While I was typing up this reply and not touching the anything
```
1 240 0
1 466 4
1 29 1
1 180 24
1 481 0
1 2 0
1 481 0
1 481 80
1 481 0
1 119 39
1 6 0
1 119 0
1 481 310
1 25 0
1 79 4
1 320 1
1 160 39
1 5 0
1 481 0
1 0 310
1 481 0
1 119 106
1 481 0
1 3 0
1 361 19
1 29 12
1 481 0
1 129 39
1 19 0
1 1 0
1 1 0
1 4 0
1 360 0
```

demitrix, 2024-06-06:
Phantom touches are resolved for sure. Rotation behavior remains the same. No rotation works as expected. Rotation applied and the slider responds to where it is when not rotated.

<details>
<summary> No rotation</summary>

```
XPT2046(raw_x=-1, raw_y=-1, x=-1, y=-1, state=RELEASED)
XPT2046_TP_DATA(x=666, y=2092, z=1409)
XPT2046(raw_x=38, raw_y=163, x=38, y=163, state=PRESSED)
XPT2046_TP_DATA(x=668, y=2097, z=1420)
XPT2046(raw_x=38, raw_y=163, x=38, y=163, state=PRESSED)
XPT2046_TP_DATA(x=667, y=2090, z=1433)
XPT2046(raw_x=38, raw_y=163, x=38, y=163, state=PRESSED)
XPT2046_TP_DATA(x=653, y=2102, z=1422)
XPT2046(raw_x=37, raw_y=164, x=37, y=164, state=PRESSED)
XPT2046_TP_DATA(x=657, y=2098, z=1417)
XPT2046(raw_x=38, raw_y=163, x=38, y=163, state=PRESSED)
XPT2046_TP_DATA(x=692, y=2117, z=1449)
XPT2046(raw_x=40, raw_y=165, x=40, y=165, state=PRESSED)
XPT2046_TP_DATA(x=701, y=2116, z=1474)
XPT2046(raw_x=40, raw_y=165, x=40, y=165, state=PRESSED)
XPT2046_TP_DATA(x=734, y=2137, z=1462)
XPT2046(raw_x=42, raw_y=166, x=42, y=166, state=PRESSED)
XPT2046_TP_DATA(x=793, y=2163, z=1485)
XPT2046(raw_x=46, raw_y=168, x=46, y=168, state=PRESSED)
XPT2046_TP_DATA(x=792, y=2166, z=1494)
XPT2046(raw_x=46, raw_y=169, x=46, y=169, state=PRESSED)
XPT2046_TP_DATA(x=837, y=2190, z=1489)
XPT2046(raw_x=48, raw_y=170, x=48, y=170, state=PRESSED)
XPT2046_TP_DATA(x=853, y=2197, z=1479)
XPT2046(raw_x=49, raw_y=171, x=49, y=171, state=PRESSED)
XPT2046_TP_DATA(x=864, y=2209, z=1501)
XPT2046(raw_x=50, raw_y=172, x=50, y=172, state=PRESSED)
XPT2046(raw_x=50, raw_y=172, x=50, y=172, state=RELEASED)
XPT2046_TP_DATA(x=969, y=2221, z=1427)
XPT2046(raw_x=56, raw_y=173, x=56, y=173, state=PRESSED)
XPT2046_TP_DATA(x=955, y=2218, z=1437)
XPT2046(raw_x=55, raw_y=173, x=55, y=173, state=PRESSED)
XPT2046(raw_x=55, raw_y=173, x=55, y=173, state=RELEASED)
XPT2046_TP_DATA(x=1118, y=2213, z=1265)
XPT2046(raw_x=65, raw_y=172, x=65, y=172, state=PRESSED)
XPT2046_TP_DATA(x=1141, y=2210, z=1465)
XPT2046(raw_x=66, raw_y=172, x=66, y=172, state=PRESSED)
XPT2046_TP_DATA(x=1177, y=2205, z=1579)
XPT2046(raw_x=68, raw_y=172, x=68, y=172, state=PRESSED)
XPT2046(raw_x=68, raw_y=172, x=68, y=172, state=RELEASED)
XPT2046_TP_DATA(x=1224, y=2229, z=1123)
XPT2046(raw_x=71, raw_y=174, x=71, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1258, y=2219, z=1383)
XPT2046(raw_x=73, raw_y=173, x=73, y=173, state=PRESSED)
XPT2046_TP_DATA(x=1283, y=2232, z=1590)
XPT2046(raw_x=74, raw_y=174, x=74, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1285, y=2243, z=1586)
XPT2046(raw_x=75, raw_y=175, x=75, y=175, state=PRESSED)
XPT2046_TP_DATA(x=1293, y=2236, z=1633)
XPT2046(raw_x=75, raw_y=174, x=75, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1302, y=2240, z=1625)
XPT2046(raw_x=76, raw_y=174, x=76, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1334, y=2232, z=1627)
XPT2046(raw_x=77, raw_y=174, x=77, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1346, y=2239, z=1670)
XPT2046(raw_x=78, raw_y=174, x=78, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1362, y=2235, z=1707)
XPT2046(raw_x=79, raw_y=174, x=79, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1395, y=2237, z=1699)
XPT2046(raw_x=81, raw_y=174, x=81, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1439, y=2239, z=1580)
XPT2046(raw_x=84, raw_y=174, x=84, y=174, state=PRESSED)
XPT2046_TP_DATA(x=1446, y=2228, z=1593)
XPT2046(raw_x=84, raw_y=173, x=84, y=173, state=PRESSED)
[... 576 lines cut ...]
XPT2046(raw_x=216, raw_y=187, x=133, y=216, state=PRESSED)
XPT2046_TP_DATA(x=3688, y=2394, z=1915)
XPT2046(raw_x=216, raw_y=186, x=134, y=216, state=PRESSED)
XPT2046_TP_DATA(x=3702, y=2395, z=1925)
XPT2046(raw_x=217, raw_y=187, x=133, y=217, state=PRESSED)
XPT2046_TP_DATA(x=3698, y=2390, z=1840)
XPT2046(raw_x=216, raw_y=186, x=134, y=216, state=PRESSED)
XPT2046_TP_DATA(x=3687, y=2384, z=1895)
XPT2046(raw_x=216, raw_y=186, x=134, y=216, state=PRESSED)
XPT2046_TP_DATA(x=3726, y=2389, z=1904)
XPT2046(raw_x=218, raw_y=186, x=134, y=218, state=PRESSED)
XPT2046_TP_DATA(x=3713, y=2385, z=1920)
XPT2046(raw_x=217, raw_y=186, x=134, y=217, state=PRESSED)
XPT2046_TP_DATA(x=3756, y=2383, z=1916)
XPT2046(raw_x=220, raw_y=186, x=134, y=220, state=PRESSED)
XPT2046_TP_DATA(x=3751, y=2380, z=1883)
XPT2046(raw_x=220, raw_y=185, x=135, y=220, state=PRESSED)
XPT2046_TP_DATA(x=3786, y=2368, z=1817)
XPT2046(raw_x=222, raw_y=184, x=136, y=222, state=PRESSED)
XPT2046(raw_x=222, raw_y=184, x=136, y=222, state=RELEASED)
XPT2046(raw_x=222, raw_y=184, x=136, y=222, state=RELEASED)
XPT2046(raw_x=222, raw_y=184, x=136, y=222, state=RELEASED)
XPT2046(raw_x=222, raw_y=184, x=136, y=222, state=RELEASED)
```
</details>

kdschlosser, 2024-06-06:
rotated what degree?

kdschlosser, 2024-06-06:
from what I am seeing is you have rotated the display 90° and the numbers are aligning properly.. if you have a native resolution of 240 x 320 and you touch the display at position 20, 200 that would translate to position 120, 20 when rotating the display 90°

What I am wondering is if LVGL is handling the change to the indev. I have to look at the code in LVGL to see what it is doing with respect to the indev drivers.

kdschlosser, 2024-06-06:
Yup, that's what it is. LVGL is handling the rotation internally. I removed the transforming of the pointer coordinates due to LVGL handling this internally.

kdschlosser, 2024-06-06:
I think that should finish this driver up.

If you still get phantom touches at all you can change the threshold value for the Z axis. it is defaulted to 400 and you can up that number if you want to filter out any random touches if they occur. You can set the threshold per instance or globally across all XPT2046 indevs.

To set it at the instance level..

```python

indev1 = xpt2046.XPT2046(...)
indev1.touch_threshold= 600

indev2 = xpt2046.XPT2046(...)
indev2.touch_threshold= 800
```

and to set it globally.

```python

xpt2046.XPT2046.touch_threshold = 600
indev1 = xpt2046.XPT2046(...)
indev2 = xpt2046.XPT2046(...)
indev3 = xpt2046.XPT2046(...)
```

kdschlosser, 2024-06-06:
Also I wanted to say thanks for adding the expandable/collapsible code blocks. It makes it a lot nicer when scrolling through an issue.

I like to use those when dealing with large code blocks or pasting logging data. 👍 for doing that.

demitrix, 2024-06-06:
Yeah realized things were getting lengthy and wanted to clean it up haha. I'll give this a shot tomorrow and see what happens 🤞

demitrix, 2024-06-06:
Think everything looks good here. Closing this issue.

closed 2024-06-06 by demitrix

kdschlosser, 2024-06-06:
rotation is working properly?

demitrix, 2024-06-06:
Sure is. I think everything is solid. While there may be improvements down the road, everything is in working order as of now.

ikkesia, 2024-06-06:
Can someone tell me how to set the parameters for touch calibration ?

With the old touch driver I used  :
`touch = xpt2046(cs=32 ,cal_x0=3840, cal_x1 = 393, cal_y1=220, cal_y0 = 3719)`

kdschlosser, 2024-06-06:
the calibration program has not had all of the kinks ironed out with it yet. I am going to be working on that soon.

ikkesia, 2024-06-07:
So, if I understand well, I can't enter the calibration values manually ?

I'm doing some tests and like @demitrix I confirm that it works fine. Unfortunately I detect a misalignment between the graphic position and the touch pointer, which effectively makes it unusable.

kdschlosser, 2024-06-07:
You can enter them manually but they don't use the typical calibration coordinates that you have seen in the past. I went for an easier approach. which is recording the left, right top and bottom boundaries. It is very rudimentary as it doesn't handle if the touch panel has been rotated at all. to handle the rotation of the display I needed to have the ability to store floats into the NVS on the esp32. I did add that ability but have not gotten to updating the calibration so it is more complete.

if you wanted to pass the left, right top and bottom boundaries that can be done by doing the following.

```python
class TouchCal:
   
    def __init__(self),

        left = 0
        right = 0
        top = 0
        bottom = 0

    def save():
        pass

cal = TouchCal()

import xpt2046

xpt2046.XPT2046(spi_bus, touch_cal=cal)
```

you change the values in the `__init__` to what you need and pass an instance of that class to the touch driver as seen above. It's very basic at the moment but at least it is something. I have been focusing on getting the bus drivers working properly and fixing some things in MicroPython that caused serious limitations on being able to use things like the touch, display and an SD card being on the same SPI bus... I still have to fix the SDCard driver that is built into MicroPython to get this to work properly.. 

I also want to update the SPI display bus driver so it can take an instance of the SPI driver from MicroPython. Being able to do that is going to take a little bit more work.

kdschlosser, 2024-06-09:
@ikkesia 

I updated the calibration routine. The new design uses a more complex way of handling the touch panel to screen alignment. I never looked at the xpt2046 driver code in the official binding so I do not know what their calibration was all about. I can tell you that what is being used is not something that is considered typical.

My design doesn't not require any hard coding of the calibration numbers either.  The calibration data gets stored in non volatile memory so it is persistent between power cycles and restarts. This is a better design for a production environment where the user is able to calibrate the display without needing to change and flash any source code.

It is easy to launch the screen calibration

```
# display driver and display bus code
import machine
import xpt2046
import task_handler

spi_bus = machine.SPI(...)
indev = xpt2046.XPT2046(spi_bus)

th = task_handler.TaskHandler()

if not indev.is_calibrated:
    indev.calibrate()

```

You have to make sure that the task handler has been started prior to running the calibration. 

I do have to make some tweaks to it still and I am going to mess about with that now.

ikkesia, 2024-06-09:
@kdschlosser

I would have worked with the manual setting as you suggested, but having a calibration routine is great.
Thank you very much for this update.

Unfortunately I can't try it quickly because my development kit broke. I'm waiting for a new one at the beginning of the week.

ste7anste7an, 2024-06-09:
For m the calibration is not working:
```
indev.calibrate()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pointer_framework.py", line 45, in calibrate
  File "touch_calibrate.py", line 275, in __init__
  File "touch_calibrate.py", line 117, in build_crosshair
ValueError: need more than 2 values to unpack

```

[7 more replies]