lvgl-micropython/lvgl_micropython #48: Cannot build for ESP32

ikkesia, 2024-06-01:
On MACOS I'm trying to build for ESP32 GENERIC board.
My run command is :

`python make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC  DISPLAY=ili9341 INDEV=xpt2046`

Below is the building output log :

````
setting up ESP-IDF v5.0.4
this might take a bit...

Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: macos
Selected targets are: esp32c2, esp32s3, esp32c3, esp32, esp32h2, esp32s2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230419 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Installing Python environment and packages
Python 3.10.6
pip 24.0 from /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
Requirement already satisfied: pip in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (24.0)
Requirement already satisfied: setuptools in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (70.0.0)
Upgrading pip and setuptools...
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.0.txt
Destination: /Users/tbe/.espressif/espidf.constraints.v5.0.txt.tmp
Done
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Ignoring windows-curses: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 3)) (70.0.0)
Requirement already satisfied: packaging in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 4)) (24.0)
Requirement already satisfied: click in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 7)) (8.1.7)
Requirement already satisfied: pyserial in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 8)) (3.5)
Requirement already satisfied: future in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 9)) (0.18.2)
Requirement already satisfied: cryptography in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 10)) (36.0.2)
Requirement already satisfied: pyparsing in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 11)) (3.0.9)
Requirement already satisfied: pyelftools in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 12)) (0.27)
Requirement already satisfied: idf-component-manager in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.5.2)
Requirement already satisfied: esp-coredump in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 14)) (1.11.0)
Requirement already satisfied: esptool in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 15)) (4.7.0)
Requirement already satisfied: kconfiglib in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 18)) (14.1.0)
Requirement already satisfied: freertos_gdb in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 22)) (1.0.3)
Requirement already satisfied: cffi>=1.12 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from cryptography->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 10)) (1.16.0)
Requirement already satisfied: requests<3 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (2.32.3)
Requirement already satisfied: urllib3<2 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.26.18)
Requirement already satisfied: requests-file<2 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.5.1)
Requirement already satisfied: requests-toolbelt in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.0.0)
Requirement already satisfied: schema in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (0.7.7)
Requirement already satisfied: six in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (1.16.0)
Requirement already satisfied: tqdm<5 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (4.66.4)
Requirement already satisfied: colorama in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (0.4.6)
Requirement already satisfied: pyyaml>5.2 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (6.0.1)
Requirement already satisfied: cachecontrol>0.12.6 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (0.14.0)
[... cut ...]
Requirement already satisfied: certifi>=2017.4.17 in /Users/tbe/.espressif_v5.0.4/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from requests<3->idf-component-manager->-r /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt (line 13)) (2024.2.2)
Installing Python packages
Constraint file: /Users/tbe/.espressif/espidf.constraints.v5.0.txt
Requirement files:
- /Users/tbe/esp/kds/lvgl_micropython/lib/esp-idf/tools/requirements/requirements.core.txt
All done! You can now run:
. ./export.sh
Getting ESP-IDF build Environment
bash: line 1: gmake: command not found
````

How can I fix it ?

kdschlosser, 2024-06-02:
ok that should be fixed now.

ikkesia, 2024-06-02:
```
/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:398:13: error: 'dma_chan' undeclared (first use in this function)
398 |             dma_chan = 1;
|             ^~~~~~~~
/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/machine_hw_spi.c:398:13: note: each undeclared identifier is reported only once for each function it appears in
[1221/1830] Building C object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/Users/tbe/esp/lvgl_micropython/lib/micropython/lib/littlefs/lfs2.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_84047 and /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_84047
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
gmake: *** [Makefile:66: all] Error 1
```

kdschlosser, 2024-06-02:
that issue should now be solved.

kdschlosser, 2024-06-02:
I updated the MicroPython version to 1.23.0 I am going to be working on the esp-idf version to see if I can get that up to 5.2.1

closed 2024-06-10 by ikkesia