lvgl-micropython/lvgl_micropython #49: RPEL not working in a ESP32-S2 custom board

Alby1987, 2024-06-04:
I'm trying to make a Lvgl for MicroPython using your software. Using a precompiled Micropython for my Esp32_generic_s2 works, but when building the firmware with lvgl with your I get a non working RPEL and Pykmr says it can not connect. This is my command

make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S2 DISPLAY=ili9488 INDEV=gt911

kdschlosser, 2024-06-07:
I added a new build command to enable the serial output over the USB OTG (JTAG) pins.

Add `--USB-OTG` to the end of your build command. That will fix your problem.

kdschlosser, 2024-06-07:
You need to clone the repo again as I just added that feature like 2 minutes ago.

Alby1987, 2024-06-07:
I've just tried it, it works! Thanks! Now fighting against initializing my ILI9488, but that should be just a matter of testings...

closed 2024-06-07 by Alby1987

kdschlosser, 2024-06-07:
The bus drivers look pretty good. I believe the only one that has not been tested is the I2C driver at this point. Some of the display IC drivers have been tested. I believe the ILI9488 has already been tested and it was considered to be working. If you have something going on with it that's not working properly let me know and I will straighten it out.

fabse-hack, 2024-09-01:
@kdschlosser with the LOLIN_S2_MINI is the same problem. 

- With the ESP32_S2_GENERIC and --usb-otg is possible to open the REPL ....
- LOLIN_S2_MINI not have the option --usb-otg and is not possible to open REPL ...

btw is `--usb-otg` .... not `--USB-OTG` (i think there is general problem in the command line with lower() and upper()