lvgl-micropython/lvgl_micropython #51: ESP32 I2S

marksull, 2024-06-09:
in `builder/esp32.py` this code will create a build failure:

```
    if '#define MICROPY_PY_MACHINE_I2S (0)' not in data:
        data = data.replace(
            '#ifndef MICROPY_PY_MACHINE_I2S',
            '#define MICROPY_PY_MACHINE_I2S (0)\n'
            '#ifndef MICROPY_PY_MACHINE_I2S'
        )
        with open(mpconfigport_path, 'wb') as f:
            f.write(data.encode('utf-8'))
```

If under `lib/micropython/ports/esp32/mpconfigport.h` I define the following to enable I2S.

```
#define MICROPY_PY_MACHINE_I2S (1)
```

If I remove the code above in `esp32.py`,  I2S is compiled in correctly and works as expected. I didn't want to submit a PR to remove that code because I genuinely don't understand the intent behind the action as you were obviously trying to solve a particular problem.

kdschlosser, 2024-06-09:
That code is there because I2S fails with MIcroPython 1.22.x, just a few days ago I upgraded the repo so it is using MicroPython 1.23.0 and the I2s problem has been listed as being fixed in this version. I have not had the chance to test the I2S to see if it is working properly or not. 

https://github.com/micropython/micropython/issues/13104

This problem has existed since 1.21.0 which is when MicroPython was upgraded to use ESPIDF 5.0.x. Never got a difinitive list of what ESP32 series SOC's had the issue so I turned it off for all of them. I do know 100% that the problem was with the ESP32 SOC's and the I2S would need to be checked using that MCU to see if the problem has in fact been taken care of.

What version of the ESP32 are you using? Are you able to check to see if there are no issues with it running on an ESP32_GENERIC board??

marksull, 2024-06-09:
I can confirm 1.23.0 with your package builds fine when I2S is enabled now. Thank you.

closed 2024-06-09 by marksull