lvgl-micropython/lvgl_micropython #52: Build error on spi_bus.c and modlcd_bus.c

getznerh, 2024-06-11:
On the actual version there are build errors on /lvgl_micropython/ext_mod/lcd_bus/esp32_src/spi_bus.c
On line 181 the variable ret has to be declared with  mp_lcd_err_t ret = 
and on /lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c on line 63 there is a ',' to much (we are on C and not on python ;-))

with these changes the firmware compiles without errors and the new SPI version is working OK

kdschlosser, 2024-06-11:
There are lots and lots of issues at the moment. I am in the middle of doing a very large update to the binding. One of the things that has not been able to be done is using an SPI touch screen on the same bus as a display. Or using an SPI  SDCard reader on the same bus as a display. This is because of how MicroPython is coded and it only allowing a single device per bus without having to write all kinds of involved code in Python to make it work and even then you cannot have an SDCard reader sharing a bus with anything. It had to be on it's own SPI bus.

The big issue is there is no consistency between the SPI interfaces between the different MCU's. No common back end API, only the Python API has consistency to it. This is making it a really large job because I have to create and put in place a common back end across all of the MCU's.

If you want to use this binding I recommend cloning the repo and then checking out this commit.
https://github.com/kdschlosser/lvgl_micropython/commit/c4f225a2f7052f750b53823f729718a0be6103f9

```
git checkout c4f225a2f7052f750b53823f729718a0be6103f9
git pull
```

That should compile without any errors and allow you to run the binding. Sharing of the SPI bus only works for the ESP32 and only for the touch and display not the SDCard.

kdschlosser, 2024-06-19:
If everything is good could you please close this issue...

closed 2024-12-03 by kdschlosser