lvgl-micropython/lvgl_micropython #53: Not really an issue, with CYD

zelogik, 2024-06-13:
Hello,

I have passed (lost) many time without success trying to get lvgl and micropython, or lv_micropython working on a CYS (Cheap Yellow Screen), ESP-2432s028, both touch and screen.

And I have found an impressive work on that repository, and I have read that the SPI is rewrited now.
But my simple question, is that driver is compatible with something like:

disp = ili9341(spihost=esp.HSPI_HOST, clk=14, dc=2, cs=15, rst=12, miso=16, mosi=13, backlight=-1, power=-1, rot=0xC0, width=320, height=240)
touch = xpt2046(spihost=esp.VSPI_HOST, cs=33, clk=25, mosi=32, miso=39)

xpt2046 and ili9341 don't have the same SPI, and neither ports.

Need to wait some day that the SPI interface is fonctionning, or can I help you ?

Regards

kdschlosser, 2024-06-13:
I am almost done rewriting the SPI interface in MicroPython. I am working on it at this moment.

This is not like the official binding for MicroPython. The Python code used is going to be different. It offers more flexibility for the advanced user and makes it easier to use for the novice. There is a unified API across all of the display drivers and also the touch drivers. It becomes really easy if you want to swap out displays. This is able to be done without having to compile and flash MicroPython. 

It also allows for faster development because you can compile the binding to run on either Linux or macOS and run it on the desktop. The same code that is written for the desktop gets used on your board. So need to keep on uploading the Python source code to your MCU to test to make sure it is working right.

As for your display.. It will work with it no worries there....
You will use the code below to get it working.

```python
from micropython import const  # NOQA
import machine  # NOQA
import lcd_bus

# Display settings
_WIDTH = const(240)
_HEIGHT = const(320)

# Display SPI bus settings
_LCD_HOST = const(1)
_LCD_FREQ = const(80000000)
_LCD_MISO = const(16)
_LCD_MOSI = const(13)
_LCD_SCK = const(14)
_LCD_CS = const(15)

# display additional pins
_LCD_RST = const(12)
_LCD_DC = const(2)
_LCD_BKL = None
_LCD_PWR = None

# touch panel SPI settings
_TP_HOST = const(2)
_TP_FREQ = const(1000000)
_TP_MISO = const(39)
_TP_MOSI = const(32)
_TP_SCK = const(25)
_TP_CS = const(33)

# create the SPI bus for the display
lcd_spi_bus = machine.SPI(
    _LCD_HOST,
    _LCD_FREQ,
    mosi=_LCD_MOSI,
    miso=_LCD_MISO,
    sck=_LCD_SCK
)

# create the SPI device on the bus for the display
display_bus = lcd_bus.SPIBus(
    spi_bus=lcd_spi_bus,
    dc=_LCD_DC,
    freq=_LCD_FREQ,
    cs=_LCD_CS
)

import ili9341  # NOQA
import lvgl as lv  # NOQA

# create the display driver
display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_LCD_RST,
    power_pin=_LCD_PWR,
    backlight_pin=_LCD_BKL,
    color_space=lv.COLOR_FORMAT_RGB565,
    rgb565_byte_swap=True
)

# start the display driver
display.set_power(True)
display.init()
display.set_backlight(100)

import xpt2046  # NOQA

# create the SPI bus for the touch panel
touch_spi_bus = machine.SPI(
    _TP_HOST,
    _TP_FREQ,
    mosi=_TP_MOSI,
    miso=_TP_MISO,
    sck=_TP_SCK,
    cs=_TP_CS
)

# create the touch driver
touch = xpt2046.XPT2046(touch_spi_bus)

# set the rotation of the display so it is in landscape/widescreen orientation
display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA

# starting the task handler. 
# This takes care of calling lv.task_handler and managing the time 
import task_handler  # NOQA
th = task_handler.TaskHandler()

# write your code here....
```

kdschlosser, 2024-06-13:
you might need to adjust the frequency for the SPI bus used for the display. It should run at 80Mhz without an issue but if there is a problem then you will want to bump it down to 40Mhz.

zelogik, 2024-06-13:
WONDERFULL!!!! a big thank for your work!

but ... (I always have a but), there is some "bug"
The touch is rotated with the screen (ie view picture)

an almost working code under, correcting some typo you have made on the previous code (for other people)
<details close>
  <summary>Hello World (ie: code_block is not markdown readable...</summary>
from micropython import const  # NOQA
from machine import Pin, SPI # NOQA
import lcd_bus

_WIDTH = const(320)
_HEIGHT = const(240)

_LCD_HOST = const(1)
_LCD_FREQ = const(20000000)
_LCD_MISO = const(16)
_LCD_MOSI = const(13)
_LCD_SCK = const(14)
_LCD_CS = const(15)

_LCD_RST = const(12)
_LCD_DC = const(2)
_LCD_BKL = const(21)
_LCD_PWR = None

_TP_HOST = const(2)
_TP_FREQ = const(1000000)
_TP_MISO = const(39)
_TP_MOSI = const(32)
_TP_SCK = const(25)
_TP_CS = const(33)

lcd_spi_bus = SPI(
    _LCD_HOST,
    _LCD_FREQ,
    mosi=_LCD_MOSI,
    miso=_LCD_MISO,
    sck=_LCD_SCK
)

display_bus = lcd_bus.SPIBus(
    spi_bus=lcd_spi_bus,
    dc=_LCD_DC,
    freq=_LCD_FREQ,
    cs=_LCD_CS
)

import ili9341  # NOQA
import lvgl as lv  # NOQA

display = ili9341.ILI9341(
    data_bus=display_bus,
    display_width=_WIDTH,
    display_height=_HEIGHT,
    reset_pin=_LCD_RST,
    power_pin=_LCD_PWR,
    backlight_pin=_LCD_BKL,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

display.set_power(True)
display.init()
display.set_backlight(100)

import xpt2046  # NOQA

touch_spi_bus = SPI(
    _TP_HOST,
    _TP_FREQ,
    mosi=_TP_MOSI,
    miso=_TP_MISO,
    sck=_TP_SCK,
    cs=_TP_CS
)

touch = xpt2046.XPT2046(touch_spi_bus)

display.set_rotation(lv.DISPLAY_ROTATION._0)  # NOQA

import task_handler  # NOQA
th = task_handler.TaskHandler()

green_led = Pin(16, Pin.OUT)

green_led.off()

def handler( data ):
    obj = data.get_target()
    # if type( obj ) == type( lv.button() ):
    print( lv.button() )
    print(type(obj))

screen1 = lv.obj( )
screen1.set_size( 250, 240 )
screen1.center()
screen1.add_event_cb( handler, lv.EVENT.CLICKED, None )

leftButton = lv.button( screen1 )
leftButton.add_flag( lv.obj.FLAG.EVENT_BUBBLE )
leftButton.set_size( 80, 40 )
leftButton.center()
leftButton.set_pos( -50, 0 )

leftLabel = lv.label( leftButton )
leftLabel.center()
leftLabel.set_text( 'left' )

rightButton = lv.button( screen1 )
rightButton.add_flag( lv.obj.FLAG.EVENT_BUBBLE )
rightButton.set_size( 80, 40 )
rightButton.center()
rightButton.set_pos( 50, 0 )

rightLabel = lv.label( rightButton )
rightLabel.center()
rightLabel.set_text( 'right' )

lv.screen_load(screen1)
</details>

Current rotations:

-  Number 1
_WIDTH = const(240)
_HEIGHT = const(320)
and
display.set_rotation(lv.DISPLAY_ROTATION._90)  # NOQA

- Number 2 / 3
don't remember values, just that i need to push on the "2" to get an almost clean screen like on "3"
 

- Number 4
_WIDTH = const(320)
_HEIGHT = const(240
and
display.set_rotation(lv.DISPLAY_ROTATION._0)  # NOQA

[image]

zelogik, 2024-06-13:
When touch and screen will be in sync, just need to see how to test under linux without uploading on the MCU.

ie: Thanks for your explanation, but it's not my first project on a MCU with C/C++, but It's been over 10-15 years since I last touched python/micropython, and as I don't like display/frontend, python should be easier to debug/deploy :D

kdschlosser, 2024-06-14:
OK so your display is setup so the display is rotated from the touch screen right out of the gate.

I have to think about the easiest way to handle that situation. I am going to have to build in a rotation mechanism into the touch that is separate from the display.

Give me a few minutes to get that done.

kdschlosser, 2024-06-14:
OK I added the ability to set the starting rotation of the XPT2046 driver.

so you need to add `startup_rotation=lv.DISPLAY_ROTATION._90` to the constructor for the xpt2046 driver.

zelogik, 2024-06-14:
Thanks, seem to work.

But now is touch.calibrate() is functional ?
Can't take photo now, but I have a big target on top left, with a small one inside but not centered, and the label Touch Screen Calibration.
impossible to get touch response from here..

kdschlosser, 2024-06-14:
I haven't tested the calibration at all. I quickly hammered the code out so I am sure there are issues with it's alignment and possibly how I am handling the sizing of the crosshair. I don't even know if the crosshair looks right because it is being rendered real time. I am sure there are issues I am going to have to iron out. I also don't know if that calculation is going to do the trick either. I am hoping it will work well enough.

The other thing is the ESP32 doesn't support storing floats in NVS. I extended the functionality of the NVS to support it and I don't know if that is working correctly either as I have not yet tested it. I add a new feature and then I get side tracked with issues that get opened. I was in the middle of fixing an issue with the RGB Bus and got side tracked with the XPT2046 driver not working and that led to the need to write the calibration differently due to the large number of those specific types of touch panels being off kilter from the display.

zelogik, 2024-06-15:
Take your time, i'm already impressed by the work you have done with this project.
I have never been this far with python and lvgl with this CYD.

And I have to relearn everything in python.

kdschlosser, 2024-06-18:
I wanted to let you know that I have not forgotten about ya. I had some other things to deal with. I have one other thing that I need to fix and then I will jump on getting the calibration to run properly. It shouldn't take me too long to fix that other issue (I hope)

kdschlosser, 2024-06-19:
OK so I got the thing to compile properly now. I am going to mess about with the touch calibration in a little bit. It shouldn't take me that long to get it all dialed in.

closed 2024-12-03 by kdschlosser