lvgl-micropython/lvgl_micropython #54: Compile proble using ESP-IDF 5.2.1 and MicroPython 1.23.0

kdschlosser, 2024-06-15:
There is a compile issue with nimble and I have opened an issue with MicroPython for the problem. Hopefully they will fix it pretty quickly and get a release made for it. Otherwise I am going to have to patch the code. Not something I really want to do.

https://github.com/micropython/micropython/issues/15287

zelogik, 2024-06-15:
So maybe my new error at compile time is related...

<details error>
  <summary>Error linking</summary>
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/logik/lvgl_micropython/lib/esp-idf/components/esp_coex/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.17+0x0): undefined reference to `btdm_rf_bb_reg_init'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/logik/lvgl_micropython/lib/esp-idf/components/esp_coex/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.17+0x16): undefined reference to `btdm_rf_bb_reg_init'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_exchange_mtu+0x0): undefined reference to `mp_bluetooth_gattc_exchange_mtu'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_write+0x0): undefined reference to `mp_bluetooth_gattc_write'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_read+0x0): undefined reference to `mp_bluetooth_gattc_read'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_descriptors+0x0): undefined reference to `mp_bluetooth_gattc_discover_descriptors'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_characteristics+0x4): undefined reference to `mp_bluetooth_gattc_discover_characteristics'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gattc_discover_services+0x0): undefined reference to `mp_bluetooth_gattc_discover_primary_services'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_set_buffer+0x0): undefined reference to `mp_bluetooth_gatts_set_buffer'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_notify_indicate+0x0): undefined reference to `mp_bluetooth_gatts_notify_indicate'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_write+0x0): undefined reference to `mp_bluetooth_gatts_write'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_read+0x0): undefined reference to `mp_bluetooth_gatts_read'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x10): undefined reference to `mp_bluetooth_gatts_register_service_begin'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x14): undefined reference to `mp_bluetooth_gatts_register_service'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gatts_register_services+0x18): undefined reference to `mp_bluetooth_gatts_register_service_end'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_passkey+0x0): undefined reference to `mp_bluetooth_gap_passkey'
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(modbluetooth.c.obj):(.literal.bluetooth_ble_gap_pair+0x0): undefined reference to `mp_bluetooth_gap_pair'
[... cut ...]
/home/logik/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/logik/lvgl_micropython/lib/micropython/extmod/modbluetooth.c:654:(.text.bluetooth_ble_gap_connect+0x9c): undefined reference to `mp_bluetooth_gap_peripheral_connect'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

</details>

with more that 375 pull request and 1.5k issues, I don't know if micropython will be as reactive as you....

kdschlosser, 2024-06-16:
yes I will have this fixed in the next 8 to 10 hours

kdschlosser, 2024-06-18:
This should be fixed.

closed 2024-06-18 by kdschlosser