lvgl-micropython/lvgl_micropython #55: RGB -> BGR on ili9341

zelogik, 2024-06-15:
Hi,

I have read than putting:
color_byte_order=BYTE_ORDER_BGR, in the contructor of the display should work 2 weeks ago (in another issue on this repo)

But have you done some work there too ? because with or without putting that, the Red is displayed Blue, and blue is red. I can have a look at it if you say that you have touched something on that driver :D

kdschlosser, 2024-06-16:
you need to set `rgb565_byte_swap` to `True`

zelogik, 2024-06-16:
With a constructor like that:
`
display = ili9341.ILI9341(
...
    color_space=lv.COLOR_FORMAT.RGB565,
    color_byte_order=ili9341.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)
`
Tested with rgb565_byte_swap  True|False, color_byte_order commented|RBG|BGR

exactly the same wrong color order ...

kdschlosser, 2024-06-16:
I am going to have to make a change to the code, give me a little while to do that.

kdschlosser, 2024-06-18:
OK I made the changes needed to the driver IC. so this should be all set.

commit 371a3ee0 (2024-06-18): reverts some of the changes made to the ili9341 driver.

I had hard coded `color_space` to BGR when it turns out that there are variants that use RGB. So I am defaulting `color_space` to RGB which is what is done across the board for all other display drivers. The users that have the BGR type displays IC's will need to set this parameter.

Fixes #55
closed 2024-06-18 by kdschlosser