lvgl-micropython/lvgl_micropython #56: Error comping

ikkesia, 2024-06-16:


zelogik, 2024-06-16:
Same as #54 
@ikkesia , kdschlosser is doing many change on the code, give him some time :D

ikkesia, 2024-06-16:
No pressure on @kdschlosser,  just report in case it's helpful.

kdschlosser, 2024-06-18:
This is fixed.

kdschlosser, 2024-06-18:
sorry didn't mean to close this, I thought it was a different issue.

Clone the repo again and give it a try. It should be fixed. If everything is good then close the issue please.

ikkesia, 2024-06-18:
Error after resizing partition.

```
MPY ili9341.py                                               
GEN /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/frozen_content.c
[1062/1891] Generating ../../genhdr/moduledefs.collected
Module registrations updated
[1890/1891] Generating binary image from built executable                                                                                                                           
esptool.py v4.7.0
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
[1891/1891] cd /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py && /Users/tbe/.espressif_v5.2.0/python_env/idf5.2_py3.10_env/bin/python /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py/CMakeFiles/app_check_size
cd /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py && /Users/tbe/.espressif_v5.2.0/python_env/idf5.2_py3.10_env/bin/python /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
Error: app partition is too small for binary micropython.bin size 0x27c2a0:
- Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x17c2a0)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_71701 and /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_71701
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1

***** Resizing Partition *****

***** Running build again *****

idf.py -D MICROPY_BOARD=ESP32_GENERIC -D MICROPY_BOARD_DIR="/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC"  -DUSER_C_MODULES=../../../../../ext_mod/micropython.cmake -D MICROPY_FROZEN_MANIFEST=/Users/tbe/esp/lvgl_micropython/build/manifest.py -B build-ESP32_GENERIC build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
WARNING: Python interpreter "/Users/tbe/.espressif_v5.2.0/python_env/idf5.2_py3.10_env/bin/python" used to start idf.py is not from installed venv "/Users/tbe/.espressif/python_env/idf5.2_py3.10_env"
Executing action: all (aliases: build)
Running ninja in directory /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC
Executing "ninja all"...
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[2/2] idf (5.2.0)           
-- Project sdkconfig file /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig
Loading defaults file /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined...
/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:100 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT
/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:101 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT
/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:144 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
-- App "micropython" version: v1.24.0-preview-42-g2b0e64bea-d
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/tbe/esp/lvgl_micropython/lib/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
Including User C Module(s) from ../../../../../ext_mod/micropython.cmake
Found User C Module(s): usermod_lcd_bus, usermod_lvgl, lvgl_interface, usermod_lcd_utils, usermod_nvs
[... cut ...] b/micropython/ports/esp32/build-ESP32_GENERIC/genhdr/mpversion.h
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_79155 and /Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_79155
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1

```

ikkesia, 2024-06-18:
`python make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC DISPLAY=ili9341 INDEV=xpt2046`

Commit #e12b838. Compilation completed successfully.

closed 2024-06-18 by ikkesia

kdschlosser, 2024-06-18:
I am still working on the last bit of getting the build system to work properly.

I added the ability to overclock the flash and the SPIRAM on the ESP32-S3.

If the MCU has octal SPIRAM and octal flash the actual speed they end up being at is 240Mhz for each. That is a HUGE increase from the 80Mhz it is currently at. The reason why this is so important is due to a limiting factor if using SPIRAM to hold the frame buffer and wanting to use double buffering. If you have the display set to say 80Mhz you will end up with anomalies on the display. This is because the RAM is not able to keep up with the 80MHZ write if the ram is being read from and written to at the same time the transfer is taking place. The 80MHZ bandwidth ends up getting split between the read and write that is taking place. so the highest SPI speed you would be able to set for the display would be 40Mhz. This solves that specific problem. 

As always there is a cost to doing that. The cost is an increase in temperature of the ESP32. it will raise the temps by 20°C and if the ambient temperature is high enough the ESP32 can overheat. Nothing that adding some passive cooling wouldn't handle but it is something that has to be considered.

ikkesia, 2024-06-18:
For ESP32-S3 users this will certainly be a fabulous improvement.

Regarding commit #ec841b6 "**hopefully fixes the ESP32 build**" maybe you need to work a little bit more.

```
micropython.bin binary size 0x27c2a0 bytes. Smallest app partition is 0x27d000 bytes. 0xd60 bytes (0%) free.
Warning: The smallest app partition is nearly full (0% free space left)!
Project build complete. To flash, run:
idf.py flash
or
idf.py -p PORT flash
or
python -m esptool --chip esp32 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or from the "/Users/tbe/esp/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC" directory
python -m esptool --chip esp32 -b 460800 --before default_reset --after no_reset write_flash "@flash_args"
bootloader  @0x001000    24800  (    3872 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2605728  (    3424 remaining)
total                  2671264

esptool.py v4.7.0
A fatal error occurred: Please specify the chip argument (choose from esp8266, esp32, esp32s2, esp32s3beta2, esp32s3, esp32c3, esp32c6beta, esp32h2beta1, esp32h2beta2, esp32c2, esp32c6, esp32h2, esp32p4)

```

kdschlosser, 2024-06-18:
yeah I know. It should be fixed now. I am being lazy and I don't want to go and look at the docs for the esptool.

kdschlosser, 2024-06-19:
It is compiling on the CI so everything should be good to go now.

ikkesia, 2024-06-19:
Just compiled. 
:ok_hand:

ste7anste7an, 2024-06-23:
I just pulled the latest version from github and get the IRAM error:
```
 python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM DISPLAY=ili9341 INDEV=xpt2046 LV_CFLAGS="-DLV_COLOR_DEPTH=16"
```

with this error:
```
 cat /home/stefan/projects/esp32/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC-SPIRAM/log/idf_py_stderr_output_14003
make[1]: warning: -j6 forced in submake: resetting jobserver mode.
/home/stefan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: micropython.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/stefan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: IRAM0 segment data does not fit.
/home/stefan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: region `iram0_0_seg' overflowed by 3128 bytes
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/micropython.elf.dir/build.make:5218: micropython.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:2065: CMakeFiles/micropython.elf.dir/all] Error 2
make[1]: *** [Makefile:136: all] Error 2
```

Is there a simple fix for this IRAM overflow error?