lvgl-micropython/lvgl_micropython #57: Issues list Error compiling :undefined reference to 'mp_bluetooth_xxx'

huochaigunkeji, 2024-06-16:
Build Command:
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=ft6x36

ESP-IDF v5.2

error:
/root/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/lvgl_micropython/lib/micropython/extmod/modbluetooth.c:654:(.text.bluetooth_ble_gap_connect+0x9c): undefined reference to `mp_bluetooth_gap_peripheral_connect'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: Leaving directory '/home/lvgl_micropython/lib/micropython/ports/esp32'
ninja failed with exit code 1, output of the command is in the /home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_202662 and /home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_202662
make: *** [Makefile:66: all] Error 1

zelogik, 2024-06-17:
Same as #54, and probably #56.
Big refactoring in the code

kdschlosser, 2024-06-18:
OK this should be fixed.

Clone the repo again and give it a try. If it is fixed then close the issue please.

kdschlosser, 2024-06-19:
This is all fixed now. If you could close this issue that would be great.

huochaigunkeji, 2024-06-22:
Thank you, I have git pull esp-idf v5
2 and micropython master, this issue did not occur during compilation. But another problem arose.
errors mesage:
Traceback (most recent call last):
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3366, in <module>
gen_obj(obj_name)
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3325, in gen_obj
locals_dict_entries = ",\n    ".join(gen_obj_methods(obj_name)),
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3203, in gen_obj_methods
obj_metadata[obj_name]['members'].update({method_name_from_func_name(method.name): func_metadata[method.name] for method in get_methods(obj_name)})
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3203, in <dictcomp>
obj_metadata[obj_name]['members'].update({method_name_from_func_name(method.name): func_metadata[method.name] for method in get_methods(obj_name)})
KeyError: 'lv_animimg_get_src'
-- Configuring incomplete, errors occurred!
See also "/home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/CMakeFiles/CMakeOutput.log".
OUTPUT:
RESULT: 1
CMake Error at /home/lvgl_micropython/ext_mod/lvgl/micropython.cmake:37 (message):
Failed to generate

huochaigunkeji, 2024-06-22:
Thank you, I have git pull esp-idf v5
2 and micropython master, this issue did not occur during compilation. But another problem arose.
errors mesage:
Traceback (most recent call last):
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3366, in
gen_obj(obj_name)
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3325, in gen_obj
locals_dict_entries = ",\n ".join(gen_obj_methods(obj_name)),
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3203, in gen_obj_methods
obj_metadata[obj_name]['members'].update({method_name_from_func_name(method.name): func_metadata[method.name] for method in get_methods(obj_name)})
File "/home/lvgl_micropython/gen/python_api_gen_mpy.py", line 3203, in
obj_metadata[obj_name]['members'].update({method_name_from_func_name(method.name): func_metadata[method.name] for method in get_methods(obj_name)})
KeyError: 'lv_animimg_get_src'
-- Configuring incomplete, errors occurred!
See also "/home/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_C3/CMakeFiles/CMakeOutput.log".
OUTPUT:
RESULT: 1
CMake Error at /home/lvgl_micropython/ext_mod/lvgl/micropython.cmake:37 (message):
Failed to generate

closed 2024-06-22 by huochaigunkeji

kdschlosser, 2024-06-22:
when you clone the repo do not update any submodules. Just run the build script and let it do its thing. It will handle collecting everything that is needed to compile.

huochaigunkeji, 2024-06-23:
> when you clone the repo do not update any submodules. Just run the build script and let it do its thing. It will handle collecting everything that is needed to compile.

Thank you, I dit it.

kdschlosser, 2024-06-23:
???

huochaigunkeji, 2024-06-23:
> ???

Thank you. Following the steps you mentioned, I have successfully compiled it.I compiled the ESP32C3 firmware and there is also this issue[#63](https://github.com/kdschlosser/lvgl_micropython/issues/63)