lvgl-micropython/lvgl_micropython #58: Compile problem not completely gone...

zelogik, 2024-06-18:
Error now with fresh clone repo:
```
[1890/1891] Generating binary image from built executable                                                                                                                                     
esptool.py v4.7.0
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
[1891/1891] cd /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py && /home/hideit/.espressif/python_env/idf5.2_py3.11_env/bin/python /home/hideit/Arduino/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py/CMakeFiles/app_check_size
cd /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esptool_py && /home/hideit/.espressif/python_env/idf5.2_py3.11_env/bin/python /home/hideit/Arduino/lvgl_micropython/lib/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/micropython.bin
Error: app partition is too small for binary micropython.bin size 0x27c270:
- Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x17c270)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_76435 and /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_76435
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
make: Leaving directory '/home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32'
```
... many output... 

```

-- Build files have been written to: /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC
[1/250] Generating ../../partition_table/partition-table.bin
FAILED: partition_table/partition-table.bin /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin
cd /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/partition_table && /home/hideit/.espressif/python_env/idf5.2_py3.11_env/bin/python /home/hideit/Arduino/lvgl_micropython/lib/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 4MB -- /home/hideit/Arduino/lvgl_micropython/build/partitions-4MiB.csv /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin && /usr/bin/cmake -E echo "Partition table binary generated. Contents:" && /usr/bin/cmake -E echo "*******************************************************************************" && /home/hideit/.espressif/python_env/idf5.2_py3.11_env/bin/python /home/hideit/Arduino/lvgl_micropython/lib/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 4MB -- /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/partition_table/partition-table.bin && /usr/bin/cmake -E echo "*******************************************************************************"
Partition vfs invalid: Offset 0x28d270 is not aligned to 0x1000
[2/250] cd /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/main_esp32 && /usr/bin/cmake -E make_directory /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/genhdr && /home/hideit/.espressif/python_env/idf5.2_py3.11_env/bin/python /home/hideit/Arduino/lvgl_micropython/lib/micropython/py/makeversionhdr.py /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/genhdr/mpversion.h
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stderr_output_85524 and /home/hideit/Arduino/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC/log/idf_py_stdout_output_85524
```

zelogik, 2024-06-18:
link with commit a4c3252 ?

kdschlosser, 2024-06-18:
I know the problem exists. I am hammering it out now. The partition files that come with MicroPython are very limiting and I wanted to make sure to squeeze every drop of space I am able to so the user is able to use it. Things like having the application partition set to a size of 3,145,728 bytes (3MiB) and the program size is 2,594,480 bytes is wasteful. You end up having 551,248 bytes of space that ends up going unused. There is no way to access that at runtime either, so you cannot do anything with it. It is best to tighten of the partition to align with what is actually needed. The partition sizes need to be a multiple of 4096 so what needs to be done is the size of the compiled application needs to be normalized using `(int(app size / 4096) + 1) * 4096` which will give us a size of 2,596,864 for the partition which leaves 2,384 bytes of extra space that is technically lost. That is a whole lot less than the 551,248 bytes that were previously lost.

kdschlosser, 2024-06-18:
what the goal I am trying to get to is to eliminate the need to compile a second time that would occur if nothing in changed in `lv_config.h`. For the most part there is going to be a pretty concrete firmware size for this. So I am trying to get things dialed in as close as possible right out of the gate to eliminate that second compile.

kdschlosser, 2024-06-18:
if you compile using the code form this point in the repo it should compile without any issues.

https://github.com/kdschlosser/lvgl_micropython/tree/f0e1c33bda58327878fb33f0aec08f40eaa9495d

kdschlosser, 2024-06-19:
It should compile properly now. I just got the last update all patched up.

zelogik, 2024-06-19:
So:
At the end of the compile we get:
python -m esptool --chip esp32 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin

- the path seem to be wrong, ie: missing "lib/micropython/ports/esp32/"
- And the compiled images, don't work, impossible to get REPL, screen kept black too...

kdschlosser, 2024-06-19:
use the master branch. That issue should be fixed.

zelogik, 2024-06-19:
+1 
Can close the issue

closed 2024-06-19 by zelogik

kdschlosser, 2024-06-19:
all good now?