lvgl-micropython/lvgl_micropython #62: Can't access REPL or filesystem with ESP32-S3

dikei100, 2024-06-19:
Hi,

I was able to successfully build und flash an ESP32-S3 with octal-SPIRAM, but I am not able to connect to the board after flashing the firmware. I think I am missing something, but maybe the specific board I am using is just unsupported.

I am using a T-Display S3 Pro (Specifications at https://github.com/Xinyuan-LilyGO/T-Display-S3-Pro and https://www.lilygo.cc/products/t-display-s3-pro)

I am using this command:

python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 --flash-size=16

I also tried to add --usb-otg flag, but no success:

python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 --usb-otg --flash-size=16

Error message in for Thonny:
Internal error (serial.serialutil.SerialTimeoutException: Write timeout).

Maybe someone has an idea, what I am missing here? The precompiled Micropython binary version 1.23 for Generic ESP32-S3 boards with SPIRAM works fine.

Thanks in advance! :)

zelogik, 2024-06-19:
Read correctly the last after the compilation done.
Got the same problem with esp32.

Make full flash reset, and reupload the firmware. 
As code have been improved to use in a better way the flash partition.

@kdschlosser, it can be a problem... change the partition could be problematic if we just want to use a new lvgl/micropython image but we don't want to change files inside flash partition, no?

kdschlosser, 2024-06-20:
It the sizes of the partitions and they MUST change because it won't fit on the ESP32 unless they get changed. MicroPython has a lot of the application partition sizes set to 2mb but when LVGL is compiled in with MicroPython the resulting binary size is 2.5mb. so it will not fit. The partition files need to altered to get it to fit.

What I had changes that I believe is causing the issue is I added support for overclocking the flash and the SPIRAM on the esp32-s3. It's how I went about doing that is what I believe is causing the problems.

I changed up the code for that I am waiting for the CI to finish building to make sure it at least compiles properly.

kdschlosser, 2024-06-20:
It successfully compiled for the ESP32. so you can give it another go and clone the repo to see if it is working now.

dikei100, 2024-06-20:
Thanks for the fast response!

Tried to reclone the repo and recompile, but no success. Micropython still can't be accessed, neither through Thonny now through picocom.

Here is the log from Thonny:

09:53:04.500 [MainThread] ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/thonny/plugins/micropython/mp_back.py", line 138, in __init__
    self._process_until_initial_prompt(
  File "/usr/lib/python3.12/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 278, in _process_until_initial_prompt
    self._process_output_until_active_prompt(
  File "/usr/lib/python3.12/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 861, in _process_output_until_active_prompt
    self._write(RAW_MODE_CMD)
  File "/usr/lib/python3.12/site-packages/thonny/plugins/micropython/mp_back.py", line 461, in _write
    return self._connection.write(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/thonny/plugins/micropython/serial_connection.py", line 88, in write
    size = self._serial.write(data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/serial/serialposix.py", line 636, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

To erase flash I use the command from the output given after compilation finishes:

/home/user/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool -p /dev/ttyACM0 -b 460800 erase_flash

Output:

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 74:4d:bd:87:73:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 36.9s
Hard resetting via RTS pin...

(PS: why is it changing from ESP-IDF 5.2 to 4.7?)

To flash the device:

/home/user/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool --chip esp32s3 -b 921600 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 "/home/user/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-SPIRAM_OCT-16.bin"

Output:

esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 74:4d:bd:87:73:34
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00275fff...
Compressed 2580288 bytes to 1492589...
Wrote 2580288 bytes (1492589 compressed) at 0x00000000 in 23.0 seconds (effective 897.6 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

After that of course I reset the device to exit bootloader-mode.

Could the SY6970 chip for phone OTG inside the device be a problem? (though I have no problem with the chip while using the original micropython binary).

kdschlosser, 2024-06-20:
OH you are not specifying --octal_flash are you?

also what board are you using. I want to see if it uses OTG or not.

kdschlosser, 2024-06-20:
also can you try flashing using this command...

```
/home/user/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool --chip esp32s3 -b 921600 --before default_reset --after no_reset write_flash --flash_mode qio --flash_size 16MB --flash_freq 80m 0x0 "/home/user/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-SPIRAM_OCT-16.bin"
```

dikei100, 2024-06-20:
Oh, no, I thought defining BOARD_VARIANT=SPIRAM_OCT would be enough! Just tried with --octal-flash, but still doesn't work. But interestingly the display now starts to turn on and off now...

The board seems to be a custom board, the T-Display S3 Pro from Lilygo. The used MPU is an ESP32-S3R8.  On their website they also write that they use an additional SY6970 chip for phone OTG.

While flashing I get these infos from esptool.py:

Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz

On their github repository (link in my first post at the top) they recommend these settings for flashing with Arduino IDE:

USB CDC On Boot: Enabled
CPU Frequency: 240 MHz WiFi
USB DFU On Boot: Disabled
Core Debug Level: None
Events Run On: Core 1
Flash Mode: QIO 80 Mhz
Flash size: 16MB
USB Firmware: MSC On Boot Disabled
Partition Scheme: 16M Flash (3MB APP/9.9MB FATFS)
PSRAM: OPI PSRAM
Upload Mode: UART0 / Hardware CDC
USB Mode: Hardware CDC and JTAG

Maybe that helps?

(Stillt getting no connection with the new flash command btw)

kdschlosser, 2024-06-20:
I am going to make some changes. hold on for a bit,
.

kdschlosser, 2024-12-01:
Delete your clone of the repo and clone it again and then try and build it again. There have been a lot of changes since you opened this issue and it might be solved.

kdschlosser, 2024-12-01:
You should be using this build command...

```
python3 make.py esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 --usb-otg --flash-size=16 --octal-flash
```

after it is done compiling it will tell you the command you need to run in order to flash the ESP32. use the provided command. Personally I dislike Thonny it has too many issues with it. things like it doesn't perform hard resets, only soft resets which only cleans out the memory and doesn't reset the hardware states.

I suggest that after you flash the ESP32 you use an actual terminal program to connect to the ESP32 to see if it is running. There is a utility that was written by the MicroPython devs that handles uploading files, i don't know if it is cross platform or not but it is something that is worth looking into. 

I don't like the way MicroPython handles transferring files. One of these days I am going to write a way to do this without having to mess about with the repl and sending all kind of extra data back and forth and the inability to negotiate what the transfer speed should be. I have already written a cross platform driver that will show the ESP32 as a removable drive in an OS where files can be dragged and dropped as well as directories as well. At the time the data transfer rate was only 115200 so it was horribly slow because of the sheer amount of data going back and forth through the REPL. The ability to change the serial speed to a higher speed has been added since then, I just have not gotten back to that driver to get it to work better. One day I will get there. Changes need to be written into the C code in MicroPython to make it run better.

closed 2024-12-03 by kdschlosser