lvgl-micropython/lvgl_micropython #63: ESP32_C3 firmware is permanently rebooting no REPL

getznerh, 2024-06-19:
I use the last build and comile for ESP32_C3 with
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_C3 DISPLAY=st7789
firmware compiles without error, upload to board without error
python -m esptool --chip esp32c3 --port COM11  -b 921600 --before default_reset --after no_reset --no-stub write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m 0x0 "lvgl_micropy_ESP32_GENERIC_C3-4.bin"

after board-reset:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1028
load:0x403cc710,len:0xa08
load:0x403ce710,len:0x2c08
entry 0x403cc710
E (24) flash_parts: partition 0 invalid magic number 0xfdcc
E (24) boot: Failed to verify partition table
E (24) boot: load partition table error!

kdschlosser, 2024-06-19:
OK this is an issue with the flash memory. I probably have the wrong mode set for it.

maro48m, 2024-07-08:
I have similar issue  (maybe even the same:) ): after building C3:
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_C3 --partition-size=2700000 --flash-size=4 DISPLAY=gc9a01 INDEV=cst816s 
board is stuck in reboot loop with errors:

`ELF file SHA256:

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381006
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x89c
load:0x403ce710,len:0x2b14
entry 0x403cc710

A fatal error occurred. [...]

MPY version : a61c446-dirty on 2024-07-08
IDF version : 11eaf41b
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
`
After that i've got core dump.
I'm building against latest git repositories.

kdschlosser, 2024-07-08:
I need you to decode the core dump

maro48m, 2024-07-09:
Hi Kevin,

I've tried decoding that data, but for me it seems like it's crashing in bootloader: there is no backtrace addres list in this dump:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381006
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x89c
load:0x403ce710,len:0x2b14
entry 0x403cc710

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : a61c446-dirty on 2024-07-09
IDF version : 11eaf41b
Machine     : ESP32C3 module with ESP32C3

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4004c48c  RA      : 0x4004c4b0  SP      : 0x3fcde2b0  GP      : 0x3fc97200  
TP      : 0x00000000  T0      : 0x4005890e  T1      : 0x00000000  T2      : 0x00000018  
S0/FP   : 0x000000ff  S1      : 0x00000050  A0      : 0x000000ff  A1      : 0x3fcde2e0  
A2      : 0x0012ad40  A3      : 0x03ff0000  A4      : 0x0000e000  A5      : 0x60ffe01c  
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x0000000c  S3      : 0x3fcde434  
S4      : 0x42000000  S5      : 0x3c210000  S6      : 0xffff0000  S7      : 0x3fcd6000  
S8      : 0x00080020  S9      : 0x00010020  S10     : 0x00000006  S11     : 0x3fcde3cc  
T3      : 0x60008000  T4      : 0x60008000  T5      : 0xffff01ff  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x60ffe01c  
MHARTID : 0x00000000  

Stack memory:
3fcde2b0: 0x0000000c 0x00000050 0x00000001 0x420d56d8 0x400391e6 0x00000000 0x00000000 0x60008000
3fcde2d0: 0x00000001 0x000001e0 0x00000006 0x00000050 0x00000001 0x000001e0 0x00000003 0x000000a0
3fcde2f0: 0x0000000c 0x4220e9c7 0x00000000 0x40380d50 0x40380c46 0x4220e9c7 0x00000001 0x403cf714
3fcde310: 0x04083fd1 0xd46ca6c5 0x31303a6f 0x0020e9a8 0x59820a56 0xf1dab2c6 0x0ea4d4d1 0x00210000
3fcde330: 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcdf000
3fcde350: 0x00004308 0x3ff1c14c 0x0000ffff 0xffffffff 0xffffffff 0x3fcde4b0 0xffffffff 0x403cfaf8
3fcde370: 0xbf75892a 0x7179415c 0x00010000 0x002dd000 0x00010000 0x2f0206e9 0x40380c46 0x000000ee
3fcde390: 0x03030005 0x0000c700 0x01000000 0x3c210020 0x0006f150 0x3fc96a00 0x00000ea0 0x42000020
3fcde3b0: 0x0020e9a8 0x3fc978a0 0x00002c44 0x40380000 0x000168a0 0x50000000 0x00000018 0x00000000
3fcde3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde410: 0x00000000 0x00000000 0x00000000 0x00010020 0x0007f178 0x00080020 0x0028e9d0 0x0029161c
3fcde430: 0x002a7ec4 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde450: 0x00000000 0x00000000 0x00000000 0x00297f00 0x04083fd1 0xd46ca6c5 0x31303a6f 0x30fe7ad1
3fcde470: 0x59820a56 0xf1dab2c6 0x0ea4d4d1 0xe77568eb 0x400481b0 0x0000003a 0x00000000 0x3fce0000
3fcde490: 0x00000000 0x0000003a 0x00004310 0x403cc796 0x418cdc22 0xd84ada26 0x00000000 0xffffffff
3fcde4b0: 0x00000000 0x00000000 0x00010000 0x002dd000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde530: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde550: 0x600c5200 0x600c5200 0x00000000 0x4004a296 0x97b34785 0x450100e7 0x3fcde668 0x00000000
3fcde570: 0xffffffff 0x00000000 0x00010000 0x2f0203e9 0x403cc710 0x403ce710 0x00002b14 0x000000ee
3fcde590: 0x03030005 0x0000c700 0x01000000 0x3c004310 0x7533885e 0x8e0d8e5b 0xb4de241f 0x11a741d2
3fcde5b0: 0x29350fb0 0xafa09c61 0x8a9f2adc 0x1bb09593 0x72b327f5 0x00000000 0x00000000 0x00000000
3fcde5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcde700 0x00000101 0x00000002
3fcde5f0: 0x8e0d8e5b 0xb4de241f 0x11a741d2 0x29350fb0 0xafa09c61 0x8a9f2adc 0x1bb09593 0x72b327f5
3fcde610: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde630: 0x00876713 0x8082dbb8 0x00000000 0x3a000000 0x00000080 0x00000000 0x00000000 0x00000000
3fcde650: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x80180200
3fcde670: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde690: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 

Rebooting...

i've checked entry point address (last addres before crash) 0x403cc710, it's from bootloader:
lvgl_micropython/lib/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

kdschlosser, 2024-07-09:
have you pulled the most recent code changes yet? I would delete you local clone and clone it again and build it again. I did fix some things that might have caused this problem. I would start off there and see if that solves the issue.

If it doesn't can you give me the brand and model number of the board you are using.

maro48m, 2024-07-09:
I've checked with fresh clone of repository - it doesn't help - errors are roughly the same.
I'm using esp32c3 board with builtin gc9a01 display and cst816s touch from aliexpress.

esptool identifies chip as:
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 68:67:25:6b:79:50
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB

I have no problem using generic micropython on it and loading native esp-idf compiled program.

kdschlosser, 2024-07-09:
what are you using for a compile command?

kdschlosser, 2024-07-09:
I just updated the repo. I am hoping the change will fix your issue.

maro48m, 2024-07-10:
i'm using 
`python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC DISPLAY=gc9a01 INDEV=cst816s`
Latest build (again clean clone) did not help.
Earlier i was using command to flash bootloader, micropython and partition table
Now i've used (as suggested by make tool) one generated file from lvgl_micropython/build dir

Is there any more information i could provide you or maybe change some esp-idf settings  to test?

kdschlosser, 2024-07-10:
IDK what the flash size is for the board. If you can add the  `--flash-size={MB}` parameter to the build command... You have to change `{MB}` to the number of megabytes your board has. This would make sure that it is assigning the correct amount of flash memory. That's a starting point.

Now you said that it was working with vanilla MicroPython.... What version??

maro48m, 2024-07-11:
According to esptool my board has 4mb flash.
With vanilla MP i'm using ESP32_GENERIC_C3-20240602-v1.23.0.bin from micopython site - it's latest available stable build.
I've tried building with --flash-size=4 and without - the same effect.

kdschlosser, 2024-07-11:
OK lemme get a binary built for ya and you can see if that works. we need to eliminate the build environment as a possible cause. If it works then we know there is an issue because of macOS and that will narrow down where we need to look for a problem. If it doesn't then we know it is not going to be something specific to macOS and once again that changes where to look for the issue.

maro48m, 2024-07-11:
I'm not building on macOS - sorry if i had not specified that.
I'm using windows 11 with installed WSL debian distro - on that debian i've got build toolchain.

kdschlosser, 2024-07-12:
i got you mixed up with another user. I'm sorry about that.

kdschlosser, 2024-07-12:
and flash size is only going to be dictated by how the binary is built. only the spiram is auto detectable. So we need to make sure how how much flash your board has.

kdschlosser, 2024-07-12:
if you can also provide me with the code that runs when you are booting the ESP32.

maro48m, 2024-07-12:
Kevin,

My chip is ESP32c3 mini 1u - according to specs it has 4MB flash.
When it boots after uploading lvgl_micropython it gives this:

ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381006
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x89c
load:0x403ce710,len:0x2b14
entry 0x403cc710

then error message, then core dump, then reboot

kdschlosser, 2024-07-12:
what is the code you are running or does it reboot without running any code?

maro48m, 2024-07-12:
No code at all.
As i understood micropython debugging - if firmware starts properly it should give ELF file SHA and there should backtrace list of addreses that caused error. If you look at my error few comments ago - threre is nothing like that in my case.

kdschlosser, 2024-07-12:
You are going to have to give me some time to see if I am able to locate the exact issue. I want to make sure that the USB is getting enabled properly. I have a sneaky suspicion that there is something going on there that is causing the issue. 

OH!!!!

I cannot believe I forgot about this...

add `--usb-otg` to the build command when you compile. See if that fixes it. I am thinking it will.

maro48m, 2024-07-12:
I'm getting error "make: unrecognized option '--usb-otg'"
But i've checked make.py - and this option seems to be parsed only for s3 and s2 boards.

kdschlosser, 2024-07-12:
OK i need to make a code change real quick.,..

kdschlosser, 2024-07-12:
OMG I cannot believe I missed that!!!!

This is WRONG

```
 python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC DISPLAY=gc9a01 INDEV=cst816s
```

This is right

```
 python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s --flash-size=4
```

HOLY COW!!!

maro48m, 2024-07-13:
You didn't missed that, I'm using correct board:
`sudo python3 make.py esp32 submodules clean mpy_cross --flash-size=4 BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s`
that's my exact command line

kdschlosser, 2024-07-13:
are you sure???

https://github.com/kdschlosser/lvgl_micropython/issues/63#issuecomment-2221260153

It's not what you said there...

Spefx, 2024-07-13:
Thanks for your time and support -- I have come issue after trying to flash my ESP32-C3 board and running into the same issue, after building and loading the firmware when connecting via mpremote it seems to constantly be rebooting.

for clarity, the specific board i am using is the [seeed XIAO ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html?gad_source=1&gclid=CjwKCAjwy8i0BhAkEiwAdFaeGFZ3Cpi16UOyiPo09HFrWHeKX6GUzF3q2SbRZeT2UXgqAAIpO_EAihoCIVEQAvD_BwE)

the command that i used during build was:

` python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s --flash-size=4`

When connecting with mpremote with the goal of accessing REPL after flashing the firmware I am seeing the following:

```
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381006
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x89c
load:0x403ce710,len:0x2b14
entry 0x403cc710
```

I have looked to include the --usb-otg command as reccomended above however this is generating:  
`make: unrecognized option '--usb-otg"`

I am somewhat new to the world of microcontrollers so please forgive me if I have used the wrong terminology..

kdschlosser, 2024-07-13:
the osb-otg is not being disabled for them C3. so that option doesn't work. I couldn't remember how I set that up and when I went and looked at the code it only works for the S2 and S3.

maro48m, 2024-07-13:
> are you sure???
> 
> [#63 (comment)](https://github.com/kdschlosser/lvgl_micropython/issues/63#issuecomment-2221260153)
> 
> It's not what you said there...

I'm absolutely sure i use proper board.
I've must have copied wrong command in my previous comment - i've tried few boards before decided to bother You.

kdschlosser, 2024-07-13:
OK I just wanted to make sure it was correct.

I am trying to figure out why the C3 is having that issue. There is nothing in the code that is specific to the C3 and everything compiles and works on other ESP32 versions...

tempdban, 2024-08-30:
try this:
esp_idf:5.2.2
cd lib/micropython/
git checkout master #may need force checkout

try build again.
my build output:
https://file.io/6GOxyJpo9LMH

maro48m, 2024-09-01:
@tempdban i don't really know how to force specific esp_idf version,
also trying to checkout master in lib/micropython gives me error
```
~/lvgl_micropython/lib/micropython$ git checkout master
error: pathspec 'master' did not match any file(s) known to git
```
i've tried this on lvgl dir that i've pulled from git earlier and also on fresh dir made with git checkout

tempdban, 2024-09-01:
> @tempdban i don't really know how to force specific esp_idf version, also trying to checkout master in lib/micropython gives me error
>
> ```
> ~/lvgl_micropython/lib/micropython$ git checkout master [...]

Follow esp-idf doc install 5.2.2 to ~./espressif/

Your submodule is not initialized.

kdschlosser, 2024-09-01:
you don't need to do anything with the submodules. just clone the repo and run the `make.py` script. that is all that needs to be done..

kdschlosser, 2024-09-01:
the ESP-IDF gets set up in a partial sandbox environment so it is not going to be initialized in the shell even after running the make.py script. There will be the `.espressif ` folder in your home directory and that's it.

maro48m, 2024-09-22:
Hi,
i've tried to check if recent changes may fix my problem with board reset, but lates pull is not building at all for C3.
I've done clean clone, deleted .espressif, went through newes build commands (especialy problem with not initialised berkley-db).
Now i get error:
`MPY spi3wire.py
error compiling spi3wire.py:
Traceback (most recent call last):
File "spi3wire.py", line 80, in Spi3Wire
SyntaxError: invalid arch`

My current command:
`sudo python3 make.py esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s`

the same command for BOARD=ESP32_GENERIC works

maro48m, 2024-10-06:
> Hi, i've tried to check if recent changes may fix my problem with board reset, but lates pull is not building at all for C3. I've done clean clone, deleted .espressif, went through newes build commands (especialy problem with not initialised berkley-db). Now i get error: `MPY spi3wire.py error compiling spi3wire.py: Traceback (most recent call last): File "spi3wire.py", line 80, in Spi3Wire SyntaxError: invalid arch`
> 
> My current command: `sudo python3 make.py esp32 BOARD=ESP32_GENERIC_C3 DISPLAY=gc9a01 INDEV=cst816s`
> 
> the same command for BOARD=ESP32_GENERIC works, but i haven't checked it it launches after flash to actual board.

I've found solution for this problem:
in file 

`api_drivers\common_api_drivers\frozen\other\spi3wire.py`

 three methods are marked with `@micropython.viper`
after removing these decorators code builds for ESP32C3...

unfortunatelly it still is not launchind after flashing, so i cannot check if removing these decorators breaks something else in build micropython..

kdschlosser, 2024-10-06:
OK so the C3 doesn't support using the viper code emitter.

closed 2024-12-17 by kdschlosser