lvgl-micropython/lvgl_micropython #64: REPL not opening

arunkm490, 2024-06-21:
i built a firmware for ESP32 GENERIC S3 (ESP32-S3-xiao seeed studio)
python3 make.py esp32 submodules clean mpy_cross BOARD=ESP32_GENERIC_S3 DISPLAY=ili9488 INDEV=xpt2046
by using this command.
binary file created successfully.

after flashing the firmware.bin ,I can't open the repl.

fabse-hack, 2024-08-31:
how you flash it? maybe you can tell us your esptool command?

kdschlosser, 2024-09-01:
Thonny is one of the issues. It doesn't do hard resets only soft resets and that causes issues. My suggestion for you is to use a terminal program like putty to connect to the ESP32. There is also mpremote as well. 

Does your ESP have one or 2 USB ports? do you know what GPIO's the USB is connected to? Is it 19 and 20?

fabse-hack, 2024-09-01:
hmmmmm another think is: there is a missing `--usb-otg` in your command. But i'm not 1000% sure about it

kdschlosser, 2024-09-01:
That is why I was asking if the USB is connected to pins 19 and 20. if they are then `--osb-otg` would need to be added to get the repl to work.. 

pins 19 and 20 even when not used for USB would not be usable in MicroPython. It sets up those pins as USB pins. This doesn't allow them to be used for anything else and this is not ideal. so instead of loosing 2 pins for the REPL not 4 are lost. This is not ideal and is why I fixed the issue.

closed 2024-12-17 by kdschlosser