lvgl-micropython/lvgl_micropython #67: bash: line 1: /: is a directory

dcmcshan, 2024-06-26:
Am attempting to build/deploy with this command.

python3 make.py esp32 clean deploy BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM DISPLAY=st7735 PORT=/dev/cu.usbmodem2101 --flash-size=4

It seems to build fine, but ends with 

"bash: line 1: /: is a directory"

and does not install the firmware.

kdschlosser, 2024-06-26:
where did you get the "PORT=" build command from You should probably view the readme file located here

https://github.com/kdschlosser/lvgl_micropython/blob/main/README.md

It goes over all of the build commands and provides examples of how to use them.

dcmcshan, 2024-06-26:
% python3 make.py esp32 clean deploy BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM DISPLAY=st7735 PORT=/dev/cu.usbmodem2101 --flash-size=4
Cleaning build....
Getting ESP-IDF build Environment
idf.py -D MICROPY_BOARD=ESP32_GENERIC_S3 -D MICROPY_BOARD_DIR="/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3"  -D MICROPY_BOARD_VARIANT=SPIRAM -B build-ESP32_GENERIC_S3-SPIRAM -p /dev/ttyUSB0 -b 460800 fullclean
Executing action: fullclean
Executing action: remove_managed_components
Done
Generating build files....
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/display/display_driver_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/esp32/touch_cal_data.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/touch_calibration/touch_calibrate.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/_indev_base.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/button_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/encoder_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/keypad_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/frozen/indev/pointer_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/py_api_drivers/fs_driver.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/i2c.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/io_expander_framework.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/common_api_drivers/frozen/other/task_handler.py
/Users/danielmcshan/GitHub/lvgl_micropython/api_drivers/common_api_drivers/display/st7735.py
Compiling....
idf.py -D MICROPY_BOARD=ESP32_GENERIC_S3 -D MICROPY_BOARD_DIR="/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3"  -DUSER_C_MODULES=../../../../../ext_mod/micropython.cmake -D MICROPY_FROZEN_MANIFEST=/Users/danielmcshan/GitHub/lvgl_micropython/build/manifest.py -D MICROPY_BOARD_VARIANT=SPIRAM -B build-ESP32_GENERIC_S3-SPIRAM build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Executing action: all (aliases: build)
Running cmake in directory /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/danielmcshan/.espressif/python_env/idf5.2_py3.12_env/bin/python -DESP_PLATFORM=1 -DMICROPY_BOARD=ESP32_GENERIC_S3 -DMICROPY_BOARD_DIR=/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/boards/ESP32_GENERIC_S3 -DUSER_C_MODULES=../../../../../ext_mod/micropython.cmake -DMICROPY_FROZEN_MANIFEST=/Users/danielmcshan/GitHub/lvgl_micropython/build/manifest.py -DMICROPY_BOARD_VARIANT=SPIRAM -DCCACHE_ENABLE=0 /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32"...
-- The ASM compiler identification is GNU                          
-- Found assembler: /Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Building ESP-IDF components for target esp32s3
Processing 4 dependencies:
[4/4] idf (5.2.0)                  
-- Project sdkconfig file /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/sdkconfig
Loading defaults file /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/sdkconfig.combined...
/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/sdkconfig.combined:100 CONFIG_ESP32_WIFI_IRAM_OPT was replaced with CONFIG_ESP_WIFI_IRAM_OPT
/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/sdkconfig.combined:101 CONFIG_ESP32_WIFI_RX_IRAM_OPT was replaced with CONFIG_ESP_WIFI_RX_IRAM_OPT
/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/sdkconfig.combined:148 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
/var/folders/sg/7k1f9lns75sdtd39z4tnhvjr0000gn/T/kconfgen_tmpgqwkp_4b:167 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n
/var/folders/sg/7k1f9lns75sdtd39z4tnhvjr0000gn/T/kconfgen_tmpgqwkp_4b:169 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_16MB=n
-- Check size of time_t - done               
-- Found Python3: /Users/danielmcshan/.espressif/python_env/idf5.2_py3.12_env/bin/python (found version "3.12.3") found components: Interpreter
-- App "micropython" version: v1.23.0-dirty                        
-- Adding linker script /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/esp-idf/esp_system/ld/memory.ld
[... cut ...]
python -m esptool --chip esp32s3 -b 460800 --before default_reset --after no_reset write_flash "@flash_args"
bootloader  @0x000000    19152  (   13616 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2510592  (    8448 remaining)
total                  2576128

esptool.py v4.7.0
Wrote 0x274f00 bytes to file /Users/danielmcshan/GitHub/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM-SPIRAM-4.bin, ready to flash to offset 0x0
bash: line 1: /: is a directory

kdschlosser, 2024-06-26:
add `deploy` to your build command.

kdschlosser, 2024-06-26:
scratch that. I just saw you have it in there.

dcmcshan, 2024-06-27:
Some guy on forum.lvgl.io :) :)

https://github.com/kdschlosser/lvgl_micropython/blob/main/README.md

kdschlosser, 2024-06-27:
I know I know. I was out of my mind for a second there. I didn't realize that I had pushed that change to the repo. When I realized it I deleted the comment.

You problem should be fixed now. I just pushed some commits and one of them addressed the issue you were having.

kdschlosser, 2024-06-29:
I am going to need a little bit of time. Please read...  https://github.com/kdschlosser/lvgl_micropython/discussions/69