lvgl-micropython/lvgl_micropython #70: Issue using new SPI.Bus and SPI.Device on ESP32

ikkesia, 2024-07-09:
Last commit [@dd293a4](https://github.com/kdschlosser/lvgl_micropython/commit/dd293a43afbda649b5af8696e0c060e0c31d27bb)

Using these new features I get an error
```
Traceback (most recent call last):
  File "<stdin>", line 89, in <module>
  File "display_driver_framework.py", line 218, in __init__
ValueError: 258(esp_lcd_new_panel_io_spi)
```
Maybe I made the wrong calls.

Below is my code

```
from micropython import const
from machine import SPI
import time
import sys
import lcd_bus
import lvgl as lv

# Display settings
_WIDTH = const(240)
_HEIGHT = const(320)

# Display SPI bus settings
_LCD_SPI_HOST = const(1)
_LCD_SPI_FREQ = const(40_000_000)
_LCD_MOSI_PIN = const(23)
_LCD_MISO_PIN = const(25)
_LCD_SCLK_PIN = const(19)
_LCD_CS_PIN = const(22)
# display additional pins
_LCD_RST_PIN = const(18)
_LCD_DC_PIN = const(21)
_LCD_BKL_PIN = const(5)
_LCD_PWR_PIN = None

# touch panel SPI settings
_TP_SPI_HOST = const(1)
_TP_SPI_FREQ = const(1_000_000)
_TP_CS_PIN = const(32)
_TP_INT_PIN = const(-1)

class TouchCal:
    def __init__(self):
        self.alphaX = 1.088837
        self.betaX = 0.005908419
        self.deltaX = -7.80586
        self.alphaY = 0.00886263
        self.betaY = -1.136386
        self.deltaY = 353.614

    def save():
        pass

# Create SPI bus
spi_bus = SPI.Bus(
    host=_LCD_SPI_HOST,
    miso=_LCD_MISO_PIN,
    mosi=_LCD_MOSI_PIN,
    sck=_LCD_SCLK_PIN
)

# Create touch controller SPI Device
touch_spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=_TP_SPI_FREQ,
    cs=_TP_CS_PIN
)

# Create display SPI Device
display_spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=_LCD_SPI_FREQ,
    cs=_LCD_CS_PIN
)

display_bus = lcd_bus.SPIBus(
    spi_bus=display_spi_device,
    freq=_LCD_SPI_FREQ,
    dc=_LCD_DC_PIN		#,
    #cs=_LCD_CS_PIN
)

import ili9341
try:
    display = ili9341.ILI9341(
        data_bus=display_bus,
        display_width=_WIDTH,
        display_height=_HEIGHT,
        reset_pin=_LCD_RST_PIN,
        reset_state=ili9341.STATE_LOW,
        power_pin=_LCD_PWR_PIN,
        backlight_pin=_LCD_BKL_PIN,
        backlight_on_state=ili9341.STATE_HIGH,
        color_space=lv.COLOR_FORMAT.RGB565,
    #	 color_byte_order=ili9341.BYTE_ORDER_BGR,
        rgb565_byte_swap=False
    )
except Exception as ex:
    sys.print_exception(ex)

# start display driver
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
#display.set_backlight(50)

import xpt2046

cal=TouchCal()
try:
    indev = xpt2046.XPT2046(touch_spi_device, startup_rotation=lv.DISPLAY_ROTATION._90
                            , touch_cal=cal, debug=True)
    print('is_calibrate is', indev.is_calibrated)

except Exception as ex:
    sys.print_exception(ex)
```

kdschlosser, 2024-07-09:
Since the touch panel and the display share the same SPI bus that is what you pass to `lcd_bus.SPIBus` hence the keyword `spi_bus` being used. 

```python
from micropython import const
from machine import SPI
import time
import sys
import lcd_bus
import lvgl as lv

# Display settings
_WIDTH = const(240)
_HEIGHT = const(320)

# Display SPI bus settings
_LCD_SPI_HOST = const(1)
_LCD_SPI_FREQ = const(40_000_000)
_LCD_MOSI_PIN = const(23)
_LCD_MISO_PIN = const(25)
_LCD_SCLK_PIN = const(19)
_LCD_CS_PIN = const(22)
# display additional pins
_LCD_RST_PIN = const(18)
_LCD_DC_PIN = const(21)
_LCD_BKL_PIN = const(5)
_LCD_PWR_PIN = None

# touch panel SPI settings
_TP_SPI_HOST = const(1)
_TP_SPI_FREQ = const(1_000_000)
_TP_CS_PIN = const(32)
_TP_INT_PIN = const(-1)

class TouchCal:
    def __init__(self):
        self.alphaX = 1.088837
        self.betaX = 0.005908419
        self.deltaX = -7.80586
        self.alphaY = 0.00886263
        self.betaY = -1.136386
        self.deltaY = 353.614

    @staticmethod
    def save():
        pass

# Create SPI bus
spi_bus = SPI.Bus(
    host=_LCD_SPI_HOST,
    miso=_LCD_MISO_PIN,
    mosi=_LCD_MOSI_PIN,
    sck=_LCD_SCLK_PIN
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_SPI_FREQ,
    dc=_LCD_DC_PIN,
    cs=_LCD_CS_PIN
)

import ili9341
try:
    display = ili9341.ILI9341(
        data_bus=display_bus,
        display_width=_WIDTH,
        display_height=_HEIGHT,
        reset_pin=_LCD_RST_PIN,
        reset_state=ili9341.STATE_LOW,
        power_pin=_LCD_PWR_PIN,
        backlight_pin=_LCD_BKL_PIN,
        backlight_on_state=ili9341.STATE_HIGH,
        color_space=lv.COLOR_FORMAT.RGB565,
    #	 color_byte_order=ili9341.BYTE_ORDER_BGR,
        rgb565_byte_swap=False
    )
except Exception as ex:
    sys.print_exception(ex)
    raise RuntimeError

# start display driver
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
#display.set_backlight(50)

import xpt2046

# Create touch controller SPI Device
touch_spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=_TP_SPI_FREQ,
    cs=_TP_CS_PIN
)

cal = TouchCal()
try:
    indev = xpt2046.XPT2046(
        touch_spi_device, startup_rotation=lv.DISPLAY_ROTATION._90, 
        touch_cal=cal, debug=True
    )
    print('is_calibrate is', indev.is_calibrated)

except Exception as ex:
    sys.print_exception(ex)

```

ikkesia, 2024-07-09:
With this code the system crashes and reboot
```
MPY: soft reboot
is_calibrate is True

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : a61c446-dirty on 2024-07-09
IDF version : 11eaf41b
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x401a36c3  PS      : 0x00060430  A0      : 0x801bce94  A1      : 0x3ffd0400  
A2      : 0x3ffd5500  A3      : 0x00000003  A4      : 0x3ffd5420  A5      : 0x3ffd56b0  
A6      : 0x00000018  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffd03e0  
A10     : 0x00000000  A11     : 0x3ffbbbdc  A12     : 0x00000002  A13     : 0x3ffd6138  
A14     : 0x3ffbbbdc  A15     : 0x3ffbd954  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x401a36c0:0x3ffd0400 0x401bce91:0x3ffd0490 0x40182ff1:0x3ffd04d0 0x4018b7c5:0x3ffd04f0 0x4018b88d:0x3ffd0510 0x40085aa6:0x3ffd0530 0x401830db:0x3ffd05d0 0x4018b7c5:0x3ffd0640 0x4018b88d:0x3ffd0660 0x40085aa6:0x3ffd0680 0x401830db:0x3ffd0720 0x4018b7c5:0x3ffd0750 0x4018b88d:0x3ffd0770 0x40085aa6:0x3ffd0790 0x401830db:0x3ffd0830 0x4018b7c5:0x3ffd0860 0x401b76e3:0x3ffd0880 0x401b772d:0x3ffd08c0 0x4018b7c5:0x3ffd08e0 0x400f6379:0x3ffd0900 0x40121357:0x3ffd0940 0x401213b6:0x3ffd0970 0x4012136d:0x3ffd09c0 0x401461d1:0x3ffd09f0 0x40145db9:0x3ffd0a30 0x400d5419:0x3ffd0a70 0x400f72bd:0x3ffd0a90 0x400d5555:0x3ffd0ad0 0x4018b7c5:0x3ffd0b10 0x4018b88d:0x3ffd0b30 0x40085aa6:0x3ffd0b50 0x401830db:0x3ffd0bf0 0x4018b7c5:0x3ffd0c20 0x401b76e3:0x3ffd0c40 0x401b772d:0x3ffd0c80 0x4018b7c5:0x3ffd0ca0 0x4018b84d:0x3ffd0cc0 0x4018c40c:0x3ffd0cf0 0x4018c589:0x3ffd0d70 0x4019f82f:0x3ffd0d90 0x40193c13:0x3ffd0dc0 0x40182fb5:0x3ffd0de0 0x4018b7c5:0x3ffd0e00 0x4018b88d:0x3ffd0e20 0x40085aa6:0x3ffd0e40 0x401830db:0x3ffd0ee0 0x4018b7c5:0x3ffd0f10 0x4018b7da:0x3ffd0f30 0x40197c1b:0x3ffd0f50 0x40197d5e:0x3ffd0fe0 0x4019f3f6:0x3ffd1030

ELF file SHA256: e9d5f21b1

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4892
ho 0 tail 12 room 4
load:0x40078000,len:14892
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3372
entry 0x400805b0
MicroPython a61c446-dirty on 2024-07-09; Generic ESP32 module with ESP32
Type "help()" for more information.
```

kdschlosser, 2024-07-09:
If you can decode that backtrace it would be helpful for me to locate the zero division error that is occuring.

kdschlosser, 2024-07-09:
you will need to modify the paths in the command below. once you do that and run it it will decode the backtrace. Paste that into a post for me.

```
~/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32-elf-addr2line -aspCfire "lib/micropython/ports/esp32/{build_folder}/micropython.elf"  0x401a36c0:0x3ffd0400 0x401bce91:0x3ffd0490 0x40182ff1:0x3ffd04d0 0x4018b7c5:0x3ffd04f0 0x4018b88d:0x3ffd0510 0x40085aa6:0x3ffd0530 0x401830db:0x3ffd05d0 0x4018b7c5:0x3ffd0640 0x4018b88d:0x3ffd0660 0x40085aa6:0x3ffd0680 0x401830db:0x3ffd0720 0x4018b7c5:0x3ffd0750 0x4018b88d:0x3ffd0770 0x40085aa6:0x3ffd0790 0x401830db:0x3ffd0830 0x4018b7c5:0x3ffd0860 0x401b76e3:0x3ffd0880 0x401b772d:0x3ffd08c0 0x4018b7c5:0x3ffd08e0 0x400f6379:0x3ffd0900 0x40121357:0x3ffd0940 0x401213b6:0x3ffd0970 0x4012136d:0x3ffd09c0 0x401461d1:0x3ffd09f0 0x40145db9:0x3ffd0a30 0x400d5419:0x3ffd0a70 0x400f72bd:0x3ffd0a90 0x400d5555:0x3ffd0ad0 0x4018b7c5:0x3ffd0b10 0x4018b88d:0x3ffd0b30 0x40085aa6:0x3ffd0b50 0x401830db:0x3ffd0bf0 0x4018b7c5:0x3ffd0c20 0x401b76e3:0x3ffd0c40 0x401b772d:0x3ffd0c80 0x4018b7c5:0x3ffd0ca0 0x4018b84d:0x3ffd0cc0 0x4018c40c:0x3ffd0cf0 0x4018c589:0x3ffd0d70 0x4019f82f:0x3ffd0d90 0x40193c13:0x3ffd0dc0 0x40182fb5:0x3ffd0de0 0x4018b7c5:0x3ffd0e00 0x4018b88d:0x3ffd0e20 0x40085aa6:0x3ffd0e40 0x401830db:0x3ffd0ee0 0x4018b7c5:0x3ffd0f10 0x4018b7da:0x3ffd0f30 0x40197c1b:0x3ffd0f50 0x40197d5e:0x3ffd0fe0 0x4019f3f6:0x3ffd1030
```

ikkesia, 2024-07-09:
`
tbe@MacBook-Pro build-ESP32_GENERIC % /Users/tbe/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-addr2line -aspCfire "micropython.elf"  0x401a36c0:0x3ffd0400 0x401bce91:0x3ffd0490 0x40182ff1:0x3ffd04d0 0x4018b7c5:0x3ffd04f0 0x4018b88d:0x3ffd0510 0x40085aa6:0x3ffd0530 0x401830db:0x3ffd05d0 0x4018b7c5:0x3ffd0640 0x4018b88d:0x3ffd0660 0x40085aa6:0x3ffd0680 0x401830db:0x3ffd0720 0x4018b7c5:0x3ffd0750 0x4018b88d:0x3ffd0770 0x40085aa6:0x3ffd0790 0x401830db:0x3ffd0830 0x4018b7c5:0x3ffd0860 0x401b76e3:0x3ffd0880 0x401b772d:0x3ffd08c0 0x4018b7c5:0x3ffd08e0 0x400f6379:0x3ffd0900 0x40121357:0x3ffd0940 0x401213b6:0x3ffd0970 0x4012136d:0x3ffd09c0 0x401461d1:0x3ffd09f0 0x40145db9:0x3ffd0a30 0x400d5419:0x3ffd0a70 0x400f72bd:0x3ffd0a90 0x400d5555:0x3ffd0ad0 0x4018b7c5:0x3ffd0b10 0x4018b88d:0x3ffd0b30 0x40085aa6:0x3ffd0b50 0x401830db:0x3ffd0bf0 0x4018b7c5:0x3ffd0c20 0x401b76e3:0x3ffd0c40 0x401b772d:0x3ffd0c80 0x4018b7c5:0x3ffd0ca0 0x4018b84d:0x3ffd0cc0 0x4018c40c:0x3ffd0cf0 0x4018c589:0x3ffd0d70 0x4019f82f:0x3ffd0d90 0x40193c13:0x3ffd0dc0 0x40182fb5:0x3ffd0de0 0x4018b7c5:0x3ffd0e00 0x4018b88d:0x3ffd0e20 0x40085aa6:0x3ffd0e40 0x401830db:0x3ffd0ee0 0x4018b7c5:0x3ffd0f10 0x4018b7da:0x3ffd0f30 0x40197c1b:0x3ffd0f50 0x40197d5e:0x3ffd0fe0 0x4019f3f6:0x3ffd1030
`

```
0x401a36c0: machine_hw_spi_device_transfer at machine_hw_spi.c:315
0x401bce91: mp_machine_spi_transfer at machine_spi.c:66
 (inlined by) mp_machine_spi_write_readinto at machine_spi.c:103
0x40182ff1: fun_builtin_3_call at objfun.c:92
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b88d: mp_call_method_n_kw at runtime.c:733
0x40085aa6: mp_execute_bytecode at vm.c:1042
0x401830db: fun_bc_call at objfun.c:267
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b88d: mp_call_method_n_kw at runtime.c:733
0x40085aa6: mp_execute_bytecode at vm.c:1042
0x401830db: fun_bc_call at objfun.c:267
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b88d: mp_call_method_n_kw at runtime.c:733
0x40085aa6: mp_execute_bytecode at vm.c:1042
0x401830db: fun_bc_call at objfun.c:267
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x401b76e3: mp_call_method_self_n_kw at objboundmeth.c:70
0x401b772d: bound_meth_call at objboundmeth.c:83
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x400f6379: lv_indev_t_read_cb_callback at lv_mp.c:?
0x40121357: indev_read_core at lv_indev.c:?
0x401213b6: lv_indev_read at ??:?
0x4012136d: lv_indev_read_timer_cb at ??:?
0x401461d1: lv_timer_exec at lv_timer.c:?
0x40145db9: lv_timer_handler at ??:?
0x400d5419: lv_task_handler at lv_mp.c:?
0x400f72bd: mp_lv_task_handler at lv_mp.c:?
0x400d5555: lv_fun_builtin_var_call at lv_mp.c:?
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b88d: mp_call_method_n_kw at runtime.c:733
0x40085aa6: mp_execute_bytecode at vm.c:1042
0x401830db: fun_bc_call at objfun.c:267
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x401b76e3: mp_call_method_self_n_kw at objboundmeth.c:70
0x401b772d: bound_meth_call at objboundmeth.c:83
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b84d: mp_call_function_1 at runtime.c:695
0x4018c40c: mp_call_function_1_protected at runtime_utils.c:33
0x4018c589: mp_sched_run_pending at scheduler.c:111
 (inlined by) mp_handle_pending at scheduler.c:240
0x4019f82f: mp_hal_delay_ms at mphalport.c:171
0x40193c13: time_sleep_ms at modtime.c:136
0x40182fb5: fun_builtin_1_call at objfun.c:68
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b88d: mp_call_method_n_kw at runtime.c:733
0x40085aa6: mp_execute_bytecode at vm.c:1042
0x401830db: fun_bc_call at objfun.c:267
0x4018b7c5: mp_call_function_n_kw at runtime.c:717
0x4018b7da: mp_call_function_0 at runtime.c:691
0x40197c1b: parse_compile_execute at pyexec.c:123
0x40197d5e: do_reader_stdin at pyexec.c:280
 (inlined by) pyexec_raw_repl at pyexec.c:513
0x4019f3f6: mp_task at main.c:157
```

kdschlosser, 2024-07-09:
OK i just pushed a commit that should fix it.

kdschlosser, 2024-07-09:
knowing how to decode the backtrace is helpful huh?

ikkesia, 2024-07-10:
Your latest commit fixes the problem.

Of course, knowing that it is possible to decode the backtrace is relevant. I learned that it can be done from your post not long ago.

I then explored in more detail [here](https://github.com/micropython/micropython/wiki/ESP32-debugging)

closed 2024-07-10 by ikkesia

kdschlosser, 2024-07-10:
ok cool so this is all set now? You display is working properly?

ikkesia, 2024-07-11:
I tested ESP32 with ili9341 and xpt2046 which share the same SPI bus. Everything works properly.
I still want to try sharing an sdcard  and I'm getting ready to do so.

The problem remains on the touch panel in case of rotation as I described #65

kdschlosser, 2024-07-11:
The rotation stuff I have to work through. I am going to order an SPI display that has an xpt2046 touch panel. I don't know why the rotation is getting all jacked up.

if you don't rotate the display does it render properly to the screen? If it doesn't then flip flow the width and the height and see if it displays properly.. if it does then do the rotation and see if everything works properly.

ikkesia, 2024-07-13:
I did a bit of investigating on the touch panel rotation issue. For now only with calibrated touch; I have already verified that without calibration data it does not work.

I have identified two causes :

- the position of the instruction that imposes the rotation

   `display.set_rotation(lv.DISPLAY_ROTATION._90)`
   
   must be run after instantiating the xpt2046 driver.
   Otherwise the _normalize() function in xpt2046 will produce incorrect values, based on **_orig_width** and **_orig_height** values, which are not the original ones, but those set by the display rotation instruction.

- LVGL already applies its own rotation calculation of the mapped coordinates provided by _calc_coords() in pointer_framework.py. To make all three possible rotations work properly I modified _calc_coords() as below
  ```
  def _calc_coords(self, x, y):
        if self.is_calibrated:
            cal = self._cal
            x = int(round(x * cal.alphaX + y * cal.betaX + cal.deltaX))
            y = int(round(x * cal.alphaY + y * cal.betaY + cal.deltaY))

        if (
            self._startup_rotation == lv.DISPLAY_ROTATION._90 or
            self._startup_rotation == lv.DISPLAY_ROTATION._270
        ):
            x = self._orig_width - x
            y = self._orig_height - y
        elif(self._startup_rotation == lv.DISPLAY_ROTATION._180):
            pass

        return x, y
```

kdschlosser, 2024-07-14:
You have to make sure that you are using the native orientation of the display and set the startup rotation of the touch driver to align with the native orientation of the display.

There is always some confusion as to how the rotation works. A display might be sold as 480 x 320 but at a rotation of 0 degrees if you set the width to 480 and the height to 320 what is seen on the display ends up being what looks like noise. If this happens then you need to set the width to 320 and the height to 480. do not change the rotation at all, leave it at zero. Next is to initialize the touch driver AFTER the `init` method is called for the display driver. Test the touch and see how it is aligned. Do not set any calibration at all. if the touch is not working properly then try setting the startup rotation for the touch driver to 90, 180 and 270. The touch might not be aligned which is fine, so long as the touch is registering in the same orientation as the display. It is best to use the slider widget to test the touch driver.

Once those things are done THEN you can set the rotation of the display and the touch should also rotate at the same time.

The xpt2046 driver is a real sore point for me because the display manufacturers don't set them up so it matches the display orientation. It can even be mirrored as if the touch was attached to the display flipped over either vertically or horizontally. This is incredibly infuriating because it makes it exceedingly difficult to make an easy API to adjust without having to add custom code for the user. It all depends on how the touch screen gets connected to the interface IC chip. It's like they don't know how to follow the instructions for the IC chip which tells them what needs to be connected and where. They know it will work connecting it any which way so instead of routing the tracers on the PCB so they can connect it properly they just connect to whichever pin in the IC they are able to so they don't have to change the design to get them to connect how they should be.

So the above code works for your use case but it will not work for someone else. This is the reason why I wanted the drivers written in Python code. You can subclass the driver and override the default functionality.

```python

import xpy2046
import lvgl as lv

class XPT2046(xpt2046.XPT2046):
    def _calc_coords(self, x, y):
        if self.is_calibrated:
            cal = self._cal
            x = int(round(x * cal.alphaX + y * cal.betaX + cal.deltaX))
            y = int(round(x * cal.alphaY + y * cal.betaY + cal.deltaY))

        if (
            self._startup_rotation == lv.DISPLAY_ROTATION._90 or
            self._startup_rotation == lv.DISPLAY_ROTATION._270
        ):
            x = self._orig_width - x
            y = self._orig_height - y
      
        return x, y

touch = XPT2046(touch_spi, ...)

```

so you don't need to alter any of the files in lvgl_micropython. you need to subclass and override to change the behavior. I wrote it this way for this specific reason. So if you decide to use a newer version of lvgl_micropython you do not have to go and alter any of the build files. it will work out of the box if you add the code above to your script.

I put a considerable amount of thought for these kinds of edge cases where a user might need to adjust how it works. It is the reason why there is a method for calculating the coordinates and I didn't add it to the code that interfaces with LVGL or the code that collects the coords from the IC. It's so you can easily override only the code that needs to be overridden.

It also allows you to define your own custom calibration routine by doing this as well. You have the ability to override every aspect of the calibration if you want.

ikkesia, 2024-07-14:
Thanks for the great suggestions!

In fact my display(native w249xh320)/touch panel kit has pixels(0.0) at the top-left and touch(0.0) at the bottom-left.

However, now I understand a little better how your drivers work and I can adapt them more  easily to my needs.

Below is the override solution that works for me 

```
    def _calc_coords(self, x, y):
        if self.is_calibrated:
            cal = self._cal
            x = int(round(x * cal.alphaX + y * cal.betaX + cal.deltaX))
            y = int(round(x * cal.alphaY + y * cal.betaY + cal.deltaY))
        else:
            y = self._orig_height - y
            
        if (
            self._startup_rotation == lv.DISPLAY_ROTATION._90 or
            self._startup_rotation == lv.DISPLAY_ROTATION._270
        ):
            x = self._orig_width - x
            y = self._orig_height - y
            
        return x, y
```