lvgl-micropython/lvgl_micropython #71: arduino giga - cant compile

0x6679, 2024-07-09:
you may say im noob (and you are right) but i just cant compile that firmware for the arduino giga board. could you help me? it would be a huge help if you could add a binary or just send it to me. thank you in advance! im really lost

kdschlosser, 2024-07-09:
MicroPython doesn't support that board to my knowledge.

0x6679, 2024-07-09:
it supports it. i have micropython on my board already. i wanna have lvgl too, but i cant compile the code 

ports/stm32 
make BOARD=ARDUINO_GIGA

kdschlosser, 2024-07-09:
OK so you have an STM32 and not an Arduino.

Unfortunatly there is not much support for the STM32 line of MCU's. I have not found anyonme that is willing to spend the time to help out with writing proper bus drivers. The only thing that works are SPI displays and that is only able to be done using a single frame buffer with no DMA memory used. It's not very good from the performance side of things.

0x6679, 2024-07-10:
i have an stm32 arduino (there are a lot) with H7 series processor (contains an M7 and an M4 core) i wanna run micropython on the m7 with lvgl. i have seen people on forums achieving that. could you compile that firmware for me? 
ports/stm32
make BOARD=ARDUINO_GIGA

it would mean a lot i i could import lvgl :D

0x6679, 2024-07-10:
whole lot of people on the internet got it working and we are in the lvgl_micropython repo. and in the ports i can see a folder called stm32 and inside that folder there is a boards folder where i can see the arduino giga board. i would like to compile that. thats the reason im trying to reach out for anybody who wrote that port. maybe they can compile it so i can have lvgl on my giga board with micropython

0x6679, 2024-07-10:
lol im in the wrong repo XDDDDDDD

closed 2024-07-10 by kdschlosser