lvgl-micropython/lvgl_micropython #74: Error calling lv.oj() - reboot esp32s3

marilantagliari, 2024-07-11:
Compiled in MacOS IDF_ESP:

`MicroPython a61c446-dirty on 2024-07-09; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

import lvgl as lv
lv.init()
s = lv.obj()

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : a61c446-dirty on 2024-07-09
IDF version : 11eaf41b
Machine : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

Core 1 register dump:
PC : 0x42030659 PS : 0x00060830 A0 : 0x8202dcdb A1 : 0x3fcbd550
A2 : 0x00000000 A3 : 0x00000000 A4 : 0x00000000 A5 : 0x3fcbd84c
A6 : 0x00000001 A7 : 0x3fcbd550 A8 : 0x00000000 A9 : 0x3fcbd530
A10 : 0x3c254550 A11 : 0x00000038 A12 : 0x00000000 A13 : 0x3fcbd84c
A14 : 0x00000001 A15 : 0x3fcbd530 SAR : 0x00000002 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000036 LBEG : 0x400570e8 LEND : 0x400570f3 LCOUNT : 0x00000000

Backtrace: 0x42030656:0x3fcbd550 0x4202dcd8:0x3fcbd590 0x4202c8bf:0x3fcbd5d0 0x4201482e:0x3fcbd610 0x42006105:0x3fcbd650 0x420be92d:0x3fcbd690 0x420065a1:0x3fcbd6b0 0x42014876:0x3fcbd700 0x420bd471:0x3fcbd730 0x420be92d:0x3fcbd750 0x420be9f5:0x3fcbd770 0x40379aa2:0x3fcbd790 0x420b61bb:0x3fcbd830 0x420be92d:0x3fcbd880 0x420be942:0x3fcbd8a0 0x420cad93:0x3fcbd8c0 0x420caed6:0x3fcbd950 0x420d256e:0x3fcbd9a0

ELF file SHA256: 117aa061b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382ce6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
MicroPython a61c446-dirty on 2024-07-09; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

`
This is the backtrace:

MacBook-Air-de-Marilan:lvgl_micropython marilantagliari$ xtensa-esp32s3-elf-addr2line -pfiaC -e lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/micropython.elf 0x42030656:0x3fcbd550 0x4202dcd8:0x3fcbd590 0x4202c8bf:0x3fcbd5d0 0x4201482e:0x3fcbd610 0x42006105:0x3fcbd650 0x420be92d:0x3fcbd690 0x420065a1:0x3fcbd6b0 0x42014876:0x3fcbd700 0x420bd471:0x3fcbd730 0x420be92d:0x3fcbd750 0x420be9f5:0x3fcbd770 0x40379aa2:0x3fcbd790 0x420b61bb:0x3fcbd830 0x420be92d:0x3fcbd880 0x420be942:0x3fcbd8a0 0x420cad93:0x3fcbd8c0 0x420caed6:0x3fcbd950 0x420d256e:0x3fcbd9a0 0x42030656: lv_obj_mark_layout_as_dirty at ??:? 0x4202dcd8: lv_obj_class_init_obj at ??:? 0x4202c8bf: lv_obj_create at ??:? 0x4201482e: mp_lv_obj_create at lv_mp.c:? 0x42006105: lv_fun_builtin_var_call at lv_mp.c:? 0x420be92d: mp_call_function_n_kw at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/runtime.c:717 0x420065a1: make_new at lv_mp.c:? 0x42014876: obj_make_new at lv_mp.c:? 0x420bd471: type_call at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/objtype.c:1003 0x420be92d: mp_call_function_n_kw at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/runtime.c:717 0x420be9f5: mp_call_method_n_kw at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/runtime.c:733 0x40379aa2: mp_execute_bytecode at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/vm.c:1042 0x420b61bb: fun_bc_call at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/objfun.c:267 0x420be92d: mp_call_function_n_kw at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/runtime.c:717 0x420be942: mp_call_function_0 at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/py/runtime.c:691 0x420cad93: parse_compile_execute at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:123 0x420caed6: do_reader_stdin at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:280 (inlined by) pyexec_raw_repl at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/shared/runtime/pyexec.c:513 0x420d256e: mp_task at /Users/marilantagliari/dev_app/LCD_Tracker/lvgl_micropython/lib/micropython/ports/esp32/main.c:157

any suggestion?
Did you thing this is an error related with display driver or an issue with compile environment ?

kdschlosser, 2024-07-11:
I am not even sure how you managed to get it to compile. The macOS build CI is currently failing. I believe I just fixed it. The CI is running now so I will know in about 15 minutes.

I will compile a binary for you in a bit. you can flash that binary and see if it works. If it does then we know it's the environment that is causing a hiccup.

kdschlosser, 2024-07-11:
what build command are you using?

How much flash does your board have? How much ram does it have?

marilantagliari, 2024-07-12:
> what build command are you using?
> 
> How much flash does your board have? How much ram does it have?

Command: make -C ports/esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=gt911

LCD module uses ESP32-S3-WROOM-1, 512KB SRAM, 384KB ROM, 8M PSRAM,Flash size is 16MB

kdschlosser, 2024-07-12:
AHhhh and there is the issue. your entire build command is wrong...

it should be...

```
python3 make.py esp32 mpy_cross submodules BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=gt911 --flash-size=16 --oct-flash
```

You may or may not need the `--oct-flash`. you would have to look of the exact model number of the ESP32 you are using, specifically the numbers/letters after the `-`. I believe there are 2 versions of the 16mb of flash, one that is quad and one that is octal. I have not seen the quad variant at all so chances are you have an octal. The ESP32 is able to auto detect the ram size but not the flash size so this needs to be set at build time.

we can also give your setup a significant speed boost as well. We can chat about that later once we get things going for ya.

marilantagliari, 2024-07-13:
> AHhhh and there is the issue. your entire build command is wrong...
> 
> it should be...
> 
> ```
> python3 make.py esp32 clean mpy_cross submodules BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT DISPLAY=st7796 INDEV=gt911 --flash-size=16 --oct-flash
> ```
> 
> You may or may not need the `--oct-flash`. you would have to look of the exact model number of the ESP32 you are using, specifically the numbers/letters after the `-`. I believe there are 2 versions of the 16mb of flash, one that is quad and one that is octal. I have not seen the quad variant at all so chances are you have an octal. The ESP32 is able to auto detect the ram size but not the flash size so this needs to be set at build time.
> 
> we can also give your setup a significant speed boost as well. We can chat about that later once we get things going for ya.

Thanks a lot Kevin, but still getting error.. sorry for the wrong command, but your command has a param --octal-flash, in my case is quad, I flashed with octal param and got error rebooting the esp32.. 

In my case the LCD is a RGB 3wires (ST7701), and I've been trying to use your drive RGB_DISPLAY.py, maybe this error is causing by wrong driver?

the error got in esp32:

MicroPython a61c446-dirty on 2024-07-13; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import lvgl as lv
>>> lv.init()
>>> s = lv.obj()

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : a61c446-dirty on 2024-07-13
IDF version : 11eaf41b
Machine     : Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42030659  PS      : 0x00060830  A0      : 0x8202dcdb  A1      : 0x3fcbd550  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3fcbd84c  
A6      : 0x00000001  A7      : 0x3fcbd550  A8      : 0x00000000  A9      : 0x3fcbd530  
A10     : 0x3c254550  A11     : 0x00000038  A12     : 0x00000000  A13     : 0x3fcbd84c  
A14     : 0x00000001  A15     : 0x3fcbd530  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000036  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42030656:0x3fcbd550 0x4202dcd8:0x3fcbd590 0x4202c8bf:0x3fcbd5d0 0x4201482e:0x3fcbd610 0x42006105:0x3fcbd650 0x420be92d:0x3fcbd690 0x420065a1:0x3fcbd6b0 0x42014876:0x3fcbd700 0x420bd471:0x3fcbd730 0x420be92d:0x3fcbd750 0x420be9f5:0x3fcbd770 0x40379aa2:0x3fcbd790 0x420b61bb:0x3fcbd830 0x420be92d:0x3fcbd880 0x420be942:0x3fcbd8a0 0x420cad93:0x3fcbd8c0 0x420caed6:0x3fcbd950 0x420d256e:0x3fcbd9a0

ELF file SHA256: 39bb62f6c

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382ce6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e30
entry 0x403c989c
MicroPython a61c446-dirty on 2024-07-13; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.

kdschlosser, 2024-07-13:
at least we know you are not using the correct flash size. as I said you may or may not need the --oct-flash in the build command. Your MCU is the first I have seen that is not octal.

what code are you running when the ESP32 boots?

Carglglz, 2024-07-13:
@marilantagliari 
You need to setup the display with `lv.display_create` + flush callback + buffers before creating any lvgl object. 

e.g. this happens in unix port too:
``` micropython
MicroPython v1.24.0-preview.111.g2046793aca.dirty on 2024-07-11; darwin [GCC 4.2.1] version
Type "help()" for more information.
>>> import lvgl as lv
>>>
>>> lv.init()
>>> obj = lv.obj()
Segmentation fault: 11
```

marilantagliari, 2024-07-13:
> at least we know you are not using the correct flash size. as I said you may or may not need the --oct-flash in the build command. Your MCU is the first I have seen that is not octal.
> 
> what code are you running when the ESP32 boots?

the compilation and flash is working fine... thanks Kevin.
My problem now is the drive for my st7701, I'm not able to write this driver but I've been learning about this.

marilantagliari, 2024-07-13:
> @marilantagliari You need to setup the display with `lv.display_create` + flush callback + buffers before creating any lvgl object.
> 
> e.g. this happens in unix port too:
> 
> ```
> MicroPython v1.24.0-preview.111.g2046793aca.dirty on 2024-07-11; darwin [GCC 4.2.1] version
> Type "help()" for more information.
> >>> import lvgl as lv
> >>>
> >>> lv.init()
> >>> obj = lv.obj()
> Segmentation fault: 11
> ```

Thanks Carglglz, is true... after lv.display_create works fine... but I'm trying the example written by Kevin and got error in allocate memory framebuffer.... but I still need a right driver for LCD ST7701 spi rgb.. :-(

_BUFFER_SIZE = const(25600)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

Traceback (most recent call last):
  File "<stdin>", line 61, in <module>
  File "display_driver_framework.py", line 169, in __init__
MemoryError: memory allocation failed, allocating 2971407374 bytes

kdschlosser, 2024-07-13:
The ST7701 display is an RGB Bus display and as such you need to allocate 2 frame buffers to SPIRAM in order for it to work. Unless the display you are using is incredibly small.

kdschlosser, 2024-07-13:
this is also conflicting..,..

```
_BUFFER_SIZE = const(25600)
fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

Traceback (most recent call last):
File "", line 61, in
File "display_driver_framework.py", line 169, in init
MemoryError: memory allocation failed, allocating 2971407374 bytes

```

The error is saying it is happening in the display driver but the code says that you are doing it yourself and the display driver is not handling the buffer creation.

closed 2024-11-05 by kdschlosser