lvgl-micropython/lvgl_micropython #78: Staying in bootloader

dcmcshan, 2024-07-25:
Fresh build, following directions.

Any idea why my ESP32-S3 stays in DFU?  I cannot seem to get it to actually boot this image - it boots the stock micropython image well enough.  Chip I have is ESP32-S3-WROOM-1-N16R8

```
% /Users/danielmcshan/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool --chip esp32s3 -p /dev/tty.usbmodem101 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 8MB --flash_freq 80m --erase-all 0x0 /Users/danielmcshan/GitHub/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-8.bin 
esptool.py v4.7.0
Serial port /dev/tty.usbmodem101
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: dc:da:0c:17:d6:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.9s
Compressed 2523536 bytes to 1463417...
Wrote 2523536 bytes (1463417 compressed) at 0x00000000 in 15.1 seconds (effective 1333.1 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

dcmcshan, 2024-07-25:
This is what bootloader says:
`rst:0x3 (RTC_SW_SYS_RST),boot:0x0 (DOWNLOAD(USB/UART0))
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40041a76
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3810,len:0xf3c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xaac
load:0x403cc700,len:0x2ba4
entry 0x403c9880`

dcmcshan, 2024-07-25:
I thought maybe the no_reset was perhaps the issue, but it's not. 

`% python -m esptool --chip esp32s3 -b 921600 write_flash --flash_mode dout --flash_size 16MB --flash_freq 80m 0x0 build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-16_OCTFLASH.bin
esptool.py v4.7.0
Found 2 serial ports
Serial port /dev/cu.usbmodem101
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: dc:da:0c:17:d6:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00268fff...
Compressed 2524224 bytes to 1463207...
Wrote 2524224 bytes (1463207 compressed) at 0x00000000 in 18.5 seconds (effective 1090.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

kdschlosser, 2024-07-25:
ESP32-S3-WROOM-1-N16R8

You have a board that has 16mb of flash and 8mb or ram... The image you made if doe a board that has 8mb of flash and 8mb of ram

add the option `--flash-size=16` when you compile to fix the issue. You also want to check and see if you have octal SPIRAM and octal flash.

dcmcshan, 2024-07-25:
python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --octal-flash --flash-size 16

python -m esptool --chip esp32s3 -p /dev/tty.usbmodem101 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dout --flash_size 16MB --flash_freq 80m --erase-all 0x0 /Users/danielmcshan/GitHub/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-16_OCTFLASH.bin

Was my last (still unsuccessful) attempt.

dcmcshan, 2024-07-25:
Yes, N16R8 has octal flash/spiram.

python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --octal-flash --flash-size=16

Does this look correct?

kdschlosser, 2024-07-25:
There are 2 versions of the 16/8.. one has quad flash and octal spiram and the other has octal flash and octal spiram. 

ESP32-S3-WROOM-2-N16R8V: 16mb octal flash and 8mb octal spiram
ESP32-S3-WROOM-1-N16R8: 16mb quad flash and 8mb octal spiram

so you might not have octal flash. which is what I am thinking is the case I would remove the `--octal-flash` command line switch.

kdschlosser, 2024-07-25:
The build command should also read

```
python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16
```

kdschlosser, 2024-07-25:
what is the display IC you are using and what is the touch IC you are using?

dcmcshan, 2024-07-25:
Subtle.  

I am no V.  ESP32-S3-WROOM-1-N16R8

No difference, though.  Please check my commands:

`% python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16`

% /Users/danielmcshan/.espressif/python_env/idf5.2_py3.12_env/bin/python -m esptool --chip esp32s3 -p /dev/tty.usbmodem1101 -b 460800 --before default_reset --after no_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m --erase-all 0x0 /Users/danielmcshan/GitHub/lvgl_micropython/build/lvgl_micropy_ESP32_GENERIC_S3-SPIRAM_OCT-16.bin
esptool.py v4.7.0
Serial port /dev/tty.usbmodem1101
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: dc:da:0c:17:d6:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.8s
Compressed 2522112 bytes to 1462642...
Wrote 2522112 bytes (1462642 compressed) at 0x00000000 in 15.1 seconds (effective 1339.0 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

It is a custom board, and it has boot/reset buttons.  With the standard version, I reset and it boots.

I am using st7785, so I'm guessing ili9XXX of some fashion.

kdschlosser, 2024-07-25:
since your MCU is staying in bootloader mode you need to press the reset button after you flash the firmware to it and then open up a serial connection to the board to see if you get the REPL.

what are you using for a touch IC?

your build command you want as follows.

```
python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DSIPLAY=st7785 INDEV={indev_ic}
```

replacing `{indev_ic}` with the model number of the touch IC you are using. If ypou are not using a touchscreen then you can remove that config option.

I am guessing that the ESP32 you are using is a devkit of some kind. could you please provide me a link to where you got it from. I need to look up a spec on it to see how they have the uart connected.

kdschlosser, 2024-07-25:
are you using the USB-OTG or do you have a uart connected to the chip? If you are using I think GPIO's 18 and 19 (maybe 19 and 20 can't remember off the top of my head) for connecting to a PC then you need to add `--usb-otg` to the build command.

dcmcshan, 2024-07-25:
Ah, that sounds promising... building now.  It's a custom board using the USB port.  Serial port is not connected (well, it's connected to test points).   Inputs are 4 microswitch buttons.  Nothing cool like touch...

dcmcshan, 2024-07-25:
> I am guessing that the ESP32 you are using is a devkit of some kind. could you please provide me a link to where you got it from. I need to look up a spec on it to see how they have the uart connected.

I could bring you one ;)

dcmcshan, 2024-07-25:
What do you think about CONFIG_ESP_CONSOLE_USB_CDC?  
What is the best way to set CONFIG_ESP_CONSOLE*  in your scheme?

kdschlosser, 2024-07-25:
If you have one of the board you are willing to get to me so I can test it to see what is going on we can do that. It sure would make it a whole lot easier doing that...

dcmcshan, 2024-07-26:
19 and 20.   Good call!

￼

dcmcshan, 2024-07-26:
For reference, head of micropython builds and works as expected.

What version micropython are you baselining?

```
MicroPython v1.24.0-preview.148.g4a134d212 on 2024-07-25; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> 
```

dcmcshan, 2024-07-26:
As an aside, --after hard_reset will put the ESP32-S3 into run mode without having to press the reset button.  I have confirmed on the ESP32_GENERIC_S3-SPIRAM_OCT-20240602-v1.23.0.bin build.

kdschlosser, 2024-07-26:
I changed the `no_reset` to `hard_reset` as per your suggestion..

https://github.com/kdschlosser/lvgl_micropython/commit/550b78a22ee2767d51b71ef1a339dc639fa7d99a

kdschlosser, 2024-07-26:
See if using the release version of MicroPython 1.23.0 works. If it does then I am going to have to dive into what I am changing that is breaking it.

The reason why it is disabled is because GPIO 19 and 20 were getting locked and couldn't be used for anything. There are more boards that do not use those pins for USB than there are boards that do. The pins shouldn't be locked to being USB only as a default. I have to check to see if I have inadvertently turned something else off. People have reported that using the --usb-otg parameter does correct the output. I am not sure why it's not in your case.

kdschlosser, 2024-07-26:
I also did a rework on both the usb otg and jtag handling.

By default if the MCU supported jtag it would use that in combination with a hardware uart for the repl. If the MCU didn't support jtag then it would use otg if the MCU supports it. once again in combination with the uart repl.

I am going to add in code to disable the hardware uart repl if the user wants that done. That will free up those 2 pins.

I have not cared for the inability to turn on and off the hardware uart repl at runtime. It is only able to be done at compile time. There are benefits to being able to do it at runtime. It's the same thing for the usb-otg or jtag as well.

dcmcshan, 2024-07-26:
[image]
I pulled HEAD from Github.  And I think it worked!  Will have some breakfast, then redo everything from scratch to make sure it was not something I was fiddling with... but looks good, whatever it was.

kdschlosser, 2024-07-26:
hopefully it will work out for ya..

closed 2024-07-26 by dcmcshan