lvgl-micropython/lvgl_micropython #79: AttributeError: type object 'SPI' has no attribute 'MSB'

dcmcshan, 2024-07-26:
Trying out your new SPI architecture.  Am really looking forward to this, and hope that it will make reinitializing display after soft reset actually work.  

Here is what I have so far:

        self.spi_bus = SPI.Bus( host=1, mosi=self.PIN_MOSI, miso=self.PIN_MISO, sck=self.PIN_SCK )

        self.spi_device = SPI.Device( spi_bus=self.spi_bus, freq=20000000, cs=3, polarity=0, phase=0, bits=8, first_bit=SPI.MSB )

        self.display = ILI9341(self.spi_device)

But I get this error:

AttributeError: type object 'SPI' has no attribute 'MSB'

Please advise!
Thank you.

dcmcshan, 2024-07-26:
I found your use_example...

        self.spi_bus = SPI.Bus( host=1, mosi=self.PIN_MOSI, miso=self.PIN_MISO, sck=self.PIN_SCK )

        print ("Setup display")
        self.display = ILI9341.ILI9341(spi_bus=self.spi_bus,
                                        display_width= 80,
                                        display_height= 160,
                                        # we are going to let the driver handle the allocation of the frame buffers
                                        frame_buffer1=None,
                                        frame_buffer2=None,
                                        reset_pin=self.PIN_LCD_RESET,
                                        reset_state=ILI9341.STATE_HIGH,
                                        power_on_state=ILI9341.STATE_HIGH,
                                        backlight_pin= self.PIN_LCD_BACKLIGHT,
                                        backlight_on_state=ILI9341.STATE_HIGH,
                                        offset_x=_OFFSET_X,
                                        offset_y=_OFFSET_Y,
                                        color_space=lv.COLOR_FORMAT.RGB565,
                                        rgb565_byte_swap=True
                                       )

This isn't working yet (is resetting...) but I'm just letting your know what path I'm on

dcmcshan, 2024-07-26:
self.spi_bus = SPI.Bus( host=1,
                                mosi=self.PIN_MOSI,
                                miso=self.PIN_MISO,
                                sck=self.PIN_SCK )

        print ("Setup display")
        time.sleep(1)
        self.display = ILI9341.ILI9341(data_bus=self.spi_bus,
                                        display_width= 80,
                                        display_height= 160,
                                        reset_pin=self.PIN_LCD_RESET,
                                        reset_state=ILI9341.STATE_HIGH,
                                        power_on_state=ILI9341.STATE_HIGH,
                                        backlight_pin= self.PIN_LCD_BACKLIGHT,
                                        offset_x=0,
                                        offset_y=0,
                                        color_space=lv.COLOR_FORMAT.RGB565,
                                        rgb565_byte_swap=True
                                       )

Setup display
Traceback (most recent call last):
  File "<stdin>", line 239, in <module>
  File "<stdin>", line 75, in __init__
  File "display_driver_framework.py", line 191, in __init__
AttributeError: 'Bus' object has no attribute 'allocate_framebuffer'

Ideas?

kdschlosser, 2024-07-26:
OK this is what you need to do...

```python
import lcd_bus

self.spi_bus = SPI.Bus(
    host=1,
    mosi=self.PIN_MOSI,
    miso=self.PIN_MISO,
    sck=self.PIN_SCK
)

self.display_bus = lcd_bus.SPIBus(
    spi_bus=self.spi_bus
)

self.display = ILI9341.ILI9341(
    data_bus=self.display_bus,
    display_width=80,
    display_height=160,
    reset_pin=self.PIN_LCD_RESET,
    reset_state=ILI9341.STATE_HIGH,
    power_on_state=ILI9341.STATE_HIGH,
    backlight_pin=self.PIN_LCD_BACKLIGHT,
    offset_x=0,
    offset_y=0,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
    )
```

dcmcshan, 2024-07-26:
This compiles, but unfortunately, I need to use Pin 36 to enable it, and that is now invalid (see #80)

closed 2024-07-26 by dcmcshan