lvgl-micropython/lvgl_micropython #80: ValueError: invalid pin

dcmcshan, 2024-07-26:
MicroPython a61c446-dirty on 2024-07-26; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import machine
>>> machine.PWM(machine.Pin(37))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid pin

Is something using pin 37?

dcmcshan, 2024-07-26:
>>> machine.PWM(machine.Pin(36))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid pin

dcmcshan, 2024-07-26:
>>> machine.PWM(machine.Pin(35))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid pin

kdschlosser, 2024-07-26:
Your comparing it to a version of micropython that is not yet released. One of the things you have to remember is that your board doesn't have SPIRAM and those pins are usually used for the SPIRAM. 

I am going to have to make some decent sized modifications to the MicroPython ESP32 port to free up those pins.

GPIO pins  35, 36, and 37 are used for the octal SPIRAM. those pins should only end up getting reserved if you set your board variant to `SPIRAM_OCT`.. I am going to double check the code and see if that is what is actually being done...

kdschlosser, 2024-07-26:
add `CONFIG_SPIRAM_MODE_OCT=n` to your build command

kdschlosser, 2024-07-26:
I am not sure why you are compiling the board with `BOARD_VARIANT=SPIRAM_OCT`..

from your first post, this is what the firmware is reporting.....
```
MicroPython a61c446-dirty on 2024-07-26; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
```

dcmcshan, 2024-07-26:
Good catch.  That's another beer I owe you... :)

closed 2024-07-26 by dcmcshan

kdschlosser, 2024-07-27:
don't drink. Haven't had any alcohol in over 10 years.