lvgl-micropython/lvgl_micropython #81: fsdriver memory allocation failed

dcmcshan, 2024-07-26:
Moving right along...

This worked under lv_micropython and does not under lvgl_micropython.

I know that fond loading was under some development in lv_micropython.  Do you have an example of how you think fonts should be loaded and used?  

File "fs_driver.py", line 91, in fs_register
MemoryError: memory allocation failed, allocating 1297567784 bytes

`
import lvgl as lv
import sys
import fs_driver
import uos

lv.init()

fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')

fs_drv = lv.fs_drv_t()
fs_driver.fs_register(fs_drv, 'S')

FONTS_LVGL_PATH="/fonts"

def load(name, lvgl_path: str = FONTS_LVGL_PATH):
    font_path = f"S:{lvgl_path}/{name}"
    
    # Check if the file exists before creating the binfont
    if name in uos.listdir(lvgl_path):
        #print(f"Font file '{font_path}' exists.")
        try:
            font = lv.binfont_create(font_path)
        except Exception as e:
            print(e)
            return None
        return font
    else:
        print(f"Font file '{font_path}' does not exist.")
        return None  # Return None to indicate that the font could not be loaded

conseil10 = load("conseil10.font")
conseil20 = load("conseil20.font")
conseil30 = load("conseil30.font")
conseil40 = load("conseil40.font")
montserrat10 = load("montserrat10.font")
`

dcmcshan, 2024-07-26:
I think this is not actually a problem with this.

closed 2024-07-26 by dcmcshan

kdschlosser, 2024-07-27:
I have not tested the file system driver at all. it should be tested at some point. If there is an error with it then there is an error. But the error you are having might be because you don't have enough memory? The allocation amount seems like it is too much.... I also want to mess with the file system driver and how the memory is being allocated. I want to be able to provide the user with a choice of where they want to allocate the memory from, internal ram or psram. This would be done in C code to provide that functionality so the overhead would be minimal in terms of how the data is packaged. It would be a memoryview object that points to the location of the C array, It's more memory efficient this way...