lvgl-micropython/lvgl_micropython #82: Simple LVGL Test

dcmcshan, 2024-07-26:
Here is a simple test for LVGL.  I am running ST7785 through this interface.

[image]

It's wired like this: 
[image]

This code runs, but I see nothing on display.  Wondering if you can take a quick look to see if anything looks obviously wrong.

`import lvgl as lv
import ILI9341
import lcd_bus
from machine import SPI, Pin

# Initialize LVGL
lv.init()

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=11,
    miso=13,
    sck=12
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=14,
    freq=1000000
)

# Initialize the ILI9341 display driver
display = ILI9341.ILI9341(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=21,
    reset_state=ILI9341.STATE_HIGH,
    power_on_state=ILI9341.STATE_HIGH,
    backlight_pin=40,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

# Initialize display
display.init()
display.set_backlight(100)

Pin(36).on()

# Create a screen and load it
scr = lv.obj()
lv.screen_load(scr)

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)
`

dcmcshan, 2024-07-26:
If it's helpful, this works under baseline micropython: 
```

import time
from machine import Pin, SPI

class ST7735:
    def __init__(self, spi, cs, dc, rst):
        self.spi = spi
        self.cs = Pin(cs, Pin.OUT)
        self.dc = Pin(dc, Pin.OUT)
        self.rst = Pin(rst, Pin.OUT)
        self.width = 128
        self.height = 160
        self.init_display()

    def init_display(self):
        self.reset()
        
        self.write_cmd(0x01)  # Software reset
        time.sleep_ms(150)
        
        self.write_cmd(0x11)  # Sleep out
        time.sleep_ms(500)
        
        self.write_cmd(0x3A, b'\x05')  # Set color mode to 16-bit
        self.write_cmd(0x36, b'\xC0')  # Set MADCTL: row/col address, bottom-top refresh
        self.write_cmd(0x21)  # Invert display

        self.write_cmd(0x29)  # Display on

    def reset(self):
        self.rst.value(0)
        time.sleep_ms(50)
        self.rst.value(1)
        time.sleep_ms(50)

    def write_cmd(self, cmd, data=None):
        self.dc.value(0)
        self.cs.value(0)
        self.spi.write(bytes([cmd]))
        self.cs.value(1)
        if data:
            self.write_data(data)

    def write_data(self, data):
        self.dc.value(1)
        self.cs.value(0)
        self.spi.write(data)
        self.cs.value(1)

    def set_window(self, x0, y0, x1, y1):
        self.write_cmd(0x2A, bytes([0, x0, 0, x1]))
        self.write_cmd(0x2B, bytes([0, y0, 0, y1]))
        self.write_cmd(0x2C)

    def fill_color(self, color):
        self.set_window(0, 0, self.width - 1, self.height - 1)
        color_bytes = color.to_bytes(2, 'big')
        chunk = bytearray()
        for _ in range(32):
            chunk.extend(color_bytes)
        for _ in range((self.width * self.height) // 32):
            self.write_data(chunk)

    def test(self):
        colors = [
            (self.rgb565(255, 0, 0), "red"),
            (self.rgb565(0, 255, 0), "green"),
            (self.rgb565(0, 0, 255), "blue")
        ]
        
        print("Starting display test...")
        for color, name in colors:
            self.fill_color(color)
            time.sleep(1)
            response = input(f"Is the background {name}? (y/n): ").lower()
            if response != 'y':
                print(f"Test failed at {name} color.")
                return False
        
        print("All colors displayed correctly!")
        return True

    def rgb565(self, r, g, b):
        """Convert RGB888 to BGR565 by swapping red and blue components."""
        return ((b & 0xf8) << 8) | ((g & 0xfc) << 3) | (r >> 3)

    def hex_to_rgb(self, hex_color):
        """Convert a hex color to an RGB tuple."""
        r = (hex_color >> 16) & 0xFF
        g = (hex_color >> 8) & 0xFF
        b = hex_color & 0xFF
        return r, g, b

    @property
    def fill(self):
        raise NotImplementedError("Getter for 'fill' is not implemented.")

    @fill.setter
    def fill(self, hex_color):
        r, g, b = self.hex_to_rgb(hex_color)
        self.fill_color(self.rgb565(r, g, b))
        
    def exists(self):
        return 0 # I don't think we can actually know that it exists

def main():
    from MCT import MCT
    mct = MCT()
    mct.display = ST7735(mct.spi,
                         dc=Pin(mct.PIN_LCD_RS),
                         cs=Pin(mct.PIN_LCD_CS),
                         rst=Pin(mct.PIN_LCD_RESET))

    if mct.display.test():
        print("Display test passed successfully!")
    else:
        print("Display test failed. Please check your connections and try again.")

if __name__ == "__main__":
    main()

```

dcmcshan, 2024-07-26:
Looking into your world a bit more, I see that you have a separate driver for st7735.  lv_micropython lived in ILI9XXX.  I will try that.

dcmcshan, 2024-07-26:
I have changed to ST7735, but still no display.

```
import lvgl as lv
import ST7735
import lcd_bus
from machine import SPI, Pin

# Initialize LVGL
lv.init()

Pin(36).on() # Enable LED

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=11,
    miso=13,
    sck=12
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=14,
    freq=1000000
)

# Initialize the st7735 display driver
display = ST7735.ST7735B(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=21,
    reset_state=ST7735.STATE_HIGH,
    power_on_state=ST7735.STATE_HIGH,
    backlight_pin=40,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

# Initialize display
display.init()
display.set_backlight(100)

# Create a screen and load it
scr = lv.obj()
lv.screen_load(scr)

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)

```

dcmcshan, 2024-07-26:
Have added the enable as "power_in", but I still have no display. 

```
import lvgl as lv
import ST7735
import lcd_bus
from machine import SPI, Pin

# Initialize LVGL
lv.init()

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=11,
    miso=13,
    sck=12
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=14,
    freq=1000000
)

# Initialize the st7735 display driver
display = ST7735.ST7735B(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=21,
    reset_state=ST7735.STATE_HIGH,
    power_pin=36,
    power_on_state=ST7735.STATE_HIGH,
    backlight_pin=40,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

display.set_power(True)

# Initialize display
display.init()
display.set_backlight(100)

# Create a screen and load it
scr = lv.obj()
lv.screen_load(scr)

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)

```

dcmcshan, 2024-07-27:
One things I notice is there is no CS in my code, so I try this:

```

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=20000000
)
```

but to no more avail

dcmcshan, 2024-07-27:
For reference, and to make the comparison between what works and doesn't, I have used the same defines for pins.  I think everything is setup the same way, so I suspect the different must be in the driver logic, which I'll look into next.

```
import lvgl as lv
import ST7735
import lcd_bus
from machine import SPI, Pin

PIN_LCD_CS = const(10)
PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_RESET = const(21)
PIN_EN_LED = const(36)
PIN_LCD_BACKLIGHT = const(40)
    
# Initialize LVGL
lv.init()

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=PIN_MOSI,
    miso=PIN_MISO,
    sck=PIN_SCK
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=20000000
)

# Initialize the st7735 display driver
display = ST7735.ST7735B(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=PIN_LCD_RESET,
    reset_state=ST7735.STATE_LOW,
    power_pin=PIN_EN_LED,
    power_on_state=ST7735.STATE_HIGH,
    backlight_pin=PIN_LCD_BACKLIGHT,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

display.set_power(True)

# Initialize display
display.init()
display.set_backlight(100)

# Create a screen and load it
scr = lv.obj()
lv.screen_load(scr)

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import utime as time
time_passed = 1000

while True:
    start_time = time.ticks_ms()
    time.sleep_ms(1)  # sleep for 1 ms
    lv.tick_inc(time_passed)
    lv.task_handler()
    end_time = time.ticks_ms()
    time_passed = time.ticks_diff(end_time, start_time)
```

kdschlosser, 2024-07-27:
import lvgl as lv
import st7735
import lcd_bus
from machine import SPI, Pin

# Initialize LVGL
lv.init()

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host=1,
    mosi=11,
   #  according to the diagrams there is no miso pin connected. 
    miso=-1,
    sck=12
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=14,
    cs=-1, # just for sake of making sure....
    # If you arte using hardware SPI pins that align with the 
    # host that you are using you should be able to push this 
    # up to 80mhz. 
    # Otherwise 40mhz should work. 
    freq=40000000 
)

# Initialize the ST7735 display driver
display = st7735.ST7735(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=21,
    # You can try changing this to STATE_LOW. 
    # Several people have had to set it to LOW
    reset_state=ILI9341.STATE_HIGH,  
    power_on_state=ILI9341.STATE_HIGH,
    backlight_pin=40,
    offset_x=0,
    offset_y=0,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
)

# You have supplied a power pin so you have to turn the display on
display.set_power(True)

# Initialize display
display.init()
display.set_backlight(100)

Pin(36).on()  #   <==== What is this for?

# Create a screen and load it

scr = lv.screen_active()

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import task_handler

th = task_handler.TaskHandler()

kdschlosser, 2024-07-27:
Also, You need to stop using lv_micropython as a reference. This repo and the "official" binding may as well be on the other side of the planet from each other. They work completely different. The official binding only has support for SPI displays where as this repo has support for SPI, RGB, I8080, I2C and very soon LED Matrix. The display drivers are unaware of what the data connection is to the display. It is done this way because a lot of driver IC's support multiple ways of connecting them and I didn't want to have to duplicate code by having to write a driver for SPI and then another one for I8080. The way it is now the display driver code is able to be used with any data connection type.

kdschlosser, 2024-07-27:
The other HUGE thing is the ability to hot plug displays. Without ever powering off the ESP or reset it you can swap the display to something else and it can be a completely different bus on different pins and be a different driver IC. 

Another HUGE thing is rapid prototyping. You are able to completely design your GUI with running and testing it and you never have to flash a firmware or upload any script to the ESP. Not until you are finished developing. You can run Micropython on Unix and the driver API for creating an application window on a PC is the same as the driver API that is used for an actual display. 

The code below you can run on an ESP32 or it can be run on Unix/macOS. No need to change it at all. it will run properly on both.
```python
import sys
import lvgl as lv
import lcd_bus

if sys.platform in ('darwin', 'unix'):
    import sdl_display

    display_bus = lcd_bus.SDLBus(flags=0)
    display_driver = sdl_display.SDLDisplay
    rgb565_byte_swap = False
 
else:
    import st7735
    from machine import SPI, Pin

    # Initialize the SPI bus
    spi_bus = SPI.Bus(
        host=1,
        mosi=11,
        # according to the diagrams there is no miso pin connected. 
        miso=-1,
        sck=12
    )

    # Initialize the display bus
    display_bus = lcd_bus.SPIBus(
        spi_bus=spi_bus,
        dc=14,
        cs=-1, # just for sake of making sure....
        # If you arte using hardware SPI pins that align with the 
        # host that you are using you should be able to push this 
        # up to 80mhz. 
        # Otherwise 40mhz should work. 
        freq=40000000 
    )
    
    display_driver = st7735.ST7735
    rgb565_byte_swap = True
    Pin(36).on()  #   <==== What is this for?

display = display_driver(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    reset_pin=21,
    # You can try changing this to STATE_LOW. 
    # Several people have had to set it to LOW
    reset_state=ILI9341.STATE_HIGH,  
    power_on_state=ILI9341.STATE_HIGH,
    backlight_pin=40,
    offset_x=0,
    offset_y=0,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=rgb565_byte_swap,
)

# You have supplied a power pin so you have to turn the display on
display.set_power(True)

# Initialize display
display.init()
display.set_backlight(100)

# Create a screen and load it
scr = lv.screen_active()

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello, LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import task_handler

th = task_handler.TaskHandler()

```

dcmcshan, 2024-07-27:
That is awesome.  And I will make extensive use of it!  Do you have a web version as well?  I have questionable visions of letting customers make their own UIs...

I only mentions lv_micropython because that is what I am porting from and it was working.  We have made the tragic (and unexpected) blunder of updating both hardware (ESP32 to ESP32S3) and software (lv_micropython to lvgl_micropython).  When I decided to upgrade to the S3, I tacitly assumed that lv_micropython would be a zero cost effort and would come along for the ride.

kdschlosser, 2024-07-27:
The problems with lv_micropython are too numerous to list them all. but the biggest one is how micropython's build system has been updated in order to compile lvgl with it. The way I wrote this library is that the library depends on micropython and not the other way around which is how lv_micropython is written. I am able to change out the version of micropython and also the espidf that is being using very easily. It is just a matter of changing the links for the submodules...

That was the original reason for making this library. It has since expanded to include extra functionality and drivers for displays which I was not originally going to do. 

I have not messed around with making it work as a javascript. Yes it can be done but I have not done it yet. It will work that way if I spend the time to get the build system coded for it.

kdschlosser, 2024-07-27:
Once the new LVGL studio is finished and I get a copy of the file spec for the save file I will be able to write a parser that will read that save file and build the UI using it. So you visions of letting customers make their own UI should be put to rest in the near future... They will be able to design the UI using WYSIWYG interface on a PC and they will just upload the save file and any images the UI uses and the UI will run.

closed 2024-11-05 by kdschlosser