lvgl-micropython/lvgl_micropython #83: SPI.Bus weirdness

dcmcshan, 2024-07-27:
SPI.Bus initialization has some weird things for me...

The code below 
- sometimes works fine, 
- sometimes causes reset
- sometimes gives me this error:
```
MPY: soft reboot
PIN_MOSI = <class 'int'>
PIN_MISO = <class 'int'>
PIN_SCK = <class 'int'>
Traceback (most recent call last):
  File "<stdin>", line 262, in <module>
  File "<stdin>", line 71, in __init__
TypeError: can't convert str to int
>>> 
```

I've had issues with SPI bus retaining state after soft_resets before, and suspect it's something similar.  But this is pretty weird.  It's like the kernel itself is getting corrupted?

Is there a way to reset/release the SPI bus?

```

        print(f"PIN_MOSI = {type(PIN_MOSI)}")
        print(f"PIN_MISO = {type(PIN_MISO)}")
        print(f"PIN_SCK = {type(PIN_SCK)}")
        
        # Initialize the SPI bus
        self.spi_bus = SPI.Bus(
            host=1,
            mosi=PIN_MOSI,
            miso=PIN_MISO,
            sck=PIN_SCK
        )

        print(f"spi_bus = {self.spi_bus}")
```

dcmcshan, 2024-07-27:
And now, just to be interesting, 

```
MPY: soft reboot
PIN_MOSI = <class 'int'>
PIN_MISO = <class 'int'>
PIN_SCK = <class 'int'>
Traceback (most recent call last):
  File "<stdin>", line 302, in <module>
  File "<stdin>", line 262, in main
  File "<stdin>", line 71, in __init__
TypeError: can't convert float to int
````

kdschlosser, 2024-07-27:
That last error is an error in your main.py file on line 71. in your `__init__` function.

Without seeing all of the code I am not going to be able to offer assistance with this...

dcmcshan, 2024-07-27:
It's not actually an error, is my point.  It's weirdness that happens after soft reboot.

Another below.  What I'm wondering is whether with this new SPI architecture if there is a way to reset the SPI bus before we initialize it.

MPY: soft reboot
PIN_MOSI = <class 'int'>
PIN_MISO = <class 'int'>
PIN_SCK = <class 'int'>
spi_bus = <Bus>
Traceback (most recent call last):
  File "<stdin>", line 124, in <module>
  File "MCT.py", line 67, in __init__
  File "MCT.py", line 124, in init_display
  File "display_driver_framework.py", line 169, in __init__
MemoryError: memory allocation failed, allocating 335946543 bytes

kdschlosser, 2024-07-27:
unless I see all of the code that is being run I cannot determine what is happening.

kdschlosser, 2024-07-27:
also what is the purpose of doing a soft reboot? if you have to do a soft reboot then you also have the ability to shut down the display driver and the SPI bus gracefully before the reset takes place so it is a non issue.

```python

th.disable()
th.deinit()
del th

fb1 = display_driver._frame_buffer1
fb2 = display_driver._frame_buffer2
del display_driver

display_bus.free_framebuffer(fb1)
if fb2 is not None:
    display_bus.free_framebuffer(fb1)

del fb1
del fb2

del display_bus
spi_bus.deinit()
del spi_bus()

reboot()
```

kdschlosser, 2024-07-27:
and I would actually call a `gc.collect()` just before the reboot happens.

dcmcshan, 2024-07-27:
I suppose it's just an IDE quirk. Thonny "soft reboots" when you click the "stop/restart backend" button.  

If I change code in a module and I need it to reimport it, it's easiest to just click that button and start fresh.  Only, of course, it doesn't start fresh...

dcmcshan, 2024-07-27:
I'd be interested to know better development habits...

kdschlosser, 2024-07-27:
Oh lord. Don't use Thonny.  It doesn't play nice with Micropython and LVGL. You are better to just write the firmware using esptool and then use putty to connect to the MCU. I have a modified version of ampy to send files to the ESP32. It doesn't have any issues with connecting. I fixed those.

I also have boot script you can load onto your ESP so that you will not be limited to 115200 when sending files. You can jack it up to almost 1 mbit and the files will transfer without any issues.

Carglglz, 2024-07-27:
> also what is the purpose of doing a soft reboot? if you have to do a soft reboot then you also have the ability to shut down the display driver and the SPI bus gracefully before the reset takes place so it is a non issue.

@kdschlosser 
there is a bug in `lv_binding_micropython` in `init/deinit` so right now soft reset is not supported see https://github.com/lvgl/lv_binding_micropython/issues/343, also I think this is the same as #66 and #74, it will raise a memory error or crash. 

test this, soft-reset and run it again...
```py 
import lvgl as lv
print(f"LVGL INIT STATE: {lv.is_initialized()}")
if not lv.is_initialized():
     lv.init()
     print(f"LVGL INIT STATE: {lv.is_initialized()}")

print(f"LVGL INIT STATE: {lv.is_initialized()}")
lv_display = lv.display_create(320, 240)

lv.deinit()

print(f"LVGL INIT STATE: {lv.is_initialized()}")
```

> also what is the purpose of doing a soft reboot?

This allows to run multiple lvgl-micropython tests on devices with the micropython test suite. 👍🏼

kdschlosser, 2024-07-27:
This issue only  exists when people use Thonny. That seems the be the current commonality between them. Solution. Stop using Thonny.

Carglglz, 2024-07-27:
> This issue only exists when people use Thonny 

No, it is the soft reset, Thonny seems to use soft reset between runs, so that's why it happens there. 
Again check https://github.com/lvgl/lv_binding_micropython/issues/343 and the script above, something is wrong with the root pointers...

kdschlosser, 2024-07-27:
I do not know of any reason why someone would want to perform a soft reset with an ESP32 it's either being put to sleep or it's running.

It states in the micropython documentation that the software reset is to ONLY restart the python interpreter. It does NOT clear memory that is used and it does NOT reset the hardware. In other words it restarts the REPL. There are things that micropython does where it stored information in the global namespace in C code so resetting the interpreter is not going to clear those things out. micropython has the SPI done in this manner as well as Pins and other hardware related things.

There is a hardware reset and that is what should be used if wanting to restart the device. 

This is not a bug it is by design. The software reset only has the purpose of stopping the running code on the device. So if there is a loop that gets stuck that kind of thing. It is so the connection between a PC and the ESP doesn't get broken... That is the ONLY purpose it has.

This is under the REPL section of the MicroPython Documentation

https://docs.micropython.org/en/latest/reference/repl.html#soft-reset

> A soft reset will reset the python interpreter, but tries not to reset the method by which you’re connected to the MicroPython board (USB-serial, or Wifi).

Things like frame buffers are created in C code and the pointers to them are also stored in C code. Those pointers gets passed into the interpreter so if the interpreter gets reset only the addresses that are stored in the interpreter gets deleted and not the actual buffer. The buffer still resides in memory and it remains allocated. A software reset is not going to clear that information. Most everything that is done hardware wise is going to persist and that in and of itself is going to cause issues. software resetting the interpreter is not supported.

Carglglz, 2024-07-28:
> I do not know of any reason why someone would want to perform a soft reset with an ESP32 it's either being put to sleep or it's running. 

Testing, e.g. micropython test suite soft resets between tests. 

But ignoring the use case, 

> It states in the micropython documentation that the software reset is to ONLY restart the python interpreter. It does NOT clear memory that is used and it does NOT reset the hardware. In other words [...]

I'm not sure about the memory bit, see

`ports/esp32/main.c`
```C
soft_reset_exit:

    #if MICROPY_BLUETOOTH_NIMBLE
    mp_bluetooth_deinit();
    #endif

    #if MICROPY_PY_ESPNOW
    espnow_deinit(mp_const_none);
    MP_STATE_PORT(espnow_singleton) = NULL;
    #endif

    machine_timer_deinit_all();

    #if MICROPY_PY_THREAD
    mp_thread_deinit();
    #endif

    // Free any native code pointers that point to iRAM.
    if (MP_STATE_PORT(native_code_pointers) != MP_OBJ_NULL) {
        size_t len;
        mp_obj_t *items;
        mp_obj_list_get(MP_STATE_PORT(native_code_pointers), &len, &items);
        for (size_t i = 0; i < len; ++i) {
            heap_caps_free(MP_OBJ_TO_PTR(items[i]));
        }
    }

    gc_sweep_all();

    mp_hal_stdout_tx_str("MPY: soft reboot\r\n");

    // deinitialise peripherals
    machine_pwm_deinit_all();
    // TODO: machine_rmt_deinit_all();
    machine_pins_deinit();
    machine_deinit();
    #if MICROPY_PY_SOCKET_EVENTS
    socket_events_deinit();
    #endif

    mp_deinit();
    fflush(stdout);
    goto soft_reset;
}
```

> Most everything that is done hardware wise is going to persist and that in and of itself is going to cause issues. software resetting the interpreter is not supported.

Well the script above doesn't set any hardware nor allocates any frame buffer, yet it fails so I still believe there is a bug in `init/deinit` code (especially when it can be fixed by https://github.com/lvgl/lv_binding_micropython/issues/343#issuecomment-2185427218)

kdschlosser, 2024-07-28:
are you shutting down the SPI?
are you shutting down the display bus?
are you shitting down the display driver?
are you shutting down the touch driver?
are you shitting down the I2C or SPI for the touch?
are you freeing the frame buffer(s)?

I am willing to be NO to every single one of those things.

The reason why I am willing to bet that is because Thonny is passing an escape sequence to the REPL terminal that does the restart. It is NOT calling a function in your code that would shut these things down. There is NO way for me to determine that the interpreter is being restarted and because there is no way of knowing that I am not able to perform a cleanup of anything. You are more than welcome to reach out to the developers of MicroPython and ask them to add the ability to register callback functions for a soft restart this way if the interpreter is restarted by something like Thonny then the callback will get called and you can perform the proper shutdown of the display and associated parts and pieces. That is going to be a larger code change than I am willing to make for it to only support a single use case. Yes I said single use case read below to find out why.

There are ways to perform testing without needing to use the tests that are built into MicroPython. Those tests are written to test MicroPython and only MicroPython. MicroPython is a dependency of this project and the tests that are built into that dependency are not going to be written to work with the code in this library. There are ways of testing to make sure that it is working properly and it doesn't do any soft resets. See here...

https://github.com/lvgl/lvgl/tree/master/tests/micropy_test

I am the one that wrote that testing package and it works and is used each and every single time someone makes a PR or a commit gets pushed to LVGL.

kdschlosser, 2024-07-28:
and to your statement of no code allocation...

https://github.com/micropython/micropython/blob/6007f3e2062cc65fc8416f241c682e37eb956c11/ports/esp32/machine_hw_spi.c#L136

That right there. A variable set in the global namespace that is tasked with holding all of the SPI instances that have been started, Memory IS allocated for the objects that are stored in there. 

That allocation occurs right here...

https://github.com/micropython/micropython/blob/6007f3e2062cc65fc8416f241c682e37eb956c11/ports/esp32/machine_hw_spi.c#L275

kdschlosser, 2024-07-28:
If the ability to get a notification on soft reset existed what I would end up doing is performing a hard reset once the notification was received. In the end that is what needs to be done if you want to perform a reset. 

This is a 100% gurantee that all memory and hardware settings will be cleaned out for the program to restart into a clean slate and not have any issues.

Carglglz, 2024-07-28:
> you can perform the proper shutdown of the display and associated parts and pieces.
> the freeing of those resources will ONLY occur is deinit gets called.
> That is the responsibility of the user. 

What I'm saying is that even in the case you can clean and free those resources (not even setting hardware, buffers, etc) before doing a soft-reset , after a soft-reset `lvgl.init()/lvgl.deinit()` won't work and will raise a memory error or crash.

Have you tested this script, soft-reset and run again?
```py

import lvgl as lv
print(f"LVGL INIT STATE: {lv.is_initialized()}")
if not lv.is_initialized():
     lv.init()
     print(f"LVGL INIT STATE: {lv.is_initialized()}")

print(f"LVGL INIT STATE: {lv.is_initialized()}")
lv_display = lv.display_create(320, 240)

lv.deinit()

print(f"LVGL INIT STATE: {lv.is_initialized()}")

```

> https://github.com/lvgl/lvgl/tree/master/tests/micropy_test

I'm aware of this but AFAIK this isn't intended to run on devices, only unix port...

> There are ways to perform testing without needing to use the tests that are built into MicroPython. Those tests are written to test MicroPython and only MicroPython

Not exactly, it can run any micropython/test code and compare stdout output with an expected output, and can run on devices too ( and it's nice to have tests that can be run both on unix and on devices...)

kdschlosser, 2024-07-28:
As I have stated in my previous comments. The soft reset is ONLY for the purposes of restarting the REPL. It is not intended to be used as a "restart" that is what a hard reset is for. There are things that are not going to be able to get cleaned up properly unless a hard reset is performed. 

The test program that comes with MicroPython is written for the purpose of testing MicroPython. It has not been written for testing anything other than MicroPython. The fact that you can use it in SOME cases to test user code is because the user code isn't going anything that is really advanced. It is not going to work with ALL user code and nor should it. The test program is written for the sole purpose of testing the internals of MicroPython and nothing else.   

And this is the output I get from running that code...

```
LVGL INIT STATE: False
LVGL INIT STATE: True
LVGL INIT STATE: True
LVGL INIT STATE: False
```

closed 2024-11-05 by kdschlosser

kdschlosser, 2024-12-30:
when the bus has not devices attached to it it will automatically deinit.  so when you delete all references to the `SPI.Device` instances or call the `deinit` method for all `SPI.Device` instances that are connected to an `SPI.Bus` instance that instance will automatically deinit. 

Optionally if you delete the `SPI.Bus` instance it will deinit all of the attached devices for you and deinit the bus.

If you are using an `SPI.Device` instance with an `lcd_bus.SPIBus` the ` `lcd_bus.SPIBus` instance should get torn down if you delete the  `SPI.Bus` instance or if you delete or deinit the  `SPI.Device` instance. 

The reason why it is done this way is to allow for "hot plugging" devices onto an SPI Bus without having to create a new `SPI.Bus` instance. The instance can always exist and the underlying esp bus driver will get created once a device gets added and torn down when there are no more devices on the bus.

I put a very large amount of thought into the handling of the `SPI.Bus`/`SPI.Device` relationship and how to handle the starting and stopping of the underlying hardware drivers. This is what I had come up with as being the best possible way of handling it. It allows you to put the ESP32 to sleep without the need to recreate the `SPI.Bus` instance when coming out of sleep. You will only have to recreate the devices attached to the bus as those can only be initialized a single time. I looked into making the `SPI.Device` instances and the `lcd_bus.*` instances so they could be started and stopped without needing to create a new instance but the issue is the very large amount of memory that needs to be used to store all of the config parameters seemed very wasteful. It's a fairly large amount of memory used to hold that information, probably something along the lines of a couple hundred bytes. When most of the parameters that get passed to the `lcd_bus.*` constructors can be placed as constants using `micropython.const` it really doesn't make any sense to keep all the information resident in memory. It doesn't take that long for the code to run to create a new instance for it. The `SPI.Bus` driver needs < 30 bytes to hold the pin numbers and other config data. Not a large amount which is the reason why I opted to make it so it can start and stop. There is actually less memory that is used in a "hot plug" kind of a scenario when the user has no device plugged into it.

Basically when you create the spi bus the hardware is not active until you add a device to the bus.

kdschlosser, 2024-12-30:
OH, as a note. if you use `del` to delete an instance the deinit may not happen right away. The deinit is only going to happen when the garbage collection runs. so you will want to manually run the garbage collection after you delete anything that is associated with the hardware.

I also wanted to mention another thing. The reason why I do not allow using `machine.Pin` instances with any of the drivers is because MicroPython is written in a manner that doesn't properly tear down the pins even when their use goes out of scope or the `machine.Pin` instance gets deleted. The hardware is never able to go back to it's "unused" state. This is the reason why I altered the spi so it is not able to take a `machine.Pin instance` for the pins. It must be passed an integer instead. This way any pins that are used get set back to the unused state when any of the hardware drivers I wrote get torn down.

wxzed, 2024-12-31:
@kdschlosser  I also want the system to work properly after a soft reset and re-uploading the code. However, even after using gc, it still doesn't work. After a hard reset, the code runs fine for the first time. But after a soft reset and re-uploading, it behaves abnormally.
```
from machine import SPI,SDCard
import gc
import time
import machine

class TF_card(object):
    def __init__(self):
        print("TF card init\n")
        self.spi_bus = SPI.Bus(host = 1, mosi = 42, miso = 41, sck = 44)
        print(" SDCard(spi_bus\n")
        self.sd = SDCard(spi_bus = self.spi_bus, cs = 40, freq = 1000000)
        #vfs.mount(self.sd, "/sd")

    def deinit(self):
        print("deinit")
        self.sd.deinit()
        del self.sd
        del self.spi_bus
        gc.collect()
tf_card = TF_card()
time.sleep(1)
tf_card.deinit()
del tf_card
gc.collect()
time.sleep(1)
while True:
    pass
```

Carglglz, 2024-12-31:
@wxzed yes, at least the lvgl part (see this commit https://github.com/lvgl/lv_binding_micropython/pull/341/commits/f76c25fbc)

kdschlosser, 2024-12-31:
if you want the binding to work properly after a soft reset then you have to handle tearing down everything manually prior to the soft reset occurring. I am guessing that Thonny is being used and in that case you are basically SOL because there is no indication of when the soft reset is occurring so you cannot take care of tearing down the drivers. This is a bug in Thonny as it should not be performing a soft reset. It should be performing a hard reset and reconnecting the serial once the hard reset is performed. They opted to do a soft reset so they wouldn't have to content with remaking the serial connection.

Open an issue with the people over at Thonny to get the problem fixed as it is a problem with their software.

Just as a note, if you use a `machine.Pin` object in MicroPython you will still be faced with soft reset issues because it doesn't reset the hardware back to it's "unused" state even if you delete the instance. This is done so a user has the ability to pass a `machine.Pin` instance to any of the hardware driver classes in MicroPython. If the `machine.Pin` instance reset the hardware state back to unused once the instance was garbage collected you would have problems with this kind of code..

```python
i2c = machine.I2C(sca=machine.Pin(10), sdl=machine.Pin(11))
```

What happens above is the I2C driver collects the pin number from those `machine.Pin` instances and then sets up the driver. The driver modifies the hardware states so it will work with I2C. The `machine.I2C` constructor then exits and those 2 `machine.Pin` instances end up getting flagged for garbage collection. Once they get garbage collected if the instances were written to reset the hardware pin states back to being unused guess what would happen? The I2C driver would then stop working. 

So that is also going to cause problems because the machine.Pin instances don't get reset either. Nothing hardware related gets reset when performing a soft reset. Hence the name "soft" reset, as in software reset not hardware. a "hard" reset is a hardware reset. 

The only way this could be handled properly is if a user was able to register a callback function that would get called when a soft reset is performed. That is something that would need to be added to MicroPython. The answer you will get for asking to have that done is to use the hard reset instead of the soft. This makes the entire issue come full circle back to it being a problem in Thonny and that is where the problem needs to be fixed.

kdschlosser, 2024-12-31:
People are failing to understand what "soft reset" and "hard reset" are. a soft reset only restarts the REPL and clears the heap. The hardware remains unchanged. This is not a function of the MCU. it is something that MicroPython does. There is only one type of reset available for say the ESP32 and that is a hardware reset. That is the ONLY way the ESP32 has been designed to be reset and is the ONLY kind of reset that should be performed. The soft reset should never have been made a public function. It was written into MicroPython to be able to switch from a normal REPL to a RAW REPL for the purpose of doing things like uploading files. the soft reset gets performed by using a control key in the REPL so there really was never any reason to expose the soft reset function so it can be used in Python code.

I do not believe that anyone has opened an issue with Thonny about this problem. Not one that is explained in a manner that the developers would understand. Nothing comes up about this problem when I search for "soft reset". 

This is what I am going to do. I will open an issue and fully explain what is happening and why it is happening. If they decide to not correct the issue then I am going to remove the soft reset function from being able to be called. Yes it will break using Thonny entirely. It will be easier to say that Thonny is not compatible with this binding then it is to have to continually explain this soft reset issue. It is actually really starting to grate on me having to explain it over and over and over again.

kdschlosser, 2025-01-01:
I opened an issue for Thonny, maybe they will fix it and maybe they will not. If they do not fix it I am removing the ability to soft reset the MCU's....

https://github.com/thonny/thonny/issues/3423

0ut4t1m3, 2025-01-01:
If you do go the route of disabling soft reboot could you please add a build flag to re-enable it? I'm able to work around issues with soft rebooting (I just avoid initilising any busses under certain boot conditions) so would prefer still having it as an option if possible.

kdschlosser, 2025-01-01:
I get a lot of people that complain about the soft rebooting issue. It's driving me bananas. I can fix it by patching MicroPython by adding a callback registry for a soft reset. This way all of the hardware can be proper reset. It defeats the purpose tho because it basically ends up being a hard reset at that point. The only hardware that wouldn't get reset would be the uart/otg. everything else would need to be reset. It's by far a whole lot easier to handle it in thonny. having to increase the firmware size to work around a big in the coding of an IDE that some people use is really not the correct thing to do.

I am hoping that my request for them to fix it doesn't fall on deaf ears. Another thing that should be added to thonny is the ability to set the bitrate the serial connection runs at. This is something that is able to be adjusted in MicroPython at compile time and it would be nice if it could also be changed in thonny. Last I checked it was hard coded, maybe they added the feature?

0ut4t1m3, 2025-01-01:
Yeah there's a ton of things they could do better, especially around USB native connections. I see a lot less issues with UART based boards and the fixed baudrate kills me on larger applications.
I do like how it manages the filesystem on the device though and being able to run unsaved scripts in the editor directly are a couple of reasons I still fall back to Thonny, it's just simple and does everything I need in one app. I know you can't fix any of this just thought some context be useful.

On your callback idea I like the idea as it makes everything "just work" and would stop a lot of the "can't init after soft reboot" issues you get but I understand not wanting to need to add in special handlers for one specific IDE. Alternatively would it be possible to signal that you don't care about any existing instances during init of the bus?

kdschlosser, 2025-01-01:
Let me take some time and look at the code again to see how I might be able to reset the hardware state. I have to look at how the soft reset is being performed to see if python objects allocated on the heap are getting garbage collected the way they should be. If they are then I will need to make some changes to the SPI driver to do a proper tear down. The user is going to need to keep a reference to the SPI.Bus object in order for this to work properly. Making the change will cause code like this to not work anymore.

```
def init():

   hw_spi_bus = machine.SPI.Bus(...)
   display_bus = lcd_bus.SPIBus(hw_spi_bus, ...)
   display=some_display_driver.DisplayDriver(display_bus, ...)
   
   return display
   
disp = init()
```

this is because the reference to `machine.SPI.Bus` would go out of scope once the function exists resulting in that object getting garbage collected and when the garbage collection takes place the tear down of the SPI bus would occur. The code would need to be changed to...

```
def init():

   hw_spi_bus = machine.SPI.Bus(...)
   display_bus = lcd_bus.SPIBus(hw_spi_bus, ...)
   display=some_display_driver.DisplayDriver(display_bus, ...)
   
   return hw_spi_bus, display
   
bus, disp = init()

```

so the reference to `machine.SPI.Bus` does go out of scope so it doesn't end up getting garbage collected. We don't need to worry about `lcd_bus.SPIBus` because a reference is held inside the display driver for that. Even though the `machine.SPI.Bus` instance is passed when creating the LCD bus the Python object that gets passed ends up getting converted into a C object so it no longer inside the scope of Python so the reference is not monitored. 

Right now the way the SPI driver is written is no tear down will occur for an SPI bus if the object gets garbage collected. The hardware side of things still stays active. This is how it is done in vanilla MicroPython. I honestly do not know why it was written like that but it was. This is same thing is done in several places for different types of hardware. The biggest issue is the `machine.Pin` objects. In order to make this all work properly any of the MicroPython classes that get passed a `machine.Pin` instance would need to be changed to remove that from being able to be passed. If I add the tear down and reset of the `machine.Pin` instances you could no longer do this...

```
i2c = machine.I2C(sda=machine.Pin(10), scl=machine.Pin(11), ....)
```

you would have to do this..
```
sda = machine.Pin(10)
scl = machine.Pin(11)
i2c = machine.I2C(sda=sda, scl=scl, ....)
```

and the references to `sdl` and `scl` would need to have their references held the entire time the `i2c` reference exists. This results in excess memory use that doesn't really need to exist.

kdschlosser, 2025-01-01:
OK i just updated the repo, maybe it will work... Maybe it won't.. only time will tell..

kdschlosser, 2025-01-04:
I can confirm the changes I made DO allow the soft reset to work properly.  In order for the soft reset to work properly the board needs to fully initialize before the soft reset occurs. so you cannot perform a soft reset until after the display drivers have fully loaded. If the display drivers are not fully initialized a crash will occur like it has previously.

I replaced the spi3wire driver with one that is written in C code. This new one offers for control and I am able to handle the tare down of the driver when a soft reset occurs. This new driver has been tested and it does work properly. There is a change to the API. the parameter names used in the constructor are different. I added a stub file for the spi3wire driver so make sure to check to see what the API names are.

I still have yet to update the I2C driver so it can do a correct tare down. This is a little bit more complex because there is a python wrapper around the `machine.I2C` implementation. I need to update the C code for `machine.I2C` like what was done with `machine.SPI` so I will be able to do away with the `i2c.I2C` python wrapper.

I need to also add some C code to terminate any running timers. That needs to be done prior to the hardware being torn down. I still have to add in testing to see if a driver is valid prior to trying to transmit anything. I believe that is why a crash might still be happening. I am going to attempt to get rid of the crash that occurs if the drivers are not 100% initialized. I might not be able to work around that issue. I think that when I make the changes above it should take care of the issue. we will see what happens.