lvgl-micropython/lvgl_micropython #86: build option

chieve-bale, 2024-07-29:
Thanks for your effort.
I found you creat a python script (make.py) to auto build micropython firmware with lvgl as a submodule.
make.py let it  is convenient to build firmware, but it also let  it confusing with too much parameters.
Personally i prefer to write my specific parameters in to cmake file or sdkconfig file.

So can i  clone this git and write some thing in cmake file so that I can make firmware under` micropython/ports/esp32/ ` and only run `make BOARD=s3 USER_C_MODULE=path_to_lvgl/lvgl.cmake`, even just include `lvgl.cmake `in `mpconfigboard.cmake` and  simplely run `make BOARD=s3`   ? 

['s3'](https://github.com/chieve-bale/esp32s3_v1.23.git) is my specific board sdkconfig and cmake settings which was modified from [ESP32_GENERIC_S3](https://github.com/micropython/micropython/tree/master/ports/esp32/boards/ESP32_GENERIC_S3) under micropython/ports/esp32/boards/, I try to put all modification in this directory include sdkconfig(idf config file) ，mpconfigboard.cmake (included by CMakeLists.txt) and mpconfigboard.h (included by mpconfigport.h)

kdschlosser, 2024-07-29:
you can run it using cmake or make if you want. If you believe it is easier to build having to manually create a partition file, having to manually go and edit the header files in micropython to add the include guards to the ones that are missing them. Those kinds of things.. Then by all means go right ahead.

the build command that is used is no different then what you would do to build MicroPython..

```
make -C mpy_cross

make -C ports/esp32 submodules

make -C ports/esp32 clean BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT LV_CFLAGS="..." LV_PORT=esp32 USER_C_MODULE=../../../../../ext_mod/micropython.cmake

make -C ports/esp32 BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT LV_CFLAGS="..." LV_PORT=esp32 USER_C_MODULE=../../../../../ext_mod/micropython.cmake
```

if you believe that is going to be easier then by all means. You can do that if you want. more power to ya.

kdschlosser, 2024-07-29:
created files like board config file and partition file get placed into the build directory under the root of the repo. They get cleaned up when the build finishes because there is no reason to save them and confuse people with additional files that have been output.

closed 2024-07-30 by chieve-bale