lvgl-micropython/lvgl_micropython #89: ESP32-S3 with GC9A01 displayed as a mirror image

217heidai, 2024-07-31:
[image]

```python
import lvgl as lv
import gc9a01
import lcd_bus
from machine import SPI, Pin
from micropython import const

_WIDTH = const(240)
_HEIGHT = const(240)

_SPI_HOST = const(2)
_SPI_SCK = const(18)
_SPI_MOSI = const(17)
_SPI_MISO = const(-1)

_LCD_FREQ = const(80000000)
_LCD_DC = const(13)
_LCD_CS = const(14)
_LCD_RST = const(15)
_LCD_BACKLIGHT = const(-1)

# Initialize LVGL
#lv.init()

# Initialize the SPI bus
spi_bus = SPI.Bus(
    host = _SPI_HOST,
    mosi = _SPI_MOSI,
    miso = _SPI_MISO,
    sck = _SPI_SCK
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus = spi_bus,
    dc = _LCD_DC,
    cs = _LCD_CS,
    freq=_LCD_FREQ 
)

# Initialize the GC9A01 display driver
display = gc9a01.GC9A01(
    data_bus = display_bus,
    display_width = _WIDTH,
    display_height = _HEIGHT,
    reset_pin = _LCD_RST,
    reset_state = gc9a01.STATE_LOW,
    power_on_state = gc9a01.STATE_HIGH,
    backlight_pin=None,
    offset_x=0,
    offset_y=0,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True
)

# Initialize display
display.init()

scr = lv.screen_active()

# Create a label on the screen
label = lv.label(scr)
label.set_text("Hello LVGL!")
label.align(lv.ALIGN.CENTER, 0, 0)

print("LVGL test script executed successfully")

import task_handler

th = task_handler.TaskHandler()

```

kdschlosser, 2024-07-31:
when did you clone the repo? I just fixed this issue within the last couple of days.

217heidai, 2024-08-01:
It works well now. Thank you for the update.

closed 2024-08-01 by 217heidai

kdschlosser, 2024-08-01:
cool glad to hear it works for ya...

kdschlosser, 2024-08-01:
If you don't mind me asking.. What board are you using?

217heidai, 2024-08-01:
[LuatOS ESP32S3-CORE](https://wiki.luatos.com/chips/esp32s3/board.html) with 8MB psram and 16MB flash