lvgl-micropython/lvgl_micropython #92: lv.task_handler()

dcmcshan, 2024-08-07:
Hi there.

Wondering if you can help me better understand task_handler.  In particular, I have this code below which sets the bg to blue and then green with some manual lv.task_handler() calls in between.  The result is just a blue screen.  The green is ineffective.

This is just an example.  My desire is to run lv.task_handler in a thread, but that seems to be not working, so I have boiled down my lack of understanding to this example.

```

        # Initialize the SPI bus
        try:
            import MCT
            self.spi_bus = SPI.Bus(
                host=1,
                mosi=MCT.PIN_MOSI,
                miso=MCT.PIN_MISO,
                sck=MCT.PIN_SCK
            )  
        except Exception as e:
            #sys.print_exception(e)
            print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
            machine.reset()

        # Initialize the display bus
        self.display_bus = lcd_bus.SPIBus(
            spi_bus=self.spi_bus,
            dc=MCT.PIN_LCD_RS,
            cs=MCT.PIN_LCD_CS,
            freq=40000000
        )

        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=None,
            frame_buffer2=None,
            reset_pin=MCT.PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_pin=MCT.PIN_EN_LED, #FIXME: Not LED!
            power_on_state=ST7735.STATE_HIGH,
            backlight_pin=MCT.PIN_LCD_BACKLIGHT,
            backlight_on_state=ST7735.STATE_HIGH,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )
        
        lv.init()
        
        self.display.set_power(True)
        self.display.init()
        self.display.set_params(self.display._INVON)
        
        scr = lv.screen_active()
        scr.set_style_bg_color(lv.color_make(0, 0, 0xff), 0)
        lv.task_handler()
        scr.set_style_bg_color(lv.color_make(0, 0xff, 0), 0)
        lv.task_handler()
```

dcmcshan, 2024-08-07:
I think it needs an lv.tick_inc to do something new.

kdschlosser, 2024-08-07:
The tack_handler that is built into lvgl is pretty simple in how it works. All it does is it iterates over an array of lv.timer_t structures and checks to see if each one has had it's interval met or exceeded and it if has then it calls the callback function attached to the timer. 

In order for this to work LVGL needs to have some kind of time keeping ability. Since LVGL is written to be unaware of the hardware in which it is running on that timekeeping needs to be done by the user. There is a function that is used to tell lvgl how much time in milliseconds has passed between calling the function. as an example...

```python
import time

start_time = time.ticks_ms()
while True:
    end_time = time.ticks_ms()
    diff = time.ticks_diff(end_time, start_time)
    if diff > 0:
        start_time = end_time
        lv.tick_inc(diff)
        lv.task_handler()
```

This is able to be done but it has a downside. First is it ties up the main thread in a loop so the repl will no longer work. The second is code like you have posted will not work using this because this code has to go after your GUI code so only the last color will end up being displayed...

The other issue is what is called spinning wheels. That loop is going to consume 100% of the processor time this is because it is a continual loop that basically does nothing for an entire millisecond. LVGL is pre set so it is only going to update the display every 33 milliseconds. the better way to handle this is going to be using threads and creating thread workers. This is an important concept because it allows you to inject code that mneeds to be run into specific threads from any other thread. This is an important thing to do because LVGL is NOT thread safe. If you tell LVGL to refresh the display and then in another thread you are updating a GUI element you can end up with memory corruption. This is how you keep that from happening...

```python
import _thread
import time

class ThreadFunc(object):
    
    def __init__(self, func, *args, **kwargs):
        self.func = func
        self.args = args
        self.kwargs = kwargs
        
        self.exception = None
        self.result = None
        self.lock = _thread.allocate_lock()
        self.lock.acquire()
    
    def run(self):
        try:
            self.result = self.func(*self.args, **self.kwargs)
        except Exception as err:  # NOQA
            self.exception = err

        self.lock.release()

class ThreadWorker(object):
    
    def __init__(self):
        self._exit_event = False
        self._loop_lock = _thread.allocate_lock()
        self._queue_lock = _thread.allocate_lock()
        self._queue = []
        self._thread = _thread.start_new_thread(self.loop, ())
    
    def exit(self):
        with self._queue_lock:
            del self._queue[:]
        
        self._exit_event = True
        if self._loop_lock.locked():
            self._loop_lock.release()
            
        while self._exit_event:
            time.sleep_ms(3)
            
    def add_nowait(self, func, *args, **kwargs):
        tf = ThreadFunc(func, *args, **kwargs)
        
        with self._queue_lock:
            self._queue.append(tf)
        
        if self._loop_lock.locked():
            self._loop_lock.release()
        
    def add_wait(self, func, *args, **kwargs):
        tf = ThreadFunc(func, *args, **kwargs)
        
        with self._queue_lock:    
            self._queue.append(tf)
            
        if self._loop_lock.locked():
            self._loop_lock.release()
                
        tf.lock.acquire()
        
        if tf.exception is not None:
            raise tf.exception
        
        return tf.result
            
    def loop(self):
        self._loop_lock.acquire()
        
        while not self._exit_event:
            self._loop_lock.acquire()

            while self._queue:
                with self._queue_lock:
                    try:
                        tf = self._queue.pop(0)
                    except IndexError:
                        break
                
                tf.run()

        self._exit_event = False

lvgl_thread_worker = ThreadWorker()

class LVGLRefreshTask(object):
    
    def __init__(self, display_):
        self.display = display_
        display_.delete_refr_timer()
        
        self.start_time = time.ticks_ms
        
    def __call__(self):
        end_time = time.ticks_ms()
        diff = time.ticks_diff(end_time, self.start_time)
        if diff:
            self.start_time = end_time
            lv.tick_inc(diff)
            lv.task_handler()
        
        lv.refr_now(self.display)
       
        return None

lvgl_display_refr = LVGLRefreshTask(display)

# any time you want to refresh the display you do it by calling

# lvgl_thread_worker.add_wait(lvgl_display_refr)
# or 
# lvgl_thread_worker.add_nowait(lvgl_display_refr)

# Anything that has any modification to anything lvgl related must be 
# handed off to that thread worker. so if you want to update the text for a label

# lvgl_thread_worker.add_nowait(label.set_text, 'some text to be updated')

# if you want to get the text from a label

# label = lvgl_thread_worker.add_wait(label.get_text)
```

by having this thread worker you have the ability to update anything and everything in LVGL from any thread without ever having to worry about memory corruption or crashes. You do not use the `task_handler` module if you are using this. 

One of the things you need to remember is if you have a thread that is running amuck and doesn't have any kind of a stall in it then the thread worker may not ever get switched to to process things. The way the thread worker is written it pauses without sleeping and it will only wake up if there is actual work that needs to be done.

kdschlosser, 2024-08-07:
That thread worker code block is pseudo code. I have not tested it and it is likely to need some tweaking to get running properly. It is intended to be used as a general idea and not actual running code.

dcmcshan, 2024-08-07:
This is really important to know.  Thank you.  I have written a small "realtime exec" for micropython.  In quotes because of course it's not actually realtime.  I register functions/methods and the scheduler actually calls them based on their deadlines.  It's pretty simple, but nice to be organized.

This is the sort of statistics the module tracks. 

```
                                                     ______Delta______ ___Exec Time___
t/T Func                               n     Delay   Min   Max   Avg   Min   Max   Avg          Run  %Run     Hz   Goal Blown    Next
🚀 t0  idle                             0        1     0     0     0     0     0     0          0ms    0%   0.00 1000.0      0     7073   6323
🚀 t2  update_display                   0      200     0     0     0     0     0     0          0ms    0%   0.00   5.0      0    12814    592
🚀 t4  update_state                     0      500     0     0     0     0     0     0          0ms    0%   0.00   2.0      0    13482    -56
🚀 t5  monitor                          0     1000     0     0     0     0     0     0          0ms    0%   0.00   1.0      0    13986   -550
🚀 t3  update                           0     1000     0     0     0     0     0     0          0ms    0%   0.00   1.0      0    13639   -193
🚀 t1  collect                          0     1000     0     0     0     0     0     0          0ms    0%   0.00   1.0      0     8076   5390
```

And, assuming that I schedule rate monotonically, I can then determine if the threads are schedulable:

```
simulate_and_compute_schedulability
2 update_display             63.163 ms /  200 ms =   0.32   0.32 
4 update_state               15.359 ms /  500 ms =   0.03   0.35 
5 monitor                     0.052 ms / 1000 ms =   0.00   0.35 
3 update                     18.631 ms / 1000 ms =   0.02   0.37 
1 collect                    12.338 ms / 1000 ms =   0.01   0.38 
System is schedulable.                               0.38 < 0.74
```

I think the scheduler works well enough.  What doesn't work is starting things from repl.  And now, I suspect that is because it's not thread safe.  While all those tasks are in the thread, the repl is not!   

Now, the behavior I am seeing is that my USB CDC serial connection resets.  No core dump, just resets.  I thought it might be a bug in USB CDC, like maybe it's getting starved.  But lvgl not being thread safe is a far more compelling theory.  I will do some experiments and see how it goes.

Thank you!

dcmcshan, 2024-08-07:
I tried moving my "UI" class which creates various lvgl labels into the main class so it's all in the same thread, but that doesn't seem to address the USB CDC hanging up issue.  Still, good to know...

dcmcshan, 2024-08-07:
Here is some test code based on what you suggest.  Interestingly, I get the same reset behavior as I do with my scheduler.  I sometimes do see the label on the screen, then it resets.

Any ideas?

```

import _thread
import time
import lvgl as lv  # Ensure lvgl is imported appropriately
from MCT_DISPLAY import MCT_DISPLAY

class ThreadFunc:
    def __init__(self, func, *args, **kwargs):
        self.func = func
        self.args = args
        self.kwargs = kwargs
        self.exception = None
        self.result = None
        self.lock = _thread.allocate_lock()
        self.lock.acquire()

    def run(self):
        try:
            self.result = self.func(*self.args, **self.kwargs)
        except Exception as err:
            self.exception = err
        self.lock.release()

class ThreadWorker:
    def __init__(self):
        self._exit_event = False
        self._loop_lock = _thread.allocate_lock()
        self._queue_lock = _thread.allocate_lock()
        self._queue = []
        self._thread = _thread.start_new_thread(self.loop, ())

    def exit(self):
        with self._queue_lock:
            del self._queue[:]
        self._exit_event = True
        if self._loop_lock.locked():
            self._loop_lock.release()
        while self._exit_event:
            time.sleep_ms(3)

    def add_nowait(self, func, *args, **kwargs):
        tf = ThreadFunc(func, *args, **kwargs)
        with self._queue_lock:
            self._queue.append(tf)
        if self._loop_lock.locked():
            self._loop_lock.release()

    def add_wait(self, func, *args, **kwargs):
        tf = ThreadFunc(func, *args, **kwargs)
        with self._queue_lock:
            self._queue.append(tf)
        if self._loop_lock.locked():
            self._loop_lock.release()
        tf.lock.acquire()
        if tf.exception is not None:
            raise tf.exception
        return tf.result

    def loop(self):
        self._loop_lock.acquire()
        while not self._exit_event:
            self._loop_lock.acquire()
            while self._queue:
                with self._queue_lock:
                    try:
                        tf = self._queue.pop(0)
                    except IndexError:
                        break
                tf.run()
        self._exit_event = False

lvgl_thread_worker = ThreadWorker()

class LVGLRefreshTask:
    def __init__(self, display_):
        self.display = display_
        self.start_time = time.ticks_ms

    def __call__(self):
        end_time = time.ticks_ms()
        diff = time.ticks_diff(end_time, self.start_time)
        if diff:
            self.start_time = end_time
            lv.tick_inc(diff)
            lv.task_handler()
        lv.refr_now(self.display)
        return None

lvgl_display_refr = LVGLRefreshTask(MCT_DISPLAY())

if __name__ == "__main__":
    import time
    
    lv.init()
    display = MCT_DISPLAY()
    
    # Create a label
    label = lv.label(lv.screen_active())
    label.set_text("Time will be updated here")
    label.align(lv.ALIGN.CENTER, 0, 0)
    label.set_style_text_color(lv.color_make(0, 0, 0),0)  #
    
    def update_label():
        lvgl_thread_worker.add_nowait(label.set_text, f"{time.ticks_ms()}")
    
    try:
        while True:
            update_label()
            lvgl_thread_worker.add_nowait(lvgl_display_refr)
            time.sleep(1)
    except KeyboardInterrupt:
        lvgl_thread_worker.exit()

```

kdschlosser, 2024-08-07:
we have discovered there seem to be an issue with the task_handler included in the binding and threads. I have yet tested it to see what the issue is but I have a suspicion that the problem is not in the binding and the issue is actually in MicroPython itself. If you are willing to test that out using vanilla micropython and the task_handler module with all of the lvgl related bits removed from it....

From my understanding if the  task_handler is running as soon as a thread is launched a core panic occurs.

dcmcshan, 2024-08-07:
Do you have such a task_handler that you'd like me to test?  I can do so immediately.

kdschlosser, 2024-08-07:
Yeah let me put it together for ya and tell you want to do.

You will need to compile v1.23.0 of Micropython for your MCU and flash it. I will send the code for the task handler in a few minutes and tell you what to do from there,..

kdschlosser, 2024-08-07:
OK... Take the code below and put it into a file mnamed `main.py` and upload it to your ESP32. Then restart the ESP with the serial terminal connected and see what the output is...

```python
import micropython  # NOQA
import sys

from machine import Timer  # NOQA

_default_timer_id = 0

class _DefaultUserData(object):
    pass

def _default_exception_hook(e):
    sys.print_exception(e)
    TaskHandler._current_instance.deinit()  # NOQA

class TaskHandler(object):
    _current_instance = None

    def __init__(
        self,
        duration=1000,
        timer_id=_default_timer_id,
        max_scheduled=2,
        exception_hook=_default_exception_hook
    ):
        if TaskHandler._current_instance is not None:
            self.__dict__.update(TaskHandler._current_instance.__dict__)
        else:
          
            TaskHandler._current_instance = self

            self._callbacks = []

            self.duration = duration
            self.exception_hook = exception_hook

            self._timer = Timer(timer_id)

            # Allocation occurs here
            self._task_handler_ref = self._task_handler
            self.max_scheduled = max_scheduled

            self._timer.init(
                mode=Timer.PERIODIC,
                period=self.duration,
                callback=self._timer_cb
            )
            self._scheduled = 0

    def add_event_cb(self, callback, event, user_data=_DefaultUserData):
        for i, (cb, evt, data) in enumerate(self._callbacks):
            if cb == callback:
                evt = event
                if user_data != _DefaultUserData:
                    data = user_data

                self._callbacks[i] = (cb, evt, data)
                break
        else:
            if user_data == _DefaultUserData:
                user_data = None

            self._callbacks.append((callback, event, user_data))

    def remove_event_cb(self, callback):
        for i, obj in self._callbacks:
            if obj[0] == callback:
                self._callbacks.remove(obj)
                break

    def deinit(self):
        self._timer.deinit()
        TaskHandler._current_instance = None

    def disable(self):
        self._scheduled += self.max_scheduled

    def enable(self):
        self._scheduled -= self.max_scheduled

    @classmethod
    def is_running(cls):
        return cls._current_instance is not None

    def _task_handler(self, _):
        self._scheduled -= 1
        print('Scheduled Task Called...')

    def _timer_cb(self, _):
        if self._scheduled < self.max_scheduled:
            try:
                micropython.schedule(self._task_handler_ref, 0)
                self._scheduled += 1
            except:  # NOQA
                pass

import _thread
import time

th = TaskHandler()

def run_thread():
    while True:
        time.sleep(1)
        print('Thread loop running')
        

t = _thread.start_new_thread(run_thread, ())

```

kdschlosser, 2024-08-07:
here this is a better script to use...

```python

mport micropython  # NOQA

from machine import Timer  # NOQA

class TaskHandler(object):
    _current_instance = None

    def __init__(
        self,
        duration=1000,
        timer_id=0,
        max_scheduled=2,
    ):
    
        TaskHandler._current_instance = self

        self.duration = duration

        self._timer = Timer(timer_id)

        # Allocation occurs here
        self._task_handler_ref = self._task_handler
        self.max_scheduled = max_scheduled
        self._scheduled = 0

        self._timer.init(
            mode=Timer.PERIODIC,
            period=self.duration,
            callback=self._timer_cb
        )

    def _task_handler(self, _):
        self._scheduled -= 1
        print('Scheduled Task Called...')

    def _timer_cb(self, _):
        if self._scheduled < self.max_scheduled:
            try:
                micropython.schedule(self._task_handler_ref, 0)
                self._scheduled += 1
            except:  # NOQA
                pass

import _thread
import time

th = TaskHandler()

def run_thread():
    while True:
        time.sleep(1)
        print('Thread loop running')

t = _thread.start_new_thread(run_thread, ())
```

dcmcshan, 2024-08-07:
Seems to run cleanly on both stock kernel and lvgl.

dcmcshan, 2024-08-07:
>>> Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running
Scheduled Task Called...
Thread loop running

kdschlosser, 2024-08-08:
really???

OK...

Then I am not sure why someone has reported that it doesn't work... very strange...

Try using the binding with this code...

```python

import task_handler

th = task_handler.TaskHandler()

def run_thread():
    while True:
        time.sleep(1)
        print('Thread loop running')

t = _thread.start_new_thread(run_thread, ())
```

let me know if it crashes...

dcmcshan, 2024-08-08:
That does crash after about 6 seconds.

>>> Thread loop running
Thread loop running
Thread loop running
Thread loop running
Thread loop running
Thread loop running
Thread loop running

Connection lost -- read failed: [Errno 6] Device not configured

Use Stop/Restart to reconnect.

Process ended with exit code 1.

dcmcshan, 2024-08-08:
But not always? (it's currently running)

dcmcshan, 2024-08-08:
I can also confirm that instantiating TaskHandler breaks this slightly more complicated with display driver.  This is all a bit work-in-progress with all the debugging  but you get the idea.  Ideally, I'd like to run update_display in a thread of some fashion.

```

class MCT_DISPLAY:
    _instance = None
    _lock = None

    def __new__(cls, *args, **kwargs):
        if cls._instance is None:
            cls._instance = super(MCT_DISPLAY, cls).__new__(cls)
        return cls._instance
    
    def __del__(self):
        print ("deleting MCT_DISPLAY")
        del self.spi_bus

    def __init__(self):
        if hasattr(self, '_initialized') and self._initialized:
            return
        
        self._initialized = True
        
        self._lock = _thread.allocate_lock()
        
        self.last_update = time.ticks_ms()
        
        # Initialize the SPI bus
        try:
            self.spi_bus = SPI.Bus(
                host=1,
                mosi=PIN_MOSI,
                miso=PIN_MISO,
                sck=PIN_SCK
            )  
        except Exception as e:
            sys.print_exception(e)
            print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
            machine.reset()

        # Initialize the display bus
        self.display_bus = lcd_bus.SPIBus(
            spi_bus=self.spi_bus,
            dc=PIN_LCD_RS,
            cs=PIN_LCD_CS,
            freq=40000000
        )

        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=None,
            frame_buffer2=None,
            reset_pin=PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_pin=PIN_EN_LED, #FIXME: Not LED!
            power_on_state=ST7735.STATE_HIGH,
            backlight_pin=PIN_LCD_BACKLIGHT,
            backlight_on_state=ST7735.STATE_HIGH,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )
        
        lv.init()
        
        self.display.set_power(True)
        self.display.init()
        self.display.set_params(self.display._INVON)
        
        scr = lv.screen_active()
        scr.set_style_bg_color(lv.color_make(0, 0, 0), 0)
        
    def fill(self,color):
        scr = lv.screen_active()
        scr.set_style_bg_color(hex_to_lv_color(color), 0)
        
    def wait_for_lock(name = ""):
        if not MCT_DISPLAY._lock:
            MCT_DISPLAY._lock = _thread.allocate_lock()
        #print(f"{name} waiting for lock {MCT_DISPLAY._lock}")
        while MCT_DISPLAY._lock.locked():
            time.sleep_ms(1)
        
        #print(f"{name} acquiring lock")
        MCT_DISPLAY._lock.acquire()
        
    def release_lock(name = ""):
        #print(f"{name} releasing lock")
        MCT_DISPLAY._lock.release()
        
    def update_display(self):
        MCT_DISPLAY.wait_for_lock("display")
            
        current_ticks_ms = time.ticks_ms()
        if self.last_update:
            self.update_delta = time.ticks_diff(current_ticks_ms,self.last_update)
        else:
            self.update_delta = 0
        lv.tick_inc(self.update_delta)
        self.last_update = current_ticks_ms
        
        lv.task_handler()
        
        MCT_DISPLAY.release_lock("display")
            
    def exists(self):
        pass

    def clock(self):
        import time
        
        # Create a label
        label = lv.label(lv.screen_active())
        label.set_text("Time")
        label.align(lv.ALIGN.CENTER, 0, 0)
        label.set_style_text_color(lv.color_make(0xff, 0, 0),0)  #
        
        def update_label():
            MCT_DISPLAY.wait_for_lock("label")
            t = time.ticks_ms()
            print(t)
            label.set_text(f"{t}")
            MCT_DISPLAY.release_lock("label")
            lv.tick_inc(1000)
            lv.task_handler()
            
            
        def threaded_update(update_function, interval):
            while True:
                update_function()
                time.sleep(interval)
         
        _thread.start_new_thread(lambda: threaded_update(update_label, 1), ())
        #_thread.start_new_thread(lambda: threaded_update(self.update_display, 1), ())

if __name__ == "__main__":

    display = MCT_DISPLAY()
    
    lv.init()
    
    # This works as expected
    label = lv.label(lv.screen_active())
    label.set_text("Test")
    
    lv.tick_inc(1000)
    lv.task_handler()
    
    # This works as expected
    start = time.ticks_ms()
    while time.ticks_ms() < start + 10000:
        label.set_text(f"{time.ticks_ms()}")
        
        lv.tick_inc(1000)
        lv.task_handler()
    
    def update_label(t = None):
        t = time.ticks_ms()
        print(t)
        label.set_text(f"{t}")
    
    def threaded_update(update_function, interval):
        while True:
            try:
                update_function()
                time.sleep(interval)
            except Exception as e:
                sys.print_exception(e)
                
    _thread.start_new_thread(lambda: threaded_update(update_label, 1), ())
    
    import task_handler
    th = task_handler.TaskHandler() # This causes crash
            
    print ("MCT_DISPLAY Done")
    
```

kdschlosser, 2024-08-08:
So what is happening is this. The timer makes the callback to the task handler and then the task handler schedules to have a callback made in the main thread. This is when `lv.task_handler()` gets called. In order for that to happen there needs to be some kind of stall that happens where tasks are able to change. If the task_handler ends up consuming a lot of time either from the amount of time LVGL is taking to render or how long it is taking to write the data to the display that could be causing the watchdog timer to get upset...

Lets try bumping the timeout for the watchdog up a couple of notches.

What is the ESP32 you are using?

kdschlosser, 2024-08-08:
I am going to look into making it possible to run LVGL on the second core in order to speed things up.

kdschlosser, 2024-08-08:
OK rebuild the binding adding

```
CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=n CONFIG_ESP_TASK_WDT_PANIC=n CONFIG_ESP_TASK_WDT_EN=n CONFIG_ESP_INT_WDT=n CONFIG_ESP_TASK_WDT_INIT=n CONFIG_BOOTLOADER_WDT_ENABLE=n CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=n
```

onto the end of your build command...

dcmcshan, 2024-08-08:
Am working with ESP32-S3-WROOM-N16R8.  Have pulled lvgl_micropython and rebuilding now.

I do not know that it's a problem with TaskHandler as such, as when I call lv.task_handler() on my own, it still breaks. Here is another data point, before I try new kernel with options:

In this code, I am running both the label update and the lv.task_handler() in the same thread.  It works fine with the lv.task_handler() directly in the loop (as commented out).  But it fails calling the object method which has the same calls.

```
    def update_display(self, delay_ms = 50):
        lv.tick_inc(delay_ms)
        lv.task_handler()
```

```
    def update_label_loop(delay_ms = 50):
        while True:
            update_label()
            display.update_display(delay_ms)
            #lv.tick_inc(delay_ms)
            #lv.task_handler()
            time.sleep_ms(delay_ms)
                
    _thread.start_new_thread(update_label_loop, ())
```

dcmcshan, 2024-08-08:
Build fails for me.  I believe I have pulled latest head.

% git pull
Already up to date.

```
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(machine_wdt.c.obj):(.literal.machine_wdt_make_new+0x14): undefined reference to `esp_task_wdt_reconfigure'
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(machine_wdt.c.obj):(.literal.machine_wdt_make_new+0x18): undefined reference to `esp_task_wdt_add_user'
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(machine_wdt.c.obj):(.literal.machine_wdt_feed+0x0): undefined reference to `esp_task_wdt_reset_user'
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(machine_wdt.c.obj): in function `mp_machine_wdt_make_new_instance':
/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/machine_wdt.c:51:(.text.machine_wdt_make_new+0x34): undefined reference to `esp_task_wdt_reconfigure'
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/machine_wdt.c:62:(.text.machine_wdt_make_new+0x50): undefined reference to `esp_task_wdt_add_user'
/Users/danielmcshan/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32s3/libmain_esp32s3.a(machine_wdt.c.obj): in function `mp_machine_wdt_feed':
/Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/machine_wdt.c:73:(.text.machine_wdt_feed+0x9): undefined reference to `esp_task_wdt_reset_user'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/log/idf_py_stderr_output_51508 and /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM/log/idf_py_stdout_output_51508
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1

danielmcshan@Daniels-Laptop lvgl_micropython % python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM --flash-size=16 DISPLAY=ST7735 --usb-otg CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=n CONFIG_ESP_TASK_WDT_PANIC=n CONFIG_ESP_TASK_WDT_EN=n CONFIG_ESP_INT_WDT=n CONFIG_ESP_TASK_WDT_INIT=n CONFIG_BOOTLOADER_WDT_ENABLE=n CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=n

```

dcmcshan, 2024-08-08:
This code implements a thread lock for the label writing and the display updating.  It even outputs cute little emoji.  It works as expected without lv.task_handler().  With lv.task_handler() it crashes.  

Point being that I believe that lthis code is properly semaphored, so it's not that the lv queue is being messed with while we are handling it.

```
import sys
import time
from machine import SPI
import machine
import _thread

import lvgl as lv
import lcd_bus

class MCT_DISPLAY:
    _instance = None
    _lock = None

    def __new__(cls, *args, **kwargs):
        if cls._instance is None:
            cls._instance = super(MCT_DISPLAY, cls).__new__(cls)
        return cls._instance
    
    def __del__(self):
        print ("deleting MCT_DISPLAY")
        del self.spi_bus

    def __init__(self):
        if hasattr(self, '_initialized') and self._initialized:
            return
        
        self._initialized = True
        
        self._lock = _thread.allocate_lock()
        
        self.last_update = time.ticks_ms()
        
        # Initialize the SPI bus
        try:
            self.spi_bus = SPI.Bus(
                host=1,
                mosi=PIN_MOSI,
                miso=PIN_MISO,
                sck=PIN_SCK
            )  
        except Exception as e:
            sys.print_exception(e)
            print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
            machine.reset()

        # Initialize the display bus
        self.display_bus = lcd_bus.SPIBus(
            spi_bus=self.spi_bus,
            dc=PIN_LCD_RS,
            cs=PIN_LCD_CS,
            freq=40000000
        )

        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=None,
            frame_buffer2=None,
            reset_pin=PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_pin=PIN_EN_LED, #FIXME: Not LED!
            power_on_state=ST7735.STATE_HIGH,
            backlight_pin=PIN_LCD_BACKLIGHT,
            backlight_on_state=ST7735.STATE_HIGH,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )
        
        lv.init()
        
        self.display.set_power(True)
        self.display.init()
        self.display.set_params(self.display._INVON)
        
        scr = lv.screen_active()
        scr.set_style_bg_color(lv.color_make(0, 0, 0), 0)
        
    def fill(self,color):
        scr = lv.screen_active()
        scr.set_style_bg_color(hex_to_lv_color(color), 0)
        
    def nowait_for_lock(name = ""):
        if not MCT_DISPLAY._lock:
            MCT_DISPLAY._lock = _thread.allocate_lock()
        print(f"{name} 🔐")
        if MCT_DISPLAY._lock.locked():
            return False
        
        print(f"{name} 🔒")
        MCT_DISPLAY._lock.acquire()
        return True
        
    def wait_for_lock(name = ""):
        if not MCT_DISPLAY._lock:
            MCT_DISPLAY._lock = _thread.allocate_lock()
        print(f"{name} 🔑")
        while MCT_DISPLAY._lock.locked():
            time.sleep_ms(1)
            if time.ticks_ms()%1000 == 0: print ("🔐", end="")
        
        print(f"{name} 🔒")
        MCT_DISPLAY._lock.acquire()
        
    def release_lock(name = ""):
        print(f"{name} 🔓")
        MCT_DISPLAY._lock.release()
        
    def update_display(self, delay_ms = 50):
        lv.tick_inc(delay_ms)
        lv.task_handler()
        
    def exists(self):
        pass

    def clock(self):
        import time
        
        # Create a label
        label = lv.label(lv.screen_active())
        label.set_text("Time")
        label.align(lv.ALIGN.CENTER, 0, 0)
        label.set_style_text_color(lv.color_make(0xff, 0, 0),0)  #
        
        def update_label():
            MCT_DISPLAY.wait_for_lock("label")
            t = time.ticks_ms()
            print(t)
            label.set_text(f"{t}")
            MCT_DISPLAY.release_lock("label")
            lv.tick_inc(1000)
            lv.task_handler()
            
            
        def threaded_update(update_function, interval):
            while True:
                update_function()
                time.sleep(interval)
         
        _thread.start_new_thread(lambda: threaded_update(update_label, 1), ())
        #_thread.start_new_thread(lambda: threaded_update(self.update_display, 1), ())

def update_display(timer=None):
    MCT_DISPLAY.wait_for_lock("d")
    lv.tick_inc(1000)
    #lv.task_handler()
    MCT_DISPLAY.release_lock("d")
    
def update_display_loop():
    while True:
        update_display()
        time.sleep(1)
            
_thread.start_new_thread(update_display_loop, ())

if __name__ == "__main__":

    display = MCT_DISPLAY()
    
    lv.init()
    
    # This works as expected
    label = lv.label(lv.screen_active())
    label.set_text("Test")

    def update_label():
        t = time.ticks_ms()
        label.set_text(f"{t}")
        
    def update_label_loop():
        while True:
            MCT_DISPLAY.wait_for_lock("l")
            update_label()
            MCT_DISPLAY.release_lock("l")
            time.sleep(1)
            
    _thread.start_new_thread(update_label_loop, ())
            
    print ("MCT_DISPLAY Done")
    

```

It even outputs cute emoji (it's running slowly)

```
d 🔑
d 🔒
d 🔓
l🔑
l 🔒
l 🔓
d 🔑
d 🔒
d 🔓
l 🔑
l 🔒
l 🔓
d 🔑
d 🔒
d 🔓
l 🔑
l 🔒
l 🔓
```

dcmcshan, 2024-08-08:
For the time being, I'm going to switch everything over to use lv.timer_create, though this is suboptimal to the solution I desire, so I'd like to figure out a way to make it thread safe.

kdschlosser, 2024-08-08:
Lets step back from this for a minute and make it a simpler thing to test...

don't change anything with the code other than adding the missing constants and correcting any typos I may have. Keep the core of how it works alone. I need the code to rune exactly how it is to determine the location of the issue.

```python
import machine

import lcd_bus

spi_bus = machine.SPI.Bus(
    host=1,
    mosi=PIN_MOSI,
    miso=PIN_MISO,
    sck=PIN_SCK
)

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=40000000
)

import st7735
import lvgl as lv

display = st7735.ST7735R_Red(
    data_bus=self.display_bus,
    display_width=80,
    display_height=160,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=PIN_LCD_RESET,
    reset_state=st7735.STATE_LOW,
    power_pin=PIN_EN_LED,  # FIXME: Not LED!
    power_on_state=st7735.STATE_HIGH,
    backlight_pin=PIN_LCD_BACKLIGHT,
    backlight_on_state=st7735.STATE_HIGH,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
    color_byte_order=st7735.BYTE_ORDER_BGR
)

display.set_power(True)
display.init()
display.invert_colors()

scr = lv.screen_active()

import time
import _thread

lock = _thread.allocate_lock()

def run_thread():
    start_time = time.ticks_ms()
    while True:
        end_time = time.ticks_ms()
        diff = time.ticks_diff(end_time, start_time)
        
        if diff >= 100:
            start_time = end_time
            lv.tick_inc(diff)
            
            lock.acquire()
            print('calling task handler')
            lv.task_handler()
            lock.release()

t = _thread.start_new_thread(run_thread, ())

import random

color_start_time = time.ticks_ms()
while True:
    color_end_time = time.ticks_ms()
    if time.ticks_diff(color_end_time, color_start_time) >= 150:
        color_start_time = color_end_time
    
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        lock.acquire()
        print('changing color')
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        lock.release()

```

dcmcshan, 2024-08-08:
With minimal changes, this works fine.

dcmcshan, 2024-08-08:
I have successfully integrated your example into my OO/thread world.  It works as advertised. So that's pretty great.

```
if __name__ == "__main__":
    
    from MCT_DISPLAY import MCT_DISPLAY
    display = MCT_DISPLAY()
            
    import random

    import lvgl as lv
    scr = lv.screen_active()

    def change_color():
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        MCT_DISPLAY._lock.acquire()
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        MCT_DISPLAY._lock.release()
          
    import mct_thread
    c = mct_thread.start_new_thread(change_color, (), 1000)

    mct_thread.start()
    mct_thread.taskShow()
```

dcmcshan, 2024-08-08:
However... adding this label causes it to crash.

```

if __name__ == "__main__":
    
    from MCT_DISPLAY import MCT_DISPLAY
    display = MCT_DISPLAY()
            
    import random

    import lvgl as lv
    scr = lv.screen_active()

    def change_color():
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        MCT_DISPLAY._lock.acquire()
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        MCT_DISPLAY._lock.release()
          
    import mct_thread
    c = mct_thread.start_new_thread(change_color, (), 1000)

    mct_thread.start()
    mct_thread.taskShow()
    
    MCT_DISPLAY._lock.acquire()
    temp_label = lv.label(scr)
    MCT_DISPLAY._lock.release()
```

dcmcshan, 2024-08-08:
Interestingly, when I put it in the straight python (no OO) world, it works fine.

scr = lv.screen_active()

import time
import _thread

lock = _thread.allocate_lock()

```

def run_thread():
    start_time = time.ticks_ms()
    while True:
        end_time = time.ticks_ms()
        diff = time.ticks_diff(end_time, start_time)
        
        if diff >= 100:
            start_time = end_time
            lv.tick_inc(diff)
            
            lock.acquire()
            print('calling task handler')
            lv.task_handler()
            lock.release()

t = _thread.start_new_thread(run_thread, ())

temp_label = lv.label(scr)

import random

color_start_time = time.ticks_ms()
while True:
    color_end_time = time.ticks_ms()
    if time.ticks_diff(color_end_time, color_start_time) >= 150:
        color_start_time = color_end_time
    
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        lock.acquire()
        print('changing color')
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        lock.release()
```

kdschlosser, 2024-08-08:
give this a try...

```python
scr = lv.screen_active()

import time
import _thread

lock = _thread.allocate_lock()

def run_thread():
    start_time = time.ticks_ms()
    while True:
        end_time = time.ticks_ms()
        diff = time.ticks_diff(end_time, start_time)
        
        if diff >= 100:
            start_time = end_time
            lv.tick_inc(diff)
            
            lock.acquire()
            print('calling task handler')
            lv.task_handler()
            lock.release()

t = _thread.start_new_thread(run_thread, ())

lock.acquire()
temp_label = lv.label(scr)
temp_label.set_text("1")
temp_label.center()

lock.release()

import random

count = 1

color_start_time = time.ticks_ms()
while True:
    color_end_time = time.ticks_ms()
    if time.ticks_diff(color_end_time, color_start_time) >= 150:
        color_start_time = color_end_time
    
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        count += 1
        lock.acquire()
        print('changing color')
        temp_label.set_text(str(count))
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        lock.release()
```

kdschlosser, 2024-08-08:
what do you mean by no OO??

dcmcshan, 2024-08-08:
So, that works in straight "functional" python, but for reasons I do not understand, fails when I wrap it up in a class.

So this, for instance, crashes with the label, runs fine without.  It's very peculiar.

```
import sys
import time
from machine import SPI
import machine
import _thread

import lvgl as lv
import ST7735
import lcd_bus

PIN_EN_LED = const(36)
PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_CS = const(10)
PIN_LCD_RESET = const(21)
PIN_LCD_BACKLIGHT = const(40)

def hex_to_lv_color(hex_value):
    # Extract the red, green, and blue components from the integer
    r = (hex_value >> 16) & 0xFF
    g = (hex_value >> 8) & 0xFF
    b = hex_value & 0xFF
    
    # Call the lv.color_make function with the extracted values
    return lv.color_make(r, g, b)

class MCT_DISPLAY:
    _instance = None
    _lock = _thread.allocate_lock()

    def __new__(cls, *args, **kwargs):
        if cls._instance is None:
            cls._instance = super(MCT_DISPLAY, cls).__new__(cls)
        return cls._instance
    
    def __del__(self):
        print ("deleting MCT_DISPLAY")
        del self.spi_bus

    def __init__(self):
        if hasattr(self, '_initialized') and self._initialized:
            return
        
        self._initialized = True
        
        self.last_update = time.ticks_ms()
        
        # Initialize the SPI bus
        try:
            self.spi_bus = SPI.Bus(
                host=1,
                mosi=PIN_MOSI,
                miso=PIN_MISO,
                sck=PIN_SCK
            )  
        except Exception as e:
            sys.print_exception(e)
            print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
            machine.reset()

        # Initialize the display bus
        self.display_bus = lcd_bus.SPIBus(
            spi_bus=self.spi_bus,
            dc=PIN_LCD_RS,
            cs=PIN_LCD_CS,
            freq=40000000
        )

        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=None,
            frame_buffer2=None,
            reset_pin=PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_pin=PIN_EN_LED,  # FIXME: Not LED!
            power_on_state=ST7735.STATE_HIGH,
            backlight_pin=PIN_LCD_BACKLIGHT,
            backlight_on_state=ST7735.STATE_HIGH,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )
        
        lv.init()
        
        self.display.set_power(True)
        self.display.init()
        self.display.set_params(self.display._INVON)
        
        scr = lv.screen_active()
        scr.set_style_bg_color(lv.color_make(0, 0, 0), 0)
        
        self.update_delay_ms = 5
        from machine import Timer
        self.lv_timer = Timer(-1)
        
        import mct_thread
        mct_thread.start_new_thread(self.update_display, None, 100)
        
    def fill(self,color):
        scr = lv.screen_active()
        scr.set_style_bg_color(hex_to_lv_color(color), 0)

    def update_display(self, diff=100):
        lv.tick_inc(diff)
        self._lock.acquire()
        lv.task_handler()
        self._lock.release()
        
    def exists(self):
        pass

if __name__ == "__main__":
    
    from MCT_DISPLAY import MCT_DISPLAY
    display = MCT_DISPLAY()
            
    import random

    import lvgl as lv
    scr = lv.screen_active()

    def change_color():
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        
        try:
            count += 1
        except:
            count = 1
            MCT_DISPLAY._lock.acquire()
            temp_label = lv.label(scr)
            temp_label.center()
            MCT_DISPLAY._lock.release()
            
        MCT_DISPLAY._lock.acquire()
        temp_label.set_text(str(count))
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        MCT_DISPLAY._lock.release()
          
    import mct_thread
    c = mct_thread.start_new_thread(change_color, (), 1000)

    mct_thread.start()
    mct_thread.taskShow()
    
    #this resets
    
```

dcmcshan, 2024-08-09:
FWIW, here is a version that uses _thread rather than my fancy version which schedules.  It still crashes in the same manner.

```
import sys
import time
from machine import SPI
import machine
import _thread

import lvgl as lv
import ST7735
import lcd_bus

PIN_EN_LED = const(36)
PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_CS = const(10)
PIN_LCD_RESET = const(21)
PIN_LCD_BACKLIGHT = const(40)

class MCT_DISPLAY:
    _instance = None
    _lock = _thread.allocate_lock()

    def __new__(cls, *args, **kwargs):
        if cls._instance is None:
            cls._instance = super(MCT_DISPLAY, cls).__new__(cls)
        return cls._instance
    
    def __del__(self):
        print ("deleting MCT_DISPLAY")
        del self.spi_bus

    def __init__(self):
        if hasattr(self, '_initialized') and self._initialized:
            return
        
        self._initialized = True
        
        self.last_update = time.ticks_ms()
        
        # Initialize the SPI bus
        try:
            self.spi_bus = SPI.Bus(
                host=1,
                mosi=PIN_MOSI,
                miso=PIN_MISO,
                sck=PIN_SCK
            )  
        except Exception as e:
            sys.print_exception(e)
            print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
            machine.reset()

        # Initialize the display bus
        self.display_bus = lcd_bus.SPIBus(
            spi_bus=self.spi_bus,
            dc=PIN_LCD_RS,
            cs=PIN_LCD_CS,
            freq=40000000
        )

        self.display = ST7735.ST7735R_Red(
            data_bus=self.display_bus,
            display_width=80,
            display_height=160,
            # we are going to let the driver handle the allocation of the frame buffers
            frame_buffer1=None,
            frame_buffer2=None,
            reset_pin=PIN_LCD_RESET,
            reset_state=ST7735.STATE_LOW,
            power_pin=PIN_EN_LED,  # FIXME: Not LED!
            power_on_state=ST7735.STATE_HIGH,
            backlight_pin=PIN_LCD_BACKLIGHT,
            backlight_on_state=ST7735.STATE_HIGH,
            offset_x=26,
            offset_y=1,
            color_space=lv.COLOR_FORMAT.RGB565,
            rgb565_byte_swap=True,
            color_byte_order=ST7735.BYTE_ORDER_BGR
        )
        
        lv.init()
        
        self.display.set_power(True)
        self.display.init()
        self.display.set_params(self.display._INVON)
        
        scr = lv.screen_active()
        scr.set_style_bg_color(lv.color_make(0, 0, 0), 0)
        
        self.update_delay_ms = 5
        from machine import Timer
        self.lv_timer = Timer(-1)
    
        _thread.start_new_thread(self.update_display_loop, ()) 
        
    def fill(self,color):
        scr = lv.screen_active()
        scr.set_style_bg_color(color, 0)

    def update_display(self, diff=100):
        lv.tick_inc(diff)
        self._lock.acquire()
        lv.task_handler()
        self._lock.release()
        
    def update_display_loop(self):
        while True:
            self.update_display()
            time.sleep_ms(100)
        
    def exists(self):
        pass

if __name__ == "__main__":

    display = MCT_DISPLAY()
            
    import random

    import lvgl as lv
    scr = lv.screen_active()

    def change_color():
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        
        try:
            count += 1
        except:
            count = 1
            MCT_DISPLAY._lock.acquire()
            temp_label = lv.label(scr)
            temp_label.center()
            MCT_DISPLAY._lock.release()
            
        MCT_DISPLAY._lock.acquire()
        temp_label.set_text(str(count))
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        MCT_DISPLAY._lock.release()
          
          
    def change_color_loop():
        while True:
            change_color()
            time.sleep_ms(1000)
          
    c = _thread.start_new_thread(change_color_loop, ())
    
    #this resets
```

dcmcshan, 2024-08-09:
In fact, just creating the label will cause the crash.  No label, and it runs fine.

```
    def change_color():
        r = random.randint(0, 255)
        g = random.randint(0, 255)
        b = random.randint(0, 255)
        
        try:
            count += 1
        except:
            count = 1
            MCT_DISPLAY._lock.acquire()
            #temp_label = lv.label(scr)  # << uncommenting this causes crash
            MCT_DISPLAY._lock.release()
            
        MCT_DISPLAY._lock.acquire()
        scr.set_style_bg_color(lv.color_make(r, g, b), 0)
        MCT_DISPLAY._lock.release()
```

dcmcshan, 2024-08-09:
Further more, the second thread isn't necessary.  I think maybe task_handler just really doesn't like being in a thread?

```

    #c = _thread.start_new_thread(change_color_loop, ())
    change_color_loop()
```

dcmcshan, 2024-08-09:
So, it's not anything about the OO.  It's about how far down the call stack.  In the code below, everything is the same, except loop_1 calls  task_handler directly,, but loop_2 calls a function.

```
def update_display():
    lv.tick_inc(100)
    _lock.acquire()
    lv.task_handler()
    _lock.release()
    
def update_display_loop_1():
    while True:
        lv.tick_inc(100)
        _lock.acquire()
        lv.task_handler()
        _lock.release()
        time.sleep_ms(100)
        
def update_display_loop_2():
    while True:
        update_display()
        time.sleep_ms(100)
        
_thread.start_new_thread(update_display_loop_1, ()) #works
#_thread.start_new_thread(update_display_loop_2, ()) #crashes
```

kdschlosser, 2024-08-09:
You might need to increase the stack memory. I am thinking that might be what is causing the problem... The default stack size is 5120 bytes and the minimum is 4096 bytes. This may not be enough for LVGL to do what it needs to do. 

To set the stack size use the following code..

```python

import _thread

_thread.stack_size(10240)
_thread.start_new_thread(some_func, ())

```

kdschlosser, 2024-08-09:
another thing is I have not had much luck using `__new__` in MicroPython. For the purposes of what you are dong you don't need it anywho

```python

class SomeClass(object):
    _instance = None
    
    def __init__(self, param1, param2):
        if SomeClass._instance is not None:
            self.__dict__.update(SomeClass._instance.__dict__)
        else:
            self.param1 = param1
            self.param2 = param2
```

The actual proper way to create an instance singleton is using meta classes...
This will not work on MicroPython but it is still good to know how to do it for CPython.

```python

# This instance single creates a single instance based on 
# the arguments passed when constructing the instance.
# this is handy if you are wanting to represent devices and you only want one
# instance made for each device. This allows for you to collect the same 
# instance for a device no matter where in the code you construct the instance.
# so long as the parameters passed are the same. 

class InstanceSingletonMeta(type):
    # this doesn't actually hold anything. It is here only for the purposes
    # of making an IDE happy
    _instances = {}
    
    def __init__(cls, name, bases, dct):
        super().__init__(name, bases, dct)
        
        setattr(cls, '_instances', {})
        
    def __call__(cls, *args, **kwargs):
        key = list(args)
        
        for k in sorted(list(kwargs.keys())):
            key.append(kwargs[k])
            
        if key not in cls._instances:
            cls._instances[key] = super().__call__(*args, **kwargs)
            
        return cls._instances[key]
    

# this class you would subclass if you wanted to 
# make an instance singleton class
class InstanceSingleton(metaclass=InstanceSingletonMeta):
    pass

```

dcmcshan, 2024-08-09:
Increasing stack size seems to have sorted it out.  Very strange and very silent error.  Might want to just make that default if it's what it needs.

dcmcshan, 2024-08-09:
Why did this not raise a stack overflow exception?

kdschlosser, 2024-08-09:
The stack size is going to be dictated by what the user needs to do. So it is really up to the user to set it.

The cool thing is we got it sorted out. Took me a minute to think of what could be causing the problem. I am going to let the other user that was having the exact same issue know how to fix it as well.

kdschlosser, 2024-08-09:
your guess is as good as mine as to why it just crashes without any error.

closed 2024-09-06 by kdschlosser