lvgl-micropython/lvgl_micropython #93: Why is micropython heap with SPIRAM on N16R8 only 128k?

dcmcshan, 2024-08-10:
I'm unfamiliar with how micropython allocates SPIRAM, but my naive assumption is that with 8MB SPIRAM on the N16R8 I would have more than 128K heap?
```

>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 127936, used: 126336, free: 1600, max new split: 69632
 No. of 1-blocks: 2167, 2-blocks: 365, max blk sz: 282, max free sz: 86
```

As an aside, it's also weird to me that on soft-reset, micropython now has 64K heap. You may use this in your fight against soft reset ;)
```

>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 38832, free: 25168, max new split: 69632
 No. of 1-blocks: 559, 2-blocks: 117, max blk sz: 98, max free sz: 393
```

also, why does mem_info report differently than gc.mem_*?

```

>>> import micropython; micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 47808, free: 16192, max new split: 69632
 No. of 1-blocks: 749, 2-blocks: 133, max blk sz: 98, max free sz: 203
>>> import gc; gc.mem_alloc()
55792
>>> import gc; gc.mem_free()
93616

```

kdschlosser, 2024-08-10:
try using vanilla micropython and see if it reports the same thing

kdschlosser, 2024-08-10:
I have an N16R8 and also an N32R8 and I know that both of them report the proper amount of memory. It might be something with the board you are using?  That is the reason why I am suggesting to try vanilla MicroPython and see if it is reporting the same low numbers or not. If it is doing it with vanilla MicroPython then we will mosey on over to their repo and open up an issue there.

dcmcshan, 2024-08-10:
I will investigate, but I also see this below.  But if you are seeing the proper amount of memory...

https://github.com/orgs/micropython/discussions/12316

> On ESP32, MicroPython will request a small initial heap (64KiB) from the RTOS, and then grow it as necessary by adding new "splits".
> It will aim to grow the heap by doubling in size each time (but will grow by more if the current allocation requires it).
> gc.mem_free() returns how much RAM is free in all heap splits, plus the largest free block of RTOS memory that could be made into a new "split" on request. This is a reasonable heuristic for "total free memory".
> micropython.mem_info() prints separate values for how much memory is free in the existing MicroPython memory heap, and what size a "max new split" could be if added on demand. There might be still more available after that, in the next contiguous block.

kdschlosser, 2024-08-10:
I think my memory checks may have been when using an older version of MicroPython. I am not 100% sure if that is the case.

Based on that description `gc.mem_free` is not reporting correctly. It should be showing the total amount of available memory...

kdschlosser, 2024-08-10:
ok looking at the micropython code it suggests that the amount of memory returned from `gc.mem_free` should be the amount of free space in the heap and also the largest free contiguous block of memory available outside the heap.

But what is odd is micropython is not using any of the multi heap functions for allocation....

Now sure what the deal is with that. malloc and other memory functions do not allocate to a split heap. I am not sure what they are trying to do with the multi heap thing.

kdschlosser, 2024-08-10:
I opened up an issue on MicroPythons repo to see what they have to say... There is something definatly up with what is being done. Something is not working properly.

Are you having issues allocating memory?

You can try setting up an SPI display bus and then allocating say a 4mb frame buffer in SPIRAM to see if it will allow it.

dcmcshan, 2024-08-10:
Here is result from stock N16R8.  I think the stock kernel I downloaded is only 8MB FLASH aware.

```
MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
>>> import micropython; micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 7168, free: 56832, max new split: 188416
 No. of 1-blocks: 129, 2-blocks: 15, max blk sz: 19, max free sz: 3175
>>> import gc; gc.mem_free()
243280
>>> gc.mem_alloc()
13392
>>> import esp; esp.flash_size()
8388608
```

dcmcshan, 2024-08-10:
As for allocating a frame buffer, I'm not sure that I'm doing it right, please advise.

```
>>> import MCT_DISPLAY
>>> display = MCT_DISPLAY.MCT_DISPLAY()
>>> import lcd_bus
>>> display.display_bus.allocate_framebuffer(1024*1024*4, lcd_bus.MEMORY_SPIRAM | lcd_bus.MEMORY_DMA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: Unable to allocate frame buffer
```

kdschlosser, 2024-08-10:
you cannot allocate the frame buffer after the display driver has been loaded. You cannot start the display driver if you are allocating a buffer of that size..

What you need to do is you need to set up a test script that only starts the SPI bus and then the lcd bus then create the frame buffer. only use the SPIRAM flag and not the DMA one.

dcmcshan, 2024-08-10:
Please review this code and advise whether this should be working.

```
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
MemoryError: Unable to allocate frame buffer
```

```

import machine
from machine import SPI

import lcd_bus

PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_CS = const(10)
PIN_LCD_RESET = const(21)
PIN_LCD_BACKLIGHT = const(40)

# Initialize the SPI bus
try:
    spi_bus = SPI.Bus(
        host=1,
        mosi=PIN_MOSI,
        miso=PIN_MISO,
        sck=PIN_SCK
    )  
except Exception as e:
    sys.print_exception(e)
    print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
    machine.reset()

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=40000000
)

fb = display_bus.allocate_framebuffer(1024*1024*4, lcd_bus.MEMORY_SPIRAM | lcd_bus.MEMORY_DMA)

import ST7735
import lvgl as lv

display = ST7735.ST7735R_Red(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=PIN_LCD_RESET,
    reset_state=ST7735.STATE_LOW,
    backlight_pin=PIN_LCD_BACKLIGHT,
    backlight_on_state=ST7735.STATE_HIGH,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
    color_byte_order=ST7735.BYTE_ORDER_BGR,
    framebuffer1=fb
)

display.init()
display.invert_colors()

```

dcmcshan, 2024-08-10:
Oops, left the DMA flag in, but still fails to allocate the buffer.

dcmcshan, 2024-08-10:
This more modest (and accurate) request also fails.

```
import machine
from machine import SPI

import lcd_bus

PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_CS = const(10)
PIN_LCD_RESET = const(21)
PIN_LCD_BACKLIGHT = const(40)

# Initialize the SPI bus
try:
    spi_bus = SPI.Bus(
        host=1,
        mosi=PIN_MOSI,
        miso=PIN_MISO,
        sck=PIN_SCK
    )  
except Exception as e:
    sys.print_exception(e)
    print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
    machine.reset()

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=40000000
)

import lvgl as lv
fb_size = 80*60*lv.color_format_get_size(lv.COLOR_FORMAT.RGB565)
fb1 = display_bus.allocate_framebuffer(fb_size, lcd_bus.MEMORY_SPIRAM)
fb2 = display_bus.allocate_framebuffer(fb_size, lcd_bus.MEMORY_SPIRAM)

import ST7735
import lvgl as lv

display = ST7735.ST7735R_Red(
    data_bus=display_bus,
    display_width=80,
    display_height=160,
    # we are going to let the driver handle the allocation of the frame buffers
    frame_buffer1=None,
    frame_buffer2=None,
    reset_pin=PIN_LCD_RESET,
    reset_state=ST7735.STATE_LOW,
    backlight_pin=PIN_LCD_BACKLIGHT,
    backlight_on_state=ST7735.STATE_HIGH,
    offset_x=26,
    offset_y=1,
    color_space=lv.COLOR_FORMAT.RGB565,
    rgb565_byte_swap=True,
    color_byte_order=ST7735.BYTE_ORDER_BGR,
    framebuffer1=fb
)

display.init()
display.invert_colors()

```

kdschlosser, 2024-08-10:
this is what you need to do...

```
from machine import SPI, const

import lcd_bus

PIN_MOSI = const(11)
PIN_SCK = const(12)
PIN_MISO = const(13)
PIN_LCD_RS = const(14)
PIN_LCD_CS = const(10)
PIN_LCD_RESET = const(21)
PIN_LCD_BACKLIGHT = const(40)

# Initialize the SPI bus
try:
    spi_bus = SPI.Bus(
        host=1,
        mosi=PIN_MOSI,
        miso=PIN_MISO,
        sck=PIN_SCK
    )  
except Exception as e:
    import machine
    sys.print_exception(e)
    print ("💥💥💥 Unable to SPI, You should Hard Reset 💥💥💥")
    machine.reset()

# Initialize the display bus
display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    dc=PIN_LCD_RS,
    cs=PIN_LCD_CS,
    freq=40000000
)

fb1 = display_bus.allocate_framebuffer(4194304, lcd_bus.MEMORY_SPIRAM)
    
```

do you end up with a memory allocation failure when running that?

If you are then we need to figure out why.

Now. the N16R8 MCU's there are 2 different versions made by Espressif

ESP32-S3-WROOM-1-N16R8
ESP32-S3-WROOM-2-N16R8V

The difference between the 2 is the first one is octal SPIRAM and quad flash and the second one is octal for both.

Another thing that I have seen being used is the ESP32-S3-WROOM-1-N16 being used which has 16mb of quad flash and there is an external chip for the SPIRAM and that chip is quad speed. You might have this setup...

If you add --DEBUG to the build command it will enable debugging messages. do not run any Python code. Just flash the firmware and then connect a terminal program to it and press the reset button. Copy and paste the output of the debugging information so I can have a look at it. I suspect we are going to see that there is a SPIRAM failure. If that happens we can compile it using quad SPIRAM and see how it goes.

To correct the issue with the flash not being the proper size add --flash-size=16 to your build command... If you have an MCU that has octal flash then we add '--octal-flash' to the build command..

We also have the ability to pass any of the ESP-IDF `CONFIG_*` parameters. A list of the config options are located in the [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32/api-reference/kconfig.html)

The default settings for the S3 are

```
CONFIG_SPIRAM=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=8192
CONFIG_SPIRAM_MODE_QUAD=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
```
Is you supply the board variant SPIRAM_OCT all the above options apply and also these..

```
CONFIG_SPIRAM_MODE_QUAD=n
CONFIG_SPIRAM_MODE_OCT=y
```

if you add `CONFIG_SPIRAM_IGNORE_NOTFOUND=n` to the build command the ESP should fail to boot and there will be an error.

dcmcshan, 2024-08-10:
I get this with your code.  Pretty certain this is -N16R8, no V

Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
MemoryError: Unable to allocate frame buffer

dcmcshan, 2024-08-10:
`python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM --flash-size=16 DISPLAY=ST7735 --usb-otg `

is my command line

kdschlosser, 2024-08-10:
There is no SPIRAM board variant. there is only SPIRAM_OCT

Maybe that is your issue?

```
python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ST7735 --usb-otg
```

kdschlosser, 2024-08-10:
are you not seeing the proper amount of flash even when using `--flash-size=16`???

kdschlosser, 2024-08-10:
both of the 8mb SPIRAM variants use octal SPI for the memory. so you need to use SPIRAM_OCT as the board variant no matter what.

kdschlosser, 2024-08-10:
> The difference between the 2 is the first one is octal SPIRAM and quad flash and the second one is octal for both.

I misspoke there. I meant to say flash not SPIRAM. That's my fault. sorry for that.

kdschlosser, 2024-08-10:
from the data sheets for the WROOM1 and WROOM2

WROOM1
```
ESP32-S3-WROOM-1-N16R8 16 MB (Quad SPI) 8 MB (Octal SPI)
```

WROOM2
```
ESP32-S3-WROOM-2-N16R8V 16 MB (Octal SPI) 8 MB (Octal SPI)
```

dcmcshan, 2024-08-10:
-SPIRAM_OCT didn't seem to attach to USB serial properly on my board,

But, I get this on a stock N16R8:

dcmcshan, 2024-08-10:
I am concerned that my PCB designer did not account for Octal SPIRAM...

kdschlosser, 2024-08-11:
But now it's reading properly?

```>>> import micropython; micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 10896, free: 53104, max new split: 8257536   < --- proper memory amount
 No. of 1-blocks: 224, 2-blocks: 31, max blk sz: 32, max free sz: 3307
>>> import gc; gc.mem_free()
8308960                                                                                            < --- proper memory amount
>>> gc.mem_alloc()
17936
>>> import esp; esp.flash_size()
16777216                                                                                          <--- proper flash size
```

dcmcshan, 2024-08-11:
I should have included the preface that I am using your kernel on a N16R8 dev board.  It has no display, but I figured I'd look at memory.

```
LVGL MicroPython 1.23.0 on 2024-08-10; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
```

dcmcshan, 2024-08-11:
Stock kernel I was using wasn't Octal SPIRAM.  I think because we had a discussion a couple weeks ago and you'd said that the N16R8V was Octal, but it's just the lower voltage option.  As you say, and as I learned, all R8 are octal.

[image]

kdschlosser, 2024-08-11:
so the problem was the board variant not being set properly? Everything is working as it should yes?

kdschlosser, 2024-08-11:
If it is then the model number of the ESP32 that you are using is ESP32-S3-WROOM-1-N16R8
That MCU is quad flash and octal SPIRAM.

If you want to dink with it a bit to get better performance from it we can do that is you like...

use this as the build command....

```
python3 make.py esp32 mpy_cross submodules BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --usb-otg --flash-size=16 DISPLAY=st7735 CONFIG_FREERTOS_HZ=1000 CONFIG_ESPTOOLPY_FLASHMODE_QIO=y CONFIG_ESPTOOLPY_FLASHFREQ_120M=y CONFIG_IDF_EXPERIMENTAL_FEATURES=y CONFIG_SPIRAM_SPEED_120M=y CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y CONFIG_SPIRAM_RODATA=y CONFIG_ESP32S3_DATA_CACHE_LINE_64B=y
```

That will overclock the spi for the ram and the flash to 120mhz.

dcmcshan, 2024-08-11:
I think everything kernel wise is as it should be.  My board, on the other hand, is miswired.  I'll remove a series resistor and see if it behaves properly...

kdschlosser, 2024-08-11:
just to let you know I also do board design so if you need any help with it at all lemme know.

dcmcshan, 2024-08-11:
I absolutely will take you up on that.  ESP32 has proven to be a lot trickier than I would have imagined.

kdschlosser, 2024-08-11:
PM me on the forum with the board designs you have and I can take a quick look at them to see if anything jumps out at me as to why you are having the issue you are having.

closed 2024-09-30 by dcmcshan