lvgl-micropython/lvgl_micropython #95: its rlottie supported?

jgpeiro, 2024-08-18:
I enabled the 
#define MICROPY_RLOTTIE  1 
and then I call
python3 make.py esp32 clean BOARD=ESP32_GENERIC_S3 VARAINT=SPIRAM --flash-size=16
but it dosent works, it says:
/workspace/lvgl_micropython/lib/lvgl/src/libs/rlottie/lv_rlottie.c:12:10: fatal error: rlottie_capi.h: No such file or directory

How can I add rlottie to the firmware?
I found also this repo that does it, without micropython, but I dont know how to adapt or follow these steps.
https://github.com/0015/esp_rlottie/tree/main

kdschlosser, 2024-08-18:
you need to have rlottie installed into your system to compile it into micropython.

jgpeiro, 2024-08-18:
how can I install for esp32?

kdschlosser, 2024-08-19:
```shell
sudo apt-get install librlottie-dev
```

jgpeiro, 2024-08-19:
I got the same error. These are the steps to reproduce it:

Clone the lvgl_micropython (I directly open the repo in a workspace using gitpod)

install the esp-idf requirements:
    sudo apt-get update
    sudo apt-get install git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
    lib/esp-idf ((v5.2)) $ ./install.sh
    lib/esp-idf ((v5.2)) $ . ./export.sh

Install rlottie:
    sudo apt-get install librlottie-dev

Comment these lines in the build/esp32.py cause I need the classic SPI:
    #for file in os.listdir(src_path):
    #    src_file = os.path.join(src_path, file)
    #    dst_file = os.path.join(dst_path, file)
    #    shutil.copyfile(src_file, dst_file)

Comment these lines in /workspace/lvgl_micropython/ext_mod/lcd_bus/micropython.cmake cause I dont need it:
    set(LCD_SOURCES
        #${CMAKE_CURRENT_LIST_DIR}/modlcd_bus.c
        #${CMAKE_CURRENT_LIST_DIR}/lcd_types.c
        #${CMAKE_CURRENT_LIST_DIR}/esp32_src/i2c_bus.c
        #${CMAKE_CURRENT_LIST_DIR}/esp32_src/spi_bus.c
        #${CMAKE_CURRENT_LIST_DIR}/esp32_src/i80_bus.c
        #${CMAKE_CURRENT_LIST_DIR}/esp32_src/rgb_bus.c
        #${CMAKE_CURRENT_LIST_DIR}/esp32_src/spi_3wire.c
    )

Configure lvgl in lv_conf.h:
#define MICROPY_COLOR_DEPTH  16

#define LV_USE_ASSERT_STYLE         1

#define LV_FONT_MONTSERRAT_12 0
#define LV_FONT_MONTSERRAT_14 0
#define LV_FONT_MONTSERRAT_16 1
#define LV_FONT_MONTSERRAT_18 1
#define LV_FONT_MONTSERRAT_20 1

#define LV_FONT_DEFAULT &lv_font_montserrat_16

build the firmware (without rlottie just to be sure everything is ok):
/workspace/lvgl_micropython (main) $ python3 make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC_S3 VARAINT=SPIRAM --flash-size=16

This worked fine and produced the image/s:
micropython.bin binary size 0x25c300 bytes

Now I enable rlottie in lv_conf.h:
#define MICROPY_RLOTTIE  1

And launch the command again:
/workspace/lvgl_micropython (main) $ python3 make.py esp32 submodules mpy_cross clean BOARD=ESP32_GENERIC_S3 VARAINT=SPIRAM --flash-size=16

And the build failed:
...
[1124/1966] Generating ../../genhdr/qstr.i.last                                                                              
FAILED: genhdr/qstr.i.last /workspace/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/genhdr/qstr.i.last
/bin/sh esp-idf/main_esp32s3/CMakeFiles/qstr.i.last-93c470f.sh 4401b261f41c5d0a
/workspace/lvgl_micropython/lib/lvgl/src/libs/rlottie/lv_rlottie.c:12:10: fatal error: rlottie_capi.h: No such file or directory
12 | #include <rlottie_capi.h>
|          ^~~~~~~~~~~~~~~~
compilation terminated.
Command '['/home/gitpod/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc', '-E', '-I/workspace/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3/config', '-I/workspace/lvgl_micropython/lib/micropython', '-I/workspace/lvgl_micropython/lib/micropython/lib/berkeley-db-1.xx/include', '-I/workspace/lvgl_micropython/ext_mod/lcd_bus', '-I/workspace/lvgl_micropython/ext_mod/lcd_bus/esp32_include',
...

kdschlosser, 2024-08-19:
OK so you are modifying the build script and it's not working? ¯\_(ツ)_/¯

jgpeiro, 2024-08-19:
I didnt think that these modifications can produce the error...I can build an image after the modifications as you see. I will try without these modifications and I will be happy if that was the problem.

jgpeiro, 2024-08-19:
so, I just restore these files and tryed again and I got the same error...
The only modification I have now is lv_conf.h with the lines I mention avobe.

kdschlosser, 2024-08-20:
```
python3 make.py esp32 submodules mpy_cross BOARD=ESP32_GENERIC_S3 VARAINT=SPIRAM --flash-size=16 LV_CFLAGS="-I/usr/include -llibrlottie"
```

You have to tell the compiler where to look for the header file and also tell it to link the library

kdschlosser, 2024-08-20:
also there is no `SPIRAM` board variant for the S3. There is only `SPIRAM_OCT`. `SPIRAM` is assumed for the s3.

jgpeiro, 2024-08-20:
ok, finally I make it work. Thanks

kdschlosser, 2024-08-20:
well be informative here. what did you do to get it to work?

jgpeiro, 2024-08-20:
Install esp32 idf prerequisites:
sudo apt-get update
sudo apt-get install git wget flex bison gperf python3 python3-pip python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
Install esp32 idf
/workspace/lvgl_micropython/lib/esp-idf/ ./install.sh
/workspace/lvgl_micropython/lib/esp-idf/. ./export.sh

Comment these spi features:
/workspace/lvgl_micropython/builder/esp32.py
    #for file in os.listdir(src_path):
    #    src_file = os.path.join(src_path, file)
    #    dst_file = os.path.join(dst_path, file)
    #    shutil.copyfile(src_file, dst_file)

Comment these lcd features:    
/workspace/lvgl_micropython/ext_mod/lcd_bus/micropython.cmake
    set(LCD_SOURCES
    #    ${CMAKE_CURRENT_LIST_DIR}/modlcd_bus.c
    #    ${CMAKE_CURRENT_LIST_DIR}/lcd_types.c
    #    ${CMAKE_CURRENT_LIST_DIR}/esp32_src/i2c_bus.c
    #    ${CMAKE_CURRENT_LIST_DIR}/esp32_src/spi_bus.c
    #    ${CMAKE_CURRENT_LIST_DIR}/esp32_src/i80_bus.c
    #    ${CMAKE_CURRENT_LIST_DIR}/esp32_src/rgb_bus.c
    #    ${CMAKE_CURRENT_LIST_DIR}/esp32_src/spi_3wire.c
    )

Clone the rlottie component for esp32-idf from this repo
/workspace/lvgl_micropython/lib/esp-idf/components ((v5.2)) $ git clone https://github.com/0015/esp_rlottie.git

Add the rlottie package requirement to the cmake
/workspace/lvgl_micropython/lib/micropython/ports/esp32/esp32_common.cmake
    REQUIRES
        ${IDF_COMPONENTS}
        esp_rlottie

Modify lv_conf.h to enable RLOTTIE and other details (increase memory for lottie render)
/workspace/lvgl_micropython/lib/lv_conf.h
    #define MICROPY_RLOTTIE  1
    #define MICROPY_COLOR_DEPTH  16
    #define MICROPY_MEM_SIZE  1024

Then the compilation fails, I think this function cant be translated to micropython, so I simply removed from the header.
/workspace/lvgl_micropython/lib/lvgl/src/widgets/animimage/lv_animimage.h
//const void ** lv_animimg_get_src(lv_obj_t * img);

And build the project.
python3 make.py esp32 submodules mpy_cross BOARD=ESP32_GENERIC_S3 --flash-size=16

kdschlosser, 2024-08-21:
It's failing the build because the build failed while still in the cmake scripts at some point in time. If you go to `lib/micropython/ports/esp32` and run this command `rm -rf build_ESP32*` it will delete the build folder and then it will compile without issue. This is a problem I still have to address...

Just to let you know you also do not have to do anything with the esp-idf. It all gets handled by the build script. It will clone the repo and set up the proper build environment.

kdschlosser, 2024-08-21:
Also the solution you are using might be adding to the size of the flash because of additional code that is added by using the rlottie component. I have started to mess about with trying to figure out how to link rlottie properly. I may end up doing like what I did with SDL which is cloning it and building it from scratch. This will also provide the latest version of it instead of the old version that is stored in the package manager for the OS. debian like OS's are horrible when it comes to updating the versions of a package.

jgpeiro, 2024-08-21:
> It's failing the build because the build failed while still in the cmake scripts at some point in time. If you go to `lib/micropython/ports/esp32` and run this command `rm -rf build_ESP32*` it will [...]

I dont know why it failed during build, but I cant find any reason for this particular function rather than its return type. So I suspected it cant be managed by the wrapper engine.

I think I knew that the build scripts manage the esp-idf installation too, but I got some problems and found this path and continue using it. I think my error was simply missing "sudo apt-get update" initially.

> Also the solution you are using might be adding to the size of the flash because of additional code that is added by using the rlottie component. I have started to mess about with trying to figure [...]

I didnt pay attentio to the flash usage, but in the solution I posted avobe Im not installing rlottie into the system with "sudo apt-get install librlottie-dev"...not sure if the additional code you mention can be due two rlotties on the image... but Im only using the esp component.

anyway, Im using micropython for a long time, and lvgl is a cool library cause has lots of features and cause it can be integrated into micropython...but it was always late with the miropython versions (not to mention the API changes between versions) and your work solves this problem perfectly.
Regarding the new SPI classes, I had problems with some LCD boards cause they share the SPI bus with the LCD and the TSC, and the solution was to reinit the SPI in the input driver callback (cause some TSC sensors can work at the same baudrate of the LCD), so your solution of SPI bus and SPI device looked a good approach to me...but I cant make work at the first time and it was a pitty that I cant use the "standart" micropython SPI. Even its supposed to be available under SPI.device, it didnt accept my bytearray buffers. So I decided to skip all of it.

kdschlosser, 2024-08-21:
The SPI works. It has been tested many times and it works without issue.

What is more then likely happening is you are not using it properly.

with the SPI you need to keep in mind that there are 2 distinct parts of it. You have the SPI Bus and then you have the devices that reside on the bus...

so when you want to use the SPI you need to create the SPI.Bus object and that object then gets passed to the `lcd_bus.SPIBus`. I do need to change the name of that class because it is a little bit misleading. Now in order to get the touch to work you need to create an `SPI.Device` object that can be passed to the touch driver. 

You can see here that this user has tested the sharing of the SPI bus for the display and also the touch panel..

https://github.com/lvgl-micropython/lvgl_micropython/issues/70#issuecomment-2221002792

This is the code that is used  for it So yours should also be similar to this as well. I do want to note that I have not finished up getting the touch calibration working properly. So don't use that portion of the code seen below.

```
from micropython import const
from machine import SPI
import time
import sys
import lcd_bus
import lvgl as lv

# Display settings
_WIDTH = const(240)
_HEIGHT = const(320)

# Display SPI bus settings
_LCD_SPI_HOST = const(1)
_LCD_SPI_FREQ = const(40_000_000)
_LCD_MOSI_PIN = const(23)
_LCD_MISO_PIN = const(25)
_LCD_SCLK_PIN = const(19)
_LCD_CS_PIN = const(22)
# display additional pins
_LCD_RST_PIN = const(18)
_LCD_DC_PIN = const(21)
_LCD_BKL_PIN = const(5)
_LCD_PWR_PIN = None

# touch panel SPI settings
_TP_SPI_HOST = const(1)
_TP_SPI_FREQ = const(1_000_000)
_TP_CS_PIN = const(32)
_TP_INT_PIN = const(-1)

class TouchCal:
    def __init__(self):
        self.alphaX = 1.088837
        self.betaX = 0.005908419
        self.deltaX = -7.80586
        self.alphaY = 0.00886263
        self.betaY = -1.136386
        self.deltaY = 353.614

    @staticmethod
    def save():
        pass

# Create SPI bus
spi_bus = SPI.Bus(
    host=_LCD_SPI_HOST,
    miso=_LCD_MISO_PIN,
    mosi=_LCD_MOSI_PIN,
    sck=_LCD_SCLK_PIN
)

display_bus = lcd_bus.SPIBus(
    spi_bus=spi_bus,
    freq=_LCD_SPI_FREQ,
    dc=_LCD_DC_PIN,
    cs=_LCD_CS_PIN
)

import ili9341
try:
    display = ili9341.ILI9341(
        data_bus=display_bus,
        display_width=_WIDTH,
        display_height=_HEIGHT,
        reset_pin=_LCD_RST_PIN,
        reset_state=ili9341.STATE_LOW,
        power_pin=_LCD_PWR_PIN,
        backlight_pin=_LCD_BKL_PIN,
        backlight_on_state=ili9341.STATE_HIGH,
        color_space=lv.COLOR_FORMAT.RGB565,
    #	 color_byte_order=ili9341.BYTE_ORDER_BGR,
        rgb565_byte_swap=False
    )
except Exception as ex:
    sys.print_exception(ex)
    raise RuntimeError

# start display driver
display.init()
display.set_rotation(lv.DISPLAY_ROTATION._90)
#display.set_backlight(50)

import xpt2046

# Create touch controller SPI Device
touch_spi_device = SPI.Device(
    spi_bus=spi_bus,
    freq=_TP_SPI_FREQ,
    cs=_TP_CS_PIN
)

cal = TouchCal()
try:
    indev = xpt2046.XPT2046(
        touch_spi_device, startup_rotation=lv.DISPLAY_ROTATION._90, 
        touch_cal=cal, debug=True
    )
    print('is_calibrate is', indev.is_calibrated)

except Exception as ex:
    sys.print_exception(ex)

```

jgpeiro, 2024-08-21:
Maybe the problem was simply I did something wrong. I dont know. I tried for few hours and finally go back to find how to avoid these features. 
Anyway, for now I have enought with the classic SPI and I need to spend my time in other parts of the code.
By the way, I cant find any free lottie icon packs so I moved to gif, and its too slow so I think I will not use any of them.

Just to comment, the project Im working on is published here:
https://hackaday.io/project/197450-ai-game-builder-with-esp32

kdschlosser, 2024-08-21:
the problem with the classic SPI is you are loosing out on some pretty large performance benefits.

kdschlosser, 2024-08-21:
lottie is for animations. are you wanting to do animated icons? if not then lottie would be overkill. GIF only has 8 bit color depth so it sucks just because of that. You can do PNG files and if you want animated icons there is APNG. APNG support is not built into LVGL. I could write a Python decoder for it, wouldn't be that hard to do.

jgpeiro, 2024-08-22:
Yes, the classic micropython SPI is not the fastest possible, but is enought for my requirements now.
About animated PNGs, its interesting, but I will give another chance to lottie files, juts need to find somewhere free packs available for download.

kdschlosser, 2024-08-22:
with respect to GIF's and JPEG's used in LVGL it is going to be incredibly slow if you use the built in decoder. This is because the way the decoders are written is by using the built in file system handling of LVGL. so in the case of say a JPEG where the file needs to be read in it's entirety in order to collect the width and height this process happens multiple times which consumes a lot of time.

There is also the need to iterate over all of the data in the file in order to locate the width and height and this has to be done one byte at a time because of the width and height not being in a static location inside of the file. Frankly the format sucks because of this. GIF's sucks because of the color depth as I already mentioned. 

lotties are nice but I think they carry with them too much additional overhead. There is a lot of parsing that needs to be done for the lottie animations because the images are stored separately and they need to be processed as such and the animation sequence is in json format and that too also need to be parsed. The single largest hangup with lotties is going to be the constant loading and unload of images from memory.

I think with the APNG files the frames themselves can be loaded and stored in memory and at the time they are needed then they can be run through a PNG decoder. this would remove a large portion of the loading and unload that is done with lotties.

Because you are dealing with an MCU it is highly recommended that you keep image size and also the number of frame down as small as possible. In order to get this to work as smooth as possible it is going to be best to pin the decoding of the PNG files to the second core. 

Part of the performance impact is MicroPython is only pinned to a single core. so you only get 1/2 of the processing power available to you. A loop can be written that is tasked with decoding the frames. This is a simple passing memory locations between the 2 cores. if something needs to be decoded then the second core is told to decode and the first core moves on, when the first core comes full loop it can check to see if there is anything that has been decoded.

jgpeiro, 2024-08-23:
the first thing I tested was a lottie file shown in the lvgl website, and it crashed in the middle of the animation. I had no enoguht memory, but I added 1 meg and still crashes. Then I tested with a different one and it worked fine, even with "big" sizes like 100x100 worked fine. The refreshing time was a little bit low, but was ok. The problem is that my applicacion has audio recording and playing and this is not acceptable cause these freezes make the audio sound very bad.
For now I leave it apart and working again in other parts leated with aiohttp communication. When I this works fine I can come back to the graphics animations. 
Regarding the second CPU, I thought that threads run on the second CPU if its available. At least this was true on the RP2040 platforms, and I hope is also the case on ESP32-S3, but never tested.

kdschlosser, 2024-08-23:
It is not the case with the ESP32. Everything is pinned to a single core.

As far as the lagging goes. I suspect that if you use the drivers that are packaged with the binding you would see a decent increase in performance. This is due to being able to use 2 frame buffers in DMA memory. This allows other things to be processed while one of the frame buffers is transmitting. the DMA memory allows the transfer to happen without the CPU needing to be used to do it,m so transferring ends up being non blocking... The bus drivers for SPI, I8080 and RGB all have support to do this.

IDK how large a display you are using but if you do the math for a 320 x 480 display @ 16 bit color using a frame buffer that is 1/10th the size you end up needing to transfer 30,720 bytes of data each tine the display gets flushed. That equates out to 3 milliseconds and that is without any overhead calculated into it. That is also with an 80mhz transfer rate as well. lets call it 3.5 milliseconds to transfer. 3.5 milliseconds is real world....so how many times could the display update in a single second? That would be 28 times if LVGL is able to fill the next frame buffer in 3.5 milliseconds or less. I can tell you that LVGL is going to take longer than that to render a lottie animation that's for sure...

What are the specs of the ESP32 you are using. I know you had said it is an S3, how much RAM and how much flash? We might be able to get a little more speed out of it by adding some additional build commands.

kdschlosser, 2024-08-23:
https://github.com/orgs/micropython/discussions/10604#discussioncomment-4814557

kdschlosser, 2024-08-23:
You might be able to turn off `MICROPY_PY_THREAD_GIL` and be able to run the sound on one thread while running the display in another. There is a change that can also be made to MicroPython so it doesn't pin threads to a single core. Currently all tasks/threads are being pinned to core 0. Should be able to allow FreeRTOS to delegate what core to add additional tasks/threads to. 

This has peaked my interest and I think I might mess around with it. I have already written up a thread worker that handles everything relates to LVGL so it all happens in a single thread. If that thread is run on the second core that would definitely offer a boost in performance.

jgpeiro, 2024-08-25:
I tested the second core and is not used, at least with the default build settings.

import machine
import time
import _thread

def foo(N=1000):
    a = 0
    for i in range( N ):
        for j in range( 100 ):
            a += 1
    return a

def task( core_id ):
    while( True ):
        t0 = time.ticks_ms()
        foo()
        t1 = time.ticks_ms()
        print( core_id, t1-t0 )
        time.sleep(0)

machine.freq( int( 240e6) )
_thread.start_new_thread(task, ("core 1",))
task( "core 0" )

"""
one core:
core 0 248
core 0 247
core 0 248
core 0 248
core 0 248
"""

"""
two cores:
core 1 248
core 0 496 << (it takes twice)
core 1 248
core 0 496
"""

kdschlosser, 2024-08-25:
I know that only one core is being used. You can see it in the MicroPython code...

https://github.com/micropython/micropython/blob/6c3dc0c0b09701b83a957368220d9c8bfe40af0a/ports/esp32/mpthreadport.c#L149

where MP_TASK_COREID is

https://github.com/micropython/micropython/blob/6c3dc0c0b09701b83a957368220d9c8bfe40af0a/ports/esp32/mphalport.h#L48

This is what starts the main loop for micropython

https://github.com/micropython/micropython/blob/6c3dc0c0b09701b83a957368220d9c8bfe40af0a/ports/esp32/main.c#L219

You can see it uses the exact same core id as the threads. It also uses the same core id for I2S and for the RMT as well. 

I know the GIL has to be turned off if wanting to use multiple cores and the user would be responsible for not corrupting the memory, what I do not know about is handling multiple cores accessing the same memory address areas and if it is possible. I have read that it is with the ESP32. I have to do some more digging into it. I also don't know if 2 threads will be able to share the same mutex (lock) objects from inside of the REPL. I have to mess about with it to see if I can get it to work.