lvgl-micropython/lvgl_micropython #96: color problem with i8080 16bit bus and RGB565

hemonu, 2024-08-18:
I use an ELECROW DLC35010R display, which includes a ESP32S3 with 16MB flash and 8MB SPIRAM. Display controller is ILI9488 connected with 16bit parallel bus. I can build your excellent software and get the display up and running, but there is one problem left: 

Color data for blue channel (display is BGR, so it's the 5 MSB) is shifted one bit to the right. So white color e.g. shows up as light yellow etc. I can read back the display ram with my one routine, but I would like to check the data arriving from LVGL in the flush_cb call back, to see if the problem is in LVGL or in the driver. When I put a print statement inside the flush_cb routine in your display_driver_framework.py, Thonny will crash instead of showing the data. 
Do you have any hint, how I can debug the color data?

Thanks for your support

Herbert

kdschlosser, 2024-08-19:
Stop using Thonny. It doesn't reset the board properly and it causes memory corruption. Thonny does a soft reset which doesn't reset anything hardware. and when the program runs again you can have all kinds of off the wall problems.

There are 2 settings that alter the byte and bit order for the colors.

if your display is connected using a 16 lane I8080 connection and you are using 16 bit color you may have to change the byte order for each pixel prior to it being sent.

When using an I8080 16 lane bus this would normally be done by taking the data 0-7 pins and moving them to the data 8-15 pins and moving the data 8-15 pins to the data 0-7 pins. If your display is one where the MCU is already attached then the problem is the display manufacturer wired the MCU to the display incorrectly. Fear not, I have corrected this problem and all you need to do is pass `True` to the `rgb565_byte_swap` parameter when constructing the display driver instance. Setting that to true will order the pins differently when making the connection to the display.

The other setting instructs the display to store the pixel data bits in reverse order. This is done by changing the `color_byte_order` parameter to `BGR` when constructing the display driver instance...

```python
import ili9488

display = ili9488.ILI9488(
    data_bus=data_bus,
    width=480,
    height=320,
    ...
    color_byte_order=ili9488.BYTE_ORDER_BGR,
    rgb565_byte_swap=True
)
display.set_power(1)
display.init()
```

You may also have to mess with the color inversion..

```python
display.invert_colors()
```

hemonu, 2024-08-19:
Problem solved. I finally found a copy/paste error in my bus definition, which resulted in DB15 and DB14 connected to the same pin. That made a quite interesting change in the 3 most significant bits of the bus ;-)
My fault ...

closed 2024-08-19 by hemonu

kdschlosser, 2024-08-19:
OK cool. at least you were able to locate the problem...