lvgl-micropython/lvgl_micropython #99: legacy SPI

dcmcshan, 2024-08-22:
Is there a proper way to create a legacy spi bus that will work with code designed for off the shelf micropython kernel?

I have this very basic driver that I'd like to work with lvgl kernel (it's not using lvgl)

```
self.display = ST7735(
    self.spi,
    dc=Pin(self.PIN_LCD_RS),
    cs=Pin(self.PIN_LCD_CS),
    rst=Pin(self.PIN_LCD_RESET)
)
```

kdschlosser, 2024-08-22:
yeah....

```python

import machine

spi_bus = machine.SPI.Bus(
    host=_HOST,
    mosi=_MOSI,
    miso=_MISO,
    sck=_SCK
)

spi_device = machine.SPI.Device(
    spi_bus=spi_bus,
    cs=_CS,
    freq=_FREQ
)
```

`machine.SPI.Device` is almost identical to the original `machine.SPI`. I say almost because the original didn't handle the CS pin and switching between devices. My version does. If wanting to use threads and 2 different threads try and send something on the bus at the same time would be a real pain in the ass to do using the original version of SPI. That is because you would be responsible for keeping track of the bus being used and you would need to initialize the SPI driver each and every time you wanted to send to a different device. This produced a lot of additional overhead that simply didn't need to be there. With the modified version you don't have to keep track of anything. Just read or write until your hearts content from multiple threads if you want, everything is handled internally.

All of the read and write methods in the `machine.SPI.Device` class are the same as they were in the original `machine.SPI` class there is no change there.

kdschlosser, 2024-08-22:
I also want to note that the st7735 driver has several different variations of it. You need to use the correct variation with your display.. In the binding there are 3 of them that I supply... 

* ST7735R_Green
* ST7735R_Red
* ST7735B

if the one you need is not available let me know and I can write it up for ya.

closed 2024-08-27 by dcmcshan