lvgl-micropython/lvgl_micropython #101: --ota

dcmcshan, 2024-08-24:
When I try an compile with --ota I get this misalignment error.

`python3 make.py esp32 clean submodules mpy_cross BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT --flash-size=16 DISPLAY=ST7735 --usb-otg --ota`

```
Partition ota_0 invalid: Offset 0x12000 is not aligned to 0x10000
[1000/1967] Generating /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/mpy-cross/build/mpy-cross                                                                           
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
makeversionhdr.py: Warning: No git repo or tag info available, falling back to mpconfig.h version info.
[1004/1967] Generating ../../genhdr/qstr.i.last                                                                        
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stderr_output_89040 and /Users/danielmcshan/GitHub/lvgl_micropython/lib/micropython/ports/esp32/build-ESP32_GENERIC_S3-SPIRAM_OCT/log/idf_py_stdout_output_89040
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1
```

dcmcshan, 2024-10-17:
Just a ping on this.  I think it's probably something simple to fix the alignment.  Thanks :)

kdschlosser, 2024-10-17:
I will fix it now.

kdschlosser, 2024-10-17:
all fixed....

dcmcshan, 2024-10-17:
I updated and reran, but I think we need to increase the partition size.  This isn't actually the chip for our board.  It's for a M5 Atom S3, but it fails same over there as well.

`python3 make.py esp32 BOARD=ESP32_GENERIC_S3 --flash-size=8 DISPLAY=ST7735 --usb-otg --ota --dual-core-threads
`

```
Error: All app partitions are too small for binary micropython.bin size 0x25cc10:
  - Part 'ota_0' 0/16 @ 0x20000 size 0x25a000 (overflow 0x2c10)
  - Part 'ota_1' 0/17 @ 0x280000 size 0x25a000 (overflow 0x2c10)
ninja: build stopped: subcommand failed.

```

kdschlosser, 2024-10-18:
That is going to happen because some MCU's don't have enough flash tom store LVGL and MicroPython. It takes up 2.5 MB of flash. so when doming OTA you need to have 5MB of flash just for the application. not including NVS space and user space. At a minimum 8MB of flash is what is going to need to be available to use OTA

kdschlosser, 2024-10-18:
So what the build script does is it will adjust the partition size so the binding is able to fit. and when it does that if the MCU being used doesn't have enough flash space you will see the error you got above.

dcmcshan, 2024-10-18:
Makes sense, but I thought I got the error while compiling with 16MB flash... I'll double check that.

kdschlosser, 2024-10-18:
are you adding the `--flash-size` build parameter? You need to have that added to set the flash size on the MCU.

dcmcshan, 2024-10-18:
Yes. I use flash-size of 8 for the Atom S3 and 16 for our board

kdschlosser, 2024-10-18:
I found the problem and it is now fixed.

closed 2024-11-05 by kdschlosser