lvgl-micropython/lvgl_micropython #105: No picture appears on the ili9341 display

Phil-CN, 2024-08-28:
I ran the following code but my display shows nothing, why is that? I am pretty sure my wiring is fine.

    from machine import Pin, PWM
    from ili9341 import ILI9341
    from lcd_bus import SPIBus
    from machine import SPI
    import lcd_bus

    import lvgl as lv

    _BUFFER_SIZE = const(30720)

    spi_bus = SPI.Bus(host=2, mosi=17, miso=7, sck=16)

    display_bus = SPIBus(
        spi_bus = spi_bus,
        freq = 400000,
        dc = 18,
        cs = 9)

    fb1 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)
    fb2 = display_bus.allocate_framebuffer(_BUFFER_SIZE, lcd_bus.MEMORY_INTERNAL | lcd_bus.MEMORY_DMA)

    lv.init()

    display = ILI9341(
        data_bus=display_bus,
        display_width=320,
        display_height=240,
        frame_buffer1=fb1,
        frame_buffer2=fb2,
        reset_pin=8,
        backlight_pin=15,
        backlight_on_state=0,
        color_space=lv.COLOR_FORMAT.RGB565)

    # backlight = PWM(Pin(15))
    # backlight.freq(4000)
    # backlight.duty(1023)

    display.init()
    display.invert_colors()
    display.set_backlight(100)

    scrn = lv.obj()
    scrn.set_style_bg_color(lv.color_hex(0xffb6c1), 0)

    slider = lv.slider(scrn)
    slider.set_size(300, 50)
    slider.center()

    label = lv.label(scrn)
    label.set_text('HELLO WORLD!')
    label.align(lv.ALIGN.CENTER, 0, -50)

closed 2024-08-28 by Phil-CN

straga, 2024-08-29:
@Phil-CN what was wrong?

Phil-CN, 2024-08-29:
> @Phil-CN出了什么问题？

Hahaha thanks bro, no problem now