lvgl-micropython/lvgl_micropython #109: LOLIN S2 MINI --usb-otg ....unkown command

fabse-hack, 2024-09-02:
so, i think is a bug. Because in the esp32.py i see, that --usb-otg refer to the esp32.

`python3 make.py esp32 submodules mpy_cross clean BOARD=LOLIN_S2_MINI --flash-size=4 DISPLAY=ili9341 INDEV=gt911 --usb-otg
`
`make: Unbekannte Option »--usb-otg«`
unknow option...... without --usb-otg the REPL is not working....

kdschlosser, 2024-09-06:
Sorry for the delay getting to ya. I missed this issue somehow.

I am going to look up your board right now to see what it is and what is needed to get it to work.

kdschlosser, 2024-09-06:
OK you should be all set. I just pushed a commit that should fix the problem.

closed 2024-09-06 by kdschlosser

kdschlosser, 2024-09-06:
as a note.. You don't need to supply the --usb-otg switch. It will work without it because that board is only made with usb-otg

fabse-hack, 2024-09-06:
your code is not working.

why i can not do it like this ?????
and i not found any reference, that LOLIN_S2_MINI use the right micropython binary for this contoller.

i change your code to this: ..... and then it's building successful. But i have to test it with my display and code.

```
def parse_args(extra_args, lv_cflags, brd):
    global board
    global usb_otg

    if brd is None:
        brd = 'ESP32_GENERIC'

    board = brd

    extra_args = common_args(extra_args)

    if board == 'ESP32_GENERIC':
        extra_args = esp32_args(extra_args)
    elif board == 'ESP32_GENERIC_S2':
        extra_args = esp32_s2_args(extra_args)
    elif board == 'ESP32_GENERIC_S3':
        extra_args = esp32_s3_args(extra_args)
    elif board == 'ESP32_GENERIC_C3':
        extra_args = esp32_c3_args(extra_args)
    elif board == 'LOLIN_S2_MINI':
        extra_args = esp32_s2_args(extra_args)
        # usb_otg = True
```

fabse-hack, 2024-09-08:
[image]
is working now

REPL is working too

thanks!

kdschlosser, 2024-09-08:
lets gets the colors all sorted out for ya. what are you using for a script to start the display?